fwupd/fwupd #6243: bcm57xx plugin does NOT work with CONFIG_IO_STRICT_DEVMEM kernel config

joeyli, 2023-10-05:
**Describe the bug**
When running fwupd, we found that the following failure messages in  /var/log/messages :

fwupd[6008]: 23:22:49:0833 FuEngine failed to add device /sys/devices/pci0000:00/0000:00:03.2/0000:03:00.0: could not mmap /sys/devices/pci0000:00/0000:00:03.2/0000:03:00.0/resource0: Invalid argument

**Steps to Reproduce**
Just running fwupd for updating firmware then checking messages log.

**Expected behavior**
The FuEngine failed should not be exposed. The bcm57xx plugin should not use mmap to access BAR space because security reason. It should access specific interface of bcm57xx driver in kernel.

**fwupd version information**
fwupd 1.8.6

# installation on openSUSE 15.5
zypper -v in fwupd

**Additional questions**

- Operating system and version: openSUSE 15.5

In plugins/bcm57xx/fu-bcm57xx-recovery-device.c::fu_bcm57xx_recovery_device_open(), it tries to mmap to BAR space through /dev/mem:

                        /* mmap */
                        g_debug("mapping BAR[%u] %s for 0x%x bytes", i, fn, (guint)st.st_size);
                        self->bar[i].buf =
                            (guint8 *)mmap(0, st.st_size, PROT_READ | PROT_WRITE, MAP_SHARED, memfd, 0);

But this mmap be blocked by kernel's CONFIG_IO_STRICT_DEVMEM config for security reason.

hughsie, 2023-10-05:
> It should access specific interface of bcm57xx driver in kernel

It does, but it falls back to the BAR access when the device is in recovery mode. See https://github.com/fwupd/fwupd/tree/main/plugins/bcm57xx#update-behavior

joeyli, 2023-10-05:
Hi Richard, 

Thanks for your information!

> > It should access specific interface of bcm57xx driver in kernel
> 
> It does, but it falls back to the BAR access when the device is in recovery mode. See https://github.com/fwupd/fwupd/tree/main/plugins/bcm57xx#update-behavior

In the above document, it said:

> On flash failure the device is nonfunctional, but is recoverable using direct BAR writes, which is typically much slower > than updating the device using the kernel driver and the ethtool API.

Does that mean the flash of bcm57xx in issue machine is broken? So bcm57xx plugin moved to mmap() for accessing BAR?

Thanks!

hughsie, 2023-10-05:
Ahh, I seem to be wrong: https://github.com/fwupd/fwupd/blob/main/plugins/bcm57xx/fu-bcm57xx-device.c

	/* we need this even for non-recovery to reset APE */
	fu_device_set_context(FU_DEVICE(self->recovery), fu_device_get_context(FU_DEVICE(self)));
	fu_device_incorporate(FU_DEVICE(self->recovery), FU_DEVICE(self));
	if (!fu_device_probe(FU_DEVICE(self->recovery), error))
		return FALSE;

So maybe the plugin shouldn't be enabled when CONFIG_IO_STRICT_DEVMEM is enabled?

superm1, 2023-12-21:
> So maybe the plugin shouldn't be enabled when CONFIG_IO_STRICT_DEVMEM is enabled?

Yeah; IMO if that mmap fails it should just silently fail probe.

closed 2024-01-02 by hughsie