fwupd/fwupd #6244: fwupd 1.9.1 fails to find device, returns early after restarting AVer camera, on kernel 4.19

rcyeh, 2023-10-05:
**Describe the bug**
I am using fwupdmgr (fwupd 1.9.1) to update an AVer camera. This succeeds when run on a kernel 5.4-based ChromeOS, but fails on the same version of ChromeOS based on kernel 4.19. fwupd 1.9.5 also fails on the latest ChromeOS based on kernel 4.19.

The failure occurs after the write, first restart, verify, second restart. fwupdmgr reports: `failed to get device before update reload: failed to wait for device: device b0a78eb71f4eeea7df8fb114522556ba8ce22074 did not come back`

**Steps to Reproduce**

`fwupdmgr local-install -v --allow-older avocor-avm-6570-firmware-2.22/avocor-avm-6570-firmware-2.22.cab`

(can be any version of firmware from 2.8 (factory version) through 2.22 (current version))

**Expected behavior**

After a successful update, `lsusb -v -d 34ad:0006` shows the device with the updated firmware version (`grep bcdDevice`).

**fwupd version information**
Please provide the version of the daemon and client.

On the working version:

```shell
fwupdmgr --version

runtime   org.freedesktop.fwupd         1.9.1
compile   com.hughsie.libxmlb           0.3.2
compile   org.freedesktop.gusb          0.3.10
compile   com.hughsie.libjcat           0.1.8
runtime   org.kernel                    5.4.217-19694-g8fbced507746
runtime   org.freedesktop.gusb          0.3.10
compile   org.freedesktop.fwupd         1.9.1
```

Failure case (1.9.1):

```
runtime   org.freedesktop.fwupd         1.9.1
compile   com.hughsie.libxmlb           0.3.2
compile   org.freedesktop.gusb          0.3.10
compile   com.hughsie.libjcat           0.1.8
runtime   org.kernel                    4.19.261-13934-g734f29d88a0b
runtime   org.freedesktop.gusb          0.3.10
compile   org.freedesktop.fwupd         1.9.1
```

Failure case (1.9.5):

```shell
runtime   org.freedesktop.fwupd         1.9.5
compile   com.hughsie.libxmlb           0.3.11
compile   org.freedesktop.gusb          0.4.5
compile   com.hughsie.libjcat           0.1.8
runtime   org.kernel                    4.19.294-15287-g1e5683f9b3be
runtime   org.freedesktop.gusb          0.4.5
compile   org.freedesktop.fwupd         1.9.5
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): included with ChromeOS

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

Successful case

```shell
fwupdmgr get-devices --show-all-devices

Google Endeavour
[... 189 lines cut ...]
Writing?                 [********************************       ] Less than one minute remaining?(fwupdmgr:17540): Fwupd-DEBUG: 13:03:20.992: Emitting ::status-changed() [device-busy]
Waiting?                 [********************************       ](fwupdmgr:17540): Fwupd-DEBUG: 13:03:21.000: Emitting ::status-changed() [device-verify]
Verifying?               [********************************       ] Less than one minute remaining?(fwupdmgr:17540): Fwupd-DEBUG: 13:03:21.000: Emitting ::status-changed() [device-read]
Reading?                 [**********************************     ] Less than one minute remaining?(fwupdmgr:17540): Fwupd-DEBUG: 13:03:48.529: Emitting ::status-changed() [device-restart]
Restarting device?       [**********************************     ](fwupdmgr:17540): Fwupd-DEBUG: 13:03:48.529: Emitting ::device-changed(b0a78eb71f4eeea7df8fb114522556ba8ce22074)
(fwupdmgr:17540): Fwupd-DEBUG: 13:03:48.529: Emitting ::status-changed() [device-busy]
Waiting?                 [**********************************     ](fwupdmgr:17540): Fwupd-DEBUG: 13:03:48.610: Emitting ::device-changed(b0a78eb71f4eeea7df8fb114522556ba8ce22074)
(fwupdmgr:17540): Fwupd-DEBUG: 13:03:48.610: Emitting ::device-changed(b0a78eb71f4eeea7df8fb114522556ba8ce22074)
(fwupdmgr:17540): Fwupd-DEBUG: 13:03:48.626: Emitting ::device-changed(b0a78eb71f4eeea7df8fb114522556ba8ce22074)
(fwupdmgr:17540): Fwupd-DEBUG: 13:03:48.626: Emitting ::changed()
(fwupdmgr:17540): Fwupd-DEBUG: 13:03:48.636: Emitting ::changed()
(fwupdmgr:17540): Fwupd-DEBUG: 13:04:48.531: Emitting ::device-changed(b0a78eb71f4eeea7df8fb114522556ba8ce22074)
(fwupdmgr:17540): Fwupd-DEBUG: 13:04:48.531: Emitting ::status-changed() [idle]

failed to get device before update reload: failed to wait for device: device b0a78eb71f4eeea7df8fb114522556ba8ce22074 did not come back

genesis-rev4 ~ # fwupdmgr --version
runtime   org.freedesktop.fwupd         1.9.5
compile   com.hughsie.libxmlb           0.3.11
compile   org.freedesktop.gusb          0.4.5
compile   com.hughsie.libjcat           0.1.8
runtime   org.kernel                    4.19.294-15287-g1e5683f9b3be
runtime   org.freedesktop.gusb          0.4.5
compile   org.freedesktop.fwupd         1.9.5
```

hughsie, 2023-10-05:
Can you get `sudo fwupdtool --plugins dfu -vv install whatever.cab` please? -- also -- anything in dmesg to do with device enumeration from the kernel?

rcyeh, 2023-10-05:
With --vv, the log file is too large (53 MB) for Github (25 MB limit). kdgwill@google.com will send you that log by email.

[fwupdtool.get-devices.after.txt](https://github.com/fwupd/fwupd/files/12823461/fwupdtool.get-devices.after.txt)
[fwupdtool.get-devices.before.txt](https://github.com/fwupd/fwupd/files/12823462/fwupdtool.get-devices.before.txt)
[dmesg.after.txt](https://github.com/fwupd/fwupd/files/12823463/dmesg.after.txt)
[dmesg.before.txt](https://github.com/fwupd/fwupd/files/12823464/dmesg.before.txt)

hughsie, 2023-10-10:
Isn't that a kernel crash?

	[785773.866503] Call Trace:
	[785773.866514]  dump_stack+0x97/0xdb
	[785773.866522]  nmi_cpu_backtrace+0x45/0x92
	[785773.866529]  ? arch_trigger_cpumask_backtrace+0x16/0x16
	[785773.866535]  nmi_trigger_cpumask_backtrace+0x4f/0xe6
	[785773.866541]  __handle_sysrq+0x122/0x124
	[785773.866548]  write_sysrq_trigger+0x32/0x3e
	[785773.866554]  proc_reg_write+0x40/0x72
	[785773.866560]  __vfs_write+0x169/0x183
	[785773.866565]  ? _copy_from_user+0x39/0x5f
	[785773.866570]  ? selinux_file_permission+0xe7/0x141
	[785773.866578]  ? security_file_permission+0x36/0xb2
	[785773.866583]  vfs_write+0x103/0x231
	[785773.866588]  __x64_sys_write+0x62/0xc3
	[785773.866593]  do_syscall_64+0x72/0x26e
	[785773.866601]  ? __do_page_fault+0x1fa/0x47d
	[785773.866609]  entry_SYSCALL_64_after_hwframe+0x44/0xa9

and then we see:

	[786074.739764] Sending NMI from CPU 1 to CPUs 0,2-7:
	[786074.739773] NMI backtrace for cpu 0 skipped: idling at intel_idle+0x81/0x11f
	[786074.739775] NMI backtrace for cpu 3 skipped: idling at intel_idle+0x81/0x11f
	[786074.739781] NMI backtrace for cpu 2
	[786074.739782] CPU: 2 PID: 12164 Comm: D-Bus thread Tainted: G           O      4.19.261-13934-g734f29d88a0b #1

closed 2024-11-11 by hughsie