fwupd/fwupd #6253: Not enough data was provided by the platform to make an HSI calculation, but device security report shows pass/fail for everything

mcatanzaro, 2023-10-10:
**Describe the bug**

fwupd is reporting an invalid HSI level:

```
$ fwupdmgr security

Not enough data was provided by the platform to make an HSI calculation.
 » https://fwupd.github.io/hsi.html#not-enough-info
```

However, when I check the device security report, it appears fwupd has all the information it needs to determine the security level should be HSI-0:

```
Device Security Report
======================

Report details
  Date generated:                                  2023-10-10 09:23:24
  fwupd version:                                   1.9.6

System details
  Hardware model:                                  To Be Filled By O.E.M. To Be Filled By O.E.M.
  Processor:                                       AMD Ryzen Threadripper 1950X 16-Core Processor
  OS:                                              Fedora Linux 39 (Workstation Edition Prerelease)
  Security level:                                  HSI:INVALID:missing-data

HSI-1 Tests
  UEFI Platform Key:                               Pass (Valid)
  UEFI Bootservice Variables:                      Pass (Locked)
  TPM v2.0:                                      ! Fail (Not Found)
  BIOS Firmware updates:                           Pass (Enabled)
  UEFI Secure Boot:                                Pass (Enabled)

HSI-2 Tests
  IOMMU Protection:                              ! Fail (Not Found)

HSI-3 Tests
  Pre-boot DMA Protection:                       ! Fail (Not Valid)
  Suspend To RAM:                                ! Fail (Enabled)
  Suspend To Idle:                               ! Fail (Not Enabled)

HSI-4 Tests
  Encrypted RAM:                                 ! Fail (Not Supported)

Runtime Tests
  Firmware Updater Verification:                   Pass (Not Tainted)
  Linux Swap:                                      Pass (Encrypted)
  Linux Kernel Lockdown:                           Pass (Enabled)
  Linux Kernel Verification:                       Pass (Not Tainted)

Host security events
  2023-07-05 06:49:50   IOMMU Protection           ! Fail (Enabled → Not Found)
  2023-07-05 06:49:50   Pre-boot DMA Protection    ! Fail (Not Enabled → Not Valid)
  2022-08-23 20:56:09   IOMMU Protection             Pass (Not Found → Enabled)
  2022-08-23 20:56:09   Pre-boot DMA Protection    ! Fail (Not Valid → Not Enabled)

For information on the contents of this report, see https://fwupd.github.io/hsi.html
```

In fact, I don't see a single data point missing here, other than the hardware model.

**Steps to Reproduce**
1. Be me, or else be a ninja with access to my office
2. Run `fwupdmgr security`

**Expected behavior**
Should probably print HSI-0 rather than HSI:INVALID, because fwupd knows the conditions for HSI-1 are not satisfied due to lack of TPM v2.0.

**fwupd version information**

```
$ fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.14
compile   com.hughsie.libjcat           0.1.14
compile   org.freedesktop.fwupd         1.9.6
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.6
runtime   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.gusb          0.4.6
runtime   org.freedesktop.fwupd         1.9.6
runtime   org.kernel                    6.5.6-300.fc39.x86_64
```

This is fwupd-1.9.6-1.fc39

<summary>**fwupd device information**</summary>

```
$ fwupdmgr get-devices --show-all-devices
To Be Filled By O.E.M. To Be Filled By O.E.M.
│
├─Unknown Device:
│     Device ID:          a29de8124928f7dc31db026069ac1d601a56056f
│     GUID:               2d9ac22e-0ffe-5691-9e21-f9fecce8761c ← GPIO\ID_AMDIF030:00
│   
├─AMD Ryzen Threadripper 1950X 16-Core Processor:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x08001137
│ │   Vendor:             Advanced Micro Devices, Inc.
│ │   GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│ │                       997104da-ec11-5e04-9d94-ac5501fbf609 ← CPUID\PRO_0&FAM_17
│ │                       5fa201c0-e152-5003-a56b-93a9dd683fc7 ← CPUID\PRO_0&FAM_17&MOD_01
│ │                       0eab26c9-612c-5655-957c-6a4bec51e214 ← CPUID\PRO_0&FAM_17&MOD_01&STP_1
│ │   Device Flags:       • Internal device
│ │ 
│ ├─GPIO controller:
│ │     Device ID:        f685512aa07369c9e77742acef941d779d31e766
│ │     GUID:             37b440a9-2473-5087-a39b-db84f32a8ed8 ← GPIO\ID_AMDI0030:00
│ │   
│ └─Secure Processor:
│       Device ID:        c54ab0237d7a8db8c717b68e0be78e4374a2a079
│       Vendor:           Advanced Micro Devices, Inc. (PCI:0x1022)
│       GUIDs:            aa0a9592-5ebf-51ed-b868-10b866630b66 ← PCI\VEN_1022&DEV_1456
│                         38f5fb89-5a5f-5bdc-90d3-e94363e1c9be ← PCI\VEN_1022&DEV_1456&SUBSYS_10221456
│       Device Flags:     • Internal device
│     
├─E387 Polaris20 XL A1 GDDR5 128Mx32 4GB:
│     Device ID:          1e8438ff6674a9df9ab19014dd190d4d8f50ea7a
│     Current version:    015.050.002.001.000000
│     Vendor:             Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002, PCI:0x1022)
│     GUIDs:              55746bd7-4196-594a-9d39-0808222765d1 ← PCI\VEN_1002&DEV_67DF
│                         e369f486-ab5e-5397-bbc0-08d9b6f0b7d9 ← PCI\VEN_1002&DEV_67DF&SUBSYS_1DA2E387
│                         884c0933-2918-5ff3-89a8-426c6686f9c7 ← AMD\113-2E3871U-O4A
│     Device Flags:       • Cryptographic hash verification is available
│   
├─SSD 970 EVO 1TB:
│     Device ID:          08719eac4fd20b7fb04497ff483fe71d94a92780
│     Summary:            NVM Express solid state drive
│     Current version:    2B2QEXE7
│     Vendor:             Samsung (NVME:0x144D)
│     Serial Number:      S467NF0K607390K
│     GUIDs:              47335265-a509-51f7-841e-1c94911af66b ← NVME\VEN_144D&DEV_A808
│                         c9d531ea-ee7d-5562-8def-c64d0d144813 ← NVME\VEN_144D&DEV_A808&SUBSYS_144DA801
│                         70520c98-99f3-5955-a1be-33b0a4cdce23 ← Samsung SSD 970 EVO 1TB
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Signed Payload
│   
├─UEFI Platform Key:
│     Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a
│     Summary:            Root Agency
│     GUID:               3e989e5c-2bc2-5087-8216-43b67436b3b4 ← UEFI\CRT_CB5B70FA28620B74EF4943C7F12EE9265087E201
│   
├─UEFI dbx:
│     Device ID:          362301da643102b9f38477387e2193e57abaa590
│     Summary:            UEFI revocation database
│     Current version:    371
│     Minimum Version:    371
│     Vendor:             UEFI:Linux Foundation
│     Install Duration:   1 second
│     GUIDs:              c6682ade-b5ec-57c4-b687-676351208742 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503
│                         f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Only version upgrades are allowed
│                         • Signed Payload
│   
└─X399 Series Chipset SATA Controller:
      Device ID:          7d29f2075dcafb4488b40c73f199cf46bb76bddb
      Current version:    02
      Vendor:             Advanced Micro Devices, Inc. [AMD] (PCI:0x1022)
      GUIDs:              37076a7c-c106-5f7d-b9ad-098dcd405a55 ← PCI\VEN_1022&DEV_43B6
                          e511e400-c9e1-523b-a3ab-326677431701 ← PCI\VEN_1022&DEV_43B6&SUBSYS_1B211062
                          1d3e715a-7904-528d-8c2f-ae4648e476e7 ← PCI\VEN_1022&DEV_1453
                          1b20334c-5fb2-5388-8b96-b8f7672d2cd8 ← PCI\VEN_1022&DEV_1453&SUBSYS_10221453
      Device Flags:       • Internal device
                          • Cryptographic hash verification is available
    
────────────────────────────────────────────────
Devices that have been updated successfully:
 • UEFI dbx (272 → 371)
```

**Additional questions**

- Operating system and version: Fedora 39
- Have you tried rebooting? Yes, I first noticed this issue several months ago and have rebooted many times since then.
- Is this a regression? Sort of, earlier versions of fwupd did display HSI-0 rather than invalid

hughsie, 2023-10-10:
What's the output of `fwupdmgr security --json` please

hughsie, 2023-10-10:
What's the output of `sudo dmesg | grep psp` and `sudo dmesg | grep ccp` please.

superm1, 2023-10-10:
> What's the output of `sudo dmesg | grep psp` and `sudo dmesg | grep ccp` please.

A better check for what you're looking for:

```
# ls /sys/bus/pci/drivers/ccp/0*
```

If your platform doesn't export security attributes you'll be missing things like `anti_rollback_status`.  For reference here is a platform I have on hand that exports everything properly.

```
# ls /sys/bus/pci/drivers/ccp/0*/
anti_rollback_status      current_link_speed  driver             iommu          max_link_speed  power        resource5                subsystem_device
ari_enabled               current_link_width  driver_override    iommu_group    max_link_width  power_state  revision                 subsystem_vendor
broken_parity_status      d3cold_allowed      enable             irq            modalias        remove       rom_armor_enforced       tsme_status
class                     debug_lock_on       firmware_node      link           msi_bus         rescan       rpmc_production_enabled  uevent
config                    device              fused_part         local_cpulist  msi_irqs        resource     rpmc_spirom_available    vendor
consistent_dma_mask_bits  dma_mask_bits       hsp_tpm_available  local_cpus     numa_node       resource2    subsystem
```

If you're missing the attributes then fwupd can't do anything, it can't evaluate the security of the system without them.

superm1, 2023-10-10:
@hughsie I think if anything the action here is on gnome-control-center to clarify in the device security report what missing data means.

mcatanzaro, 2023-10-10:
Note: since reporting this bug, I have enabled IOMMU in my UEFI firmware settings, which it looks like I had disabled by mistake. I don't think that has had any impact on this bug report, however.

```
$ sudo dmesg | grep psp
[    4.172713] ccp 0000:08:00.2: psp enabled
[    4.173461] ccp 0000:43:00.2: psp enabled
```

```
$ sudo dmesg | grep ccp
[    4.171732] ccp 0000:08:00.2: enabling device (0000 -> 0002)
[    4.172688] ccp 0000:08:00.2: ccp enabled
[    4.172713] ccp 0000:08:00.2: psp enabled
[    4.172751] ccp 0000:43:00.2: enabling device (0000 -> 0002)
[    4.173438] ccp 0000:43:00.2: ccp enabled
[    4.173461] ccp 0000:43:00.2: psp enabled
```

```
$ sudo ls /sys/bus/pci/drivers/ccp/0*
'/sys/bus/pci/drivers/ccp/0000:08:00.2':
aer_dev_correctable	  device	   local_cpulist   remove
aer_dev_fatal		  dma		   local_cpus	   rescan
aer_dev_nonfatal	  dma_mask_bits    max_link_speed  resource
ari_enabled		  driver	   max_link_width  resource2
broken_parity_status	  driver_override  modalias	   resource5
class			  enable	   msi_bus	   revision
config			  firmware_node    msi_irqs	   subsystem
consistent_dma_mask_bits  iommu		   numa_node	   subsystem_device
current_link_speed	  iommu_group	   pools	   subsystem_vendor
current_link_width	  irq		   power	   uevent
d3cold_allowed		  link		   power_state	   vendor

'/sys/bus/pci/drivers/ccp/0000:43:00.2':
aer_dev_correctable	  device	   local_cpulist   remove
aer_dev_fatal		  dma		   local_cpus	   rescan
aer_dev_nonfatal	  dma_mask_bits    max_link_speed  resource
ari_enabled		  driver	   max_link_width  resource2
broken_parity_status	  driver_override  modalias	   resource5
class			  enable	   msi_bus	   revision
config			  firmware_node    msi_irqs	   subsystem
consistent_dma_mask_bits  iommu		   numa_node	   subsystem_device
current_link_speed	  iommu_group	   pools	   subsystem_vendor
current_link_width	  irq		   power	   uevent
d3cold_allowed		  link		   power_state	   vendor
```

I see there is indeed no `anti_rollback_status` there, but I don't understand why this would prevent us from computing that the HSI level is 0 due to absence of TPM 2.0.

> @hughsie I think if anything the action here is on gnome-control-center to clarify in the device security report what missing data means or be consistent with how fwupdmgr behaves in not showing data if there's not enough.

I don't understand. Already both gnome-control-center and fwupdmgr show nothing beyond saying that there is not enough information available. But the provided Device Security Report also shows that all required information is present.

superm1, 2023-10-10:
I can see the confusion - it's because this decision is made by the pci-psp plugin.  It has the ability to declare unsupported due to missing data.

https://github.com/fwupd/fwupd/blob/main/plugins/pci-psp/fu-pci-psp-device.c#L288

> I don't understand. Already both gnome-control-center and fwupdmgr show nothing beyond saying that there is not enough information available. But the provided Device Security Report also shows that all required information is present.

It's not though - you don't have the data reported by the pci-psp plugin which we require on AMD CPU/APU.

mcatanzaro, 2023-10-10:
> It's not though - you don't have the data reported by the pci-psp plugin which we require on AMD CPU/APU.

So I have no clue what pci-psp is, but since one of the requirements for HSI-1 is not met, there is no uncertainty as to the correct HSI level, and any missing information is therefore immaterial. Right? So problem 1: missing information from pci-psp should only be considered if it's actually material to the security level.

In fact, the info from pci-psp was apparently not required to compute any of the pass/fail checks for _any_ HSI level, because every single check at every HSI level already has either a Pass or Fail determination. So problem 2: it's not clear from the Device Security Report why pci-psp matters at all.

I'm quite confused as to the value of having HSI-INVALID at all. Why not just consider missing information to be equivalent to Fail and compute an HSI score based on that? I am thinking that gnome-control-center should just completely ignore HSI-INVALID and always show HSI-0 instead. Problem 3: it really doesn't make sense for 0 to be less bad than INVALID, as this allows manufacturers to dodge ratings entirely.

hughsie, 2023-10-10:
> Why not just consider missing information to be equivalent to Fail and compute an HSI score based on that

@superm1 I'm not against this as an idea -- it's certainly confused enough people recently.

superm1, 2023-10-10:
> So I have no clue what pci-psp is, but since one of the requirements for HSI-1 is not met, there is no uncertainty as to the correct HSI level

I disagree.  The platform very well may be properly secured it by the hardware designer, but fwupd can't confirm nor deny it.

> and any missing information is therefore immaterial. Right? So problem 1: missing information from pci-psp should only be considered if it's actually material to the security level

It absolutely is material to the security level.

> Why not just consider missing information to be equivalent to Fail and compute an HSI score based on that

The problem with having missing information fail instead of advertise missing data is it begs the question of **what can I do to fix it**?

superm1, 2023-10-10:
> I feel like we're going in circles here. If this is true, then how do you explain fwupd's own Device Security Report? It says Fail right next to TPM 2.0. There is no way the HSI level can possibly be anything other than HSI:0, because you know there is no TPM 2.0. Right? Surely that is everything you need to know to compute the HSI level? There is absolutely no way that anything related to pci-psp could possibly raise the HSI level to HSI:1, right?

In this specific case - you're right.  But the problem is if TPM2.0 was "fixed" and IOMMU was "fixed" then it looks like you could potentially pass all the way up to HSI-3, which isn't possible to evaluate.

> Nowhere in the Device Security Report references pci-psp at all. Either the Device Security Report should be fixed to indicate that pci-psp is missing and this affects the results, or pci-psp should not be considered. Right? The Device Security Report should tell you why you wound up with a particular result, but in this case it's a complete mystery.

Device Security report is generated by gnome control center, and this is a fwupd bug.  I do think that we should mirror decisions and policies between the two, but let's keep this specific discussion about how `fwupdmgr security` behaves for now.

> The user is going to ask this same question regardless of whether the result is HSI:INVALID or HSI:0. You're just making it waaay more confusing by refusing to display any result.

As illustrated by a few reports besides your own that we've seen recently I think you're right that this situation is confusing.
I'm very cognizant however that the most user facing consumer of this (GNOME control Center) is going to be showing the big scary red box when it shouldn't.

How about for a meet in the middle solution - we amend both fwupdmgr, fwupdtool (and later the gnome control center device security report) to show _what exactly_ is missing data.  Then we can keep having an "invalid" category.

mcatanzaro, 2023-10-10:
> How about for a meet in the middle solution - we amend both fwupdmgr, fwupdtool (and later the gnome control center device security report) to show what exactly is missing data.

This would help a lot.

> Then we can keep having an "invalid" category.

In your example where we don't know whether the system meets HSI:2 or HSI:3, why is it better to show HSI:Invalid than just showing HSI:2? Showing HSI:2 provides the user way more information. Honestly, same applies to the difference between HSI:0 and HSI:Invalid.

Isn't it weird if your system meets HSI:2 today, then tomorrow fwupd decides to check some new thing and determines it is missing, so the system drops from HSI:2 to HSI:Invalid?

superm1, 2023-10-10:
Well if some checks haven't been checked due to missing data then your 2 might not be a real 2..

superm1, 2023-10-10:
Not everything is considered mandatory. Missing data only applies to certain attributes.

mcatanzaro, 2023-10-18:
> Well if some checks haven't been checked due to missing data then your 2 might not be a real 2..

Proposal:

 * fwupd developers to very seriously reconsider the value of HSI:INVALID based on suggestion in https://github.com/fwupd/fwupd/issues/6253#issuecomment-1756084031as it makes it impossible for users to understand their device security status and benefit of having this state is unclear
 * gnome-control-center to treat HSI:INVALID as equivalent to HSI:0 in the meantime

superm1, 2023-11-01:
fixed by #6301

closed 2023-11-01 by superm1