fwupd/fwupd #6256: fwupd snap usage on ubuntu core systems

benjaminbeisel, 2023-10-11:
Hi,

I've tried to use the fwupd snap 1.9.6 (rev:  5236) on an Ubuntu Core system to perform an UEFI capsule (signed cab file) update.
Unfortunately it's not clear to me which interfaces (polkit, dbus, udisks,...) I need to connect and where to provide the responsible plugs/slots.
The wiki page only states udisks2 2.7.2 or later is required. Since there is no udisks snap yet, I assume I have to provide an udisks slot from an addtional own snap.

After the installation the connections look like the following
```
$ snap connections fwupd
Interface         Plug                    Slot                         Notes
bluez             fwupd:bluez             -                            -
dbus              -                       fwupd:fwupd-dbus             -
fwupd             fwupd:fwupdmgr          fwupd:fwupd                  -
hardware-observe  fwupd:hardware-observe  :hardware-observe            -
modem-manager     fwupd:modem-manager     -                            -
network           fwupd:network           :network                     -
opengl            fwupd:opengl            :opengl                      -
polkit            fwupd:polkit            -                            -
raw-usb           fwupd:raw-usb           :raw-usb                     -
shutdown          fwupd:shutdown          :shutdown                    -
udisks2           fwupd:udisks2           udisks-snap:service  -
upower-observe    fwupd:upower-observe    -                            -
```

To install a file, I have to move it to /var/snap/fwupd/common/, otherwise it's not found by the fwupd snap. 
The installation nevertheless fails with a disconnection, probably due to the missing dbus and polkit interfaces:

```
$ sudo fwupd.fwupdmgr install /var/snap/fwupd/common/BIOSMEFW.CTL6R912.cab --allow-older

** (process:7338): WARNING **: 05:17:35.906: Unable to register authentication agent: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.PolicyKit1 was not provided by any .service files
Error registering authentication agent: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.PolicyKit1 was not provided by any .service files (g-dbus-error-quark, 2)
WARNING: UEFI ESP partition not detected or configured
See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
Waiting…                 [    \                                  ]
Message recipient disconnected from message bus without replying
```

I'm happy for any hints how to use this with Ubuntu Core systems. 

Thanks

hughsie, 2023-10-11:
I think it's all supposed to "just work". @superm1 perhaps?

superm1, 2023-10-11:
What version of ubuntu core are you trying?
I think the problem in this case is missing polkit support in ubuntu core.  It starts to show up in core22 I thought.
https://forum.snapcraft.io/t/enabling-polkitd-on-ubuntu-core/31295

CC @valentindavid

valentindavid, 2023-10-11:
There is an udisks2 snap ready to be published. But the publishing is blocked unfortunately.

If you want to build udisks2 yourself: https://github.com/snapcore/udisks2-snap

Polkit is not required. It does work without.

benjaminbeisel, 2023-10-12:
Thanks for your quick support

I'm using ubuntu core 22:
core22                  20230801
snapd                   2.60.3

I've build and installed the udisks2 snap from the repo above and connected the interfaces manually:
(is it correct that the udisks2-client connects to the slot in the udisks2 snap itself?)
```
$ sudo snap connections udisks2
Interface         Plug                      Slot                  Notes
dbus              -                         udisks2:udisks2-dbus  -
hardware-observe  udisks2:hardware-observe  :hardware-observe     manual
mount-observe     udisks2:mount-observe     :mount-observe        manual
polkit            udisks2:polkit            -                     -
udisks2           fwupd:udisks2             udisks2:udisks2       manual
udisks2           udisks2:udisks2-client    udisks2:udisks2       manual
```

The connections of fwupd now look like this:
```
$ sudo snap connections fwupd
Interface         Plug                    Slot               Notes
bluez             fwupd:bluez             -                  -
dbus              -                       fwupd:fwupd-dbus   -
fwupd             fwupd:fwupdmgr          fwupd:fwupd        manual
hardware-observe  fwupd:hardware-observe  :hardware-observe  -
modem-manager     fwupd:modem-manager     -                  -
network           fwupd:network           :network           -
opengl            fwupd:opengl            :opengl            -
polkit            fwupd:polkit            -                  -
raw-usb           fwupd:raw-usb           :raw-usb           -
shutdown          fwupd:shutdown          :shutdown          -
udisks2           fwupd:udisks2           udisks2:udisks2    manual
upower-observe    fwupd:upower-observe    -                  -
```

Unfortunately behaviour did not change:
```
$ sudo fwupd.fwupdmgr install /var/snap/fwupd/common/BIOSMEFW.CTL6R912.cab --allow-older

** (process:6559): WARNING **: 07:24:38.730: Unable to register authentication agent: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.PolicyKit1 was not provided by any .service files
Error registering authentication agent: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.PolicyKit1 was not provided by any .service files (g-dbus-error-quark, 2)
WARNING: UEFI ESP partition not detected or configured
See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
Waiting…                 [    \                                  ]
Message recipient disconnected from message bus without replying
```

I'm assuming the dbus interface either in the udisks2 snap or in the fwupd snap is still required to connect to some plug?

valentindavid, 2023-10-12:
No. The dbus slots are only here to enable dbus activation, that is starting the service only when something tries to connect to it. But the real dbus connection happens through the other interface. So nothing gets connected to the dbus slots.

I do not remember if upower is needed. This one should be published. I think I have always tested with it.

I suppose the fact that ESP is not found is the issue. I am not sure why this is so. I will have to test again to see if anything has changed.

superm1, 2023-10-12:
Have you restarted the service since installing the udisks snap?

benjaminbeisel, 2023-10-12:
Good hint! Device was rebooted in the meantime, now I get a permission denied:

```
$ sudo fwupd.fwupdmgr install /var/snap/fwupd/common/BIOSMEFW.CTL6R912.cab --allow-older

** (process:4509): WARNING **: 11:50:58.408: Unable to register authentication agent: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.PolicyKit1 was not provided by any .service files
Error registering authentication agent: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.PolicyKit1 was not provided by any .service files (g-dbus-error-quark, 2)
Decompressing…           [                                       ]
Failed to create file “/boot/efi/EFI/UpdateCapsule/fwupd-e84da0b1-c6b7-460d-bd26-3be4b10a2fc8.cap.VDTZC2”: Permission denied
```

valentindavid, 2023-10-12:
It my tests I only needed access to `/boot/efi/EFI/*/fw`. But it seems there are different types of capsules. This path is missing. I need to add it to the interface.

superm1, 2023-10-12:
Thanks @valentindavid !

@benjaminbeisel If possible can you share more about what device this is?  Is this x86 or aarch64?  We haven't seen a lot of CoD supporting systems, so I'm really interested in how it goes.

benjaminbeisel, 2023-10-13:
@superm1 It's a x86 64bit industrial control with an 11th Gen Intel i7

superm1, 2023-10-13:
Got it, thanks! Do you know who the BIOS vendor is? Might be able to tell from dmidecode if you don't know.

benjaminbeisel, 2023-10-26:
Hi @superm1,
sry for the late response, the BIOS vendor is Kontron Europe

benjaminbeisel, 2023-11-21:
Hi @valentindavid,

since the PR takes some time, I proceeded with my tests and installed the fwupd snap (1.9.9) in devmode. I'm not sure if I should create a new bug for the following...let me know if I should do so!
In devmode the installation is successful according to the print, but no update happens:

```
$ sudo fwupd.fwupdmgr install /var/snap/fwupd/common/BIOSMEFW.CTL6R912.cab --allow-older

** (process:7376): WARNING **: 06:52:53.929: Unable to register authentication agent: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.PolicyKit1 was not provided by any .service files
Error registering authentication agent: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.PolicyKit1 was not provided by any .service files (g-dbus-error-quark, 2)
Waiting…                 [***************************************]
Successfully installed firmware
An update requires a reboot to complete. Restart now? [y|N]: y
Connection to ****** closed by remote host.
```

After a reboot where I don't see any update happening, I reconnect the udisks2 interface again (this is necessary after each reboot?) and try to install the same binary without the --allow-older flag:

```
$ sudo snap connect fwupd:udisks2 udisks2:udisks2
$ sudo fwupd.fwupdmgr install /var/snap/fwupd/common/BIOSMEFW.CTL6R912.cab

** (process:4959): WARNING **: 06:55:01.819: Unable to register authentication agent: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.PolicyKit1 was not provided by any .service files
Error registering authentication agent: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.PolicyKit1 was not provided by any .service files (g-dbus-error-quark, 2)
Decompressing…           [    \                                  ]
Specified firmware is older than installed '0.0.912 < 0.0.913'
```

The installed firmware is still version 913.

I checked which settings in UEFI could possibly block an update and tried the following but without success:
Chipset -> PCH-IO Configuration -> Security Configuration -> BIOS Lock [Disabled]
Advanced -> PCH-FW Configuration -> Firmware Update Configuration -> FW Update [Enabled]
Advanced -> PCH-FW Configuration -> Firmware Update Configuration -> ME FW Image Re-Flash [Enabled]

Any idea how I could debug this further?

Thanks,
Benjamin

tobuh, 2023-11-28:
Hi @superm1 @valentindavid 

I joined @benjaminbeisel in investigating this bug. So I also tried to update UEFI via fwupd:

```
rexroot@ctrlX-CORE:~$ sudo fwupd.fwupdmgr install --allow-reinstall BIOSMEFW.CTL6R912.cab

** (process:5057): WARNING **: 09:04:56.032: Unable to register authentication agent: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The na                                   me org.freedesktop.PolicyKit1 was not provided by any .service files
Error registering authentication agent: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.PolicyKit1 was not provided                                    by any .service files (g-dbus-error-quark, 2)
Decompressing?           [                                       ]
shim sbatlevel for /snap/fwupd/5382/libexec/fwupd/efi/shimx64.efi.signed has a bricking update for entry grub (missing entry in current UEFI variab                                   le)
```

Remarks: I set OnlyTrusted=false, and we currently have an older shim installed in our gadget of this device.

Question: Why is fwupd trying to update shim? As we are in context of Ubuntu Core shim comes from gadget and shouldn't be updated by fwupd? We only want to update UEFI by COD

superm1, 2023-11-29:
> Questions: Why is fwupd trying to update shim

It's trying to add it's own embedded copy of shim into the system.  This is intentional because it may not be able to chainload from any other shim already there.  For example if it has a different embedded key.

It's to allow portability of the snap outside of Ubuntu desktop that it ships with a shim in the trust chain.

That safety check is there to make sure that you don't brick your system.  It looks like it works! :stuck_out_tongue:

superm1, 2023-12-15:
As this is fixed in snapd closing bug in fwupd.  You'll need to update to snapd version that has access to that directory to fix issue.

closed 2023-12-15 by superm1