fwupd/fwupd #6264: fwupd snap can't install files in TPM/FDE confined environment

superm1, 2023-10-12:
I set up Ubuntu 23.10 beta with TPM/FDE to experiment with it and find where things break.

I couldn't get firmware updates to work with the "inbuilt" fwupd, that's https://github.com/canonical/ubuntu-desktop-installer/issues/2371

But that aside I tried to set up the snap instead and experiment.
```
sudo apt purge fwupd
sudo snap install fwupd
```

I couldn't use `fwupdtool` to install from `~`, but I guess this is expected.
```
$ sudo fwupd.fwupdtool install-blob ~/isflash.bin
Loading…                 [ -                                     ]
Failed to open file “/home/test/isflash.bin”: Permission denied
```

So I tried to move it to a directory that _should_ work.
```
$ sudo mv isflash.bin /var/snap/fwupd/common/
$ sudo fwupd.fwupdtool install-blob /var/snap/fwupd/common/isflash.bin
Loading…                 [*****                                  ]ERROR:tcti:src/tss2-tcti/tcti-device.c:452:Tss2_Tcti_Device_Init() Failed to open specified TCTI device file /dev/tpmrm0: Operation not permitted
ERROR:tcti:src/tss2-tcti/tctildr-dl.c:154:tcti_from_file() Could not initialize TCTI file: libtss2-tcti-device.so.0
Loading…                 [*****                                  ]17:09:08.521 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:08.1/0000:c2:00.0/drm/card0/card0-DP-1/drm_dp_aux1: failed to open /dev/i2c-14: Operation not permitted
Loading…                 [*****                                  ]17:09:08.558 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:08.1/0000:c2:00.0/drm/card0/card0-DP-2/drm_dp_aux2: failed to open /dev/i2c-15: Operation not permitted
Loading…                 [*****                                  ]17:09:08.591 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:08.1/0000:c2:00.0/drm/card0/card0-DP-3/drm_dp_aux3: failed to open /dev/i2c-16: Operation not permitted
Loading…                 [******                                 ]17:09:08.623 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:08.1/0000:c2:00.0/drm/card0/card0-DP-4/drm_dp_aux4: failed to open /dev/i2c-17: Operation not permitted
Loading…                 [******                                 ]17:09:08.657 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:08.1/0000:c2:00.0/drm/card0/card0-DP-5/drm_dp_aux5: failed to open /dev/i2c-18: Operation not permitted
Loading…                 [******                                 ]17:09:08.689 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:08.1/0000:c2:00.0/drm/card0/card0-DP-6/drm_dp_aux6: failed to open /dev/i2c-19: Operation not permitted
Loading…                 [******                                 ]17:09:08.720 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:08.1/0000:c2:00.0/drm/card0/card0-DP-7/drm_dp_aux7: failed to open /dev/i2c-20: Operation not permitted
Loading…                 [******                                 ]17:09:08.752 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:08.1/0000:c2:00.0/drm/card0/card0-DP-8/drm_dp_aux8: failed to open /dev/i2c-21: Operation not permitted
Loading…                 [******                                 ]17:09:08.783 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:08.1/0000:c2:00.0/drm/card0/card0-eDP-1/drm_dp_aux0: failed to open /dev/i2c-13: Operation not permitted
Loading…                 [*******                                ]17:09:08.843 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:08.3/0000:c4:00.3/usb5/5-1/5-1:1.2/0003:046D:C52B.0004/hidraw/hidraw1: failed to open /dev/hidraw1
Loading…                 [********                               ]17:09:08.896 FuEngine             failed to add device /sys/devices/virtual/msr/msr0: failed to add device using on msr: failed to open /dev/cpu/0/msr: Operation not permitted
Writing…                 [************                           ]
0.      Cancel
1.      3743975ad7f64f8d6575a9ae49fb3a8856fe186f (SSD 980 PRO 1TB)
2.      a45df35ac0e948ee180fe216a5f703f32dda163f (System Firmware)
3.      362301da643102b9f38477387e2193e57abaa590 (UEFI dbx)
Choose device [0-3]: 2
Waiting…                 [************                           ]
Error opening directory “/run/mnt/ubuntu-seed”: Permission denied
```

That specific failure path is the ESP.
```
$ sudo fwupd.fwupdtool esp-list
Selected volume: /org/freedesktop/UDisks2/block_devices/nvme0n1p1
Error opening directory “/run/mnt/ubuntu-seed”: Permission denied
```

I'm not sure what's really wrong here, the udisks service is present.
```
$ snap connections fwupd
Interface         Plug                    Slot               Notes
bluez             fwupd:bluez             :bluez             -
dbus              -                       fwupd:fwupd-dbus   -
fwupd             -                       fwupd:fwupd        -
fwupd             fwupd:fwupdmgr          -                  -
hardware-observe  fwupd:hardware-observe  :hardware-observe  -
modem-manager     fwupd:modem-manager     :modem-manager     -
network           fwupd:network           :network           -
opengl            fwupd:opengl            :opengl            -
polkit            fwupd:polkit            :polkit            -
raw-usb           fwupd:raw-usb           :raw-usb           -
shutdown          fwupd:shutdown          :shutdown          -
udisks2           fwupd:udisks2           :udisks2           -
upower-observe    fwupd:upower-observe    :upower-observe    -
```

Here is the apparmor complaints:
```
$ sudo dmesg --clear
$ sudo fwupd.fwupdtool esp-list
Selected volume: /org/freedesktop/UDisks2/block_devices/nvme0n1p1
Error opening directory “/run/mnt/ubuntu-seed”: Permission denied
$ sudo dmesg
[ 2368.488198] audit: type=1326 audit(1697130864.727:188): auid=1000 uid=0 gid=0 ses=7 subj=snap.fwupd.fwupdtool pid=6409 comm="fwupdtool" exe="/snap/fwupd/5236/bin/fwupdtool" sig=0 arch=c000003e syscall=314 compat=0 ip=0x7fe2433b9a3d code=0x50000
[ 2368.540706] audit: type=1400 audit(1697130864.779:189): apparmor="DENIED" operation="open" class="file" profile="snap.fwupd.fwupdtool" name="/run/mnt/ubuntu-seed/" pid=6409 comm="fwupdtool" requested_mask="r" denied_mask="r" fsuid=0 ouid=0
```

@valentindavid ^

superm1, 2023-10-16:
@xnox^

superm1, 2023-12-15:
@jibel, @xnox do you have any updates on what you want to do here?  There is obviously a mismatch with how Ubuntu core's FDE sets up things versus what the fwupd snap interface works today.  I think it's up to you guys to decide how you want to resolve it though.

superm1, 2024-01-16:
@xnox Any updates on this issue?

xnox, 2024-03-12:
@superm1 I am no longer @canonical maybe somebody else can look into this.

Sysosmaster-LimoenGroen, 2024-09-26:
This Bug is still present in 24.04, which is an LTS with the install options for FDE.... 

updating your firmware is kinda critical for anyone who is required to use FDE to remain compliant with ISO standards... 

@canonical when is this issue going to get the attention it does, especially now @xnox is no longer in your employment...

xnox, 2024-09-26:
@mwhudson

superm1, 2024-10-07:
(@bboozzoo if you're not aware of this you should take a look at it and related things linked to it too as it pertains to FDE)

bboozzoo, 2024-10-07:
Ah yes. This is kind of expected and the moment. Unfortunately the lack of integration makes it really tricky if one wants to use fwupd to update anything UEFI related on an FDE Ubuntu system. I'll keep this one on my radar. In any case https://github.com/fwupd/fwupd/pull/7861 should be a step in the right direction.

hughsie, 2024-11-11:
Okay, I'll close this issue; we'll need https://github.com/fwupd/fwupd/pull/7861 at the very least.

superm1, 2025-01-22:
There is some downstream confusion in other bug trackers that this being closed means that fwupd and snapd get along as it pertains to FDE now.  #7861 was the start of work for fwupd and snapd to be communicating about updates.

There will need to be more work for FDE.

I'm going to re-open this as a tracker.

hughsie, 2025-05-04:
@superm1 still valid?

superm1, 2025-05-04:
Yeah I would say this is still valid.

bboozzoo, 2025-05-06:
Just a small update, snapd 2.68.4 has finally reached the stable channel, so most cases where snapd is relevant in the dbx update chain are now covered.

tete17, 2025-05-06:
This is still a problem for me
```
miguel@miguel-XPS-13-9310:~$ which fwupd.fwupdmgr 
/snap/bin/fwupd.fwupdmgr
miguel@miguel-XPS-13-9310:~$ sudo fwupd.fwupdmgr upgrade
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 3.25.0 to 3.29.0?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • This release contains security updates as disclosed in the Dell            ║
║ Security Advisories.                                                         ║
║                                                                              ║
║ XPS 13 9310 must remain plugged into a power source for the duration of the  ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: 
Scheduling…              [                                       ]
failed to write-firmware: Failed to create '/run/mnt/ubuntu-seed/EFI/fwupd/fw': Permission denied
miguel@miguel-XPS-13-9310:~$ snap --version
snap    2.68.4
snapd   2.68.4
series  16
ubuntu  24.04
kernel  6.8.0-58-generic
```

superm1, 2025-05-06:
I think the work for snapd mentioned above only unblocks dbx updates, it _does not_ unblock capsule updates.

tete17, 2025-05-06:
I will be honest 100% firmware noob here. You take my out of my beloved containers and I am lost.
What is the difference here? Anything I can read to catch up?

hughsie, 2026-01-13:
@bboozzoo any ideas?  From an upstream PoV it does at least feel like Ubuntu is drifting away from the other distros -- it's getting increasingly hard to justify all the snap special cases. It would be **so much less code** to just go back to shipping fwupd as a native `.deb` rather than a snap package.