fwupd/fwupd #6266: Building 1.9.6 version on openSUSE fails, because location of amdgpu_drm.h

weberhofer, 2023-10-14:
Building  version 1.9.6 on openSUSE Tumbleweed fails because the `amdgpu_drm.h` is located in `libdrm/amdgpu_drm.h` instead of  `drm/amdgpu_drm.h`.

The location is currently hardcoded in `meson.build` and `plugins/amd-gpu/fu-amd-gpu-device.c`.

Would be great, if the location could be made flexible.

superm1, 2023-10-14:
Can you please share the specific build log with the failure?

We made some changes in main related to detection for that header, can you try building main too?

superm1, 2023-10-14:
Actually I think I see a way to handle this more cleanly.  Can you please test that PR on top of main?

superm1, 2023-10-15:
I realized that's completely wrong.  We're looking for **kernel headers** not **libdrm headers**; the stuff in main is as it's intended.  

```
/usr/include$ find -name "amdgpu_drm*.h"
./drm/amdgpu_drm.h
./libdrm/amdgpu_drm.h
```

Maybe you're missing kernel headers in your build process.

dataCobra, 2023-10-16:
I'm trying to build and package fwupd 1.9.6 as well for voidlinux. We also use libdrm as the location.

To make sure these headers are correct I symlinked /usr/include/libdrm to /usr/include/drm and checked the tests.

dataCobra, 2023-10-16:
I do have kernel-headers installed during installation.

If you mean the uapi drm headers they are located here on voidlinux:
`/usr/src/kernel-headers-6.5.7_1/include/uapi/drm/amdgpu_drm.h`

superm1, 2023-10-16:
You couldn't build a kernel module against that directory though, that seems wrong.

It definitely shouldn't have uapi in the installed path.

For those headers Is there a missing symlink? Or missing install step?

superm1, 2023-10-16:
If it helps on Debian and Ubuntu this is part of the `linux-libc-dev` package.
```
$ dpkg -S /usr/include/drm/amdgpu_drm.h
linux-libc-dev:amd64: /usr/include/drm/amdgpu_drm.h
```

weberhofer, 2023-10-16:
I have included the patch in the, but it still fails; in `plugins/amd-gpu/fu-amd-gpu-device.c` you'll find
```c
#include <drm/amdgpu_drm.h>
```
which on openSUSE should be
```c
#include <libdrm/amdgpu_drm.h>
```
that is added to the build environment via `BuildRequires:  pkgconfig(libdrm)`.

superm1, 2023-10-16:
@weberhofer please discard the patch.  It's wrong. 

You're pulling the wrong headers.  You need kernel headers not libdrm headers.

Kerrung, 2023-10-16:
@superm1 I have the same problem on Arch Linux. And I have the linux-headers package installed. I tried building commit [2a8aec8384040ee8d29ced64bf31fedce62c054596](https://github.com/fwupd/fwupd/commit/2a8aec83840ee8d29ced64bf31fedce62c054596)

[A terminal log.txt](https://github.com/fwupd/fwupd/files/12921129/log.txt)
[meson-log.txt](https://github.com/fwupd/fwupd/files/12921159/meson-log.txt)
[PKGBUILD.zip](https://github.com/fwupd/fwupd/files/12921165/PKGBUILD.zip)
[image]

superm1, 2023-10-16:
@Kerrung can you please ping the owner for [linux-api-headers](https://archlinux.org/packages/core/any/linux-api-headers/) to look at this issue for Arch?  It seems like that's the package that should be carrying this header from the kernel but I don't see it.

superm1, 2023-10-16:
@weberhofer looking at OpenSUSE, I think it should be coming through [linux-glibc-devel](https://software.opensuse.org/download.html?project=openSUSE%3AFactory&package=linux-glibc-devel).  Please check with the owner for this package.

Kerrung, 2023-10-16:
> @Kerrung can you please ping the owner for [linux-api-headers](https://archlinux.org/packages/core/any/linux-api-headers/) to look at this issue for Arch? It seems like that's the package that should be carrying this header from the kernel but I don't see it.

The only thing I can do is write a bug report on the Arch bug tracker.

Kerrung, 2023-10-16:
> @Kerrung can you please ping the owner for [linux-api-headers](https://archlinux.org/packages/core/any/linux-api-headers/) to look at this issue for Arch? It seems like that's the package that should be carrying this header from the kernel but I don't see it.

https://archlinux.org/packages/core/x86_64/linux-headers/
Or maybe this package is needed?

superm1, 2023-10-16:
> https://archlinux.org/packages/core/x86_64/linux-headers/
> Or maybe this package is needed?

So that package does contain an `amdgpu_drm.h`, but it's not in any include path.

```
usr/lib/modules/6.5.7-arch1-1/build/include/uapi/drm/amdgpu_drm.h
```

I'm not sure how any software can be expected to discover it in a path like that.  It would mean that `/usr/lib/modules/6.5.7-arch1-1/build/include/uapi`  needs to be added into the search path.

superm1, 2023-10-16:
@hughsie I guess reading through https://cgit.freedesktop.org/drm/libdrm/tree/include/drm/README it looks like the copy included in libdrm _is_ a kernel copy.  Maybe we should just duplicate it in fwupd too then.

Kerrung, 2023-10-16:
> So that package does contain an `amdgpu_drm.h`, but it's not in any include path.
>
> ```
> usr/lib/modules/6.5.7-arch1-1/build/include/uapi/drm/amdgpu_drm.h
> ```
>
> I'm not sure how any software can be [...]

Can you fix it?

superm1, 2023-10-16:
> @hughsie I guess reading through https://cgit.freedesktop.org/drm/libdrm/tree/include/drm/README it looks like the copy included in libdrm _is_ a kernel copy. Maybe we should just duplicate it in fwupd too then.

or we depend on `libdrm` (what my PR did).

Here's how libdrm does it.
1. They sync all the files here: https://gitlab.freedesktop.org/mesa/drm/-/tree/main/include/drm?ref_type=heads
2. They make a top level include: https://gitlab.freedesktop.org/mesa/drm/-/blob/main/meson.build?ref_type=heads#L284
3. They include it in the whole project build: https://gitlab.freedesktop.org/mesa/drm/-/blob/main/meson.build?ref_type=heads#L305

We can do the same, but it trips up our license and formatting checker, so we'd need to set the policy to ignore that directory.

Kerrung, 2023-10-16:
@superm1 I just tried this https://github.com/fwupd/fwupd/pull/6269. There are changes. This time the compilation was at least able to start, but still ended up crashing. Here are the new logs.

[log.txt](https://github.com/fwupd/fwupd/files/12922239/log.txt)
[meson-log.txt](https://github.com/fwupd/fwupd/files/12922243/meson-log.txt)
[PKGBUILD.zip](https://github.com/fwupd/fwupd/files/12922247/PKGBUILD.zip)

superm1, 2023-10-16:
Try with the new changes I just published.

Kerrung, 2023-10-16:
> Try with the new changes I just published.

Now fixed. Thanks!

Kerrung, 2023-10-16:
> Try with the new changes I just published.

@superm1 The compilation now works fine without crashing. But I still don't see my Radeon GPU in the list of devices displayed by FWUPD afterwards.

superm1, 2023-10-16:
> > Try with the new changes I just published.
> 
> @superm1 The compilation now works fine without crashing. But I still don't see my Radeon GPU in the list of devices displayed by FWUPD afterwards.

What kernel version? Can you share output for 

sudo fwupdtool get-devices --verbose

Kerrung, 2023-10-16:
> > > Try with the new changes I just published.
> > 
> > 
> > @superm1 The compilation now works fine without crashing. But I still don't see my Radeon GPU in the list of devices displayed by FWUPD afterwards.
> 
> What kernel version? Can you share output for
> 
> sudo fwupdtool get-devices --verbose

1) This version: https://archlinux.org/packages/core/x86_64/linux/

2) [log.txt](https://github.com/fwupd/fwupd/files/12922633/log.txt)

superm1, 2023-10-17:
I see it listed in your log.

```
â”‚   
â”œâ”€D09001 Lexa XL A0 GDDR5 128bit 2GB 214e/300m:
â”‚     Device ID:          ce4c74a5188d5b9cdb1e72ed32dad2d313c1c999
â”‚     Current version:    015.050.002.001.000000
â”‚     Vendor:             Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002, PCI:0x1022)
â”‚     GUIDs:              4daaf54e-4446-5b25-8979-7fc017051118 â† PCI\VEN_1002&DEV_699F
â”‚                         d2683e36-8462-55c4-9851-777449b36eda â† PCI\VEN_1002&DEV_699F&SUBSYS_10020B04
â”‚                         e4c836f2-91e6-5e6d-943e-7d35d4ed1472 â† AMD\xxx-xxx-xxx
â”‚     Device Flags:       â€¢ Cryptographic hash verification is available
```

Kerrung, 2023-10-17:
> I see it listed in your log.
>
> ```
> â”‚
> â”œâ”€D09001 Lexa XL A0 GDDR5 128bit 2GB 214e/300m:
> â”‚     Device ID:          ce4c74a5188d5b9cdb1e72ed32dad2d313c1c999
> â”‚     Current version: [...]

And how can I install firmware updates on my GPU?
sudo fwupdmgr update
It doesn't show this in the device list, at least devices not available for update.

`Devices with no available firmware updates: 
 • ADATA SX8200PNP
 • HD103SJ
 • HD204UI
 • PLEXTOR PX-128M5Pro
 • WDC WD6400AAKS-00A7B0
Devices with the latest available firmware version:
 • Unifying Receiver
 • Unifying Receiver`

superm1, 2023-10-17:
Your GPU doesn't support updates, this is only for Navi 3x and newer.

commit 96590ffd (2023-10-14): trivial: discover libdrm, and ensure it has amdgpu support using pkgconfig

Fixes: #6266
commit 10362463 (2023-10-16): trivial: discover libdrm, and ensure it has amdgpu support using pkgconfig

Fixes: #6266
commit 35114808 (2023-10-16): trivial: discover libdrm, and ensure it has amdgpu support using pkgconfig

Fixes: #6266
PR #6269 (2023-10-17): trivial: discover libdrm, and ensure it has amdgpu support using pkgconfig
I think this is the cleaner solution - it means we never need to be updating the header copy locally if we keep using future ioctls.

Fixes: #6266

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation