fwupd/fwupd #6274: After fwupd-main of T99W373 device, the carrier config seem to be not expected

Fanice-luo, 2023-10-19:
Describe the question
When I use .cab to update/downgrade T99W373 device, finished with "Successfully installed firmware", but sometimes the carrier config seem to be not expected(GCF, DF, CC are all possible).

Wrong situation 1:
  Hardware  |            manufacturer: foxconn
            |                   model: Qualcomm Snapdragon X62
            |       firmware revision: FDE.F0.3.0.1.1.GC.002
            |                          030
            |          carrier config: GCF
            | carrier config revision: 0A000802
            |            h/w revision: Qualcomm Snapdragon X62

 sudo mmcli -m 4 --firmware-status 
  ----------------------
  Firmware |     method: mbim-qdu
           | device ids: PCI\VID_105B&PID_E0D9&REV_0000&CARRIER_GCF
           |             PCI\VID_105B&PID_E0D9&REV_0000
           |             PCI\VID_105B&PID_E0D9
           |             PCI\VID_105B
           |    version: FDE.F0.3.0.1.1.CC.001.030

Wrong situation 2:
  Hardware  |            manufacturer: foxconn
            |                   model: Qualcomm Snapdragon X62
            |       firmware revision: FDE.F0.3.0.1.1.GC.002
            |                          032
            |          carrier config: CMCC
            | carrier config revision: 0A002001
            |            h/w revision: Qualcomm Snapdragon X62
sudo mmcli -m 4 --firmware-status 
  ----------------------
  Firmware |     method: mbim-qdu
           | device ids: PCI\VID_105B&PID_E0D9&REV_0000&CARRIER_CMCC
           |             PCI\VID_105B&PID_E0D9&REV_0000
           |             PCI\VID_105B&PID_E0D9
           |             PCI\VID_105B
           |    version: FDE.F0.3.0.1.1.CC.001.032

But when I modify fu-mm-decvice.c of g_timeout_add_seconds(20, fu_mm_device_get_firmware_version_mbim_timeout, &ctx)
all will be expected.Could you offer me some advice?
  Hardware  |            manufacturer: foxconn
            |                   model: Qualcomm Snapdragon X62
            |       firmware revision: FDE.F0.3.0.1.1.CC.001
            |                          030
            |          carrier config: CMCC
            | carrier config revision: 0A002001
            |            h/w revision: Qualcomm Snapdragon X62
 sudo mmcli -m 2 --firmware-status 
  ----------------------
  Firmware |     method: mbim-qdu
           | device ids: PCI\VID_105B&PID_E0D9&REV_0000&CARRIER_CMCC
           |             PCI\VID_105B&PID_E0D9&REV_0000
           |             PCI\VID_105B&PID_E0D9
           |             PCI\VID_105B
           |    version: FDE.F0.3.0.1.1.CC.001.030

Best wishes, Fanice

hughsie, 2023-10-19:
> But when I modify fu-mm-decvice.c of g_timeout_add_seconds(20, fu_mm_device_get_firmware_version_mbim_timeout, &ctx) all will be expected

That seems like a sensible thing to do. Could we make it something like 30 seconds?

Fanice-luo, 2023-10-19:
I'm afraid 30s is too long, because sometimes the expected carrier config can also be obtained at 10s. But the probability is extremely low.

Fanice-luo, 2023-10-20:
Hi @aleksander0m What would cause inconsistency between MM's carrier config and fwupd's carrier config? How should I debug it?

Fanice-luo, 2023-10-20:
Hi @hughsie For fu_mm_device_get_firmware_version_mbim_timeout is 10s, mbim-proxy log can MCFG MCFG switching sequence  is DF, GC, CMCC,here is something from mbim-proxy log.
[image]

And  fu_mm_device_get_firmware_version_mbim_timeout is 20s, it switching CMCC directly.
[image]

aleksander0m, 2023-10-24:
> Hi @aleksander0m What would cause inconsistency between MM's carrier config and fwupd's carrier config? How should I debug it?

What is the inconsistency here? Is the fwupd plugin loading one firmware version that is different to the one loaded by MM? Can you elaborate?

The "GC" carrier config is the one the device would boot with if there is no other carrier specific configuration. The "CC" carrier config would be the one associate to the CMCC operator.

ModemManager has built-in support to automatically select carrier config (using QMI PDC) whenever the SIM card operator code is read. Once this happens, the modem object is removed and it is reprobed from scratch once the new carrier config has been selected. We did this for older Foxconn models, e.g. the DW5821e and others. I don´t think we added MM-triggered automatic carrier selection for the T99W373, although I may be mistaken.

Another possibility could be that it is the modem itself switching carrier config automatically, and so it depends when you query the firmware version and you may or may not get the latest carrier config reported. If this is happening, we have a problem, because right now ModemManager does not listen to notifications reporting automatic carrier config switches (don´t even know if they are being notified at all).

Could you elaborate a bit what is happening here?

SlarkXiao, 2023-10-24:
Hi @aleksander0m ,
Yes, you are right. T99W373(SDX65) and T99W175(SDX55) won't call the MM-triggered automatic carrier selection since we did it in firmware side. But now, just like you said, the problem is we may not get the latest carrier config from device side if the time slot is not proper. 
So do we have any fix mechanism (like a retry) if the carrier is "DF" or "GC"?  AFAIK, this switch may take no more than 5 seconds from DF(->GC)->CC.

aleksander0m, 2023-10-24:
No, this is not solved with retrying to reload, or at least that is not what I would recommend to do in the first attempt.

If the module is switching carrier config internally, is that notified to the host in any way? Maybe the modem can send a specific QMI PDC indication when the switch happens, and MM could monitor that?

MM should re-probe the device completely whenever a carrier config switch is detected.

SlarkXiao, 2023-11-10:
We just get feedback from our internal team, and they said the client needs to register the indication by QMI_PDC_INDICATION_REGISTER , then they will receive the indication of QMI_PDC_CONFIG_CHANGE_IND.
So if we want to implement this, shall we add 2 QMI message in libqmi?

SlarkXiao, 2023-11-29:
@aleksander0m As I said in last, I think that solution may also don't work even if we register this indication in ModemManager side. We can't ensure that this register point would be earlier than the MCFG switch point in last boot.
We may have 2 situation as below:
a. fwupd 1st reboot->fwupd 2nd reboot->MM initial->register PDC indication->MCFG switch in firmware side->received PDC indication->MM re-probe
b. fwupd 1st reboot->fwupd 2nd reboot->MM initial->MCFG switch in firmware side->register PDC indication  //MCFG mismatch still exist
Do you have any idea?

closed 2024-11-11 by hughsie

kerixi781, 2025-03-24:
> Yes, you are right. T99W373(SDX65) and T99W175(SDX55) won't call the MM-triggered automatic carrier selection since we did it in firmware side. But now, just like you said, the problem is we may not [...]

You on the wrong way. New famely 5G card have SDX65 and SDX62, its mean T99W368 - SDX65, T99W373 - SDX62!!!