fwupd/fwupd #6278: Firmware can not be updated in legacy BIOS mode, wiki instructions don't work

Dan-Essin, 2023-10-22:
**Describe the bug**
On Debian 12, 6.1.0-13-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.55-1 (2023-09-29) x86_64 GNU/Linux

https://github.com/fwupd/fwupd/wiki/PluginFlag:legacy-bios
says:
"The WARNING: Firmware can not be updated in legacy BIOS mode will show if you are running on a system with only legacy BIOS or most commonly, a UEFI BIOS in "CSM compatibility mode". If this is detected then UEFI runtime functions are not available. This includes the ability to schedule firmware updates.

This warning can be ignored if UEFI firmware updates are not desired. It can be disabled by adding

DisabledPlugins=test;invalid;bios

to /etc/fwupd/daemon.conf"

I have made this edit and restarted fwupd, also rebooted.

It has no effect. running fwupdmgr get-upgrades gives the same message:
"WARNING: UEFI firmware can not be updated in legacy BIOS mode
See https://github.com/fwupd/fwupd/wiki/PluginFlag:legacy-bios for more information."

**Expected behavior**
Why post instructions in the wiki that don't work?

**fwupd version information**
fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.12
compile   com.hughsie.libxmlb           0.3.10
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.fwupd         1.8.12
runtime   org.kernel                    6.1.0-13-amd64

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

No idea how/when it got installed. solaar maybe? Otherwise it was part of the distro
<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

fwupdmgr get-devices --show-all-devices
WARNING: UEFI firmware can not be updated in legacy BIOS mode
See https://github.com/fwupd/fwupd/wiki/PluginFlag:legacy-bios for more information.
TOSHIBA PORTEGE R705
│
├─BRCM Bluetooth Controller BCM2070:
│     Device ID:          2ce35bbff078550994187705f561229ae850db93
│     Current version:    4.73
│     Vendor:             Broadcom Corp (USB:0x0930)
│     Serial Number:      B482FED8B18D
│     GUIDs:              28d63995-28e3-5e00-8a23-007c037a140e ← USB\VID_0930&PID_0213
│                         98bee73d-4037-597c-aff7-b4ad79cb68b4 ← USB\VID_0930&PID_0213&REV_0473
│     Device Flags:       • Updatable
│   
├─Core Processor Integrated Graphics Controller:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    02
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              b36a8fab-195a-5871-be0b-9aefd1837d6c ← PCI\VEN_8086&DEV_0046
│                         f236a225-d4c4-5ead-b81b-18e52dabfed9 ← PCI\VEN_8086&DEV_0046&REV_02
│                         06e39351-4a0b-55b0-bf9b-e37c4884e0bf ← PCI\VEN_8086&DEV_0046&SUBSYS_11790007
│                         23f90729-b68a-5da3-aa83-3b99fb73d8b1 ← PCI\VEN_8086&DEV_0046&SUBSYS_11790007&REV_02
│                         620e5d7a-bb2b-5842-8423-5f1419c428cd ← PCI\VEN_8086&DEV_0046&REV_00
│                         46df01c1-2967-5492-8fd6-67214ce3ed95 ← PCI\VEN_8086&DEV_0046&SUBSYS_11790007&REV_00
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Core™ i3 CPU M 350 @ 2.27GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x00000007
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         c0f6363a-50f2-5d01-beca-901d771eed67 ← CPUID\PRO_0&FAM_06&MOD_25
│                         42a5c023-577e-5e6b-8d6c-123dd3e6fc92 ← CPUID\PRO_0&FAM_06&MOD_25&STP_5
│     Device Flags:       • Internal device
│   
├─SSD 840 PRO Series:
│     Device ID:          602b0a6cc821d155208724f0e22f8d111542b74c
│     Summary:            ATA drive
│     Current version:    DXM06B0Q
│     Vendor:             Samsung (ATA:0x144D, OUI:002538)
│     Serial Number:      S1ATNSADC41251F
│     GUIDs:              3c8299ff-761a-5104-9d0a-a20f422d58d4 ← IDE\Samsung_SSD_840_PRO_Series______________DXM06B0Q
│                         d2ea72d7-ca60-599e-8a73-8efc96f64602 ← IDE\0Samsung_SSD_840_PRO_Series______________
│                         13406aaf-4cb7-53a9-97a4-4341a6d585e3 ← Samsung SSD 840 PRO Series
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─Unifying Receiver:
      Device ID:          41b3929e3920d086c5ecae8db4b55d57cf6a2f2a
      Summary:            Miniaturised USB wireless receiver
      Current version:    RQR24.00_B0018
      Bootloader Version: BOT03.00_B0006
      Vendor:             HIDRAW:0x046D|USB:0x046D
      Install Duration:   30 seconds
      GUIDs:              cc4cbfa9-bf9d-540b-b92b-172ce31013c1
                          279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
                          fcf55bf5-767b-51ce-9c17-f6f538c4ee9f ← HIDRAW\VEN_046D&DEV_C52B&REV_00
      Device Flags:       • Updatable
                          • Supported on remote server
                          • Unsigned Payload
    

</details>

**Additional questions**

- Operating system and version: see above
- Have you tried rebooting? Yes
- Is this a regression? What is a regression?

hughsie, 2023-10-25:
What does `fwupdmgr get-plugins` say?

G-i-o, 2023-12-06:
The above instructions did work for my @Dan-Essin .
However, the link for the wiki given by the **fwdupdmgr** ([https://github.com/fwupd/fwupd/wiki/PluginFlag:legacy-bios]()), shows an empty page, rather than the instructions above.

Luckily, I have searched for previously logged issue and found this.

hughsie, 2023-12-07:
@G-i-o I've added back the wiki page, it was deleted by @utrtrtrtrtrtttr for some reason. I suspect it's just a bot.

closed 2024-11-11 by hughsie