fwupd/fwupd #6282: Meson build error due to version is wrong

sam412081go, 2023-10-24:
**Describe the question**
I am setting another ubuntu system laptop  for LVFS plugin development, and encount a problem.
I use 
`"sudo meson ../ -Dsystemd=false -Ddocs=none -Dintrospection=false -Dudevdir=/tmp --prefix=/home/ut2204/fwupd_test2/fwupd-exe"` for meson build, but shows error message as below:
 
`Configuring 35_fwupd using configuration
Run-time dependency fwupd-efi found: NO (tried pkgconfig and cmake)
Looking for a fallback subproject for the dependency fwupd-efi

Executing subproject fwupd-efi 

../subprojects/fwupd-efi/meson.build:1:0: ERROR: Meson version is 0.61.2 but project requires >=0.62.0

A full log can be found at /home/ut2204/fwupd_test2/fwupd/build/meson-logs/meson-log.txt
`
My meson was originally 0.61.2, then I installed the new version of 0.63.2 by `pip3 install meson==0.63.2`

After installation, I check meson version again. It shows 0.63.2.
`ut2204@UH:~/fwupd_test2/fwupd$ meson --version
0.63.2
`

But meson build still shows 
`../subprojects/fwupd-efi/meson.build:1:0: ERROR: Meson version is 0.61.2 but project requires >=0.62.0`

How can I fix it?

**fwupd version information**
Please provide the version of the daemon and client if applicable to your current installation of `fwupd`.
I downloaded fwupd source code on 24/Oct/2023.

```shell
fwupdmgr --version
```

hughsie, 2023-10-24:
Can you try a `meson --wipe` perhaps?

superm1, 2023-10-24:
I suspect it's installed as user but not as root.  Needs to be installed as both

sam412081go, 2023-10-25:
@hughsie @superm1 Thanks for your help.It's ok now. I use 'sudo pip3 install meson==0.63.2' install as root.

closed 2023-10-25 by sam412081go

superm1, 2023-10-25:
> @hughsie @superm1 Thanks for your help.It's ok now. I use 'sudo pip3 install meson==0.63.2' install as root.

Thanks.
I guess we might want some other improvements for contrib/setup in the future if this keeps being a problem that comes up.