fwupd/fwupd #6290: Fail to do end-to-end testiing

FrancisLeeKinetic, 2023-10-26:
**Describe the question**
i'm engineer at Kinet-ic responsible for fwupd. I'm relatively new to fwupd, so I'd like to start by familiarizing myself with it through testing. Currently, I've been successful in offline testing using fwupdtool, but I'm facing difficulties with end-to-end online testing. 

**fwupd version information**
I've cloned source code from [](https://github.com/fwupd/fwupd.git) and checkout branch to wip/kinetic/latest that is a  per-existing branch. Then, I followed build step to build local fwupd successfully  and finish to install.

OS: Ubuntu 22.04

```shell
$fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.3
compile   com.hughsie.libxmlb           0.3.6
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.0
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.fwupd         1.9.3
runtime   org.kernel                    6.5.7-060507-generic
```
Check plugins
```shell
$fwupdmgr get-plugins 
......
kinetic_dp:
  Flags:                • Enabled
......
```
Check devices
```shell
$fwupdmgr get-devices
├─KTM50X0:
│     Device ID:          d762543f8c20f636e6fff031a000078d3e10c600
│     Summary:            DisplayPort Protocol Converter
│     Current version:    0.013.09
│     Vendor:             Kinetic Technologies (DRM_DP_AUX_DEV:0x329A, DRM_DP_AUX_DEV:0x8086)
│     GUIDs:              e08317e3-a3c8-57b2-a85c-fd3f884bed94 ← MST\VEN_8086&FAM_JAGUAR
│                         23c90ea0-0bdf-5abe-88b4-8f46e66eecec ← MST\VEN_8086&FAM_JAGUAR&CID_KTM50X0
│     Device Flags:       • Updatable
```
**End-to-end Test**
I attempted to downgrade device version to v0.013.09 and uploaded the latest version(0.013.18) FW to LVFS. Following the end-to-end step

1. Download kinet_ic-embargo.conf
2. Copy the file to /etc/fwupd/remote.d/
3. Execute 
```shell
$fwupdmgr refresh --force
WARNING: This package has not been validated, it may not work properly.
Updating lvfs
Downloading…             [           |                           ]
Successfully downloaded new metadata: 1 local device supported
```
4. Execute 
```shell
$fwupdmgr update
WARNING: This package has not been validated, it may not work properly.
Devices with no available firmware updates: 
 • KTM50X0
 • SSDSCKKW256G8
 • System Firmware
Devices with the latest available firmware version:
 • UEFI dbx

``` 
Is it because the version of fwupd is too old, causing it not to fit in LVFS, or did I miis any steps?

hughsie, 2023-10-26:
The firmware you uploaded on the LVFS targets a device with GUID  `56c9ce8f-15d4-5c4a-b63a-44189c42cd8e`  and your GUID seems to be `23c90ea0-0bdf-5abe-88b4-8f46e66eecec` -- the GUID is what matches the device to the correct firmware.

FrancisLeeKinetic, 2023-10-27:
Hi hughsie,
I modified GUID to ```23c90ea0-0bdf-5abe-88b4-8f46e66eecec``` into FW and upload to LVFS, it still showed
 ```$fwupdmgr update
WARNING: This package has not been validated, it may not work properly.
Devices with no available firmware updates: 
 • KTM50X0
 • SSDSCKKW256G8
 • System Firmware
Devices with the latest available firmware version:
 • UEFI dbx
```
Do you have any other suggections to let it work?

superm1, 2023-10-27:
> Hi hughsie,
> I modified GUID to ```23c90ea0-0bdf-5abe-88b4-8f46e66eecec``` into FW and upload to LVFS, it still showed
>  ```$fwupdmgr update
> WARNING: This package has not been validated, it may not [...]

Can you show `fwupdmgr get- releases` output after you do `fwupdmgr refresh`?

FrancisLeeKinetic, 2023-10-27:
```shell
$ fwupdmgr refresh --force --verbose
(fwupdmgr:46476): GLib-DEBUG: 12:27:04.789: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:46476): GLib-GIO-DEBUG: 12:27:04.805: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:46476): dconf-DEBUG: 12:27:04.805: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:46476): dconf-DEBUG: 12:27:04.805: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:46476): dconf-DEBUG: 12:27:04.805: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:46476): dconf-DEBUG: 12:27:04.805: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:46476): dconf-DEBUG: 12:27:04.805: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:46476): GLib-GIO-DEBUG: 12:27:04.805: _g_io_module_get_default: Found default implementation gnome (GProxyResolverGnome) for ‘gio-proxy-resolver’
(fwupdmgr:46476): dconf-DEBUG: 12:27:04.809: watch_established: "/system/proxy/" (establishing: 1)
(fwupdmgr:46476): dconf-DEBUG: 12:27:04.810: watch_established: "/system/proxy/http/" (establishing: 1)
(fwupdmgr:46476): dconf-DEBUG: 12:27:04.810: watch_established: "/system/proxy/https/" (establishing: 1)
(fwupdmgr:46476): dconf-DEBUG: 12:27:04.810: watch_established: "/system/proxy/ftp/" (establishing: 1)
(fwupdmgr:46476): dconf-DEBUG: 12:27:04.810: watch_established: "/system/proxy/socks/" (establishing: 1)
(fwupdmgr:46476): Fwupd-DEBUG: 12:27:04.815: Emitting ::status-changed() [idle]
WARNING: This package has not been validated, it may not work properly.
Updating lvfs
Fwupd-INFO: 12:27:04.822: downloading https://cdn.fwupd.org/downloads/firmware.xml.xz.jcat
(fwupdmgr:46476): Fwupd-DEBUG: 12:27:04.822: Emitting ::status-changed() [downloading]
Downloading…             [                              /        ]Fwupd-INFO: 12:27:06.026: download progress: 100%
(fwupdmgr:46476): Fwupd-DEBUG: 12:27:06.027: Emitting ::status-changed() [idle]
Fwupd-INFO: 12:27:06.027: status-code was 200
Fwupd-INFO: 12:27:06.029: changing metadata URI from https://cdn.fwupd.org/downloads/firmware.xml.xz to https://cdn.fwupd.org/downloads/firmware-06826-stable.xml.xz
Fwupd-INFO: 12:27:06.029: metadata signature of lvfs is unchanged, skipping
```
```shell
$ fwupdmgr get-releases --verbose
(fwupdmgr:46544): GLib-DEBUG: 12:29:01.622: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:46544): GLib-GIO-DEBUG: 12:29:01.642: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:46544): dconf-DEBUG: 12:29:01.642: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:46544): dconf-DEBUG: 12:29:01.642: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:46544): dconf-DEBUG: 12:29:01.642: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:46544): dconf-DEBUG: 12:29:01.642: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:46544): dconf-DEBUG: 12:29:01.642: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:46544): GLib-GIO-DEBUG: 12:29:01.642: _g_io_module_get_default: Found default implementation gnome (GProxyResolverGnome) for ‘gio-proxy-resolver’
(fwupdmgr:46544): dconf-DEBUG: 12:29:01.646: watch_established: "/system/proxy/" (establishing: 1)
(fwupdmgr:46544): dconf-DEBUG: 12:29:01.646: watch_established: "/system/proxy/http/" (establishing: 1)
(fwupdmgr:46544): dconf-DEBUG: 12:29:01.646: watch_established: "/system/proxy/https/" (establishing: 1)
(fwupdmgr:46544): dconf-DEBUG: 12:29:01.646: watch_established: "/system/proxy/ftp/" (establishing: 1)
(fwupdmgr:46544): dconf-DEBUG: 12:29:01.647: watch_established: "/system/proxy/socks/" (establishing: 1)
(fwupdmgr:46544): Fwupd-DEBUG: 12:29:01.650: Emitting ::status-changed() [idle]
WARNING: This package has not been validated, it may not work properly.
Selected device: UEFI dbx
  AppstreamId:          org.linuxfoundation.dbx.x64.firmware
  ReleaseId:            35287
  RemoteId:             lvfs
  Summary:              UEFI Secure Boot Forbidden Signature Database
  Description:          <p>Insecure versions of the Microsoft Windows boot manager affected by Black Lotus were added to the list of forbidden signatures due to a discovered security problem.This updates the dbx to the latest release from Microsoft.</p><p>Before installing the update, fwupd will check for any affected executables in the ESP and will refuse to update if it finds any boot binaries signed with any of the forbidden signatures.Applying this update may also cause some Windows install media to not start correctly.</p>
  Version:              371
  Filename:             DBXUpdate-20230509-x64.cab
  Protocol:             org.uefi.dbx
  Categories:           X-Configuration
  Categories:           X-System
  Issues:               CVE-2022-21894
  Checksum:             SHA1(bc6a604b29918d67d5fef745ad4375ca3d43d05e)
  Checksum:             SHA256(fc3feb015df2710fcfa07583d31b5975ee398357016699cfff067f422ab91e13)
  License:              LicenseRef-proprietary
  Size:                 21.2 kB
  Created:              2023-05-09
  Uri:                  https://fwupd.org/downloads/fc3feb015df2710fcfa07583d31b5975ee398357016699cfff067f422ab91e13-DBXUpdate-20230509-x64.cab
[... cut ...]
  DistroVariant:        workstation
  DistroVersion:        38
  VersionOld:           211
  Vendor:               DMC Group
  VendorId:             1758
  Flags:                none
  RuntimeVersion(org.freedesktop.fwupd): 1.9.2
  
  [Reports]
  DeviceName:           LENOVO ThinkPad E14 Gen 2
  DistroId:             ubuntu
  DistroVersion:        22.04
  VersionOld:           220
  Vendor:               Jabra
  VendorId:             21
  Flags:                none
  RuntimeVersion(org.freedesktop.fwupd): 1.9.3
```

superm1, 2023-10-27:
Did you not enable your embargo remote?

FrancisLeeKinetic, 2023-10-27:
Yes, I have. I downloaded and copied ```kinet_ic-embargo.conf``` from website to my local environment. ```/etc/fwupd/remotes.d/```
I also checked  the configuration ```Enable=true``` in ```kinet_ic-embargo.conf```.

FrancisLeeKinetic, 2023-10-27:
I moved ```kinet_ic-embargo.conf``` from /etc/fwupd/remotes.d/ to /usr/local/etc/fwupd/remotes.d/```.
Executed 
````shell
$fwupdmgr refersh
WARNING: This package has not been validated, it may not work properly.
Updating kinet_ic-embargo
Downloading…             [***************************************]
Updating lvfs
Downloading…             [************************************** ]
Successfully downloaded new metadata: 2 local devices supported

````
fwupdmgr is able to get remote ID:```kinet_ic-embargo```!!!
But fwupdmgr update still fail to update
```shell
$fwupdmgr update
WARNING: This package has not been validated, it may not work properly.
Devices with no available firmware updates: 
 • SSDSCKKW256G8
 • System Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade KTM50X0 from 0.013.09 to 0.013.18?                                   ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ The vendor did not supply any release notes.                                 ║
║                                                                              ║
║ KTM50X0 and all connected devices may not be usable while updating.          ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Authenticating…          [***************************************]
Could not check for auth: GDBus.Error:org.freedesktop.PolicyKit1.Error.Failed: Action org.freedesktop.fwupd.update-hotplug-trusted is not registered
```

hughsie, 2023-10-27:
I really think following https://lvfs.readthedocs.io/en/latest/custom-plugin.html#building-fwupd would help a lot -- installing to /usr/local isn't recommended at all.

superm1, 2023-10-27:
> I really think following https://lvfs.readthedocs.io/en/latest/custom-plugin.html#building-fwupd would help a lot -- installing to /usr/local isn't recommended at all.

On Ubuntu it works relatively well, it's more painful on Fedora

FrancisLeeKinetic, 2023-10-30:
Thanks, End-to-end testing is succesful.
By the way, after successfully completing end-to-end testing, how can I submit a request to merge the KT branch(remotes/origin/wip/kinetic/latest) into the main branch?

hughsie, 2023-10-30:
@FrancisLeeKinetic I was working with Hai Su <hsu@kinet-ic.com> but I've been ignored since basically December of 2022, so I closed the pull request. There is https://github.com/fwupd/fwupd/pull/5571 but please ask to re-open the bug if you're intending to fix all the FIXMEs. I can also rebase it against main and force push before you start if that would help.

FrancisLeeKinetic, 2023-10-31:
Hai Su has resigned from Kinetic and I'm currently responsible for this task.
Please help me to reabase it against and force push.

hughsie, 2023-10-31:
I've invited you as a committer to the repo and opened up https://github.com/fwupd/fwupd/pull/6305 -- lets continue discussion there. I might have broken something in to rebase and fixups, so testing very welcome.

closed 2023-10-31 by hughsie