fwupd/fwupd #6291: Cached metadata exists for deleted remote

d4s, 2023-10-26:
**Describe the bug**
I've added test remote by copying LVFS remote config file into `/var/lib/fwupd/remotes.d/test-lvfs.conf`, did some tests and removed configuration file for remote. Stalled metadata exists:
```
/var/lib/fwupd/metadata/test-lvfs:
total 940
-rw-r--r--. 1 fwupd fwupd 954615 Oct 19 06:19 metadata.xml.gz
-rw-r--r--. 1 fwupd fwupd   2236 Oct 19 06:19 metadata.xml.gz.jcat
```

**Steps to Reproduce**

1. `cp /etc/fwupd/remotes.d/lvfs.conf /var/lib/fwupd/remotes.d/test-lvfs.conf`
2. `fwupdmgr refresh`
3. `rm /var/lib/fwupd/remotes.d/test-lvfs.conf`

**Expected behavior**
Metadata cached for removed remote should be wiped out.

**fwupd version information**

```shell
fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.6
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.8
compile   org.freedesktop.gusb          0.4.5
runtime   org.freedesktop.gusb          0.4.5
runtime   org.kernel                    5.15.135-20747-gb9ef44e9cff8
runtime   org.freedesktop.fwupd         1.9.6
```

**Additional questions**

- Operating system and version:
ChromeOS 120
- Have you tried rebooting?
Yes
- Is this a regression?
No

hughsie, 2023-10-26:
I'd much rather have a dbus call e.g. something like `fwupdmgr delete-remote test-lvfs` -- would that work?

d4s, 2023-10-26:
For my purposes yes -- but since I know about the issue, I can cleanup it by my own.
However it doesn't solve the issue with manual removing of the remote.conf file.
How about additional check during `fwupd refresh`? It might be enough to remove metadata for unknown folders in `/var/lib/fwupd/metadata/`

hughsie, 2023-10-27:
@superm1 how do you feel about this? I'm hesitant to delete files like this -- I don't think it's valid to have /var/lib shared on something like NFS -- but it does feel a bit magic.

superm1, 2023-10-27:
A few thoughts

I personally feel like that /var/lib/fwupd is owned by fwupd process. Contents should only be removed if *fwupd* knows the remote is gone.

It might be better to put repo cache in /var/cache/fwupd.

d4s, 2023-10-27:
Voting for `/var/cache/fwupd`.
@hughsie additional thought -- it might be useful to check the timestamp of metadata and remove the outdated ones anyway