fwupd/fwupd #6303: ThinkPad Thunderbolt 4 Dock update fails

voxik, 2023-10-31:
**Describe the bug**
ThinkPad Thunderbolt 4 Dock update fails

**Steps to Reproduce**
~~~
$ fwupdmgr install -v
(fwupdmgr:3907): GLib-DEBUG: 09:12:07.584: setenv()/putenv() are not thread-safe and should not be used after threads are created
(pkttyagent:3909): GLib-GIO-DEBUG: 09:12:07.586: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:3907): GLib-GIO-DEBUG: 09:12:07.590: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:3907): dconf-DEBUG: 09:12:07.590: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:3907): dconf-DEBUG: 09:12:07.590: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:3907): dconf-DEBUG: 09:12:07.590: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:3907): dconf-DEBUG: 09:12:07.590: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:3907): dconf-DEBUG: 09:12:07.590: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:3907): GLib-GIO-DEBUG: 09:12:07.590: _g_io_module_get_default: Found default implementation gnome (GProxyResolverGnome) for ‘gio-proxy-resolver’
(fwupdmgr:3907): GLib-GIO-DEBUG: 09:12:07.591: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:3907): GLib-GIO-DEBUG: 09:12:07.591: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:3907): dconf-DEBUG: 09:12:07.592: watch_established: "/system/proxy/" (establishing: 1)
(fwupdmgr:3907): dconf-DEBUG: 09:12:07.592: watch_established: "/system/proxy/http/" (establishing: 1)
(fwupdmgr:3907): dconf-DEBUG: 09:12:07.592: watch_established: "/system/proxy/https/" (establishing: 1)
(fwupdmgr:3907): dconf-DEBUG: 09:12:07.592: watch_established: "/system/proxy/ftp/" (establishing: 1)
(fwupdmgr:3907): dconf-DEBUG: 09:12:07.592: watch_established: "/system/proxy/socks/" (establishing: 1)
(fwupdmgr:3907): Fwupd-DEBUG: 09:12:07.593: Emitting ::status-changed() [idle]
0.	Cancel
1.	f95c9218acd12697af946874bfe4239587209232 (Battery)
2.	2292ae5236790b47884e37cf162dcf23bfcd1c60 (Embedded Controller)
3.	349bb341230b1a86e5effe7dfe4337e1590227bd (Intel Management Engine)
4.	04e17fcf7d3de91da49a163ffe4907855c3648be (MZVL2512HCJQ-00BL7)
5.	0d5d05911800242bb1f35287012cdcbd9b381148 (Prometheus)
6.	a45df35ac0e948ee180fe216a5f703f32dda163f (System Firmware)
7.	64d62e2af417c24295829c53e56618e701ab2720 (ThinkPad Thunderbolt 4 Dock)
8.	362301da643102b9f38477387e2193e57abaa590 (UEFI dbx)
Choose device [0-8]: 7
0.	Cancel
1.	10.16
2.	10.13
Choose release [0-2]: 2
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade ThinkPad Thunderbolt 4 Dock from 10.7 to 10.13?                      ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Before continuing, ensure the following:                                     ║
║                                                                              ║
║ • Your computer battery life if over 25%.                                    ║
║ • Do not unplug the dock during update.                                      ║
║ • Do not put your computer into sleep or hibernate mode during update.       ║
║ • This update typically may take up to 15 minutes depending on your          ║
║ notebook and dock firmware version.                                          ║
║ • Please ensure fwupd version is 1.8.6 or above.                             ║
║                                                                              ║
║ ThinkPad Thunderbolt 4 Dock and all connected devices may not be usable      ║
║ while updating.                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Fwupd-INFO: 09:12:13.651: downloading https://fwupd.org/downloads/fef7ac126dc30be8ce1f6674e7a5b4db469e9e2ece80f431e8db4449fb74670b-non-iot-1013.cab
(fwupdmgr:3907): Fwupd-DEBUG: 09:12:18.198: Emitting ::status-changed() [downloading]
Downloading…             [        |                              ]Fwupd-INFO: 09:12:19.382: download progress: 100%
Downloading…             [***************************************]Fwupd-INFO: 09:12:21.605: download progress: 0%
Downloading…             [  -                                    ]Fwupd-INFO: 09:12:21.649: download progress: 1%
Downloading…             [                                       ]Fwupd-INFO: 09:12:21.675: download progress: 2%

... snip ...

Downloading…             [*************************************  ]Fwupd-INFO: 09:12:21.975: download progress: 98%
Downloading…             [************************************** ]Fwupd-INFO: 09:12:21.977: download progress: 100%
(fwupdmgr:3907): Fwupd-DEBUG: 09:12:21.977: Emitting ::status-changed() [idle]
Fwupd-INFO: 09:12:21.977: status-code was 200
(fwupdmgr:3907): Fwupd-DEBUG: 09:12:22.020: Emitting ::status-changed() [decompressing]
Decompressing…           [***************************************](fwupdmgr:3907): Fwupd-DEBUG: 09:12:22.192: Emitting ::device-changed(64d62e2af417c24295829c53e56618e701ab2720)
(fwupdmgr:3907): Fwupd-DEBUG: 09:12:22.193: Emitting ::status-changed() [waiting-for-auth]
Authenticating…          [***************************************](fwupdmgr:3907): Fwupd-DEBUG: 09:12:22.243: Emitting ::status-changed() [device-busy]
[... 388 lines cut ...]
      GUIDs:              7952264b-1a16-5011-8564-8a619eeec3d7 ← MST-513
                          751e5470-945a-56cc-a7b8-23a6988bb014 ← MST-dock-513
                          eca9c28d-2b8f-5be6-b7e4-28f8d48e196f ← MST-dock-vmm6212-513
                          545c04b2-ee4b-5678-9efc-968817b31e17 ← MST-cayenne-vmm6212-513
                          ad0bd980-6cbd-54ec-8055-2caa448ed482 ← MST-cayenne-513
                          b74d76ed-fa81-5c30-8a5f-5f3f484adb86 ← MST-cayenne
      Device Flags:       • Updatable
                          • Signed Payload
    
────────────────────────────────────────────────
Devices that were not updated correctly:
 • ThinkPad Thunderbolt 4 Dock (10.7 → 10.13)
Uploading firmware reports helps hardware vendors to quickly identify failing and successful updates on real devices.
Upload report now? (Requires internet connection) [Y|n]: n
Do you want to disable this feature for future updates? [y|N]: n
Declined upload
```

</details>

**Additional questions**

- Operating system and version: Fedora Rawhide
- Have you tried rebooting? Yes
- Is this a regression?

voxik, 2023-10-31:
And BTW, this later prevents me from rebooting:

~~~
$ systemctl reboot
Operation inhibited by "fwupd" (PID 3367 "fwupd", user root), reason is "Firmware Update in Progress".
Please retry operation after closing inhibitors and logging out other users.
Alternatively, ignore inhibitors and users with 'systemctl reboot -i'.
~~~

voxik, 2023-10-31:
`# fwupdtool update --force --verbose --verbose`:

[fwupdtool.log](https://github.com/fwupd/fwupd/files/13215861/fwupdtool.log)

mrhpearson, 2023-11-01:
Checking with the dock team for guidance

mrhpearson, 2023-11-02:
Docking team said that error is a timeout. They recommend power cycling the device and trying again.

voxik, 2023-11-02:
I have already tried to unplug the dock from electric power. But that did not help. Is there maybe some more specific procedure? Or some specific port I should try?

Actually the most silly thing is the 15 minutes timeout for the whole procedure :(

mrhpearson, 2023-11-03:
Afraid I don't know - let me check with the FW team for their advice.

AvlWx2014, 2023-11-07:
I'm also having this issue trying to update from 10.7 -> 10.16 (as well as a more incremental update from 10.7 -> 10.13, thought maybe trying such a big update might be the issue). Looking forward to what the FW team says about this.

saveriob, 2024-01-18:
Same issue updating from 10.9 to 10.16. Happy to provide logs or details if helpful.

mrhpearson, 2024-01-22:
Revived the thread with the FW team for their input.
I'm wondering if this is an issue with older versions and if you'll have to update to something newer from Windows for updates to then work. I'm not sure we tested from 10.7/10.9.
Will see what they recommend.

halaney, 2024-03-27:
For what its worth I've got the same upgrade issue from 10.7. I tried the windows updater as well, it freezes up too, even when running "check" instead of update.

sgallagher, 2024-05-17:
Just to add to this, I'm also experiencing the same issue trying to update from 10.7 to 10.16. It's particularly annoying because Discover is perpetually telling me that there is a firmware update and I have to exclude it in order to update other things.

soltysh, 2024-09-02:
I'm experiencing this exact problem when trying to update my dock on a t14 gen 5 with amd from 10.16 to 10.18.

matthewsecaur, 2024-12-20:
Unable to update from 10.16 to 10.18 with fwupd-1.9.27 on Fedora 41 on a ThinkPad P16v with this same error.

Bodanel, 2024-12-22:
> Unable to update from 10.16 to 10.18 with fwupd-1.9.27 on Fedora 41 on a ThinkPad P16v with this same error.

I have the same configuration and I get the same error

Ryyti, 2024-12-23:
Same issue here (timeouts in erasing).

Ubuntu 24.04.1 LTS
Upgrade ThinkPad Thunderbolt 4 Dock from 10.16 to 10.18

$ fwupdmgr --version 
Erasing…                 [ -                                     ]
compile   org.freedesktop.fwupd         1.9.27
compile   com.hughsie.libxmlb           0.3.18
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libxmlb           0.3.x
runtime   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         1.9.27
runtime   org.kernel                    6.8.0-51-generic

Rebooted and tried several times.

lennysh, 2025-05-25:
I've tried everything, but cannot update my dock from 10.15 -> 10.16 using fwupdmgr. Same timeout issues...

```shell
$ fwupdmgr update
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade ThinkPad Thunderbolt 4 Dock from 10.15 to 10.16?                     ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Before continuing, ensure the following:                                     ║
║                                                                              ║
║ • Your computer battery life if over 25%.                                    ║
║ • Do not unplug the dock during update.                                      ║
║ • Do not put your computer into sleep or hibernate mode during update.       ║
║ • This update typically may take up to 15 minutes depending on your          ║
║ notebook and dock firmware version.                                          ║
║ • Please ensure fwupd version is 1.9.2 or above.                             ║
║                                                                              ║
║ ThinkPad Thunderbolt 4 Dock and all connected devices may not be usable      ║
║ while updating.                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: Y
Erasing…                 [                                       ]
failed to write-firmware: failed to wait for erase: failed after 30 retries: SPI state is busy [0x08]

```

```
Fedora 42 (KDE)
```

```shell
$ fwupdmgr --version
Erasing…                 [ -                                     ]
compile   info.libusb                   1.0.28
compile   org.freedesktop.fwupd         2.0.7
compile   com.hughsie.libxmlb           0.3.22
compile   org.freedesktop.Passim        0.1.9
compile   com.hughsie.libjcat           0.2.3
runtime   org.freedesktop.fwupd-efi     1.6
runtime   com.hughsie.libxmlb           0.3.22
runtime   com.hughsie.libjcat           0.2.3
runtime   org.freedesktop.Passim        0.1.9
runtime   org.kernel                    6.14.4-103.bazzite.fc42.x86_64
runtime   org.freedesktop.fwupd         2.0.7
```

mrhpearson, 2025-05-26:
For docks with < 10.16 version please can you try and update using a Windows system? There are some known issues that we're not able to solve with older dock FW versions.

For those with version 10.16 seeing issues going to 10.18 - can you check your dock serial # and let me know if it starts with "ZDZA0***" or not.

Thanks

soltysh, 2025-05-26:
> For those with version 10.16 seeing issues going to 10.18 - can you check your dock serial # and let me know if it starts with "ZDZA0***" or not.

Nope, mine starts with "ZDHQ..."