fwupd/fwupd #6304: ADLINK Ampere Altra Developer Platform will not update firmware successfully

ausil, 2023-10-31:
**Describe the bug**
updating the firmware fails to set the boot target

```
[root@ryac ~]# fwupdtool install 7e0e110f8570313ac770a29c782ceac5076be5456886893717680b43bb50d38f-ComHpcAlt_tianocore_atf_2.04.100.11.cab --allow-reinstall
X11 connection rejected because of wrong authentication.
Loading…                 [*********                              ]09:19:44.575 FuUdevDevice         reading subsystem_vendor for 0xffff was invalid: value 65535 was above maximum 65534
Decompressing…           [*******************                    ]
could not set BootOrder(7): failed to write data to efivarfs: Error writing to file descriptor: Input/output error
[root@ryac ~]# efibootmgr 
BootCurrent: 0001
Timeout: 10 seconds
BootOrder: 0000,0001,0002,0003,0004,0005,0006
Boot0000* UiApp	FvVol(5c60f367-a505-419a-859e-2a4ff6ca6fe5)/FvFile(462caa21-7614-4503-836e-8ab6f4662331)
Boot0001* Fedora (Samsung SSD 970 EVO Plus 1TB)	PcieRoot(0x7)/Pci(0x5,0x0)/Pci(0x0,0x0)/NVMe(0x1,00-25-38-53-21-41-07-F8)/HD(1,GPT,da9d601d-4ad8-45b9-9f93-6d2dbbbbb47d,0x800,0x400000)/File(\EFI\fedora\shimaa64.efi) File(.걎脈鼑䵙຅᫢ⱒ뉙)
Boot0002* UEFI PXEv4 (MAC:0030643D5FB4)	PcieRoot(0x6)/Pci(0x5,0x0)/Pci(0x0,0x0)/MAC(0030643d5fb4,1)/IPv4(0.0.0.00.0.0.0,0,0){auto_created_boot_option}
Boot0003* UEFI PXEv6 (MAC:0030643D5FB4)	PcieRoot(0x6)/Pci(0x5,0x0)/Pci(0x0,0x0)/MAC(0030643d5fb4,1)/IPv6([::]:<->[::]:,0,0){auto_created_boot_option}
Boot0004* UEFI HTTPv4 (MAC:0030643D5FB4)	PcieRoot(0x6)/Pci(0x5,0x0)/Pci(0x0,0x0)/MAC(0030643d5fb4,1)/IPv4(0.0.0.00.0.0.0,0,0)/Uri(){auto_created_boot_option}
Boot0005* UEFI HTTPv6 (MAC:0030643D5FB4)	PcieRoot(0x6)/Pci(0x5,0x0)/Pci(0x0,0x0)/MAC(0030643d5fb4,1)/IPv6([::]:<->[::]:,0,0)/Uri(){auto_created_boot_option}
Boot0006  UEFI Shell	FvVol(5c60f367-a505-419a-859e-2a4ff6ca6fe5)/FvFile(7c04a583-9e3e-4f1c-ad65-e05268d0b4d1)
Boot0007* Linux Firmware Updater	HD(1,GPT,da9d601d-4ad8-45b9-9f93-6d2dbbbbb47d,0x800,0x400000)/File(\EFI\fedora\fwupdaa64.efi)
[root@ryac ~]# efibootmgr -n 0007
BootNext: 0007
BootCurrent: 0001
Timeout: 10 seconds
BootOrder: 0000,0001,0002,0003,0004,0005,0006
Boot0000* UiApp	FvVol(5c60f367-a505-419a-859e-2a4ff6ca6fe5)/FvFile(462caa21-7614-4503-836e-8ab6f4662331)
Boot0001* Fedora (Samsung SSD 970 EVO Plus 1TB)	PcieRoot(0x7)/Pci(0x5,0x0)/Pci(0x0,0x0)/NVMe(0x1,00-25-38-53-21-41-07-F8)/HD(1,GPT,da9d601d-4ad8-45b9-9f93-6d2dbbbbb47d,0x800,0x400000)/File(\EFI\fedora\shimaa64.efi) File(.걎脈鼑䵙຅᫢ⱒ뉙)
Boot0002* UEFI PXEv4 (MAC:0030643D5FB4)	PcieRoot(0x6)/Pci(0x5,0x0)/Pci(0x0,0x0)/MAC(0030643d5fb4,1)/IPv4(0.0.0.00.0.0.0,0,0){auto_created_boot_option}
Boot0003* UEFI PXEv6 (MAC:0030643D5FB4)	PcieRoot(0x6)/Pci(0x5,0x0)/Pci(0x0,0x0)/MAC(0030643d5fb4,1)/IPv6([::]:<->[::]:,0,0){auto_created_boot_option}
Boot0004* UEFI HTTPv4 (MAC:0030643D5FB4)	PcieRoot(0x6)/Pci(0x5,0x0)/Pci(0x0,0x0)/MAC(0030643d5fb4,1)/IPv4(0.0.0.00.0.0.0,0,0)/Uri(){auto_created_boot_option}
Boot0005* UEFI HTTPv6 (MAC:0030643D5FB4)	PcieRoot(0x6)/Pci(0x5,0x0)/Pci(0x0,0x0)/MAC(0030643d5fb4,1)/IPv6([::]:<->[::]:,0,0)/Uri(){auto_created_boot_option}
Boot0006  UEFI Shell	FvVol(5c60f367-a505-419a-859e-2a4ff6ca6fe5)/FvFile(7c04a583-9e3e-4f1c-ad65-e05268d0b4d1)
Boot0007* Linux Firmware Updater	HD(1,GPT,da9d601d-4ad8-45b9-9f93-6d2dbbbbb47d,0x800,0x400000)/File(\EFI\fedora\fwupdaa64.efi)

```

**Steps to Reproduce**
Steps to reproduce the behavior.
Attempt to update the system firmware

**Expected behavior**
System to be prepped for update and reboot to update the firmware

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.14
compile   org.freedesktop.Passim        0.1.3
compile   com.hughsie.libjcat           0.1.14
compile   org.freedesktop.fwupd         1.9.7
runtime   org.freedesktop.fwupd-efi     1.4
runtime   org.freedesktop.Passim        0.1.4
compile   org.freedesktop.gusb          0.4.6
runtime   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.gusb          0.4.6
runtime   org.freedesktop.fwupd         1.9.7
runtime   org.kernel                    6.5.9-300.fc39.aarch64

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

dnf

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.
[... 225 lines cut ...]
│       ├── aa78084985894e6f9ab35b5ad1f69001-6.5.6-300.fc39.aarch64.conf
│       └── aa78084985894e6f9ab35b5ad1f69001-6.5.9-300.fc39.aarch64.conf
├── lost+found
├── symvers-6.5.5-300.fc39.aarch64.xz -> /lib/modules/6.5.5-300.fc39.aarch64/symvers.xz
├── symvers-6.5.6-300.fc39.aarch64.xz -> /lib/modules/6.5.6-300.fc39.aarch64/symvers.xz
├── symvers-6.5.9-300.fc39.aarch64.xz -> /lib/modules/6.5.9-300.fc39.aarch64/symvers.xz
├── System.map-6.5.5-300.fc39.aarch64
├── System.map-6.5.6-300.fc39.aarch64
├── System.map-6.5.9-300.fc39.aarch64
├── vmlinuz-0-rescue-aa78084985894e6f9ab35b5ad1f69001
├── vmlinuz-6.5.5-300.fc39.aarch64
├── vmlinuz-6.5.6-300.fc39.aarch64
└── vmlinuz-6.5.9-300.fc39.aarch64

52 directories, 1486 files
```

**Additional questions**

- Operating system and version: Fedora 39
- Have you tried rebooting? Yes
- Is this a regression? No
- Are you using an NVMe disk? Yes
- Is secure boot enabled? No
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No

superm1, 2023-10-31:
It's not your issue, but I did note you have a warning meaning you're missing 1826a03e6.

For your issue though:
What happens if you manually change the `BootOrder` using `efibootmgr`?

hughsie, 2023-10-31:
@superm1 from a little out-of-process debugging, it seems that the platform can write to BootNext just fine, but fails when writing to BootOrder. i.e. if we merged https://github.com/fwupd/fwupd/pull/6070 I think that it would just work.

superm1, 2023-10-31:
Hmm.  I'm still worried about regressions for that one.  How about we modify #6070 to add a quirk file and in the quirk file add CHID to match HardwareId-14 for all the vendors with stuff on LVFS today to keep doing boot order. Then one by one vendors can remove themselves from the quirk file and confirm if it still works.  If it does then we can shrink that over time.  Anyone else new doesn't get added to it unless they can't fix their BIOS for $reasons.

hughsie, 2023-10-31:
> How about we modify https://github.com/fwupd/fwupd/pull/6070 to add a quirk file and in the quirk file add CHID to match HardwareId-14 for all the vendors with stuff on LVFS today to keep doing boot order

Good idea, I've done that. @ausil would it be possible to test with the  branch? I can do a srpm if that's easier.

superm1, 2023-11-01:
It's merged - just test main.

closed 2024-11-11 by hughsie