fwupd/fwupd #6308: Failed to connect to daemon

Shijikori, 2023-11-01:
**Describe the bug**
The fwupd daemon fails to start on my Framework Laptop 13 on Arch Linux.

**Steps to Reproduce**
From an recent installation without fwupd : 
`sudo pacman -S fwupd`
restart the machine
`fwupdmgr get-devices`
wait 25 seconds to get "Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)"

**Expected behavior**
<details>
<summary>something like what I get on my desktop:</summary>
```
Gigabyte Technology Co., Ltd. B650 AORUS ELITE AX
│
├─AMD Ryzen 7 7700X 8-Core Processor:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x0a601203
│     Vendor:             Advanced Micro Devices, Inc.
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         22f9ecf4-588d-5c0a-8326-6ebff3655c6d ← CPUID\PRO_0&FAM_19
│                         16dd7d28-eade-5077-adb8-0775fdfbb2e5 ← CPUID\PRO_0&FAM_19&MOD_61
│                         6c9f4098-a058-5cd6-a1ce-36bfe27404fb ← CPUID\PRO_0&FAM_19&MOD_61&STP_2
│     Device Flags:       • Internal device
│   
├─CT500MX500SSD1:
│     Device ID:          403f79a01084d71d5388eebeeb9e625891beaa07
│     Summary:            ATA drive
│     Current version:    M3CR043
│     Vendor:             Micron (ATA:0x1344, OUI:00a075)
│     Serial Number:      2138E5D32D5F
│     GUIDs:              18bc9a1d-7472-5e9d-a371-8b32d7b8e6a7 ← IDE\CT500MX500SSD1__________________________M3CR043
│                         4a382f63-5319-5261-b05d-4c598d8bcd8b ← IDE\0CT500MX500SSD1__________________________
│                         ea728598-6b78-58ff-ad8a-e82eb571af3b ← CT500MX500SSD1
│     Device Flags:       • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│   
├─HP SSD EX950 1TB:
│     Device ID:          04e17fcf7d3de91da49a163ffe4907855c3648be
│     Summary:            NVM Express solid state drive
│     Current version:    42A0S63C
│     Vendor:             Silicon Motion, Inc. (NVME:0x126F)
│     Serial Number:      HBSE40461800016
│     GUIDs:              242d4bb0-16d1-56da-8ba9-f5e8bd695382 ← NVME\VEN_126F&DEV_2262
│                         3bdea050-e1dc-51de-85f9-1ff504270e1b ← NVME\VEN_126F&DEV_2262&SUBSYS_126F2262
│                         37738301-9715-5fc7-8bdf-32d4a1ab5589 ← HP SSD EX950 1TB
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Mass storage controller:
│     Device ID:          fde8e3afecaa5293a88377c9f695fee2b994d7cf
│     Current version:    01
│     Vendor:             Advanced Micro Devices, Inc. [AMD] (PCI:0x1022)
│     GUIDs:              c12a74b6-78b0-5cfb-95a3-cd65a12f9e94 ← PCI\VEN_1022&DEV_43F6
│                         38b47a77-8710-5b30-b79b-91df5c9a6d96 ← PCI\VEN_1022&DEV_43F6&SUBSYS_1B211062
│                         67d3b17f-d200-5c56-979e-8a3a4ab46c2b ← PCI\VEN_1022&DEV_43F5
│                         6b2e126d-81a8-530b-a8d3-a93e65bd30b2 ← PCI\VEN_1022&DEV_43F5&SUBSYS_1B213328
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Navi 22 [Radeon RX 6700/6700 XT/6750 XT / 6800M/6850M XT] (Sapphire Radeon RX 6700):
│     Device ID:          594baa45f6742120d71dd82166e762f1529c2a8a
│     Current version:    113-D5122100-S03
│     Vendor:             Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002)
│     GUIDs:              7d3e32bc-efe5-5612-9086-f12d696db976 ← PCI\VEN_1002&DEV_73DF
[... 90 lines cut ...]
runtime   org.freedesktop.fwupd         1.9.6
runtime   org.kernel                    6.5.9-arch2-1

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

fwupdmgr simply fails, so this command does not work.
</details>

**Additional questions**

- Operating system and version: Arch Linux
- Have you tried rebooting? Yes
- Is this a regression? Possibly not. Only difference with my desktop is passim version, desktop has 0.1.3.

Shijikori, 2023-11-01:
yes it is an AMD Edition. Apologies for the issue, I am able to use fwupdmgr as expected after updating to the newest bios version, since Framework now says it is stable.

So, for some reason, fwupd does not work on Framework Laptop 13 AMD Edition bios version 3.02. Since Framework wants users to move to 3.03 ASAP, I would not worry about my report.

superm1, 2023-11-01:
Yeah I was trying to understand the 3.02 bug causing a fwupd issue which is why I asked.

Shijikori, 2023-11-01:
I am sorry, since there's no way to rollback the bios, I cannot help you figure it out anymore. It's not likely to matter however since we can update the bios via other means than fwupd and future production units will ship with updated firmware.

superm1, 2023-11-01:
OK.  Can you check if your historical journal for `fwupd.service` shows anything interesting?

Shijikori, 2023-11-01:
I think the best I can get you might be this 
[some-journalctl-output-with-fwupd.log](https://github.com/fwupd/fwupd/files/13223358/some-journalctl-output-with-fwupd.log)

superm1, 2023-11-01:
It seems to me that it's caused by reading aux devices (IE `/dev/drm_dp_aux0` etc) which triggers the framework BIOS bug.

@hughsie any thoughts that we might be missing a shorter timeout in a DP aux read?

It's too bad we couldn't get a `fwupdtool get-devices --verbose` output.

hughsie, 2023-11-01:
Ohh, you think that fu_udev_device_get_sysfs_attr() is blocking? We could add a timeout to those, but that would be a lot of work. Is there any way we can quirk this?

hughsie, 2023-11-01:
> drm_dp_aux0

Ohh, maybe this is the synaptics_mst plugin blocking? In which case it might be the seek() or read() there. That's easier to add a timeout for, but it would be good to know *which* is blocking. Can we ask the framework folks?

superm1, 2023-11-01:
> > drm_dp_aux0
> 
> Ohh, maybe this is the synaptics_mst plugin blocking? In which case it might be the seek() or read() there. That's easier to add a timeout for, but it would be good to know *which* is blocking. Can we ask the framework folks?

Yeah I suspect from that journal it's the mst plugin. You can see the aux timeouts from the kernel parts of it

I'd say it's more likely a read() that fails, seek is just about moving a pointer.

hughsie, 2023-11-02:
I think https://github.com/fwupd/fwupd/pull/6313 should fix this.

FranzPoize, 2023-11-10:
Hi ! I am on a framework with amd with the 3.2 BIOS

Here is my output for `sudo fwupdtool get-devices --verbose`

```
12:33:58.262 FuDebug              verbose to info (on console 1)
12:33:58.263 FuEngine             starting fwupd 1.9.8…
Chargement…              [ -                                     ]12:33:58.265 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/fwupd-tests.conf
12:33:58.265 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/lvfs-testing.conf
12:33:58.265 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/lvfs.conf
12:33:58.265 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/vendor-directory.conf
12:33:58.265 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/vendor.conf
12:33:58.265 FuRemoteList         enabled remotes: vendor-directory[1000], lvfs[0]
Chargement…              [                                       ]12:33:58.265 FuEngine             client certificate now exists: nothing to do
Chargement…              [*                                      ]12:33:58.271 FuQuirks             loading quirks from /usr/local/share/fwupd/quirks.d
12:33:58.271 FuQuirks             loading quirks from /var/local/lib/fwupd/quirks.d
Chargement…              [*                                      ]12:33:58.271 FuContext            SMBIOS Manufacturer=Framework
12:33:58.271 FuContext            SMBIOS EnclosureKind=a
12:33:58.271 FuContext            SMBIOS Family=Laptop
12:33:58.271 FuContext            SMBIOS ProductName=Laptop 13 (AMD Ryzen 7040Series)
12:33:58.271 FuContext            SMBIOS ProductSku=FRANDGCP07
12:33:58.271 FuContext            SMBIOS BiosVendor=INSYDE Corp.
12:33:58.271 FuContext            SMBIOS BiosVersion=03.02
12:33:58.271 FuContext            SMBIOS BiosMajorRelease=03
12:33:58.271 FuContext            SMBIOS BiosMinorRelease=02
12:33:58.271 FuContext            SMBIOS FirmwareMajorRelease=ff
12:33:58.271 FuContext            SMBIOS FirmwareMinorRelease=ff
12:33:58.271 FuContext            SMBIOS BaseboardManufacturer=Framework
12:33:58.271 FuContext            SMBIOS BaseboardProduct=FRANMDCP07
12:33:58.271 FuContext            failed to load fdt: cannot find /sys/firmware/fdt or override /var/local/lib/fwupd/system.dtb
12:33:58.272 FuContext            added udev subsystem watch of firmware-attributes
Chargement…              [**                                     ]12:33:58.272 FuEngine             loading metadata for remote 'vendor-directory'
12:33:58.272 FuEngine             ignoring: /usr/local/share/fwupd/remotes.d/vendor/firmware/README.md
12:33:58.272 FuEngine             ignoring: Erreur à l’ouverture du répertoire « /var/local/lib/fwupd/local.d » : Aucun fichier ou dossier de ce type
12:33:58.272 FuEngine             ignoring: Erreur à l’ouverture du répertoire « /usr/local/share/fwupd/local.d » : Aucun fichier ou dossier de ce type
Chargement…              [**                                     ]12:33:58.278 FuEngine             failed to setup backend bluez: Erreur lors de l’appel de StartServiceByName pour org.bluez : Unit dbus-org.bluez.service not found.
Chargement…              [***                                    ]12:33:58.279 FuContext            added udev subsystem watch of platform
12:33:58.279 FuContext            added udev subsystem watch of pci
12:33:58.279 FuContext            added udev subsystem watch of block for plugin android_boot
12:33:58.279 FuContext            added udev subsystem watch of i2c
12:33:58.279 FuContext            added udev subsystem watch of i2c-dev
12:33:58.279 FuContext            added udev subsystem watch of hidraw
12:33:58.279 FuContext            added udev subsystem watch of gpio for plugin gpio
12:33:58.279 FuContext            added udev subsystem watch of mei
12:33:58.279 FuContext            added udev subsystem watch of iommu for plugin iommu
12:33:58.279 FuContext            added udev subsystem watch of drm for plugin linux_display
12:33:58.279 FuContext            added udev subsystem watch of video4linux
12:33:58.280 FuContext            added udev subsystem watch of drm_dp_aux_dev for plugin mediatek_scaler
12:33:58.280 FuContext            added udev subsystem watch of msr for plugin msr
12:33:58.280 FuContext            added udev subsystem watch of mtd for plugin mtd
12:33:58.280 FuContext            added udev subsystem watch of nvme for plugin nvme
Chargement…              [***                                    ]12:33:58.280 FuContext            added udev subsystem watch of serio
12:33:58.280 FuContext            added udev subsystem watch of thunderbolt
12:33:58.280 FuContext            added udev subsystem watch of tpm for plugin tpm
12:33:58.280 FuEngine             plugins disabled: test, test_ble
Chargement…              [***                                    ]12:33:58.280 FuContext            battery threshold now 10
Chargement…              [-                                      ]12:33:58.281 FuEngine             disabling plugin because: failed to startup using flashrom: no HwIDs found
12:33:58.281 FuEngine             disabling plugin because: failed to startup using lenovo_thinklmi: unsupported manufacturer
12:33:58.282 FuEngine             disabling plugin because: failed to startup using powerd: no service that owns the name for org.chromium.PowerManager
[... cut ...]
Chargement…              [******                                 ]12:33:58.585 FuDeviceList         ::added 23ec719b6aabc2d2dac5176c232f0da7a21881b0 [Fingerprint Sensor]
Chargement…              [***************                        ]
```

And then it hangs there I cannot interrupt fwupdtool

I have not update the firmware for the fingerprint (https://knowledgebase.frame.work/en_us/updating-fingerprint-reader-firmware-on-linux-for-13th-gen-and-amd-ryzen-7040-series-laptops-HJrvxv_za) since fwupdtool always hangs

hughsie, 2023-11-10:
> starting fwupd 1.9.8

Eeek. If you do the same again, but this time under `strace` -- what is it blocking on? It might be better if you ran with `gdb`, e.g. `sudo gdb --args fwupdtool get-devices -vv` and then ctrl+c when it's hung, and then `bt` to show the backtrace.

FranzPoize, 2023-11-10:
Ok I'm not able to get the backtrace because when the program i am not able to ctrl+c (or every other combination of ctrl+something can't even kill the process...) out of it however I have the strace file and the vv trace also which seems to hang.
[strace_fwupdtool.log](https://github.com/fwupd/fwupd/files/13319468/strace_fwupdtool.log)
[vv_trace.log](https://github.com/fwupd/fwupd/files/13319469/vv_trace.log)

hughsie, 2023-11-10:
We're reading with `O_NONBLOCK` now -- and a 10ms timeout -- but that doesn't seem to affect anything:

    openat(AT_FDCWD, "/dev/drm_dp_aux5", O_RDWR|O_NONBLOCK) = 18
    lseek(18, 1280, SEEK_SET)               = 1280
    poll([{fd=18, events=POLLIN|POLLPRI|POLLERR}], 1, 10) = 1 ([{fd=18, revents=POLLIN}])
    read(18, 

I'm wondering if you get something in `dmesg` -- e.g. some kind of kernel backtrace. If this is a 3.02 firmware bug (which I think it might be) can you also get us the output of `sudo fwupdtool hwids` please -- and maybe we can just skip the read on old firmware versions.

FranzPoize, 2023-11-10:
I've rerun with gdb and putting a breakpoint in `fu_device_set_id` because that was the last log in the trace.

It hangs on `fu_device_ensure_id` line 4791 in fu_device.c

I look in the dmesg

hughsie, 2023-11-10:
> It hangs on fu_device_ensure_id line 4791 in fu_device.c

Nah, I think slightly further than that we do ->setup on the DP AUX device.

FranzPoize, 2023-11-10:
`fwupdtool hwids` output
```
hargement…              [*                                      ]
Computer Information
--------------------
BiosVendor: INSYDE Corp.
BiosVersion: 3.02
BiosMajorRelease: 3
BiosMinorRelease: 2
FirmwareMajorRelease: ff
FirmwareMinorRelease: ff
Manufacturer: Framework
Family: Laptop
ProductName: Laptop 13 (AMD Ryzen 7040Series)
ProductSku: FRANDGCP07
EnclosureKind: a
BaseboardManufacturer: Framework
BaseboardProduct: FRANMDCP07
Hardware IDs
------------
{bf106e31-bdab-5a7b-93b1-5d5b9cd376c0}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{32d49d99-414b-55d5-813b-12aaf0335b58}   <- Manufacturer + Family + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{288590d7-75da-5320-9352-f1aa8fe66828}   <- Manufacturer + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{969d92a9-8918-50f2-b387-6bd66680b897}   <- Manufacturer + Family + ProductName + ProductSku + BaseboardManufacturer + BaseboardProduct
{8ca121ed-f99f-550c-9e10-deab47ad941c}   <- Manufacturer + Family + ProductName + ProductSku
{b5f7dcc1-568c-50f8-a4dd-e39d1f93fda1}   <- Manufacturer + Family + ProductName
{a9407fd0-216c-56ad-bc90-fcdd504be3b1}   <- Manufacturer + ProductSku + BaseboardManufacturer + BaseboardProduct
{f3a409fc-1288-509b-8538-663490954390}   <- Manufacturer + ProductSku
{71cc556b-d596-5b90-a09e-a163bfa2cebb}   <- Manufacturer + ProductName + BaseboardManufacturer + BaseboardProduct
{e3046147-c2a7-586d-81c4-85209eda3481}   <- Manufacturer + ProductName
{c9178f27-c098-5e85-bb49-76890fefc169}   <- Manufacturer + Family + BaseboardManufacturer + BaseboardProduct
{b3bdb2e4-c5cb-5c1b-bdc3-e6fc132462ff}   <- Manufacturer + Family
{96ef7072-4be1-53aa-ac9a-1c095c7784f0}   <- Manufacturer + EnclosureKind
{e2e71fed-d55c-5187-9b24-d53db3ff00e8}   <- Manufacturer + BaseboardManufacturer + BaseboardProduct
{2185e781-ed4b-5fc2-8839-b5ddb1e7d649}   <- Manufacturer
```

I get some `[  101.280398] [drm:amdgpu_dm_process_dmub_aux_transfer_sync [amdgpu]] *ERROR* wait_for_completion_timeout timeout!` in dmesg but that doesn't seem related

hughsie, 2023-11-10:
> I get some [  101.280398] [drm:amdgpu_dm_process_dmub_aux_transfer_sync [amdgpu]] *ERROR* wait_for_completion_timeout timeout! in dmesg but that doesn't seem related

I think that's probably very related :) @superm1 agree?

Can you get me the output of `cat /sys/class/drm_dp_aux_dev/drm_dp_aux5/name` and also the other drm_dp_aux* ones too. Maybe we can just ignore this by HWID and DP AUX name?

FranzPoize, 2023-11-10:
`cat /sys/class/drm_dp_aux_dev/drm_dp_aux5/name`

`AMDGPU DM aux hw bus 5`

FranzPoize, 2023-11-10:
And as for debugging I think it's the read in https://github.com/fwupd/fwupd/blob/f51a04530df60a5b7d4603d20a4bc6bbe7da336a/libfwupdplugin/fu-io-channel.c#L357

hughsie, 2023-11-10:
@FranzPoize can you try the wip/hughsie/amdgpu-dpaux branch please?

superm1, 2023-11-10:
I was hoping that switching to io channel should avoid tripping the bug. I'd really like us to be resistant to this kind of issue instead of hard coding quirks

FranzPoize, 2023-11-10:
wip/hughsie/amdgpu-dpaux makes fwupdtool not hang anymore

this is the ouput

```
Chargement…              [************************************** ]
WARNING: UEFI ESP partition may not be set up correctly
Voir https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-valid pour plus d'informations.
WARNING: Ce paquet n'a pas été validé, il peut ne pas fonctionner correctement.
Framework Laptop 13 (AMD Ryzen 7040Series)
│
├─AMD Ryzen 7 7840U w/ Radeon 780M Graphics:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Version actuelle:   0x0a704103
│ │   Fournisseur:        Advanced Micro Devices, Inc.
│ │   GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│ │                       22f9ecf4-588d-5c0a-8326-6ebff3655c6d ← CPUID\PRO_0&FAM_19
│ │                       2a1df9de-fe02-59c8-8af1-32c8bb011f94 ← CPUID\PRO_0&FAM_19&MOD_74
│ │                       5a4bca32-6547-584f-bb36-920d9adede25 ← CPUID\PRO_0&FAM_19&MOD_74&STP_1
│ │   Drapeaux de périphérique:• Périphérique interne
│ │ 
│ ├─Graphics Processing Unit (GPU):
│ │     Device ID:        83eba1fa14bdefee1f6b4bcf06069b298e1c338f
│ │     Résumé:           AMD AMD_PHOENIX_GENERIC
│ │     Version actuelle: 022.012.000.024.000001
│ │     Fournisseur:      Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002)
│ │     GUID:             6c23af4f-d6cd-5ffc-a502-0b85e472e7cb ← AMD\113-PHXGENERIC-001
│ │     Drapeaux de périphérique:• Périphérique interne
│ │   
│ ├─Secure Processor:
│ │     Device ID:        c54ab0237d7a8db8c717b68e0be78e4374a2a079
│ │     Version actuelle: 00.2d.06.6c
│ │     Version du chargeur d’amorçage:00.2d.06.6c
│ │     Fournisseur:      Advanced Micro Devices, Inc. (PCI:0x1022)
│ │     GUIDs:            9eb6a793-7f97-5fb5-b49c-139d2ce3ee46 ← PCI\VEN_1022&DEV_15C7
│ │                       2e4eb727-a7c8-5eba-80cd-553b8d796d80 ← PCI\VEN_1022&DEV_15C7&SUBSYS_F1110006
│ │     Drapeaux de périphérique:• Périphérique interne
│ │   
│ └─System Management Unit (SMU):
│       Device ID:        db0330716216c629bb2c07256e5d018f499eb6ce
│       Résumé:           Microcontroller used within CPU/APU program 0
│       Version actuelle: 76.68.0
│       Fournisseur:      Advanced Micro Devices, Inc.
│       GUID:             3ac3159e-6eef-5f6b-bc8d-67686b238747 ← /sys/devices/platform/AMDI0009:00
│       Drapeaux de périphérique:• Périphérique interne
│     
├─Fingerprint Sensor:
│     Device ID:          23ec719b6aabc2d2dac5176c232f0da7a21881b0
│     Résumé:             Match-On-Chip fingerprint sensor
│     Version actuelle:   01000320
│     Fournisseur:        Goodix (USB:0x27C6)
│     Durée d'installation:10 secondes
│     Numéro de série:    UIDXXXXXXXX_XXXX_MOC_B0
│     GUID:               1e8c8470-a49c-571a-82fd-19c9fa32b8c3 ← USB\VID_27C6&PID_609C
│     Drapeaux de périphérique:• Mise à jour possible
│                         • Device can recover flash failures
│                         • Signed Payload
│   
├─SSD 980 PRO 1TB:
│     Device ID:          03281da317dccd2b18de2bd1cc70a782df40ed7e
│     Résumé:             NVM Express solid state drive
│     Version actuelle:   5B2QGXA7
│     Fournisseur:        Samsung (NVME:0x144D)
│     Numéro de série:    S5GXNX0W628556H
│     GUIDs:              bec63ed7-a95f-54fe-b8cc-8e9fee64ba5a ← NVME\VEN_144D&DEV_A80A
│                         310f81b5-6fce-501e-acfb-487d10501e78 ← NVME\VEN_144D&DEV_A80A&SUBSYS_144DA801
│                         1dc6777c-1ed3-521f-8267-b61f0c6b3902 ← Samsung SSD 980 PRO 1TB
│     Drapeaux de périphérique:• Périphérique interne
│                         • Mise à jour possible
│                         • Le système nécessite une source d'alimentation externe
[... 31 lines cut ...]
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Only version upgrades are allowed
│                         • Signed Payload
│     
└─TPM:
      Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
      Résumé:             TPM 2.0 Device
      Version actuelle:   6.3.1.610
      Fournisseur:        Microsoft (TPM:MSFT)
      GUIDs:              2a65d4b6-60a0-5e8c-acfa-d86cc3cbe4c3 ← TPM\VEN_MSFT&DEV_0001
                          31a7bb1b-f5e1-53ff-b0e5-8692a21caed6 ← TPM\VEN_MSFT&MOD_Pluton.TPM.A
                          5bea8d58-c840-513e-a5ad-0184cbaa9821 ← TPM\VEN_MSFT&DEV_0001&VER_2.0
                          e02519c6-9139-5de8-89d4-9dca2f5c3b66 ← TPM\VEN_MSFT&MOD_Pluton.TPM.A&VER_2.0
      Drapeaux de périphérique:• Périphérique interne
                          • Le système nécessite une source d'alimentation externe
                          • Needs a reboot after installation
                          • Device can recover flash failures
                          • Full disk encryption secrets may be invalidated when updating
                          • Signed Payload
```

It says UEFI partition may not be set up correctly but my esp partition does have the esp flag on

I'm trying to get fwupdmgr to work now. The service seems to be doing fine but fwupdmgr get-devices seems to hang now...

FranzPoize, 2023-11-10:
right now fwupdmgr refresh, get-update works but fwupdmgr get-devices still hangs

hughsie, 2023-11-10:
> I was hoping that switching to io channel should avoid tripping the bug

Any read seems to trigger the kernel bug -- and it looks like the process gets stuck in some unkillable state -- even with nonblocking reads :/

Shijikori, 2023-11-10:
The firmware version 3.03 does address mostly AMDGPU related bugs. So that would make sense for the Framework AMD Edition.

FranzPoize, 2023-11-10:
I have successfully updated to 3.3 using fwupd

superm1, 2023-11-10:
@FranzPoize what did you have connected (cards and dock) when you reproduced the issue on 3.2?

FranzPoize, 2023-11-10:
I think i tried with and without hub usb not a dock (which had a usb keyboard, wireless mouse and usb to jack). Usb c power adapter and a hdmi screen were always plugged.

However i think all this is wired to a usb c port on the framework

superm1, 2023-11-10:
OK, so it's possible this issue tripped specifically from the HDMI card / HDMI monitor then.  As you already upgraded I guess hard to get any more data though.

patmorgan235, 2023-11-10:
Hey, 
I was having a very similar issue. Any fwupdmgr command I ran would result in the "Failed to Connect to daemon" error message.

OS & Hardware Info:
```
Ubuntu 22.04.3 LTS x86_64
Framework Laptop 13 (AMD Ryzen 7040Series) 
AMD Ryzen 7 7840U w/ Radeon 780M Graphics (16) @ 3.300GHz
```

systemctl status fwupd.service

```
[?7h● fwupd.service - Firmware update daemon
     Loaded: loaded (/lib/systemd/system/fwupd.service; static)
     Active: activating (start) since Fri 2023-11-10 16:45:24 CST; 1min 10s ago
       Docs: https://fwupd.org/
   Main PID: 13384 (fwupd)
      Tasks: 5 (limit: 18106)
     Memory: 6.0M
        CPU: 241ms
     CGroup: /system.slice/fwupd.service
             └─13384 /usr/libexec/fwupd/fwupd

Nov 10 16:45:24  systemd[1]: Starting Firmware update daemon...
Nov 10 16:45:25  fwupd[13384]: 22:45:25:0064 FuEngine             failed to get releases for UEFI dbx: No releases found: Not compatible with org.freedesktop.fwupd version 1.7.9, requires >= 1.9.1
Nov 10 16:45:25  fwupd[13384]: 22:45:25:0066 FuEngine             failed to get releases for UEFI dbx: No releases found: Not compatible with org.freedesktop.fwupd version 1.7.9, requires >= 1.9.1
```

I had an official framework HDMI expansion module installed, after I removed it I was immediately able to run fwupdmgr refresh and complete upgrading my firmware to 3.03 . So theirs defiantly something about the Ryzen 7 7840U + 3.02 firmware + the HDMI expansion module that causes the fwupd daemon to not run/hang.

superm1, 2023-11-11:
Just to clarify, was it just having the module plugged in, or was it module + monitor?

patmorgan235, 2023-11-11:
> Just to clarify, was it just having the module plugged in, or was it module + monitor?

Just having the module plugged in

superm1, 2023-11-11:
@JohnAZoidberg could this be related to DP/HDMI firmware out of date too?

closed 2023-11-11 by hughsie

gozes, 2023-11-13:
so I have just hit this issue on a brand new install arch/endevoreos  of the amd 13. noting is plug in to it. I even took out all the framwork cards so is basically as if you had just taken out of the box but I still can't run a refresh. any ideas what i can do here?

hughsie, 2023-11-13:
Try the main branch please.

superm1, 2023-11-13:
Another workaround for older fwupd is to boot with `nomodeset` and then do your BIOS upgrade.

gozes, 2023-11-13:
> Another workaround for older fwupd is to boot with `nomodeset` and then do your BIOS upgrade.

How would I do that? Is that a config option to fwupd?

FranzPoize, 2023-11-13:
First result on Google for "arch nomodeset"
https://wiki.archlinux.org/title/kernel_mode_setting#Disabling_modesetting

superm1, 2023-11-13:
Yeah as clarified in the link I was meaning the kernel command line option for a workaround.  If you want a fwupd workaround, I think you can probably disable the synaptics mst plugin in `/etc/fwupd/fwupd.conf`.

gozes, 2023-11-13:
> Yeah as clarified in the link I was meaning the kernel command line option for a workaround.  If you want a fwupd workaround, I think you can probably disable the synaptics mst plugin in `/etc/fwupd/fwupd.conf`.

Trying it list plugins also times out so trying to find the correct name for that plugin is proving to be a challenge

gozes, 2023-11-13:
Ok found the name by looking at the source tree :)

gozes, 2023-11-13:
> Yeah as clarified in the link I was meaning the kernel command line option for a workaround.  If you want a fwupd workaround, I think you can probably disable the synaptics mst plugin in `/etc/fwupd/fwupd.conf`.

Ok that didn't work

Appending `;synaptica-mst` after `test_ble` didn't seam to work

gozes, 2023-11-13:
> Another workaround for older fwupd is to boot with `nomodeset` and then do your BIOS upgrade.

Nop that didn't work ether. I even installed fedora 39 and tried setting it there but no luck

superm1, 2023-11-13:
Can you please share your /proc/cmdline to confirm you did it right?

It's also possible you have a different issue given you can reproduce without cards connected and they were required from above reports.

gozes, 2023-11-13:
sure

this what is on my `/etc/default/grub` file 
in fedora 39 after reinstalling in and re doing all of the above until this point
```
GRUB_TIMEOUT=5
GRUB_DISTRIBUTOR="$(sed 's, release .*$,,g' /etc/system-release)"
GRUB_DEFAULT=saved
GRUB_DISABLE_SUBMENU=true
GRUB_TERMINAL_OUTPUT="console"
GRUB_CMDLINE_LINUX="rhgb quiet nomodeset"
GRUB_DISABLE_RECOVERY="true"
GRUB_ENABLE_BLSCFG=true
```

restarted after saving that file

this is the content of `/proc/cmdline`
```
BOOT_IMAGE=(hd0,gpt2)/vmlinuz-6.5.11-300.fc39.x86_64 root=UUID=deadf24e-c044-4adf-ba5b-de377ae8125e ro rootflags=subvol=root rhgb quiet
```

I don't see it listed there so I am not sure what I'm doing wrong
it

gozes, 2023-11-13:
> It's also possible you have a different issue given you can reproduce without cards connected and they were required from above reports.

could be but given that the above is related to the usb bugs in the firmware I would think they are somehow related

superm1, 2023-11-14:
> sure
>
> this what is on my `/etc/default/grub` file
> in fedora 39 after reinstalling in and re doing all of the above until this point
> ```
> GRUB_TIMEOUT=5
> GRUB_DISTRIBUTOR="$(sed 's, release .*$,,g' [...]

You need to regenerate the grub configuration.

gozes, 2023-11-14:
Actually I didn't update it just went back to complaining that is not connected to power which is lie. Discharging the battery a bit to plug it back in and try again

FranzPoize, 2023-11-14:
Yes it's written in a framework guide somewhere that being at 100% battery
and plugged in will complain about not being plugged in.

To check the version use dmidecode

gozes, 2023-11-14:
So even on power If makes to to 100% battery it will stop the up which is mentioned on the FW13 guild. Is it safe to force it?

superm1, 2023-11-14:
@hughsie remember we thought this was only a bug in old fwupd with 100%.

@gozes you're at a current version right?

quinn-dougherty, 2023-11-26:
framework 13 11th gen intel here -- getting `Unsupported daemon version 1.9.7, client version is 1.9.9` on every `fwupdgmr` command but `--help`, so I can't even `get-devices`. I'll check the page for a bios update.

Shijikori, 2023-11-27:
> So even on power If makes to to 100% battery it will stop the up which is mentioned on the FW13 guild. Is it safe to force it?

Framework recommends to simply unplug the laptop and let it fall down to something like 95% battery and then proceed with firmware updates. If the laptop gets to 100% during the update procedure, it won't complain. Though AC connection is not recognized when the battery is at 100%. That's why the firmware update cannot happen.

hughsie, 2023-11-27:
> Unsupported daemon version 1.9.7, client version is 1.9.9

The daemon is the old version, the client is the new version. You either need to reboot after upgrading or restart the fwupd service.

quinn-dougherty, 2023-11-27:
> The daemon is the old version, the client is the new version. You either need to reboot after upgrading or restart the fwupd service

Ah, the ole turning it off and turning it back on trick. Yes, this worked.