fwupd/fwupd #6318: RFE: hook this up with systemd-pcrlock to allow PCR predictions.

poettering, 2023-11-03:
**Is your feature request related to a problem? Please describe.**
systemd today merged a new component "systemd-pcrlock" (https://github.com/systemd/systemd/pull/28891) which allows building TPM2 policies against PCRs 0-7 (and others) in a relatively flexible way that caters nicely for changes in firmware versions, extension cards and so on. To deal with firmware updates the current primary mode of operation is to generate an automatic PCR policy from the relevant firmware measurements from the TPM2 event log automatically at boot, under the expectations that such a policy is removed whenever a firmware update is expected. That works but of course is less than ideal because the firmware PCR protection are disabled whenever a firmware update is expected. And of course, right now the "loosening" of the policy before a firmware upgrade is currently a manual step. (you have to call `systemd-pcrlock unlock-firmware-code` before you reboot, followed by a `systemd-pcrlock make-policy`)

**Describe the solution you'd like**
I'd like two things implemented in fwupd:

1. the first step would be to make fwupd invoke the above mentioned two commands when it drops in a firmware update capsule, in preparation of the update. It's really as simple as that. It can either invoke the tools directly, or we can add some IPC API for you that allows doing that (would probably be Varlink though, not D-Bus, since pcrlock is early-boot stuff).

2. the bigger, second step would be if fwupd would start carrying information about expected PCR measurements of firmware updates. pcrlock allows various componenents of the boot to install their own ".pcrlock" files (which are more or less trivial TCG Common Event Log JSON files which just list a series of expected measured hashes with their matching PCRs). it would be excellent if fwupd could just drop this kind of data into /var/lib/pcrlock.d/ whenever it prepares a firmware updated, and firmware vendors would then have to supply this data. This would be more than excellent since it would mean we don't have to relax access control to encrypted disks locked to TPM anymore around firmware updates. I understand fwupd already allows firmware vendors to encode info about expected PCR0 state after boot-up. It's kinda like that, except I want the actual measurements not the resulting value, and I'd like it for any PCR. Implementation of this would be relatively straightforward: whenever fwupd drops in a firmware update capsule into the ESP it would also drop in a pcrlock file into /var/lib/pcrlock.d/ for the new update and issue "systemd-pcrlock make-policy". If it doesn't have the data it would instead do "systemd-pcrlock unlock-firmware" as in the first point.

**Describe alternatives you've considered**
nothing really

hughsie, 2023-11-04:
> the first step would be to make fwupd invoke the above mentioned two commands when it drops in a firmware update capsule

Yup, we can certainly do that -- but it's not ideal -- see below.

> would probably be Varlink though, not D-Bus, since pcrlock is early-boot stuff

Buuut wouldn't fwupd be calling those methods when userspace is running (statistically including a desktop) too? Certainly having a D-Bus method to hit (even if a proxy) would make it much easier for fwupd -- we do have a policy [for firmware updates] that we don't execute random binaries or scripts in plugins -- and I don't really want vendors to see us spawn'ing stuff for systemd and then ask us to do the same to do a firmware update.

> fwupd would start carrying information about expected PCR measurements of firmware updates

So, we have some of that already. A couple of the vendors ship PCR0 measurement**s** for *some* firmware updates -- but the plural in measurements is load-bearing -- the same firmware update can apply to different models that each have different possible PCR0s.

Another question -- do we *always* want to call `unlock-firmware-code` unconditionally? I'm wondering if we need any kind of configuration controlling when we do it -- and I'm also wondering if we should find the *runtime* systemd version before attempting to unlock. Maybe that's not an issue if we have a nice D-Bus method to hit -- and maybe even for `systemd-pcrlock make-policy` too.

bluca, 2023-11-04:
> Yup, we can certainly do that -- but it's not ideal -- see below.
>
>
> Buuut wouldn't fwupd be calling those methods when userspace is running (statistically including a desktop) too? Certainly having [...]

At the moment it's command-line only, but it would be simple enough to make it socket-activated for varlink, and once that's there, to also make it D-Bus activated for non-early-boot stuff, so this shouldn't be a blocker

hughsie, 2023-11-04:
> make it socket-activated for varlink, and once that's there, to also make it D-Bus activated

Here's a deal -- you do the DBus method, I'll do the fwupd plugin :)

superm1, 2023-11-04:
In the dbus method can we have a way to discover what PCRs they're bound to? We shouldn't *need* to be doing anything for some PCRs.

poettering, 2023-11-06:
> to also make it D-Bus activated for non-early-boot stuff, so this shouldn't be a blocker

I am not sure that's true. turning it into a D-Bus service means processing multiple requests from the same process. It makes everything a *ton* more complex. Varlink means we can do Accept=yes and just have one service instance per connection. Super simple. D-Bus doesn't allow this, it means we must process all connections within thesingleton service instance. I'd *really* like to avoid that. D-Bus doesn't really scale for our little tools that you just call once and never again.

poettering, 2023-11-06:
> In the dbus method can we have a way to discover what PCRs they're bound to? We shouldn't _need_ to be doing anything for some PCRs.

I don't grok this question. pcrlock is written very defensively: it will by default try to lock down quite a log of PCRs (12 of the 24). However, that's just what it starts it algorithm from. it will remove all PCRs from the default lists:

1. where the hash value doesn't match up with measurements in the tpm event log
2. where there are measurements that it does not recognize, because nothing told it to expect this via a /var/lib/pcrlock.d/ drop-in
3. where there are drop-ins in /var/lib/pcrlock.d/ (with possibly multiple variants) that supposedly touch that PCR but which could not be found in the event log

This means that it's ultimately the /var/lib/pcrlock.d/ drop-ins that control which PCRs we log to and which ones we do not. 

If that makes any sense?

superm1, 2023-11-06:
> If that makes any sense?

So _effectively_  - if fwupd does _nothing_ today to influence this you will end up with PCR0 as part of the locked down policy specifically if the TPM event log matches.  I think that should be a majority of machines now.  We've found a lot of bugs with this area both in firmware and fwupd's analysis comparing the two and are pretty confident on these being right now most the time.

I think the other big area that needs concern from the fwupd side is that it **also** has the ability to change BIOS settings, which are going to be influence PCR 2 on OEM systems.  The system vendors don't provide any way to systematically predict what happens when you change a given BIOS setting.  It's completely up to their implementation and might be different from one model to another.

>  it will by default try to lock down quite a log of PCRs (12 of the 24).

What are the 12 that it starts with trying to do?

poettering, 2023-11-07:
> > If that makes any sense?
> 
> So _effectively_ - if fwupd does _nothing_ today to influence this you will end up with PCR0 as part of the locked down policy specifically if the TPM event log matches.

So eventually I hope you basically have three choices wit systemd-pcrlock:

0. You don't do `systemd-pcrlock lock-firmware-code`, neither manually nor automatically via the service that wraps it and calls it at boot. In this case PCR 0 is not included in the policy stored in the NV index, because no .pcrlock file is dropped into /var/lib/pcrlock.d/. (Let's call this security level 0 – no security around firmware)
1. You call `systemd-pcrlock lock-firmware-code`/enable the service for it. In this case you are in locked down mode. But now it's important to invoke `systemd-pcrlock unlock-firmware-code` before you do a firmware update. In this mode you'll have a /var/lib/pcrlock.d/ drop-in that is generated locally from tpm2 event log. (let's call this security level 1, i.e. we lock firmware versions down, but it is disabled around firmware updates)
2. fwupd provided a PC vendor .pcrlock file that it placed in /var/lib/pcrlock.d/ before doing a firmware update. In this mode we never have to relax the security lockdown. Yippieh! (let's call this security level 2, i.e. no policy relaxation)

> I think the other big area that needs concern from the fwupd side is that it **also** has the ability to change BIOS settings, which are going to be influence PCR 2 on OEM systems. The system [...]

pcrlock treats PCR 0 and 2 pretty much the same, and `systemd-pcrlock lock-firmware-code` always operates on both, i.e. generates a local .pcrlock policy drop-in from both PCRs as listed in the tpm2 event log.

> > it will by default try to lock down quite a log of PCRs (12 of the 24).
> 
> What are the 12 that it starts with trying to do?

0-5,7,11-15

Or in other words, we only exclude PCR 6, 8, 9 and the higher ones.

poettering, 2023-11-07:
> > varlink is trivial btw: you just write some trivial JSON to some socket, followed by a NUL byte. then you read back a JSON object, which is terminated by a NUL byte. If you want, you can implement this with fprintf().
> 
> Okay, that sounds easy enough. Is there an endpoint we can test with?

Nope, I need to add varlink support to the relevant operations. Should be easy, will ping you once I have something.

So does it make sense to you to ask vendors to supply the expected measurements their fimrware will make in TCG CEL-JSON format? It's a trivial JSON format, which pcrlock directly consumes. It ultimately just is a JSON array with an object for each measurement, containing pcr number and an array with the hashes for the various banks.

For example this is how this looks like for an auto-generated policy for the firmware, from my thinkpad here: https://paste.centos.org/view/71bd3a1c

as you can see it's kinda self-explanatory.

poettering, 2023-11-07:
TCG CEL is spec'ed here btw: https://trustedcomputinggroup.org/resource/canonical-event-log-format/

We use the JSON flavour, because everything else is unnecesary complicated I'd say.

we use a minor derivative of it, i.e. don't do the "recnum" field, since it doesn't really make sense for us. If it's there we simply ignore it.

hughsie, 2023-11-07:
> So does it make sense to you to ask vendors to supply the expected measurements their fimrware will make in TCG CEL-JSON format?

The very first question I 'll get is "why should we" -- i.e. what's the business justification? It was hard enough to get a new PCR0, and most vendors just ignored that request too.

poettering, 2023-11-07:
Well, on unattended systems it's kinda a necessity if you want to implement secure disk encryption. If we have to relax the security of the system to do firmware updates this is a non-trivial compromise on security I'd say. TPM2s allow you to do secure disk encryption for unattended systems relatively nicely, but without the PCR measurement data it's largely pointless, since it gives attackers a time window where they can access the system without PCR restrictions: whenever they see a firmware update coming.

So dunno, I'd presume that Red Hat would like to offer its customers secure, unattended, confidentiality of data at rest, but if we don't know the changes that firmware updates will induce ahead of time (and we won't without that measurement JSON stuff) you simply cannot reasonably do that, unless you tell customers to disable firmware upgrades (which of course is nonsensical, since precisely firmware upgrades are another piece in the puzzle to provide proper system security). So at least from my PoV the data is essential.

poettering, 2023-11-07:
> > I'd presume that Red Hat would like to offer its customers secure, unattended, confidentiality of data at rest
> 
> It's not Red Hat we need to convince, it's the Lenovo's, Dells, HPs of the world supplying the data.

Well, I'd presue Lenovo wants to offer their customer this too, no?

hughsie, 2023-11-07:
@CragW @mrhpearson does this sound interesting to you?

mrhpearson, 2023-11-07:
I've read thru this once....I suspect I will need to read thru it many more times and maybe need the sit-down idiots guide.

It seems like a 'good thing' and there is increased interest from customers on having better security (and the secure disk encryption is an important feature on the horizon). So definitely interested in understanding it better and supporting what is needed from an implementation point of view.

I will need to get input and buy-in from the FW team so I'd need a clearer guide/outline on what the impacts will be for them in generating and delivering their LVFS FW update.

My biggest concern is that we've already had plenty of issues around the PCR0 measurement. I have had a few tickets open related to this - including a recent one on why fwupdmgr verify isn't working on multiple platforms. I've found it really hard to get traction from the FW team to actually dig into this issues, understand them, and resolve them. I've gotten some "this should work" and "it works for Windows" but no real investigative effort or input on why we've had problems on LVFS.

Mark

hughsie, 2023-11-07:
@poettering are you going to blog about this? that might be an easier to digest thing for Mark and the various firmware teams.

superm1, 2023-11-08:
For now; I'm not convinced the specs have caught up enough to be able to do what you want to do.  There are potentially multiple values for PCR0 for a given BIOS, because the state of the system firmware already on the system is measured into PCR0  in some cases.  I wrote a little bit about this in the past here: https://github.com/fwupd/fwupd/discussions/2548.  I wrote there about what I think needs to happen (extensions to FIM and RIM specs and implementation of those by OEMS).

But putting PCR0 aside, I'm also concerned about PCR1 and other things in the OS influencing it.  Some OEMs will be measuring things like the boot variables into PCR1.  If the system is non-atomic something as simple as running `grub-install` or `efibootmgr` are going to potentially break those measurements.

I really like the defensive approach, but I think a full story needs to be made for the distribution to be very careful about what it does (PCR0 from FW updates, PCR1 from boot options, PCR2 from BIOS settings).

poettering, 2023-11-09:
> For now; I'm not convinced the specs have caught up enough to be able to do what you want to do. There are potentially multiple values for PCR0 for a given BIOS

That's entirely fine, and we happily can handle this, in fact it's built right into the pcrlock model to deal with this: since we also want to operate with multiple kernel versions in parallel and multiple boot loader versions and shim versions we can as easily deal with multiple firmware versions in parallel.

All we need is zero, one or multiple "alternatives" of the expected measurements, and we'll generate a TPM policy from it that can deal with anything.

1. If a firmware vendor provides zero alternatives, then this means no lockdown possible, and thus PCR0/PCR2 are not included in any policy (unless user creates a local one, based on the even log, via `systemd-pcrlock lock-firmware-code`)

2. If the firmware vendor provides one version, then pcrlock will lock against exactly that version. Perfect!

3. If the firmware vendor provides more than version, also fantastic, pcrlock will generate a TPM policy with PolicyOR elements so that any of the provided versions will function.

> , because the state of the system firmware already on the system is measured into PCR0 in some cases. I wrote a little bit about this in the past here: #2548. I wrote there about what I think needs to happen (extensions to FIM and RIM specs and implementation of those by OEMS).

I want much simpler data than FIM/RIM btw, I just want the expected measurements in TCG-CEL format. That's trivial.

> But putting PCR0 aside, I'm also concerned about PCR1 and other things in the OS influencing it. 

I would not expect vendors to supply expected PCR 1 TCG-CEL data, because configuration is not under vendor control, but under user control. Hence, just don't provide measurements on that, pcrlock will be fine: it detects that it cannot recognize the measurements in PCR1 and then automatically excludes it from TPM access policies it generates. (The user can call `systemd-pcrlock lock-firmware-config` if they want though, which generates a local .pcrlock file from the event log, but they only should do that if they understand the implications, i.e. that any config change in the firmware will cause their disk encryption key to become unavalable).

> Some OEMs will be measuring things like the boot variables into PCR1. If the system is non-atomic something as simple as running `grub-install` or `efibootmgr` are going to potentially break those measurements.

Well, that's entirely fine, too. As mentioned, I don'think firmware vendors should provide PCR 1 measurement prediction files.

I'd expect the data from PCR 0 and maybe PCR 2. (And maybe PCR 4, but only measurements up to the separator). And again, firmware vendors can supply multiple alternative versions of the expected measurements.

> I really like the defensive approach, but I think a full story needs to be made for the distribution to be very careful about what it does (PCR0 from FW updates, PCR1 from boot options, PCR2 from BIOS settings).

pcrlock covers all that very nicely already.

superm1, 2023-11-09:
> That's entirely fine, and we happily can handle this, in fact it's built right into the pcrlock model to deal with this

OK.

> If a firmware vendor provides zero alternatives, then this means no lockdown possible, and thus PCR0/PCR2 are not included in any policy (unless user creates a local one, based on the even log, via systemd-pcrlock lock-firmware-code)
> I want much simpler data than FIM/RIM btw, I just want the expected measurements in TCG-CEL format. That's trivial.

At least to set expectations; getting the vendors to provide what you want in this area is going to be pulling teeth.  The technical part of the problem is that there are multiple ways to construct multiple PCR0 values and there needs to be a standard way to reflect that.  This is what FIM/RIM should be used for.  That's how the data should be delivered by the OEMs.  We can build tools all day long that tear apart the data how we think it should be delivered and transform it into exactly what systemd needs.

> pcrlock covers all that very nicely already.
> Well, that's entirely fine, too. As mentioned, I don'think firmware vendors should provide PCR 1 measurement prediction files.

Until the OEMs have a standardized way to package this data and agree to do so I see it very likely that >99% of machines won't have PCR0 or PCR2 measurement logs.  And I think we'll need to be excluding PCR1 as well since it can be changed by tools in Linux that might not be providing predicted values.

DemiMarie, 2023-11-16:
Why does systemd-pcrlock need to be a command-line tool or a daemon?  Why can’t it be a library with a C API?  That would be trivial for an fwupd plugin to consume, after all.

superm1, 2023-11-16:
Actually I would prefer not another library to link to. We just evicted two in the last releases.

Would much rather see an interface to interact with.

DemiMarie, 2023-11-17:
Is it possible to store the old values in a backup NV index?

I want firmware updates to do the following automatically:

- Compute the new PCR0 and PCR2 values.
- Store them in a new NV index.
- Seal the key to the new NV index in addition to the current NV index.
- Run the update.

After reboot, if the new NV index is able to unlock the disk, the old NV index should be deleted from the TPM so that it cannot be used any more.

poettering, 2023-11-17:
> Why does systemd-pcrlock need to be a command-line tool or a daemon? Why can’t it be a library with a C API? That would be trivial for an fwupd plugin to consume, after all.

We try to get away from in-process linking for loose coupling things like this, it's security-wise always problematic. In particular as this stuff talks to the TPM, synchronously, and hence integrates badly with other stuff that runs an event loop. You want to run this stuff out-of-process.

IPC is the way to go really I think

poettering, 2023-11-17:
> Is it possible to store the old values in a backup NV index?
> 
> I want firmware updates to do the following automatically:
> 
>     * Compute the new PCR0 and PCR2 values.
> 
>     * Store them in a new NV index.
> 
>     * Seal the key to the new NV index in addition to the current NV index.

Such an A/B scheme is certainly thinkable, but so far not implemented, and kinda redundant. We have to support alternatives in the PCR policy anyway, for each component, and pcrlock supports that already fine, it is already a series of PolicyOR expressions of PolicyPCR expressions, stored in an NV index so that you can use it via a simple PolicyAuthorizeNV. What you are proposing is that you'd use another levle of PolicyOR on top of the PolicyAuthorizeNV, but that's really unnecessary, since you already have the PolicyOR alternatives in the policy itself.

What's crucial here to understand is that we have to support alternatives for each component of the boot already, i.e. multiple hashes for multiple firmware versions, combined with multiple hashes for multiple shim versions, combined with multiple hashes for multiple boot loader versions, combined with multiple hashes for kernels, and so on. So at *every* part of the boot you have to accept multiple PCR values. And that's exactly what pcrlock models. If you however only put a PolicyOR  at the very end of the chain instead of already at the beginning, then you would have to combinatorially explode all combinations, which would be ridiculously ineffecient. And haveing the PolicyOR both before and after the PolicyAuthorizeNV is kinda redundant, as mentioned.

If that makes any sense?

DemiMarie, 2023-11-17:
@poettering That does make sense, and actually works even better.  The one problem is that the TPM does not seem to have any sort of PolicyAND, so I am not sure how to express ((A OR B) AND (C OR D)).  How does one do that?

In any case, the broader point is that once the firmware upgrade has completed, the NV index should be changed to block booting with old firmware versions.

poettering, 2023-11-18:
If you submit two PolicyXYZ ops to the tpm2 they are combined in an AND, hence no need for an explicit PolicyAND.

poettering, 2023-11-20:
> In any case, the broader point is that once the firmware upgrade has completed, the NV index should be changed to block booting with old firmware versions.

basically, we'd expect fwupd to remove the old .pcrlock files it installed into /var/lib/pcrlock.d/ when it wants to invalidated them and then simply re-run the pcrlock policy generation. pcrlock is pretty dumb if you so will: it just takes the pcrlock data it finds, distills a policy from it, stores it in the nvindex. if you want to add allowlisted components you drop in a .pcrlock file for it, if you want to undo that you remove the .pcrlock file again.

DemiMarie, 2023-11-20:
> > Why does systemd-pcrlock need to be a command-line tool or a daemon? Why can’t it be a library with a C API? That would be trivial for an fwupd plugin to consume, after all.
> 
> We try to get away from in-process linking for loose coupling things like this, it's security-wise always problematic.

That makes sense.  Just because something can specify PCR values does not mean it should have privileges to e.g. access disk encryption keys or perform cryptographic operations using the TPM.

> In particular as this stuff talks to the TPM, synchronously, and hence integrates badly with other stuff that runs an event loop. You want to run this stuff out-of-process.

This tells me that the TPM APIs are badly designed (they should be asynchronous).

poettering, 2023-11-23:
> This tells me that the TPM APIs are badly designed (they should be asynchronous).

The provide both a synchronous and an asynchronous interface. But frankly, the async stuff is messy to use, and I think it actually makes sense to split this out into an async IPC service which internally does things synchronously.

TobiPeterG, 2025-01-02:
Has there been any progress here? :)

hughsie, 2025-05-04:
@TobiPeterG I think someone interested in the feature needs to drive it; we're all being told to work on other things at the moment so any additional help would be most welcome.