fwupd/fwupd #6323: Host Security Events full of "Supported CPU changed: Invalid → Valid"

iyanmv, 2023-11-05:
**Describe the bug**
I think since last update to 1.9.7, the Host Security Events list when running `fwupdmgr security` is full of events `Supported CPU changed: Invalid → Valid`.

```
$fwupdmgr security
Host Security ID: HSI:4 (v1.9.7)

HSI-1
✔ BIOS firmware updates:         Enabled
✔ MEI key manifest:              Valid
✔ Platform debugging:            Disabled
✔ SPI BIOS region:               Locked
✔ SPI lock:                      Enabled
✔ SPI write:                     Disabled
✔ Supported CPU:                 Valid
✔ TPM empty PCRs:                Valid
✔ TPM v2.0:                      Found
✔ UEFI bootservice variables:    Locked
✔ UEFI platform key:             Valid
✔ UEFI secure boot:              Enabled
✔ csme manufacturing mode:       Locked
✔ csme override:                 Locked
✔ csme v0:15.0.45.2411:          Valid

HSI-2
✔ BIOS rollback protection:      Enabled
✔ IOMMU:                         Enabled
✔ Intel BootGuard:               Enabled
✔ Intel BootGuard ACM protected: Valid
✔ Intel BootGuard OTP fuse:      Valid
✔ Intel BootGuard verified boot: Valid
✔ Intel GDS mitigation:          Enabled
✔ Platform debugging:            Locked
✔ TPM PCR0 reconstruction:       Valid

HSI-3
✔ Intel BootGuard error policy:  Valid
✔ Pre-boot DMA protection:       Enabled
✔ Suspend-to-idle:               Enabled
✔ Suspend-to-ram:                Disabled

HSI-4

Runtime Suffix -!
✔ Linux kernel:                  Untainted
✔ Linux kernel lockdown:         Enabled
✔ Linux swap:                    Encrypted
✔ fwupd plugins:                 Untainted

This system has HSI runtime issues.
 » https://fwupd.github.io/hsi.html#hsi-runtime-suffix

Host Security Events
  2023-11-05 09:56:02:  ✔ Supported CPU changed: Invalid → Valid
  2023-11-04 17:52:52:  ✔ Supported CPU changed: Invalid → Valid
  2023-11-04 17:50:27:  ✔ Supported CPU changed: Invalid → Valid
  2023-11-03 16:04:26:  ✔ Supported CPU changed: Invalid → Valid
  2023-11-02 20:04:48:  ✔ Supported CPU changed: Invalid → Valid
  2023-11-02 12:25:58:  ✔ Supported CPU changed: Invalid → Valid
  2023-11-02 09:01:18:  ✔ Supported CPU changed: Invalid → Valid
  2023-11-01 15:54:14:  ✔ Supported CPU changed: Invalid → Valid
  2023-11-01 08:01:20:  ✘ Supported CPU changed: Valid → Invalid
```

Also, is it possible to read all the security events? I couldn't find how.

**Steps to Reproduce**
Not sure if this is reproducible or is a combination of my CPU model, kernel version and fwupd version.

**Expected behavior**
I want to see the relevant security events

**fwupd version information**

```shell
compile   com.hughsie.libxmlb           0.3.14
compile   org.freedesktop.Passim        0.1.4
compile   com.hughsie.libjcat           0.1.14
compile   org.freedesktop.fwupd         1.9.7
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.7
runtime   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.gusb          0.4.7
runtime   org.kernel                    6.5.9-arch2-1
runtime   org.freedesktop.fwupd         1.9.7
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

**Additional questions**

- Operating system and version: Arch Linux
- Have you tried rebooting? Yes
- Is this a regression? Yes

superm1, 2023-11-05:
I suspect this is a side effect of https://github.com/fwupd/fwupd/commit/9b68d00d454d7170924d75fa34eb8deb3eff6104 if it's new to 1.9.7.

iyanmv, 2023-11-06:
Now I'm pretty sure this started with 1.9.7 and not something else (e.g. a kernel update). I have downgraded everything that got updated in between fwupd versions and I still get those messages.

So I guess two things can be done here:
1. Remove those (redundant) messages from the host security events
2. Add a command to get all security events and not just the last 9. Something like `fwupdmgr get-host-security-events`?

superm1, 2023-11-06:
Right; the messages come from the history database, here: https://github.com/fwupd/fwupd/blob/2ee11e88d55044a03ceb65bcb05dffa1a3b405ab/src/fu-engine.c#L7588

I think what's going on is that every time you restart the daemon the attribute is changing because of how it starts out `Invalid` after CPU plugin runs and goes to `Valid` after either PCI-PSP or MEI plugins run.

Is that right, you get a new message every time the daemon is restarted?

hughsie, 2023-11-06:
I can take a look at this later -- I think we actually have two attributes with the same ID (one failed, obsoleted, one success) and it's confusing various things.

superm1, 2023-11-06:
Huh, then why is there a new entry for each day?  That's confusing to me.  But yeah I think Richard is right this is probably related to logic used for obsoletes.

closed 2023-11-06 by hughsie