fwupd/fwupd #6327: FWUPD affecting dru, causing null pointer deref in the kernel

rishabh-agr, 2023-11-06:
**Describe the bug**
Seems like the start v1.9.6 fwupd is affecting the DRU in a manner that it is causing a null pointer deref.
The null ptr dereference happens in `amdgpu_dm_i2c_xfer` ([link](https://source.chromium.org/chromiumos/chromiumos/codesearch/+/main:src/third_party/kernel/upstream/drivers/gpu/drm/amd/display/amdgpu_dm/amdgpu_dm.c;drc=2de19022c5d7ff519dd5b9690f7713267bd1abfe;l=7425))
`ddc_service->link` is NULL and is dereferenced passing arguments into `dc_submit_i2c`.

**Steps to Reproduce**
```
stop fwupd
/usr/local/libexec/igt-gpu-tools/amdgpu/amd_pci_unplug --> pass
start fwupd
/usr/local/libexec/igt-gpu-tools/amdgpu/amd_pci_unplug --> panic
```

**Logs/Debugging**
Found this log message in `fwupd.log`
```failed to add device /sys/devices/pci0000:00/0000:00:08.1/0000:04:00.0/drm/card2/card2-eDP-2/drm_dp_aux0: failed to add device using on synaptics_mst: failed to open /dev/drm_dp_aux0: No such file or directory```

Issue is temporarily solved if I add this line to `fwupd.conf`
```
DisabledPlugins=mediatek_scaler;
```

**fwupd version information**
1.9.6

```shell
fwupdmgr --version
```

rishabh-agr, 2023-11-06:
version:
```
compile   org.freedesktop.fwupd         1.9.6
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.8
compile   org.freedesktop.gusb          0.4.5
runtime   org.freedesktop.gusb          0.4.5
runtime   org.kernel                    5.15.135-20720-g95a1928abe0e
runtime   org.freedesktop.fwupd         1.9.6

```

superm1, 2023-11-06:
Any chance you can test this against main to see if it's still a problem?

@hughsie I guess we should move mediatek scalar probe to the same as synaptics mst does now too.

rishabh-agr, 2023-11-07:
I tested this today and spoke with Richard about this, definitely seems to be a bug here

rishabh-agr, 2023-11-07:
Want to mention that this has also been tested with the latest kernel and the issue persists

superm1, 2023-11-07:
> I tested this today and spoke with Richard about this, definitely seems to be a bug here

OK, so tested fwupd main you mean?

superm1, 2023-11-07:
From your description I'd think this can help it from kernel side.  Can you try this?

```
diff --git a/drivers/gpu/drm/amd/display/amdgpu_dm/amdgpu_dm.c b/drivers/gpu/drm/amd/display/amdgpu_dm/amdgpu_dm.c
index 868946dd7ef1..98c85bd9136a 100644
--- a/drivers/gpu/drm/amd/display/amdgpu_dm/amdgpu_dm.c
+++ b/drivers/gpu/drm/amd/display/amdgpu_dm/amdgpu_dm.c
@@ -7433,7 +7433,7 @@ static int amdgpu_dm_i2c_xfer(struct i2c_adapter *i2c_adap,

        cmd.payloads = kcalloc(num, sizeof(struct i2c_payload), GFP_KERNEL);

-       if (!cmd.payloads)
+       if (!cmd.payloads || !ddc_service->link)
                return result;

        cmd.number_of_payloads = num;
```

superm1, 2023-11-07:
> failed to add device /sys/devices/pci0000:00/0000:00:08.1/0000:04:00.0/drm/card2/card2-eDP-2/drm_dp_aux0: failed to add device using on synaptics_mst: failed to open /dev/drm_dp_aux0: No such file or directory

Also; I'd like to better understand this device if you can share more about it.  It's the "3rd" DRM device found; and "second" eDP of that device.  Is it contrived as a result of IGT tests or is that real?

rishabh-agr, 2023-11-07:
Oh I am stupid, I didn't think you meant main branch. It seems like testing it with main solves it! We will CP that change!

Thank you for digging out the right CL! Will close the issue once I confirm it is fully solved

superm1, 2023-11-07:
OK, I think there is still a kernel bug here though too.  Let me post you one more patch.  If you can please test this against your test case.

[0001-drm-amd-display-fix-a-NULL-pointer-dereference-in-am.patch](https://github.com/fwupd/fwupd/files/13273856/0001-drm-amd-display-fix-a-NULL-pointer-dereference-in-am.patch)

rishabh-agr, 2023-11-07:
Do you want me to try out that patch against fwupd v1.9.6?

superm1, 2023-11-07:
Yes; I think we have two bugs.  One with fwupd how it was handling the DPCD probing/scan and one in the kernel.
As you have a good way to reproduce both if you can please use your failing test environment I think this is the correct patch for the kernel aspect of it.   If you can confirm it works I'll submit it upstream and get it pulled back to stable.  It looks like an unintentional regression introduced in kernel 6.0 with https://github.com/torvalds/linux/commit/22676bc500c2.

hughsie, 2023-11-07:
> @hughsie I guess we should move mediatek scalar probe to the same as synaptics mst does now too.

I guess we could, but I don't know what that would achieve -- I don't think that scaler does anything with DPCD.

superm1, 2023-11-07:
> I guess we could, but I don't know what that would achieve -- I don't think that scaler does anything with DPCD.

Oh, my memory failed me how it works because of this reported issue and turning off Mediatek scaler plugin helped.  Yeah, carry on.

ddavenport-chromium, 2023-11-07:
With the patch posted [here](https://github.com/fwupd/fwupd/issues/6327#issuecomment-1797087519) applied, I hit a panic due to a NULL pointer dereference because `ddc_service->ddc_pin` was NULL.

superm1, 2023-11-07:
> With the patch posted [here](https://github.com/fwupd/fwupd/issues/6327#issuecomment-1797087519) applied, I hit a panic due to a NULL pointer dereference because `ddc_service->ddc_pin` was NULL.

Okay thanks for trying it. Let me look more closely why that happens and respin.

superm1, 2023-11-08:
Yeah it looks like another case in `ddc_service_construct()` that can cause this.

Try this one.

[0001-drm-amd-display-fix-a-NULL-pointer-dereference-in-am.patch](https://github.com/fwupd/fwupd/files/13300218/0001-drm-amd-display-fix-a-NULL-pointer-dereference-in-am.patch)

ddavenport-chromium, 2023-11-08:
The test completes without hitting a panic with the patch in [this](https://github.com/fwupd/fwupd/issues/6327#issuecomment-1802449561) comment applied.

rishabh-agr, 2023-11-09:
Thank you for the patch @superm1. Will you be submitting this to upstream? If so, can you please CC us so we can CP it to our chromium branch?

superm1, 2023-11-09:
I [submitted it yesterday](https://lore.kernel.org/amd-gfx/20231108193157.26090-1-mario.limonciello@amd.com/T/#u), I was going to ping back to this bug when it got a R-b/A-b.  
Sorry; I don't have your email address so I couldn't CC you.  Feel free to reply to it with a `Tested-by:`

rishabh-agr, 2023-11-09:
No worries, thank you for the prompt submit! :)

superm1, 2023-11-14:
The patch is reviewed and merged in internal AMD branch.  Here is the version that is accepted (it's slightly different than you tested):
https://lore.kernel.org/amd-gfx/20231113200949.2457-1-mario.limonciello@amd.com/T/#u

Feel free to FROMLIST it for your case, or it will come to stable eventually.

closed 2023-11-14 by superm1