fwupd/fwupd #6348: redfish: Unable to get valid type 42 data from SMBIOS table

vincent-tang1, 2023-11-09:
**Describe the bug**
There are two records for type 42 data from SMBIOS table in my server as below, one is **KCS** host interface, the other is **Network** (Redfish host interface)
In redfish plugin startup, from SMBIOS discovery stage, it get incorrect type 42 data as **KCS** to cause the plguin initialization failure. 
```shell
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.5.0 present.

Handle 0x0015, DMI type 42, 17 bytes
Management Controller Host Interface
        Host Interface Type: KCS: Keyboard Controller Style

Handle 0x0037, DMI type 42, 117 bytes
Management Controller Host Interface
        Host Interface Type: Network
        Device Type: <OUT OF SPEC>
        Vendor ID: 0x11:0x09:0x18:0x14
        Protocol ID: 04 (Redfish over IP)
                Service UUID: 894feb65-2ca8-4378-a5d3-41c78ff5b59f
                Host IP Assignment Type: Static
                Host IP Address Format: IPv4
                IPv4 Address: 10.234.147.2
                IPv4 Mask: 255.255.255.0
                Redfish Service IP Discovery Type: Static
                Redfish Service IP Address Format: IPv4
                IPv4 Redfish Service Address: 10.234.147.1
                IPv4 Redfish Service Mask: 255.255.255.0
                Redfish Service Port: 443
                Redfish Service Vlan: 0
                Redfish Service Hostname:
```

**Steps to Reproduce**
1. starting fwupd daemon
2. In redfish plugin startup, from SMBIOS discovery stage,  get incorrect type 42 data as KCS to cause the redfish plugin startup abort

**Root Cause**
1. The plugin always get first record from type 42 list to cause this issue

**Expected behavior**
1. The plugin can dynamically find correct redfish (Network) host interface data from type 42 list

**fwupd version information**
```shell
fwupdmgr --version

compile   com.hughsie.libxmlb           0.3.6
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.9.7
runtime   org.freedesktop.fwupd-efi     1.9.7
compile   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.fwupd         1.9.7
runtime   org.kernel                    5.15.0-67-generic
```

hughsie, 2023-11-09:
What's the output of `sudo fwupdtool --plugins redfish get-devices -vv` please -- I'm expecting to see something about `RedfishSmbiosType42` in the logs.

vincent-tang1, 2023-11-09:
not sure it is what you want

```shell
  anchor_str: _SM3_
  entry_point_csum: 0xf3
  entry_point_len: 0x18
  smbios_major_ver: 0x3
  smbios_minor_ver: 0x5
  smbios_docrev: 0x0
  entry_point_rev: 0x1
  reserved0: 0x0
  structure_table_len: 0x1337
  structure_table_addr: 0x76ebb000
17:01:24.079 FuStruct             SmbiosStructure:
  type: 0x0
  length: 0x1a
  handle: 0x0
17:01:24.079 FuStruct             SmbiosStructure:
  type: 0x1
  length: 0x1b
  handle: 0x1
17:01:24.079 FuStruct             SmbiosStructure:
  type: 0x2
  length: 0xf
  handle: 0x2
17:01:24.079 FuStruct             SmbiosStructure:
  type: 0x3
  length: 0x16
  handle: 0x3
17:01:24.079 FuStruct             SmbiosStructure:
  type: 0xb
  length: 0x5
  handle: 0x4
17:01:24.079 FuStruct             SmbiosStructure:
  type: 0x20
  length: 0xb
  handle: 0x5
17:01:24.079 FuStruct             SmbiosStructure:
  type: 0x29
  length: 0xb
  handle: 0x6
17:01:24.079 FuStruct             SmbiosStructure:
  type: 0x29
  length: 0xb
  handle: 0x7
17:01:24.079 FuStruct             SmbiosStructure:
  type: 0x2d
  length: 0x1a
  handle: 0x8
17:01:24.079 FuStruct             SmbiosStructure:
  type: 0x26
  length: 0x12
  handle: 0x13
17:01:24.079 FuStruct             SmbiosStructure:
  type: 0x2d
  length: 0x1a
  handle: 0x14
17:01:24.079 FuStruct             SmbiosStructure:
  type: 0x2a
  length: 0x11
  handle: 0x15
17:01:24.079 FuStruct             SmbiosStructure:
  type: 0x85
  length: 0xc
  handle: 0x16
17:01:24.079 FuStruct             SmbiosStructure:
  type: 0x10
  length: 0x17
  handle: 0x17
17:01:24.079 FuStruct             SmbiosStructure:
[... 153 lines cut ...]
17:01:24.079 FuContext            SMBIOS Family=Server
17:01:24.079 FuContext            SMBIOS ProductName=FWA-6172
17:01:24.079 FuContext            SMBIOS ProductSku=FWA-6172
17:01:24.079 FuContext            SMBIOS BiosVendor=American Megatrends Inc./Advantech
17:01:24.079 FuContext            SMBIOS BiosVersion=6172000SF60V100
17:01:24.079 FuContext            SMBIOS BiosMajorRelease=05
17:01:24.079 FuContext            SMBIOS BiosMinorRelease=1d
17:01:24.080 FuContext            SMBIOS FirmwareMajorRelease=ff
17:01:24.080 FuContext            SMBIOS FirmwareMinorRelease=ff
17:01:24.080 FuContext            SMBIOS BaseboardManufacturer=Advantech
17:01:24.080 FuContext            SMBIOS BaseboardProduct=FWA-6172
```
**as you can see, type 42 first record as** (KCS)

17:01:24.079 FuStruct             SmbiosStructure:
  type: 0x2a
  length: 0x11
  handle: 0x15

**second record as**  (Network)

17:01:24.079 FuStruct             SmbiosStructure:
  type: 0x2a
  length: 0x75
  handle: 0x37

vincent-tang1, 2023-11-09:
some finding, 

fu_context_get_smbios_data() -> fu_smbios_get_data()

for fu_smbios_get_data(), if this function find first matched interface type, it will return corresponding data directly.

hughsie, 2023-11-09:
Yup, I've got a patch that makes those into an array type.

vincent-tang1, 2023-11-09:
Actually, this Interface Type KCS is AMI-BIOS built-in content

closed 2023-11-12 by hughsie

vincent-tang1, 2023-11-15:
after verified, it is work, thanks