fwupd/fwupd #6362: T14 AMD Gen3 - NixOS BIOS Update

RiSKeD, 2023-11-10:
**Describe the bug**
Trying to update the BIOS version on the T14 Gen3 from version 1.10 upwards under nixOS via fwupdmgr (by file) results in the capsule file not being found, despite being present inside the boot volume:
```shell
cat /sys/firmware/efi/efivars/FWUPDATE_DEBUG_LOG-0abba7dc-e516-4167-bbf5-4d9d1c739416

../efi/fwupdate.c:554:efi_main(): fwupd-efi version 1.3
../efi/fwupdate.c:160:fwup_populate_update_table(): Found update fwupd-30fbf24a-e1da-4840-84f6-6d7deb630e1d-0
../efi/fwupdate.c:587:efi_main(): Adding new capsule
../efi/fwupdate.c:200:fwup_search_file(): Searching Device Path: HD(1,GPT,1C06F03B-704E-4657-B9CD-681A087A2FDC)/\EFI\nixos\fw\fwupd-30fbf24a-e1da-4840-84f6-6d7deb630e1d.cap...
../efi/fwupdate.c:220:fwup_search_file(): Device Path prepared: HD(1,GPT,1C06F03B-704E-4657-B9CD-681A087A2FDC)
../efi/fwupdate.c:230:fwup_search_file(): Device supporting SFSP: PciRoot(0)/Pci(0x14,0x0)/Usb(0x6,0x2)/HD(1,GPT,1C06F03B-704E-4657-B9CD-681A087A2FDC)
../efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(1,GPT,1C06F03B-704E-4657-B9CD-681A087A2FDC) and PciRoot(0)/Pci(0x14,0x0)/Usb(0x6,0x2)/HD(1,GPT,1C06F03B-704E-4657-B9CD-681A087A2FDC)
../efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(1,GPT,1C06F03B-704E-4657-B9CD-681A087A2FDC) and Pci(0x14,0x0)/Usb(0x6,0x2)/HD(1,GPT,1C06F03B-704E-4657-B9CD-681A087A2FDC)
../efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(1,GPT,1C06F03B-704E-4657-B9CD-681A087A2FDC) and Usb(0x6,0x2)/HD(1,GPT,1C06F03B-704E-4657-B9CD-681A087A2FDC)
../efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(1,GPT,1C06F03B-704E-4657-B9CD-681A087A2FDC) and HD(1,GPT,1C06F03B-704E-4657-B9CD-681A087A2FDC)
../efi/fwupdate.c:242:fwup_search_file(): Match up! Returning \EFI\nixos\fw\fwupd-30fbf24a-e1da-4840-84f6-6d7deb630e1d.cap
../efi/fwupdate.c:302:fwup_open_file(): Found device
../efi/fwupdate.c:309:fwup_open_file(): Found volume
../efi/fwupdate.c:314:fwup_open_file(): Could not open file '\EFI\nixos\fw\fwupd-30fbf24a-e1da-4840-84f6-6d7deb630e1d.cap': Not Found
```
But approx. 1 out of 5 times it does work. We were unable to determine the cause as the setup should be identical with each execution:

**fwupd version information**
```
fwupdmgr --version

compile   org.freedesktop.fwupd         1.8.15
compile   com.hughsie.libxmlb           0.3.10
compile   com.hughsie.libjcat           0.1.13
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.3.10
runtime   com.hughsie.libjcat           0.1.13
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    6.1.55
runtime   org.freedesktop.fwupd         1.8.15
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
According to nixOS pkg specification: 
```
  services.fwupd.enable = true;
```

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices

LENOVO 21AJS02800
│
├─Unknown Device:
│     Device ID:          6b5a06645e27c4e94b6e67d5185c46d11ec9ad67
│     GUID:               c96751f4-16e8-59ea-975d-69d3dac88d7d ← GPIO\ID_INTC1055:00
│   
├─12th Gen Intel Core™ i7-1260P:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x00000423
│ │   Vendor:             Intel
│ │   GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│ │                       30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│ │                       ab855c04-4ff6-54af-8a8a-d8193daa0cd8 ← CPUID\PRO_0&FAM_06&MOD_9A
│ │                       3ebbde86-d03e-549a-a8fd-02ebf9aa537a ← CPUID\PRO_0&FAM_06&MOD_9A&STP_3
│ │   Device Flags:       • Internal device
│ │ 
│ ├─USB4 host controller:
[... 417 lines cut ...]

/boot/nixos/fw:
fwupd-30fbf24a-e1da-4840-84f6-6d7deb630e1d.cap

```

- Operating system and version:
```shell
cat /etc/os-release 
BUG_REPORT_URL="https://github.com/NixOS/nixpkgs/issues"
BUILD_ID="23.05.20230924.261abe8"
DOCUMENTATION_URL="https://nixos.org/learn.html"
HOME_URL="https://nixos.org/"
ID=nixos
LOGO="nix-snowflake"
NAME=NixOS
PRETTY_NAME="NixOS 23.05 (Stoat)"
SUPPORT_END="2023-12-31"
SUPPORT_URL="https://nixos.org/community.html"
VERSION="23.05 (Stoat)"
VERSION_CODENAME=stoat
VERSION_ID="23.05"
```
- SecureBoot is disabled.
- Boot Order Lock is disabled.

hughsie, 2023-11-10:
> compile   org.freedesktop.fwupd         1.8.15
> runtime   org.freedesktop.fwupd-efi     1.3

That's kinda old -- can you try with fwupd 1.9.x and fwupd-efi 1.4 please.

superm1, 2023-12-21:
re-open if you have more details

closed 2023-12-21 by superm1

RiSKeD, 2023-12-22:
My setup to the versions constrained me, but a newer fwupd version did the trick.