fwupd/fwupd #6366: Cannot upgrade/downgrade the HP Thunderbolt Dock G4 firmware via LVFS when the first time run.

RuminDeng, 2023-11-13:
**Describe the bug**
First time to upgrade/downgrade the HP Thunderbolt Dock G4 firmware via LVFS got Installation successfully information popped up, but the firmware didn't be updated in fact. Need to upgrade/downgrade again.

**Steps to Reproduce**
1.Install latest clean Fedora OS.
2.Copy the hp-ws-embargo.conf file to ~/.root/etc/fwupd/remotes.d 
3.Connect to a valid network.
4.Input "sudo dnf upgrade" in "Terminal" to upgrade package list.
5.Input "fwupdmgr --version" to check fwupd version.
6.Input "sudo dnf install gnome-firmware".
7.Input "$sudo fwupdmgr --force refresh" and "$fwupdmgr enable-remote lvfs-testing" to updating LVFS.
8.Attach dock to the Host, click "Firmware" icon in Application list/Activities to launch firmware update interface.
9.Select Dock: HP Thunderbolt Dock G4.
10.Select the latest OCI version v1.4.20.0 under "Available Releases" to upgrade.
11.Click "Install" option, enter account password then press Authenticate button.
12.The process bar will show Downloading and Write.
13.After finish will pop up "Installation successful" message, but check the Dock firmware version still show previous one.
14.Repet step 11~13 again, Firmware can be upgrade successful.

**Expected behavior**
Dock Firmware can be upgraded successfully by the first time.

**fwupd version information**
fwupdmgr fwupd: 1.9.7
OS: Fedora Linux 39(Workstation Edition)
gonme -firmware: v43.2

<summary>**fwupd device information**</summary>

HP Thunderbolt Dock G4

**Additional questions**

- Operating system and version: Fedora Linux 39(Workstation Edition)
- Have you tried rebooting?  Yes. Got same result.
- Is this a regression?  Seems not.

Attach the verbose log:
[gnome-firmware.log](https://github.com/fwupd/fwupd/files/13333642/gnome-firmware.log)

Attach the capture for process:
[image]
[image]
[image]
[image]

hughsie, 2023-11-13:
If you use `fwupdmgr install` do you see the same bug? i.e. is this a GUI issue or a upgrade issue?

RuminDeng, 2023-11-14:
> If you use `fwupdmgr install` do you see the same bug? i.e. is this a GUI issue or a upgrade issue?
Yes. Got same result by "fwupdmgr install".
1st time will complete within 3 mins, 2nd time will take over 10 mins, and update firmware successfully.

hughsie, 2023-11-21:
I've just tried, and I see the same behaviour on the G5 dock too. Investigating.

closed 2023-11-21 by hughsie

RuminDeng, 2023-11-22:
Hi Hughsie,
May I know how can I get the fix on my platforms? 
Need to wait for the next official release or can upgrade directly?
Thanks~

hughsie, 2023-11-22:
It'll be in the next release. If you run fedora I have a copr you can use.