fwupd/fwupd #6370: redfish: unable to write data into config file

vincent-tang1, 2023-11-14:
**Pre-condition**
use **ipmi-create-user** flag in redfish .quirk

**Describe the question**
when ipmi creating user via ipmi finished successfully, and then these crendential info will be save to configuration file (/usr/local/etc/fwupd/fwupd.conf), As result, this operation will fail because log shows "no writeable config" from below diff => ( vincent ------- fu_redfish_plugin_ipmi_create_user : 5-1, error: no writable config)
```
diff --git a/plugins/redfish/fu-redfish-plugin.c b/plugins/redfish/fu-redfish-plugin.c
index 45f6791c5..877619231 100644
--- a/plugins/redfish/fu-redfish-plugin.c
+++ b/plugins/redfish/fu-redfish-plugin.c

        /* success */
        if (!fu_plugin_set_config_value(plugin, "UserUri", uri, error))
+       {
+               g_critical("vincent ------- %s : 5-1, error: %s", __func__, (*error)->message);
                return FALSE;
+       }
        if (!fu_plugin_set_config_value(plugin, "Username", username_fwupd, error))
                return FALSE;
        if (!fu_plugin_set_config_value(plugin, "Password", password_new, error))
                return FALSE;
```

but after edit fwupd daemon permission as below
```
# systemctl edit fwupd.service

[Service]
ProtectSystem=off
...
...
```

As result, it can work and the above configuation file can successfully be written.
```
For example : 
cat /usr/local/etc/fwupd/fwupd.conf

[fwupd]
DisabledPlugins=test;test_ble
OnlyTrusted=true
AllowEmulation=false

[redfish]
UserUri=/redfish/v1/AccountService/Accounts/7
Username=fwupd
Password=pC41Zs890K0dFvg
```

Is this behavior  expection ?  Do we have to edit fwupd deamon to acheve this, or existing other methods ? thanks! 

**fwupd version information**
```
compile   com.hughsie.libxmlb           0.3.6
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.9.8
runtime   org.freedesktop.fwupd-efi     1.9.7
compile   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.fwupd         1.9.8
runtime   org.kernel                    5.15.0-67-generic
```

hughsie, 2023-11-14:
Ahh, I think we only do the right thing if installing with prefix /usr -- and you're using /usr/local/ and not setting the sysconfdir I think -- e.g.:

    #pull configuration/cache/state from /etc and /var only if prefix is /usr
    if get_option('prefix') == '/usr'
      dynamic_options += 'ConfigurationDirectory=fwupd'
      dynamic_options += 'StateDirectory=fwupd'
      dynamic_options += 'CacheDirectory=fwupd'
    endif

I'll see if we can work around this with ReadWritePaths. PR incoming!

hughsie, 2023-11-14:
@vincent-tang1 does the wip/hughsie/ProtectSystem branch work?

closed 2023-11-14 by hughsie

vincent-tang1, 2023-11-15:
it is work, thanks