fwupd/fwupd #6371: redfish: a requirement for adding OEM ipmi command

vincent-tang1, 2023-11-14:
**Pre-condition**
use "ipmi-create-user" flag in redfish.quirk

**Describe the question**
There is a requirement. Is it chance to merge below patch about ipmi oem command (enable redfish user permission) using HWID condition ('Advantech'  manufacture) into the github community ?
```
diff --git a/plugins/redfish/fu-ipmi-device.c b/plugins/redfish/fu-ipmi-device.c
index 0e44daafe..af7544ace 100644
--- a/plugins/redfish/fu-ipmi-device.c
+++ b/plugins/redfish/fu-ipmi-device.c
@@ -686,6 +686,35 @@ fu_ipmi_device_set_user_priv(FuIpmiDevice *self,
        return TRUE;
 }

+gboolean
+fu_redfish_device_set_user_group_redfish_enable_advantech(FuIpmiDevice *self, guint8 user_id, GError **error)
+{
+    const guint8 req[] = {0x39, 0x28, 0x0, user_id, 0x3, 0x1};
+    guint8 resp[0x3] = {0};
+    gsize resp_len = 0;
+
+    g_return_val_if_fail(FU_IS_IPMI_DEVICE(self), FALSE);
+    g_return_val_if_fail(user_id != 0x0, FALSE);
+
+    /* run transaction */
+    if (!fu_ipmi_device_transaction(self,
+                    0x2e,
+                    0x08,
+                    req,
+                    sizeof(req),
+                    resp,
+                    sizeof(resp),
+                    &resp_len,
+                    FU_IPMI_DEVICE_TIMEOUT,
+                    error)) {
+        g_prefix_error(error, "failed to set user %02x group enable: ", user_id);
+        return FALSE;
+    }
+
+    /* success */
+    return TRUE;
+}
+
 static void
 fu_ipmi_device_init(FuIpmiDevice *self)
 {
diff --git a/plugins/redfish/fu-ipmi-device.h b/plugins/redfish/fu-ipmi-device.h
index 1f08c0b48..7bdf58e7e 100644
--- a/plugins/redfish/fu-ipmi-device.h
+++ b/plugins/redfish/fu-ipmi-device.h
@@ -33,3 +33,7 @@ fu_ipmi_device_set_user_priv(FuIpmiDevice *self,
                             guint8 priv_limit,
                             guint8 channel,
                             GError **error);
+gboolean
+fu_redfish_device_set_user_group_redfish_enable_advantech(FuIpmiDevice *self,
+                guint8 user_id,
+                GError **error);

diff --git a/plugins/redfish/fu-redfish-plugin.c b/plugins/redfish/fu-redfish-plugin.c
index 45f6791c5..877619231 100644
--- a/plugins/redfish/fu-redfish-plugin.c
+++ b/plugins/redfish/fu-redfish-plugin.c
@@ -365,9 +369,16 @@ fu_redfish_plugin_ipmi_create_user(FuPlugin *plugin, GError **error)
                return FALSE;
        if (!fu_ipmi_device_set_user_password(device, user_id, password_tmp, error))
                return FALSE;
+       /* OEM specific */
+       if (fu_context_has_hwid_guid(fu_plugin_get_context(plugin),
+                               "18789130-a714-53c0-b025-fa93801d3995")) { /* HWID as 'Advantech' manufacture */
+               if (!fu_redfish_device_set_user_group_redfish_enable_advantech(device, user_id, error))
+                       return FALSE;
+       }
        fu_redfish_backend_set_username(self->backend, username_fwupd);
        fu_redfish_backend_set_password(self->backend, password_tmp);

```

**Note:**
The following GUID is chosen to be used
```
# fwupdtool hwids
Loading…                 [*                                      ]
Computer Information
--------------------
...
Manufacturer: Advantech
...
Hardware IDs
------------
...
{18789130-a714-53c0-b025-fa93801d3995}   <- Manufacturer
...
```

**fwupd version information**
```shell
compile   com.hughsie.libxmlb           0.3.6
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.9.8
runtime   org.freedesktop.fwupd-efi     1.9.7
compile   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.fwupd         1.9.8
runtime   org.kernel                    5.15.0-67-generic
```

hughsie, 2023-11-14:
@vincent-tang1 I've added you as a contributor - can you accept the invite, make sure you've run ./contrib/setup and then push to something like /wip/advantech/ipmi on the main remote and create a pull request please.

hughsie, 2023-11-14:
Sure, that makes sense too -- please add as a pull request and I can approve it and merge it.

vincent-tang1, 2023-11-15:
> @vincent-tang1 I've added you as a contributor - can you accept the invite, make sure you've run ./contrib/setup and then push to something like /wip/advantech/ipmi on the main remote and create a pull request please.

Thanks, i already received the invitation, but seems like that my name can not be found from contributor member list

i also tried to push the commit, but seems like authentication fail
```
root@FWA-6172-efi:~/src/fwupd_push/fwupd# git log
commit 3aefc85d37744fef62c2d3b08b4fdd10f91b0881 (HEAD -> wip/advantech/ipmi)
Author: vincent-tang1 <vincent.tang@advantech.com.tw>
Date:   Wed Nov 15 03:11:26 2023 +0000

    redfish: Use 'Advantech' IPMI OEM command to enable redfish user group permission

    See https://github.com/fwupd/fwupd/issues/6371

root@FWA-6172-efi:~/src/fwupd_push/fwupd# git branch
  main
* wip/advantech/ipmi

root@FWA-6172-efi:~/src/fwupd_push/fwupd# git push --set-upstream origin wip/advantech/ipmi
Username for 'https://github.com': vincent-tang1
Password for 'https://vincent-tang1@github.com':
remote: Support for password authentication was removed on August 13, 2021.
remote: Please see https://docs.github.com/en/get-started/getting-started-with-git/about-remote-repositories#cloning-with-https-urls for information on currently recommended modes of authentication.
fatal: Authentication failed for 'https://github.com/fwupd/fwupd.git/'
```

Could you help to check it, thanks

hughsie, 2023-11-15:
> i also tried to push the commit, but seems like authentication fail

Looks like you're using http -- you can change to authenticated by editing `.git/config` and using:

    [remote "origin"]
            url = git@github.com:fwupd/fwupd.git
            fetch = +refs/heads/*:refs/remotes/origin/*

Does that help?

vincent-tang1, 2023-11-15:
> Looks like you're using http -- you can change to authenticated by editing `.git/config` and using:
>
> ```
> [remote "origin"]
>         url = git@github.com:fwupd/fwupd.git
>         fetch = [...]

after modified, below show some error, not sure if you have some idea? thanks
```
root@FWA-6172-efi:~/src/fwupd_push/fwupd# git push --set-upstream origin wip/advantech/ipmi
The authenticity of host 'github.com (20.27.177.113)' can't be established.
ED25519 key fingerprint is SHA256:+DiY3wvvV6TuJJhbpZisF/zLDA0zPMSvHdkr4UvCOqU.
This key is not known by any other names
Are you sure you want to continue connecting (yes/no/[fingerprint])? yes
Warning: Permanently added 'github.com' (ED25519) to the list of known hosts.
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

```

vincent-tang1, 2023-11-15:
> @vincent-tang1 I've added you as a contributor - can you accept the invite, make sure you've run ./contrib/setup and then push to something like /wip/advantech/ipmi on the main remote and create a pull request please.

about coding style check, not sure if following running from my side is correct? or do i missing any step?
```
root@FWA-6172-efi:~/src/tmp/fwupd# ./contrib/setup
Using OS profile ubuntu to setup
Install build dependencies? (y/N) N
Set up dbus activated daemon and PolicyKit actions from /usr/local? (y/N) N
Install developer-friendly **unsafe** PolicyKit rules into /etc/polkit-1/rules.d? (y/N) N
Configuring git environment
Configuring pre-commit hooks
Collecting pre-commit
  Using cached pre_commit-3.5.0-py2.py3-none-any.whl (203 kB)
Collecting identify>=1.0.0
  Using cached identify-2.5.31-py2.py3-none-any.whl (98 kB)
Collecting pyyaml>=5.1
  Using cached PyYAML-6.0.1-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (705 kB)
Collecting nodeenv>=0.11.1
  Using cached nodeenv-1.8.0-py2.py3-none-any.whl (22 kB)
Collecting virtualenv>=20.10.0
  Using cached virtualenv-20.24.6-py3-none-any.whl (3.8 MB)
Collecting cfgv>=2.0.0
  Using cached cfgv-3.4.0-py2.py3-none-any.whl (7.2 kB)
Requirement already satisfied: setuptools in ./venv/lib/python3.10/site-packages (from nodeenv>=0.11.1->pre-commit) (59.6.0)
Collecting platformdirs<4,>=3.9.1
  Using cached platformdirs-3.11.0-py3-none-any.whl (17 kB)
Collecting filelock<4,>=3.12.2
  Using cached filelock-3.13.1-py3-none-any.whl (11 kB)
Collecting distlib<1,>=0.3.7
  Using cached distlib-0.3.7-py2.py3-none-any.whl (468 kB)
Installing collected packages: distlib, pyyaml, platformdirs, nodeenv, identify, filelock, cfgv, virtualenv, pre-commit
Successfully installed cfgv-3.4.0 distlib-0.3.7 filelock-3.13.1 identify-2.5.31 nodeenv-1.8.0 platformdirs-3.11.0 pre-commit-3.5.0 pyyaml-6.0.1 virtualenv-20.24.6
Requirement already satisfied: pre-commit in ./venv/lib/python3.10/site-packages (3.5.0)
Requirement already satisfied: pyyaml>=5.1 in ./venv/lib/python3.10/site-packages (from pre-commit) (6.0.1)
Requirement already satisfied: cfgv>=2.0.0 in ./venv/lib/python3.10/site-packages (from pre-commit) (3.4.0)
Requirement already satisfied: virtualenv>=20.10.0 in ./venv/lib/python3.10/site-packages (from pre-commit) (20.24.6)
Requirement already satisfied: identify>=1.0.0 in ./venv/lib/python3.10/site-packages (from pre-commit) (2.5.31)
Requirement already satisfied: nodeenv>=0.11.1 in ./venv/lib/python3.10/site-packages (from pre-commit) (1.8.0)
Requirement already satisfied: setuptools in ./venv/lib/python3.10/site-packages (from nodeenv>=0.11.1->pre-commit) (59.6.0)
Requirement already satisfied: platformdirs<4,>=3.9.1 in ./venv/lib/python3.10/site-packages (from virtualenv>=20.10.0->pre-commit) (3.11.0)
Requirement already satisfied: filelock<4,>=3.12.2 in ./venv/lib/python3.10/site-packages (from virtualenv>=20.10.0->pre-commit) (3.13.1)
Requirement already satisfied: distlib<1,>=0.3.7 in ./venv/lib/python3.10/site-packages (from virtualenv>=20.10.0->pre-commit) (0.3.7)
pre-commit installed at .git/hooks/pre-commit
Run tests locally before pushing to remote branches? THIS WILL SLOW DOWN EVERY PUSH but reduce the risk of failing CI. (y/N) y
pre-commit installed at .git/hooks/pre-push
```

hughsie, 2023-11-15:
Hmm, that all looks okay -- maybe clang-format isn't installed perhaps?

vincent-tang1, 2023-11-16:
You can also close this ticket, thanks!

closed 2023-11-16 by hughsie