fwupd/fwupd #6372: LVFS metadata file contains incorrect version information，and how to recover it.

MankWang, 2023-11-14:
**Describe the question**
Our LVFS account maintains three embargos: Quectel, Lenovo thinkpad, and Lenovo thinkcenter. But recently it was discovered that Lenovo's devices cannot display old version information.
Looking at the metadata file, it was found that Lenovo's old version information was included in Quectel's Embargo's firmware - * * *. xml. How to recover the information of each Embargo file in this situation?

Lenovo thinkpad MetadataURI=https://fwupd.org/downloads/firmware-784231c3073fa11b10610857b7b3a7ad80a0c20d.xml.gz 
Quectel MetadataURI=https://fwupd.org/downloads/firmware-ed2940fd17678bfaaec5d3a2fbfa0227c712dd10081dc12b7917b660b8d903f4.xml.gz

[image]

**fwupd version information**
compile   org.freedesktop.fwupd         1.9.7
compile   com.hughsie.libxmlb           0.3.8
compile   com.hughsie.libjcat           0.1.14
compile   org.freedesktop.gusb          0.3.10
runtime   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.fwupd         1.9.7
runtime   org.kernel                    6.5.2

hughsie, 2023-11-14:
If the firmware is only in the Quectel embargo remote (which you shouldn't really have access to FWIW) then it means the firmware is in the Quectel vendor group and has not yet been moved to the Lenovo embargo remote. Why are you using another vendors embargo remote?

MankWang, 2023-11-14:
Because lenovo uses our product. We need to provide a new version for upgrading this product. The circled content in the screenshot above should be in the metadata file of lenovo, but I don't know why it was moved to the metadata file of Quectel. Is there a way to restore this to lenovo's metadata file? 
By the way, the version in the screenshot circled above is actually still under Lenovo's Embargo. It's just that the metadata content has been moved to Quectel's metadata file.
These versions were previously visible on lenovo computers through fwupdmgr. It was only recently discovered that old version information could not be seen through fwupdmgr.After reviewing, it was discovered that these version information had been moved to the metadata file of Quectel.

hughsie, 2023-11-14:
> Because lenovo uses our product

The way the LVFS is set up means that when a file is uploaded by a ODM it is added to their embargo remote. When the ownership is changed to the OEM it means that it is also included in the OEM remote (and also remains in the ODM metadata). When the file is moves to testing/stable the firmware is *also* added to that new metadata as well, and remains in the ODM and OEM metadata too.

I don't think Quectel has change the OEM to Lenovo in the cases you're referring to, and thus Lenovo has no access to those files without the Quectel embargo remote. I can't actually tell what firmware you're referring to, as the red box you've highlighted only contains the AppStream IDs and not the GUIDs or version numbers.

I also think you've caused a security problem by pasting in the embargo URL of the Lenovo remote into a public bug -- some of those firmwares are still under embargo.

MankWang, 2023-11-15:
Thank you for the reminder. There's no problem now. Thank you.

closed 2023-11-15 by MankWang