fwupd/fwupd #6377: ThinkPad P1 Gen5 - Cannot Update Firmware

oeymere, 2023-11-14:
**Describe the question**
I have been trying to update the firmware on a new ThinkPad P1 Gen5 laptop running Fedora 38 and just updated to 39.  I am not sure if this is a bug or configuration issue so I am starting this with a question rather than a bug report.

When I run fwupdmgr update  I get the expected prompt to reboot, I accept and the laptop reboots.  On reboot I see the message "Please wait the firmware is updated" and a progress bar.  The progress bar never moves past 0% and eventually the laptop reboots again without having applied the firmware update.  Checking the history after I see:

```
└─Intel Management Engine:
  │   Device ID:          5bfc5b91c18ef8b751d3d052af77b69ebf738038
  │   Previous version:   1.27.2176
  │   Update State:       Failed
  │   Update Error:       failed to run update on reboot
  │   Last modified:      2023-11-13 22:30
  │   GUID:               2d9ea625-256b-7882-2103-da49b102ef33
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad P1 Gen5 X1 Extreme Gen5 dTBT Vpro Corporate ME Update:
        New version:      257.27.2176
        Remote ID:        lvfs
        Release ID:       49246
        Summary:          Lenovo Lenovo ThinkPad P1 Gen5 X1 Extreme Gen5 Hybrid Corporate ME Firmware
        License:          Proprietary
        Size:             13.3 MB
        Created:          2023-07-14
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N3JRO11W
        Vendor:           Lenovo
        Release Flags:    • Trusted metadata
                          • Is upgrade
        Description:      
        Lenovo ThinkPad P1 Gen5 X1 Extreme Gen5 Corporate ME Firmware, for Hybrid Vpro
        Checksum:         68b320dd67db5717062ddbc6d1ad7e42c7c4e82d90fab41c73d87685fe6446fd
```

I have checked the BIOS settings, Secure Boot and boot lock are both disabled.  Is there any other setting I can check or something else I can try?

**fwupd version information**
Please provide the version of the daemon and client if applicable to your current installation of `fwupd`.

```shell
fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.7
compile   com.hughsie.libxmlb           0.3.14
compile   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.6
runtime   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.gusb          0.4.7
runtime   org.freedesktop.fwupd         1.9.7
runtime   org.kernel                    6.5.11-300.fc39.x86_64

```

hughsie, 2023-11-14:
> Please wait the firmware is updated

Literally that text? I think that's telling us we did everything correctly and the capsule itself isn't running properly.

oeymere, 2023-11-14:
Just rebooted to get the exact text, what it says on reboot is: "Please wait while we install a system update".  It does indeed seem like fwupd is doing the right things but I have no idea why nothing happens once that update was supposed to start.  There are no logs or errors that I can see.  It just does not do anything and, I guess, times out and gives up.