fwupd/fwupd #6379: fwupd.service dies at startup since 1.9.8

bulletmark, 2023-11-14:
fwupd got updated today on Arch Linux:
```
2023-11-02 06:56:48 fwupd 1.9.6-1 -> 1.9.7-1
2023-11-15 07:09:13 fwupd 1.9.7-1 -> 1.9.8-1
```
And since today (i.e. `fwupd` 1.9.8), it now dies on startup after a 3 min timeout. Manually trying to start it does the same thing:
```
$ systemctl status fwupd
× fwupd.service - Firmware update daemon
     Loaded: loaded (/usr/lib/systemd/system/fwupd.service; static)
     Active: failed (Result: timeout) since Wed 2023-11-15 07:51:38 AEST; 2min 2s ago
       Docs: https://fwupd.org/
    Process: 6204 ExecStart=/usr/lib/fwupd/fwupd (code=killed, signal=TERM)
   Main PID: 6204 (code=killed, signal=TERM)
        CPU: 8.094s

Nov 15 07:48:38 lt systemd[1]: Starting Firmware update daemon...
Nov 15 07:48:38 lt fwupd[6204]: 21:48:38.948 FuEngine             failed to add device usb:06:00:02:04: No device GType set
Nov 15 07:51:38 lt systemd[1]: fwupd.service: start operation timed out. Terminating.
Nov 15 07:51:38 lt systemd[1]: fwupd.service: Failed with result 'timeout'.
Nov 15 07:51:38 lt systemd[1]: Failed to start Firmware update daemon.
Nov 15 07:51:38 lt systemd[1]: fwupd.service: Consumed 8.094s CPU time.
```

hughsie, 2023-11-14:
Can you get the -vv log from the daemon please

bulletmark, 2023-11-14:
Here you go: [fwupd.log.gz](https://github.com/fwupd/fwupd/files/13357989/fwupd.log.gz)

ImmortAlexGM, 2023-11-15:
Same problem here.
FuPluginTpm          failed to load eventlog: hash algorithm 0x12 size not known

hughsie, 2023-11-15:
Do we know what `hash algorithm 0x12` is? I think someone needs to add some sizes to https://github.com/fwupd/fwupd/blob/main/plugins/tpm/fu-tpm-eventlog-common.c#L43 please.

superm1, 2023-11-15:
I don't think they're the same issue by @bulletmark and @ImmortAlexGM from the above logs.  It seems to me that @bulletmark's log shows that MST plugin got stuck trying to read stuff over and over.
@bulletmark can you please try to disable `synaptics_mst` in the `/etc/fwupd/fwupd.conf` and see if that helps?

bulletmark, 2023-11-15:
@superm1 the problem goes away if I disable `synaptics_mst`.

Please note this problem only started with 1.9.8 and goes away also if I downgrade to 1.9.7 so do you agree this is a bug in 1.9.8?

superm1, 2023-11-15:
> @superm1 the problem goes away if I disable `synaptics_mst`.
> 
> Please note this problem only started with 1.9.8 and goes away also if I downgrade to 1.9.7 so do you agree this is a bug in 1.9.8?

Yes that's exactly what I thought.

In 1.9.8 if you have it enabled does it go away if you unplug external stuff (monitors or docks)? If so what stuff is plugged in that breaks it?

bulletmark, 2023-11-16:
If I unplug both of my DP connected monitors then the problem goes away.

superm1, 2023-11-16:
Specifically monitors connected to a dock? What dock?

@hughsie 
I'm wondering if this is a regression from probing cascaded synaptics hubs from all the new dp aux stuff in 1.9.8.

bulletmark, 2023-11-16:
@superm1 the dock that is reported in that log, i.e. Dell WD19TB.

superm1, 2023-11-16:
And what monitors connected to it?

Do you by chance know if they might have MST hubs internally?

bulletmark, 2023-11-16:
The monitors are both Dell U2719D model. Not sure about the MST question but searching Google says "It has a DisplayPort Out connection that supports Multi-Stream Transport (MST), which can be used to daisy-chain a second display" so I guess that means it does?

hughsie, 2023-11-16:
>  I'm wondering if this is a regression from probing cascaded synaptics hubs from all the new dp aux stuff in 1.9.8.

Certainly plausible -- the Synaptics stuff is still kinda fragile and it's plausible I've broken the cascade feature -- and none of my devices I have here support that so I'm flying a bit blind. I'm talking with them about it already as they wanted to add even more complexity for a new feature.

hughsie, 2023-11-16:
> U2719D

I've purchased this and will test myself on Monday.

hughsie, 2023-11-17:
@bulletmark does wip/hughsie/no-cascade "fix" the issue for you? I'm working on a much bigger proper fix (the plugin has been broken for a loooong time) but we need to do something in the meantime. I can build a fedora rpm if that helps.

bulletmark, 2023-11-17:
Applied commit 3ee8d20a6d0338171d183aa60ed73209fa1708a8 to Arch `fwupd` 1.9.8-1 source but it does not fix the problem. Same 3 minute timeout.

hughsie, 2023-11-17:
@bulletmark nah, that was just to fix the warning. You need the new branch please.

bulletmark, 2023-11-17:
Applied commit e9dd13518ec1fd345b9c75c93196866351c4b011 and it does fix the problem.

hughsie, 2023-11-17:
@bulletmark thanks -- there's a build with that fix https://copr.fedorainfracloud.org/coprs/rhughes/fwupd/ if that helps.

g6avk, 2023-11-17:
@hughsie here is my log as requested in the bohdi comments from your copr build.
[fwupd.log.tar.gz](https://github.com/fwupd/fwupd/files/13394762/fwupd.log.tar.gz)

hughsie, 2023-11-17:
@g6avk that seemed to exit with success -- did you reboot after installing the new copr build? If so, can you do `sudo strace /usr/libexec/fwupd/fwupd -vv` and get the new log please.

g6avk, 2023-11-17:
@hughsie I never rebooted, I did not think it was needed. I just rebooted and get the the same result. 
Attached is the new log as requested.
[fwupdlog2.tar.gz](https://github.com/fwupd/fwupd/files/13395346/fwupdlog2.tar.gz)

hughsie, 2023-11-17:
Okay, weird, that exited the same success too. Can you do that log again without the `strace` so I can see the timestamps please. Thanks!

superm1, 2023-11-17:
> is it possible we need to add more HWIDs for this perhaps

I think it's **very unlikely**.  The only IMU firmware I know about in the wild was Framework 3.02 BIOS.  If this is also a Phoenix system though we can certainly check using UMR to confirm.

@g6avk please run this command and get me the output:

```
# sudo umr -r amd15bf.gfx1101.regGFX_IMU_SCRATCH_0
```

@hughsie If we get `0xb012b00` then it's another one of them in the wild.  If not, this is a different problem.

g6avk, 2023-11-17:
> Okay, weird, that exited the same success too. Can you do that log again without the `strace` so I can see the timestamps please. Thanks!

@hughsie ok attached
[fwupdlog3.tar.gz](https://github.com/fwupd/fwupd/files/13395641/fwupdlog3.tar.gz)

g6avk, 2023-11-17:
@superm1 "command not found" I am unfamiliar with that command, sorry

superm1, 2023-11-17:
You'll have to install it.  Something like `sudo dnf install umr` should do the trick.

superm1, 2023-11-17:
I don't see any problems here:

```c
19:13:35.228 FuBackend            looking for /sys/devices/pci0000:00/0000:00:08.1/0000:08:00.0/drm/card1/card1-DP-1/drm_dp_aux0
19:13:35.248 FuBackend            failed to open device /sys/devices/pci0000:00/0000:00:08.1/0000:08:00.0/drm/card1/card1-DP-1/drm_dp_aux0: DPCD read failed: failed to read 14: Input/output error
```

g6avk, 2023-11-17:
@superm1 OK got it now, thanks,

Cannot open DRI name under debugfs: No such file or directory
ERROR: amdgpu.ko is loaded but /sys/kernel/debug/dri/0/name is not found
[ERROR]: Path <amd15bf.gfx1101.regGFX_IMU_SCRATCH_0> not found on this ASIC

superm1, 2023-11-17:
> @superm1 OK got it now, thanks,
> 
> Cannot open DRI name under debugfs: No such file or directory
> ERROR: amdgpu.ko is loaded but /sys/kernel/debug/dri/0/name is not found
> [ERROR]: Path <amd15bf.gfx1101.regGFX_IMU_SCRATCH_0> not found on this ASIC

Is your system not Phoenix?

g6avk, 2023-11-17:
I've never heard of that. I'm running Fedora 39.

superm1, 2023-11-17:
The APU in it must be something different.  @hughsie was suspecting the same issue that happened with Framework AMD laptops recently and that's definitely not what's going on with your system.

hughsie, 2023-11-17:
@g6avk ahhm your log only has `-v`, not `-vv` -- can we get the full one please. I see:

    19:13:30.855 FuMain               acquired name: org.freedesktop.fwupd
    19:13:35.228 FuBackend            looking for /sys/devices/pci0000:00/0000:00:08.1/0000:08:00.0/drm/card1/card1-DP-1/drm_dp_aux0

i.e. a 5 second (!) stall and the `-vv` should help narrow down what it is. I suspect it's a dpaux read. What's confusing is that the daemon actually loads fine -- e.g. `sudo systemctl restart fwupd.service` should have succeeded (after ~6 seconds) -- but fwupdmgr is timing out after 30 seconds. Maybe SELinux or something weird like that?

hughsie, 2023-11-17:
> i.e. a 5 second (!) stall

Ignore me -- that's the `#define FU_UDEV_BACKEND_DPAUX_RESCAN_DELAY 5` in `src/fu-udev-backend.c` -- i.e. I can't see anything wrong with that at all. Summing up it seems that running `sudo /usr/libexec/fwupd/fwupd` seems to work well, and when it runs in the systemd service something odd happens. I wonder if adding `-vv` to the Exec in `/usr/lib/systemd/system/fwupd.service` and then doing `sudo systemctl daemon-reload`, `sudo systemctl restart fwupd.service` and `sudo systemctl status fwupd.service`

g6avk, 2023-11-17:
@hughsie OK, here is a new log anyway. I am not quite sure what happened but this log is much much larger.
[fwupd4log.tar.gz](https://github.com/fwupd/fwupd/files/13397312/fwupd4log.tar.gz)

As mentioned on bodhi rolling back to the previous version get-devices worked without any issue, there are no selinux issues.

hughsie, 2023-11-17:
@g6avk the longer log is because `-vv` is much more verbose than `-v` -- and the `fwupd4log.tar.gz` looks completely normal to me. Can you try editing the systemd service? If the `systemd restart fwupd.service` works okay (and returns in ~2 seconds) then try `systemd stop fwupd.service` and then `fwupdmgr get-devices -vv` and see if it's failing to autostart or something. This is getting a bit weird.

hughsie, 2023-11-17:
The only thing that seems to link the people with failures [of the new build in the COPR] seems to be that all have AMD Ryzen CPUs - either laptops or desktops. I'll test with the [only] Lenovo AMD system I have here on Monday, as it could be just a coincidence.

g6avk, 2023-11-17:
@hughsie Very weird indeed. Here is the modified service file output:

fwupd.service - Firmware update daemon
     Loaded: loaded (/usr/lib/systemd/system/fwupd.service; static)
    Drop-In: /usr/lib/systemd/system/service.d
             └─10-timeout-abort.conf
     Active: active (running) since Fri 2023-11-17 20:57:17 GMT; 1min 24s ago
       Docs: https://fwupd.org/
   Main PID: 32899 (fwupd)
      Tasks: 6 (limit: 37581)
     Memory: 7.0M
        CPU: 193ms
     CGroup: /system.slice/fwupd.service
             └─32899 /usr/libexec/fwupd/fwupd -vv

Nov 17 20:57:22 fedorabox fwupd[32899]: 20:57:22.776 FuEngine             requirement 1.5.8 ge 1.9.9 -> org.freedesktop.fwupd passed
Nov 17 20:57:22 fedorabox fwupd[32899]: 20:57:22.776 FuEngine             9d131a0c-a606-580f-8eda-80587250b8d6 matched 1 releases
Nov 17 20:57:22 fedorabox fwupd[32899]: 20:57:22.776 FuEngine             279ed287-3607-549e-bacc-f873bb9838c4 was not found: indexed string '279ed287-3607-549e-bacc-f873bb9838c4' was unfound
Nov 17 20:57:22 fedorabox fwupd[32899]: 20:57:22.776 FuEngine             0a774df3-b16d-5724-a5c9-b112efaf7d19 was not found: indexed string '0a774df3-b16d-5724-a5c9-b112efaf7d19' was unfound
Nov 17 20:57:22 fedorabox fwupd[32899]: 20:57:22.776 FuEngine             3b81a98d-0b29-5185-a735-5f6cc6e1b0fc was not found: indexed string '3b81a98d-0b29-5185-a735-5f6cc6e1b0fc' was unfound
Nov 17 20:57:22 fedorabox fwupd[32899]: 20:57:22.776 FuEngine             c36b81ad-664f-50b7-a11d-3206f500bc8f was not found: indexed string 'c36b81ad-664f-50b7-a11d-3206f500bc8f' was unfound
Nov 17 20:57:22 fedorabox fwupd[32899]: 20:57:22.776 FuEngine             0a774df3-b16d-5724-a5c9-b112efaf7d19 was not found: indexed string '0a774df3-b16d-5724-a5c9-b112efaf7d19' was unfound
Nov 17 20:57:22 fedorabox fwupd[32899]: 20:57:22.776 FuEngine             3b81a98d-0b29-5185-a735-5f6cc6e1b0fc was not found: indexed string '3b81a98d-0b29-5185-a735-5f6cc6e1b0fc' was unfound
Nov 17 20:57:22 fedorabox fwupd[32899]: 20:57:22.776 FuEngine             c36b81ad-664f-50b7-a11d-3206f500bc8f was not found: indexed string 'c36b81ad-664f-50b7-a11d-3206f500bc8f' was unfound
Nov 17 20:57:22 fedorabox fwupd[32899]: 20:57:22.776 FuEngine             writing motd target /run/motd.d/85-fwupd

I did the systemd stop fwupd.service and then fwupdmgr get-devices -vv and no change, have to kill it.

Yes, your right other reports are AMD. It will be interesting what you find on your AMD system next week.

superm1, 2023-11-17:
If you disable `synaptics_mst` in `/etc/fwupd/fwupd.conf` and then try again, same thing?

fkru, 2023-11-17:
> The only thing that seems to link the people with failures [of the new build in the COPR] seems to be that all have AMD Ryzen CPUs - either laptops or desktops. I'll test with the [only] Lenovo AMD system I have here on Monday, as it could be just a coincidence.

JFYI: Since update to fwupd 1.9.8 "fwupdmgr get-devices" hangs, version 1.9.7 works fine (AMD Ryzen 7 PRO 4750U).

superm1, 2023-11-17:
> JFYI: Since update to fwupd 1.9.8 "fwupdmgr get-devices" hangs, version 1.9.7 works fine (AMD Ryzen 7 PRO 4750U).

What system and what's connected to it?  Can you please try to upgrade to the COPR linked above?

superm1, 2023-11-17:
On a Lenovo Z13 I just added the COPR and can reproduce the behavior **outside of systemd**.
The daemon doesn't look hung, but rather the client process hangs when calling `get-devices`.

I can see the daemon has:
```
Called SetHints()
got hint locale=en_US.UTF-8
Called GetPlugins()
Called SetFeatureFlags(1023)
Called GetDeices()
```
and then nothing.

I do see randomly battery levels before reported too (I called with `-vv`).

hughsie, 2023-11-17:
@superm1 gdb of both? I.e. where are they stuck? Curiouser and curiouser!

superm1, 2023-11-17:
```
WARNING: This package has not been validated, it may not work properly.

^C
Thread 1 "fwupdmgr" received signal SIGINT, Interrupt.
0x000000000040b426 in fu_util_build_device_tree_cb (n=0x46ead0, user_data=0x4481e0) at ../src/fu-util.c:503
503			if (c->data != NULL)
(gdb) bt
#0  0x000000000040b426 in fu_util_build_device_tree_cb (n=0x46ead0, user_data=0x4481e0) at ../src/fu-util.c:503
#1  0x00007ffff7b0a44f in g_node_traverse_post_order
    (node=0x471900, flags=flags@entry=G_TRAVERSE_ALL, func=func@entry=0x40b3da <fu_util_build_device_tree_cb>, data=data@entry=0x4481e0)
    at ../glib/gnode.c:619
#2  0x00007ffff7b0a41b in g_node_traverse_post_order
    (node=0x44e830, flags=G_TRAVERSE_ALL, func=0x40b3da <fu_util_build_device_tree_cb>, data=0x4481e0) at ../glib/gnode.c:609
#3  0x000000000040b5d3 in fu_util_build_device_tree (priv=0x4481e0, root=0x44e830, devs=0x464760) at ../src/fu-util.c:547
#4  0x000000000040ba7c in fu_util_get_devices (priv=0x4481e0, values=0x470b10, error=0x7fffffffd8b0) at ../src/fu-util.c:639
#5  0x00007ffff7df2b97 in fu_util_cmd_array_run
    (array=0x448420, priv=0x4481e0, command=0x7fffffffe28a "get-devices", values=0x7fffffffdf28, error=0x7fffffffd8b0)
    at ../src/fu-util-common.c:576
#6  0x0000000000419dc3 in main (argc=2, argv=0x7fffffffdf18) at ../src/fu-util.c:5498
(gdb) 
```

superm1, 2023-11-17:
It's f186405b1f61d4dad2818797f782039467f7d52f.  Confirmed reverting it fixes it.

hughsie, 2023-11-17:
I did not expect that! Is there a loop in the tree perhaps? i.e. can you share get-devices --json with that reverted please

superm1, 2023-11-17:
[devices.json](https://github.com/fwupd/fwupd/files/13397896/devices.json)

closed 2023-11-18 by superm1