fwupd/fwupd #6380: redfish: a requirement for adding 'Advantech' manufacture into redfish.quirk file

vincent-tang1, 2023-11-15:
Copy from
https://github.com/fwupd/fwupd/issues/6371#issuecomment-1809910904

**Describe the question**
There is a requirement. Add 'Advantech' manufacture into redfish.quirk file to set their own flag.  About HWID matches 'Advantech' manufacture and got from below
```
# fwupdtool hwids
Loading…                 [*                                      ]
Computer Information
--------------------
...
Manufacturer: Advantech
...
Hardware IDs
------------
...
{18789130-a714-53c0-b025-fa93801d3995}   <- Manufacturer
...
```

Add contents in redfish.quirk file as
```
...
# Advantech
[18789130-a714-53c0-b025-fa93801d3995]
Flags = wildcard-targets,ipmi-create-user
```

Will push a commit into remote branch /wip/advantech/redfish-quirk and then submit pull request to wait for reviewing and approving.

**fwupd version information**
```shell
fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.6
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.9.8
runtime   org.freedesktop.fwupd-efi     1.9.7
compile   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.fwupd         1.9.8
runtime   org.kernel                    5.15.0-67-generic
```

hughsie, 2023-11-15:
Sounds like a plan, PR please! Thanks.

vincent-tang1, 2023-11-16:
Please help to review the pushed commit, thanks!

hughsie, 2023-11-16:
Done! In the future if you include "Fixes https://github.com/fwupd/fwupd/issues/6380" in the commit message then GitHub is clever enough to close the issue when we merge the PR. Thanks!

closed 2023-11-16 by hughsie

vincent-tang1, 2023-11-16:
OK, thanks for the reminder!