fwupd/fwupd #6381: redfish: create-ipmi-user behavior improvement

vincent-tang1, 2023-11-15:
**Pre-condition**
use "ipmi-create-user" flag in redfish.quirk

**Describe the question**
When first time startup fwupd deamon, new user account via ipmi will create and named ''fwupd'' with random password. the following is saved configuration in OS and BMC user list querying.
```
# cat /usr/local/etc/fwupd/fwupd.conf
[fwupd]
DisabledPlugins=test;test_ble
OnlyTrusted=true
AllowEmulation=false

[redfish]
UserUri=/redfish/v1/AccountService/Accounts/7
Username=fwupd
Password=J6FJXMnrQsMHr0h

# ipmitool user list
ID  Name             Callin  Link Auth  IPMI Msg   Channel Priv Limit
1                    false   false      false      NO ACCESS
2   callback         false   true       true       CALLBACK
3   user             false   true       true       USER
4   operator         false   true       true       OPERATOR
5   administrator    false   true       true       ADMINISTRATOR
6   sasas            true    true       false      ADMINISTRATOR
7   fwupd            true    true       false      ADMINISTRATOR
8                    true    false      false      NO ACCESS
```

When next time startup fwupd deamon. it can directly use existing crendential from the configurarion file to communicate with BMC Redfish (No need create ipmi user again).

There are two scenarios. it might cause user inconvenient.

**Scenario 1:**
OS reinstall fwupd as below
```
ninja -C build install
```
as result, the configuration will be reload default, so just created 'fwupd' user is gone
```
# cat /usr/local/etc/fwupd/fwupd.conf
# use `man 5 fwupd.conf` for documentation
[fwupd]
DisabledPlugins=test;test_ble
OnlyTrusted=true
AllowEmulation=false
```
but 'fwupd' user account in BMC still exist
```
# ipmitool user list
ID  Name             Callin  Link Auth  IPMI Msg   Channel Priv Limit
1                    false   false      false      NO ACCESS
2   callback         false   true       true       CALLBACK
3   user             false   true       true       USER
4   operator         false   true       true       OPERATOR
5   administrator    false   true       true       ADMINISTRATOR
6   sasas            true    true       false      ADMINISTRATOR
7   fwupd            true    true       false      ADMINISTRATOR
```
From below source, when fwupd daemon startup, it will lead to the redfish plugin fail
```
    /* check for existing user, and if not then remember the first spare slot */
    for (guint8 i = 2; i < 0xFF; i++) {
        g_autofree gchar *username = fu_ipmi_device_get_user_password(device, i, NULL);
        if (username == NULL && user_id == G_MAXUINT8) {
            g_debug("KCS slot %u free", i);
            user_id = i;
            continue;
        }
        if (g_strcmp0(username, "fwupd") == 0) {
            g_set_error(error,
                    FWUPD_ERROR,
                    FWUPD_ERROR_NOT_SUPPORTED,
                    "fwupd user already exists in KCS slot %u",
                    (guint)i);
            return FALSE;
        }
    }
```

This result might confuse user. user need delete BMC 'fwupd' user to recovery it.
Do you think it need to be improved ? In my opinion, it might be better that overwrite the 'fwupd' BMC user. 

**Scenario 2:**
BMC reload default, BMC user list also be reload default, so just created 'fwupd' user is gone.
```
# ipmitool user list
ID  Name             Callin  Link Auth  IPMI Msg   Channel Priv Limit
1                    false   false      false      NO ACCESS
2   callback         false   true       true       CALLBACK
3   user             false   true       true       USER
4   operator         false   true       true       OPERATOR
5   administrator    false   true       true       ADMINISTRATOR
```
But the configuration in OS still exist 'fwupd' user
```
# cat /usr/local/etc/fwupd/fwupd.conf
[fwupd]
DisabledPlugins=test;test_ble
OnlyTrusted=true
AllowEmulation=false

[redfish]
UserUri=/redfish/v1/AccountService/Accounts/7
Username=fwupd
Password=J6FJXMnrQsMHr0h
```

When fwupd daemon startup, it will lead to the redfish plugin fail because all redfish communication using this crendential is failure.
This result might confuse user . User need remove 'fwupd' user content in the confguration to recovery it.

Do you think it need to be improved ? In my opinion, it might be better that below flow
1, check if the cendential is valid
2. if not, re-create ipmi user again 

**fwupd version information**
```shell
compile   com.hughsie.libxmlb           0.3.6
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.9.7
runtime   org.freedesktop.fwupd-efi     1.9.7
compile   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.fwupd         1.9.7
runtime   org.kernel                    5.15.0-67-generic
```

hughsie, 2023-11-15:
I agree it's not awesome.

> In my opinion, it might be better that overwrite the 'fwupd' BMC user

Yes, if you have permission to do that I think that make sense. I don't know if xclarity allows that, but I can certainly test.

> if not, re-create ipmi user again

I think that works.

vincent-tang1, 2023-11-17:
Seems like it should have the permission to do that because KCS interface can access any IPMI command.

Not sure your experimental result ? and do you have plan to improve these two situation ? thanks

hughsie, 2023-11-17:
> and do you have plan to improve these two situation

Next week I'm super busy, but I'm happy to review pull requests if that helps.

vincent-tang1, 2023-11-20:
The following is my draft patch for thw two situation. and it is workable from my side.
Maybe you can first review it before PR and then change or modify any code you think better, thanks

```
diff --git a/plugins/redfish/fu-redfish-plugin.c b/plugins/redfish/fu-redfish-plugin.c
index 639fd456f..ead58fe03 100644
--- a/plugins/redfish/fu-redfish-plugin.c
+++ b/plugins/redfish/fu-redfish-plugin.c
@@ -334,19 +334,11 @@ fu_redfish_plugin_ipmi_create_user(FuPlugin *plugin, GError **error)
        /* check for existing user, and if not then remember the first spare slot */
        for (guint8 i = 2; i < 0xFF; i++) {
                g_autofree gchar *username = fu_ipmi_device_get_user_password(device, i, NULL);
-               if (username == NULL && user_id == G_MAXUINT8) {
+               if ((username == NULL || g_strcmp0(username, "fwupd") == 0) && user_id == G_MAXUINT8) {
                        g_debug("KCS slot %u free", i);
                        user_id = i;
                        continue;
                }
-               if (g_strcmp0(username, "fwupd") == 0) {
-                       g_set_error(error,
-                                   FWUPD_ERROR,
-                                   FWUPD_ERROR_NOT_SUPPORTED,
-                                   "fwupd user already exists in KCS slot %u",
-                                   (guint)i);
-                       return FALSE;
-               }
        }
        if (user_id == G_MAXUINT8) {
                g_set_error_literal(error,
@@ -419,8 +411,10 @@ fu_redfish_plugin_startup(FuPlugin *plugin, FuProgress *progress, GError **error
        FuRedfishPlugin *self = FU_REDFISH_PLUGIN(plugin);
        g_autofree gchar *password = NULL;
        g_autofree gchar *redfish_uri = NULL;
+       g_autofree gchar *user_uri = NULL;
        g_autofree gchar *username = NULL;
        g_autoptr(GError) error_uefi = NULL;
+       g_autoptr(FuRedfishRequest) request = NULL;

        /* optional */
        if (!fu_redfish_plugin_discover_smbios_table(plugin, error))
@@ -482,9 +476,19 @@ fu_redfish_plugin_startup(FuPlugin *plugin, FuProgress *progress, GError **error
        if (fu_context_has_hwid_flag(fu_plugin_get_context(plugin), "wildcard-targets"))
                fu_redfish_backend_set_wildcard_targets(self->backend, TRUE);

+       user_uri = fu_plugin_get_config_value(plugin, "UserUri", NULL);
+       if (user_uri != NULL)
+       {
+               request = fu_redfish_backend_request_new(self->backend);
+               fu_redfish_request_perform(request,
+               user_uri,
+               FU_REDFISH_REQUEST_PERFORM_FLAG_LOAD_JSON,
+               NULL);
+       }
+
 #ifdef HAVE_LINUX_IPMI_H
        /* we got neither a type 42 entry or config value, lets try IPMI */
-       if (fu_redfish_backend_get_username(self->backend) == NULL) {
+       if (fu_redfish_backend_get_username(self->backend) == NULL || (fu_redfish_request_get_status_code(request) == 401)) {
                if (!fu_context_has_hwid_flag(fu_plugin_get_context(plugin), "ipmi-create-user")) {
                        g_set_error_literal(error,
                                            FWUPD_ERROR,
```

vincent-tang1, 2023-11-23:
After verified your soultion, it is workable, thanks

closed 2023-11-23 by hughsie