fwupd/fwupd #6392: fwupdmgr security returns multiple "Supported CPU" checks

Vringe, 2023-11-17:
**Describe the bug**
After upgrading vom `1.9.6` to `1.9.8`, `fwupdmgr security` shows, that my CPU is `valid`. But when running `fwupdmgr security --json` it actually returns two checks of which one is `not-valid` and one is `valid`.

```
    {
      "AppstreamId" : "org.fwupd.hsi.SupportedCpu",
      "Created" : 1700213051,
      "HsiLevel" : 1,
      "HsiResult" : "not-valid",
      "HsiResultSuccess" : "valid",
      "Name" : "Supported CPU",
      "Summary" : "Processor Security Checks",
      "Description" : "Each system should have tests to ensure firmware security.",
      "Plugin" : "cpu",
      "Uri" : "https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.SupportedCpu",
      "Flags" : [
        "obsoleted",
        "missing-data",
        "action-contact-oem"
      ],
      "Guid" : [
        "b9a2dd81-159e-5537-a7db-e7101d164d3f",
        "30249f37-d140-5d3e-9319-186b1bd5cac3",
        "ab855c04-4ff6-54af-8a8a-d8193daa0cd8",
        "3ebbde86-d03e-549a-a8fd-02ebf9aa537a"
      ]
    },
   {
      "AppstreamId" : "org.fwupd.hsi.SupportedCpu",
      "Created" : 1700213051,
      "HsiLevel" : 1,
      "HsiResult" : "valid",
      "Name" : "Supported CPU",
      "Summary" : "Processor Security Checks",
      "Description" : "Each system should have tests to ensure firmware security.",
      "Plugin" : "pci_mei",
      "Uri" : "https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.SupportedCpu",
      "Flags" : [
        "success"
      ]
    },
```

**Steps to Reproduce**
Run `fwupdmgr security --json`

**Expected behavior**
It should only return one check

**fwupd version information**
```shell
compile   com.hughsie.libxmlb           0.3.14
compile   org.freedesktop.Passim        0.1.4
compile   com.hughsie.libjcat           0.1.14
compile   org.freedesktop.fwupd         1.9.8
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.7
runtime   org.freedesktop.Passim        0.1.4
runtime   org.freedesktop.gusb          0.4.7
runtime   com.hughsie.libjcat           0.1.14
runtime   org.kernel                    6.6.1-arch1-1
runtime   org.freedesktop.fwupd         1.9.8
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): `pacman`

<details>

<summary>**fwupd device information**</summary>

[... 334 lines cut ...]
│                         • Device is usable for the duration of the update
│                         • Updatable
│
└─UEFI Device Firmware:
      Device ID:          a083ebc5138e5e071ef7270cc9a8280722cc7adf
      Summary:            UEFI ESRT device
      Current version:    3554875
      Vendor:             DMI:LENOVO
      Update State:       Success
      Problems:           • Device requires AC power to be connected
      GUID:               4f306bc6-322c-4959-8afe-d34ffb3159c2
      Device Flags:       • Internal device
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
                          • Updatable
```

</details>

**Additional questions**

- Operating system and version: Arch Linux
- Have you tried rebooting? Yes
- Is this a regression? No

closed 2023-11-17 by hughsie

Vringe, 2023-11-21:
@hughsie just tested 1.9.9 but the issue persists. Still two checks are returned.

hughsie, 2023-11-21:
Are you looking in the `SecurityAttributes` or `SecurityEvents` section?

Vringe, 2023-11-21:
It's two checks inside `SecurityAttributes` and two inside `SecurityEvents`

hughsie, 2023-11-21:
@Vringe can you try the wip/hughsie/depsolve-hsi branch? I can do you a fedora rpm if that helps.

Vringe, 2023-11-21:
I used fwupd-git to build it from the main branch. Seems to be fixed now. Thank you!