fwupd/fwupd #6397: Monitor Freezes When Using fwupdmgr

PhasecoreX, 2023-11-17:
**Describe the bug**
When running `sudo fwupdmgr refresh --force`, my monitor goes into a "locked" state. This state functions exactly like normal, except the hardware buttons on the monitor (power, input select, etc.) do not work, and when the screen loses signal and is supposed to sleep, it freezes on a black screen. The only way to wake it back up and to get the buttons to work again is to unplug the power and plug it back in.

**Steps to Reproduce**
1. Have an HP - 34" IPS 165Hz UWQHD HDR AMD FreeSync Gaming Monitor
2. Run the command `sudo fwupdmgr refresh --force`
3. Or, just open KDE Discover program (it uses `lvfs`, which I assume uses `fwupd`. This is how I first encountered this bug.)
4. Try to use the buttons on your monitor (they don't work)
5. Let the screen go to sleep (or unplug the DisplayPort)
6. No amount of doing anything, short of unplugging the power, will unfreeze it.

**Expected behavior**
Running `fwupdmgr` doesn't freeze my monitor.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version

compile   com.hughsie.libxmlb           0.3.14
compile   org.freedesktop.Passim        0.1.4
compile   com.hughsie.libjcat           0.1.14
compile   org.freedesktop.fwupd         1.9.8
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.7
runtime   org.freedesktop.Passim        0.1.4
runtime   org.freedesktop.gusb          0.4.7
runtime   com.hughsie.libjcat           0.1.14
runtime   org.kernel                    6.6.1-arch1-1
runtime   org.freedesktop.fwupd         1.9.8
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): `pacman`

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices

Gigabyte Technology Co., Ltd. X670 AORUS ELITE AX
│
├─Unknown Device:
│     Device ID:          5f7dee69f5c6dc34f6af5a8ecff65b51ce478670
│     Serial Number:      377306
│     GUID:               73fdc772-ade9-569d-b0a4-384759591b5e ← DRM\VEN_GSM&DEV_76FA
│   
├─AMD Ryzen 5 7600X 6-Core Processor:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x0a601203
│ │   Vendor:             Advanced Micro Devices, Inc.
│ │   GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│ │                       22f9ecf4-588d-5c0a-8326-6ebff3655c6d ← CPUID\PRO_0&FAM_19
│ │                       16dd7d28-eade-5077-adb8-0775fdfbb2e5 ← CPUID\PRO_0&FAM_19&MOD_61
│ │                       6c9f4098-a058-5cd6-a1ce-36bfe27404fb ← CPUID\PRO_0&FAM_19&MOD_61&STP_2
│ │   Device Flags:       • Internal device
│ │ 
│ ├─GPIO controller:
│ │     Device ID:        f685512aa07369c9e77742acef941d779d31e766
│ │     GUID:             37b440a9-2473-5087-a39b-db84f32a8ed8 ← GPIO\ID_AMDI0030:00
│ │   
│ └─Secure Processor:
│       Device ID:        c54ab0237d7a8db8c717b68e0be78e4374a2a079
│       Current version:  00.2b.00.4d
│       Bootloader Version:00.2b.00.4d
│       Vendor:           Advanced Micro Devices, Inc. (PCI:0x1022)
[... 112 lines cut ...]

- Operating system and version: `Arch Linux 6.6.1-arch1-1 (64-bit)`
- Have you tried rebooting? `Yes`
- Is this a regression? `Maybe? I don't know. It just broke one day, less than a month ago?`

**Other Notes**

I ran `sudo journalctl -o short-precise -f` to watch what was happening when I use `fwupdmgr`, compared my monitor to another monitor I had, and the only difference I see is my monitor had this extra line:
```
fwupd[4538]: 15:11:03.570 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:01.1/0000:01:00.0/0000:02:00.0/0000:03:00.0/drm/card1/card1-DP-2/drm_dp_aux1: invalid read buffer: size 0x80 must greater than 0x80.
```
As well as some extra `passimd` lines:
```
passimd[4549]: Called Publish(18, metadata.xml.xz, 0x0, 86400, 50)
passimd[4549]: added /var/lib/passim/data/8dbe9a30191371fd25ac43f41ac0ba897e5f414a11cc0c2176799733c241c5e5-metadata.xml.xz
passimd[4549]: resetting Passim-A9C1
passimd[4549]: adding subtype _8dbe9a30191371fd25ac43f41ac0ba897e5f414a11cc0c2176799733c241._sub._cache._tcp
passimd[4549]: adding subtype _8a2a7e6fb5b615da13eae58fdea7153e9ef4d98277633e93effb4eec97c6._sub._cache._tcp
passimd[4549]: signal_name: StateChanged (is)
passimd[4549]: signal_name: StateChanged (is)
passimd[4549]: signal_name: StateChanged (is)
```
I don't know if that helps, but that first error might be [from this code](https://github.com/fwupd/fwupd/blob/329b652a12deb1cdfeed32b85f55b4a46ca15916/plugins/mediatek-scaler/fu-mediatek-scaler-device.c#L187). Also, after the first run of `sudo fwupdmgr refresh --force`, I can run it again and nothing freezes. But at the same time, nothing really shows up in the journalctl, so I think that command isn't actually forcing anything to be updated.

Please let me know of anything else I can provide to help figure out what is happening. Thank you for your time.

superm1, 2023-11-17:
Can you please share the output for `sudo fwupdtool get-devices -vv` which should also reproduce this error?

PhasecoreX, 2023-11-17:
Yes, that did reproduce the error:

https://pastebin.com/wBDtfRBC

superm1, 2023-11-17:
@CragW it looks like this monitor might be triggering the Mediatek plugin.

```
23:36:41.607 FuPluginMediatekScaler DDC/CI write message:51 83 cc 90 01 e1 
23:36:41.660 FuPluginMediatekScaler DDC/CI write message:51 84 cc fe 91 af b7 
23:36:41.718 FuPluginMediatekScaler DDC/CI read buffer:
6e 82 02 40 fe 00 00 00 00 00 00 00 00 00 00 00 d2 d0 a4 0a f9 ab 2f 9a 2d 3a b5 ff 9d 37 b3 63 
23 6e a3 4e 58 ef d2 78 6d 3b de fd 7e 4d aa b6 0c 59 4f 74 3c f8 da 65 6e 2f b4 ab ad 7f 89 a1 
23:36:41.718 FuPluginMediatekScaler DDC/CI read report:6e 82 02 40 
23:36:41.718 FuPluginMediatekScaler found mediatek display controller: /dev/drm_dp_aux1, i2c-dev: /dev/i2c-9
23:36:41.718 FuPluginMediatekScaler DDC/CI write message:51 83 cc ff 00 8f 
23:36:41.777 FuPluginMediatekScaler DDC/CI read buffer:
6e 80 be 74 3c f8 da 65 6e 2f b4 ab ad 7f 89 a1 00 e6 d2 d0 a4 0a f9 ab 2f 9a 2d 3a b5 ff 9d 37 
b3 63 23 6e a3 4e 58 ef d2 78 6d 3b de fd 7e 4d aa b6 0c 59 4f 74 3c f8 da 65 6e 2f b4 ab ad 7f 
23:36:41.777 FuEngine             failed to add device 
```

@PhasecoreX 
Can you please experiment with disabling `mediatek-scaler` in `DisabledPlugins` within `/etc/fwupd/fwupd.conf`?

CragW, 2023-11-18:
> @CragW it looks like this monitor might be triggering the Mediatek plugin.

Yeah, mediatek plugin involved.

> 23:36:41.777 FuEngine             failed to add device 
> /sys/devices/pci0000:00/0000:00:01.1/0000:01:00.0/0000:02:00.0/0000:03:00.0/drm/card1/card1-DP-2/drm_dp_aux1: invalid read buffer: size 0x80 must greater than 0x80.

The plugin sends DDC/CI command to ask if is a supported monitor otherwise exit immediately, somewhat the buttons unresponse after the ask.

@GregLo007 do you know this monitor?

PhasecoreX, 2023-11-18:
I added `mediatek-scaler` to `/etc/fwupd/fwupd.conf` like so:
```
# use `man 5 fwupd.conf` for documentation
[fwupd]
DisabledPlugins=test;test_ble;mediatek-scaler
OnlyTrusted=true
AllowEmulation=false
```
However even after restarting my machine, doing `sudo fwupdtool get-devices -vv` still causes the freeze.

superm1, 2023-11-18:
Can you try it with "mediatek_scalar"? I might have mixed up the `-` versus `_`

PhasecoreX, 2023-11-18:
Ah, yes that works now with `mediatek_scaler`. I can now run `sudo fwupdtool get-devices -vv` and it doesn't freeze! Thank you!

I see in the output of that command now that the `PossiblePlugin:       mediatek_scaler` line is missing from the two sections, and the
```
FuMediatekScalerPlugin:
  Name:                 mediatek_scaler
  Flags:                ready
```
is also missing, as expected.

superm1, 2023-11-18:
Thanks for confirming.

@CragW
Should there be some VCP info commands to confirm specific manufacturer or model maybe earlier in the plugin probe?

CragW, 2023-11-18:
> @CragW Should there be some VCP info commands to confirm specific manufacturer or model maybe earlier in the plugin probe?

Yeh, some more info tests are required. Will check with mtk.

closed 2023-11-19 by hughsie

GregLo007, 2023-11-20:
Currently, this plug-in only supports the AIO model because of the GUID rule. Therefore, other monitors are not supported at the moment.

hughsie, 2023-11-20:
> because of the GUID rule

*which* GUID? At the moment the only GUID is the one we use for firmware stream matching *after* the device has been probed. Can you attach the output of `sudo fwupdtool hwids` on the target machine please.

CragW, 2023-11-20:
The data returned by vcp 0xc8 is yet confirmed for this controller specifically.  So before it is clear, we should add a check for both the host system and the controller type.

The latter to avoid accidents happen to the allowlisted host.