fwupd/fwupd #6404: fwupdmgr get-devices Does not respond

Zouhuo, 2023-11-18:
Since Update to 1.9.8, I cannot execute 'fwupdmgr get-devices'. The console just waits for a response and does not return to command line.

Linux kernel 6.6.1
Distro Manjaro
On 3 different hardware:
- Lenovo ThinkPad T14 AMD Ryzen 7 PRO 5850U.
- Aourus - Intel i9 9900k + ATI Radeon RX570
- Gigabyte - Intel i7 9700

superm1, 2023-11-18:
It should be fixed in main. Any chance you can try there and confirm?

@hughsie after fixing the mediatek thing this week I think it's worth a new respin with how prevalent some of these new bugs are.

Zouhuo, 2023-11-19:
> It should be fixed in main. Any chance you can try there and confirm?

Hmm, I'm a simple user, I don't have the environment set to compile. Maybe I can do it but not soon. And at the moment, are there other comments that mention the same thing or am I the only one?

hughsie, 2023-11-19:
@superm1 I agree. I'll merge the mediatek disable-by-default patch and then do a new release on Monday.

hughsie, 2023-11-19:
@Zouhuo if you're on Fedora, you can use: https://copr.fedorainfracloud.org/coprs/rhughes/fwupd/

Shadowstone12, 2023-11-19:
I've also had the same issue on a recent install of Arch. Haven't been able to find any warning messages running in verbose or in journalctl. This is on a new framework 13 AMD.

superm1, 2023-11-19:
> @superm1 I agree. I'll merge the mediatek disable-by-default patch and then do a new release on Monday.

If we can find the hwid that the shot device that needs that uses it could be on by default only for that so as to not regress that one device but if that isn't available in short notice it's pragmatic

Cc @CragW

CragW, 2023-11-19:
Does it help if mediatek plugin disabled? Adding it to `/etc/fwupd/fwupd.conf`.

```
DisabledPlugins=test;test_ble;mediatek_scaler
```

Zouhuo, 2023-11-19:
> Does it help if mediatek plugin disabled? Adding it to `/etc/fwupd/fwupd.conf`.
> DisabledPlugins=test;test_ble;mediatek_scaler

No, it's still the same without responding.
By the way, the distro I use is Manjaro, derivative of Arch

superm1, 2023-11-19:
> Does it help if mediatek plugin disabled? Adding it to `/etc/fwupd/fwupd.conf`.
> 
> ```
> DisabledPlugins=test;test_ble;mediatek_scaler
> ```
> 

The other issue helps with it disabled. This one I think is the client side issue with walking the topology.

hughsie, 2023-11-19:
> If we can find the hwid on short notice for the system that uses it

@CragW does the HWID make sense? My gut instinct is to do the release with the plugin disabled, and then we re-enable it for the next release when we either have the VCP or HWID checks.

sempiternal-aurora, 2023-11-20:
> It should be fixed in main. Any chance you can try there and confirm?

I just built from a local clone of the git repo, and encountered the same problem. `fwupdmgr get-devices` just runs, never generates output, and I cannot ctrl-C it, can only suspend the command with ctrl-Z.

superm1, 2023-11-20:
@sempiternal-aurora can you please try to add `mediatek-scaler` to the `DisabledPlugins` key in `/etc/fwupd/fwupd.conf` and try to restart the daemon?  If you can still reproduce the problem with that please share output for

`sudo fwupdtool get-devices --vv`

sempiternal-aurora, 2023-11-20:
Give me a sec, I still had fwupd installed the first time so it was running on the old daemon so I am just removing that and rebuilding the library

sempiternal-aurora, 2023-11-20:
Sorry, I had built the package wrong (this was my first time building a package from source, sorry). After fixing everything I can confirm that building from main with the mediatek plugin running `fwupdmgr get-devices` runs fine. Sorry about all this confusion.

superm1, 2023-11-20:
OK, it's fixed in 1.9.9.

closed 2023-11-20 by superm1