fwupd/fwupd #6405: fwupd.service: start operation timed out

pschichtel, 2023-11-19:
**Describe the bug**
The fwupd service doesn't start. When triggering using dbus via fwupdmgr it times out after 25 seconds and starting it via `systemctl start fwupd.service` also times out after a while.

this is from the journal:

```
Nov 19 05:12:46  systemd[1]: Starting Firmware update daemon...
Nov 19 05:12:46  fwupd[83885]: 04:12:46.660 FuPluginUefiCapsule  skipping device that failed coldplug: ESRT GUID '00000000-0000-0000-0000-000000000000' was not valid
Nov 19 05:12:47  fwupd[83885]: 04:12:47.549 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:01.0/0000:01:00.0/0000:02:00.0/0000:03:00.0/drm/card1/card1-DP-1/drm_dp_aux0: failed to add device using on synaptics_mst: failed to enable remote control: failure writing data register: failed to write 0x5 bytes on layer:0, relative_addr:0x0
Nov 19 05:15:46  systemd[1]: fwupd.service: start operation timed out. Terminating.
```

The device is a an apparently disconnected (according to the status) DP port on a vega64 gpu. Weirdly I only have 2 DP port on the card and that one is connected to a display.

I paused fwupd twice using gdb and in both cases the fwupd thread was write()'ing to fd 14:
`Thread 0x7f1e0e8a6380 (LWP 83885) "fwupd"           0x00007f1e105d106f in __GI___libc_write (nbytes=4, buf=0x7fffe62772d4, fd=14) at ../sysdeps/unix/sysv/linux/write.c:26`

strace also spammed this:

```
lseek(14, 1208, SEEK_SET)               = 1208
write(14, "\4\0\0\0", 4)                = 4
lseek(14, 1202, SEEK_SET)               = 1202
write(14, "\243", 1)                    = 1
lseek(14, 1202, SEEK_SET)               = 1202
read(14, "\0\0", 2)                     = 2
lseek(14, 1216, SEEK_SET)               = 1216
write(14, "\270\4\0\0", 4)              = 4
lseek(14, 1212, SEEK_SET)               = 1212
write(14, "\274\4\0\0", 4)              = 4
lseek(14, 1208, SEEK_SET)               = 1208
write(14, "\4\0\0\0", 4)                = 4
lseek(14, 1202, SEEK_SET)               = 1202
write(14, "\243", 1)                    = 1
lseek(14, 1202, SEEK_SET)               = 1202
read(14, "\0\0", 2)                     = 2
lseek(14, 1216, SEEK_SET)               = 1216
write(14, "\4\0\0\0", 4)                = 4
lseek(14, 1212, SEEK_SET)               = 1212
write(14, "\270\4\0\0", 4)              = 4
lseek(14, 1208, SEEK_SET)               = 1208
write(14, "\4\0\0\0", 4)                = 4
lseek(14, 1202, SEEK_SET)               = 1202
write(14, "\243", 1)                    = 1
lseek(14, 1202, SEEK_SET)               = 1202
read(14, "\0\0", 2)                     = 2
lseek(14, 1216, SEEK_SET)               = 1216
write(14, "\242", 1)                    = 1
lseek(14, 1212, SEEK_SET)               = 1212
write(14, "\262\4\0\0", 4)              = 4
lseek(14, 1208, SEEK_SET)               = 1208
write(14, "\1\0\0\0", 4)                = 4
lseek(14, 1202, SEEK_SET)               = 1202
write(14, "\243", 1)                    = 1
lseek(14, 1202, SEEK_SET)               = 1202
```

fds shortly before it timed out:

```
$ ll /proc/83885/fd
total 0
lr-x------ 1 root root 64 19. Nov 05:13 0 -> /dev/null
lrwx------ 1 root root 64 19. Nov 05:15 1 -> 'socket:[622965]'
lrwx------ 1 root root 64 19. Nov 05:15 10 -> 'anon_inode:[eventfd]'
lrwx------ 1 root root 64 19. Nov 05:15 11 -> 'socket:[620849]'
lrwx------ 1 root root 64 19. Nov 05:15 12 -> 'anon_inode:[eventfd]'
lrwx------ 1 root root 64 19. Nov 05:15 13 -> 'socket:[622193]'
lrwx------ 1 root root 64 19. Nov 05:15 14 -> /dev/drm_dp_aux2
lrwx------ 1 root root 64 19. Nov 05:15 2 -> 'socket:[622965]'
lrwx------ 1 root root 64 19. Nov 05:12 3 -> 'anon_inode:[eventfd]'
lr-x------ 1 root root 64 19. Nov 05:12 4 -> anon_inode:inotify
lrwx------ 1 root root 64 19. Nov 05:15 5 -> 'anon_inode:[eventfd]'
lrwx------ 1 root root 64 19. Nov 05:15 6 -> /var/lib/fwupd/pending.db
lrwx------ 1 root root 64 19. Nov 05:15 7 -> 'anon_inode:[eventfd]'
lrwx------ 1 root root 64 19. Nov 05:15 8 -> 'anon_inode:[timerfd]'
lrwx------ 1 root root 64 19. Nov 05:15 9 -> 'socket:[622980]'
```

**Steps to Reproduce**
For me it's just trying to start fwupd.service

**Expected behavior**
It should start

**fwupd version information**

`fwupdmgr --version` also hangs because the daemon doesn't come up, so this is all I got:

```
$ pacman -Qi fwupd
Name            : fwupd
Version         : 1.9.8-3
```

It's Arch without testing and I did several reboots since I had this. (not sure when this started happening, but definitely on an earlier 1.9.x version.

pschichtel, 2023-11-19:
output from gdb's `bt full`:

```
#0  0x00007f656465106f in __GI___libc_write (nbytes=1, buf=0x7ffe40706f44, fd=14) at ../sysdeps/unix/sysv/linux/write.c:26
        sc_ret = 1
        sc_cancel_oldtype = 0
        sc_ret = <optimized out>
        sc_ret = <optimized out>
        __arg2 = <optimized out>
        _a3 = <optimized out>
        _a1 = <optimized out>
        resultvar = <optimized out>
        __arg3 = <optimized out>
        __arg1 = <optimized out>
        _a2 = <optimized out>
#1  __GI___libc_write (fd=14, buf=buf@entry=0x7ffe40706f44, nbytes=nbytes@entry=1) at ../sysdeps/unix/sysv/linux/write.c:24
No locals.
#2  0x00007f6564d9ef2a in fu_synaptics_mst_connection_aux_node_write (error=0x7ffe407072a8, bufsz=1, buf=0x7ffe40706f44 "\242", offset=1202, self=0x55fffb017980) at ../fwupd-1.9.8/plugins/synaptics-mst/fu-synaptics-mst-connection.c:104
        title = 0x55fffb020050 "write@0x4b2"
        title = <optimized out>
#3  fu_synaptics_mst_connection_bus_write (error=0x7ffe407072a8, bufsz=1, buf=0x7ffe40706f44 "\242", offset=1202, self=0x55fffb017980) at ../fwupd-1.9.8/plugins/synaptics-mst/fu-synaptics-mst-connection.c:135
No locals.
#4  fu_synaptics_mst_connection_write (self=self@entry=0x55fffb017980, offset=offset@entry=1202, buf=buf@entry=0x7ffe40706f44 "\242", bufsz=bufsz@entry=1, error=error@entry=0x7ffe407072a8) at ../fwupd-1.9.8/plugins/synaptics-mst/fu-synaptics-mst-connection.c:198
No locals.
#5  0x00007f6564d9eb32 in fu_synaptics_mst_connection_rc_send_command_and_wait (self=self@entry=0x55fffb017980, rc_cmd=rc_cmd@entry=34, error=error@entry=0x7ffe407072a8) at ../fwupd-1.9.8/plugins/synaptics-mst/fu-synaptics-mst-connection.c:210
        cmd = 162
        buf = 0
        timer = 0x55fffb0159e0
#6  0x00007f6564d9ed31 in fu_synaptics_mst_connection_rc_set_command (self=self@entry=0x55fffb017980, rc_cmd=34, offset=offset@entry=1208, buf=buf@entry=0x7ffe40707044 "\002", bufsz=bufsz@entry=4, error=error@entry=0x7ffe407072a8) at ../fwupd-1.9.8/plugins/synaptics-mst/fu-synaptics-mst-connection.c:304
        chk = <optimized out>
        cur_length = 4
        cur_offset = 1208
        i = 0
        chunks = 0x55fffb015a20
#7  0x00007f6564d9effa in fu_synaptics_mst_connection_write (self=self@entry=0x55fffb017980, offset=offset@entry=1208, buf=buf@entry=0x7ffe40707044 "\002", bufsz=bufsz@entry=4, error=error@entry=0x7ffe407072a8) at ../fwupd-1.9.8/plugins/synaptics-mst/fu-synaptics-mst-connection.c:188
        node = <optimized out>
        result = <optimized out>
#8  0x00007f6564d9e78a in fu_synaptics_mst_connection_rc_get_command (self=self@entry=0x55fffb017980, rc_cmd=50, offset=offset@entry=1202, buf=buf@entry=0x7ffe407070e2 "", bufsz=bufsz@entry=2, error=error@entry=0x7ffe407072a8) at ../fwupd-1.9.8/plugins/synaptics-mst/fu-synaptics-mst-connection.c:348
        chk = 0x55fffb01ffe0
        cur_length = 2
        cur_offset = 1202
        i = 0
        chunks = 0x55fffb015a90
#9  0x00007f6564d9eac3 in fu_synaptics_mst_connection_read (self=self@entry=0x55fffb017980, offset=offset@entry=1202, buf=buf@entry=0x7ffe407070e2 "", bufsz=bufsz@entry=2, error=error@entry=0x7ffe407072a8) at ../fwupd-1.9.8/plugins/synaptics-mst/fu-synaptics-mst-connection.c:162
        node = <optimized out>
        result = <optimized out>
#10 0x00007f6564d9eb7b in fu_synaptics_mst_connection_rc_send_command_and_wait (self=self@entry=0x55fffb017980, rc_cmd=rc_cmd@entry=34, error=error@entry=0x7ffe407072a8) at ../fwupd-1.9.8/plugins/synaptics-mst/fu-synaptics-mst-connection.c:217
        cmd = 162
        buf = 0
        timer = 0x55fffb0292a0
#11 0x00007f6564d9ed31 in fu_synaptics_mst_connection_rc_set_command (self=self@entry=0x55fffb017980, rc_cmd=34, offset=offset@entry=1208, buf=buf@entry=0x7ffe407071e4 "\001", bufsz=bufsz@entry=4, error=error@entry=0x7ffe407072a8) at ../fwupd-1.9.8/plugins/synaptics-mst/fu-synaptics-mst-connection.c:304
        chk = <optimized out>
        cur_length = 4
        cur_offset = 1208
        i = 0
        chunks = 0x55fffb015ac0
#12 0x00007f6564d9effa in fu_synaptics_mst_connection_write (self=self@entry=0x55fffb017980, offset=offset@entry=1208, buf=buf@entry=0x7ffe407071e4 "\001", bufsz=bufsz@entry=4, error=error@entry=0x7ffe407072a8) at ../fwupd-1.9.8/plugins/synaptics-mst/fu-synaptics-mst-connection.c:188
        node = <optimized out>
        result = <optimized out>
#13 0x00007f6564d9e78a in fu_synaptics_mst_connection_rc_get_command (self=self@entry=0x55fffb017980, rc_cmd=50, offset=offset@entry=1200, buf=buf@entry=0x7ffe407072b4 "\377U", bufsz=bufsz@entry=1, error=error@entry=0x7ffe407072a8) at ../fwupd-1.9.8/plugins/synaptics-mst/fu-synaptics-mst-connection.c:348
        chk = 0x55fffb029200
        cur_length = 1
        cur_offset = 1200
        i = 0
        chunks = 0x55fffb015af0
#14 0x00007f6564d9eac3 in fu_synaptics_mst_connection_read (self=0x55fffb017980, offset=1200, buf=0x7ffe407072b4 "\377U", bufsz=1, error=0x7ffe407072a8) at ../fwupd-1.9.8/plugins/synaptics-mst/fu-synaptics-mst-connection.c:162
        node = <optimized out>
        result = <optimized out>
#15 0x00007f6564dea3d9 in fu_synaptics_mst_device_scan_cascade.isra.0 (self=self@entry=0x55fffb020780, layer=layer@entry=3 '\003', error=<optimized out>) at ../fwupd-1.9.8/plugins/synaptics-mst/fu-synaptics-mst-device.c:1216
        byte = "\377U\000"
[... 105 lines cut ...]
        checksums_blocked = 0x55fffad20c60
        error_quirks = 0x0
        error_json_devices = 0x0
[... cut ...]
            arg = G_OPTION_ARG_NONE, arg_data = 0x7ffe407078f4, description = 0x55fffa037f38 "Exit after the engine has loaded", arg_description = 0x0}, {long_name = 0x0, short_name = 0 '\000', flags = 0, arg = G_OPTION_ARG_NONE, arg_data = 0x0, description = 0x0, arg_description = 0x0}}
        socket_address = 0x0
        error = 0x0
        argv0_file = 0x55fffacd2dd0
        context = 0x55fffacd9590
        daemon = 0x55fffacd47c0
        argv0_monitor = 0x0
        memory_monitor = 0x0
```

pschichtel, 2023-11-19:
str1 at frame #27:

```
FuDpauxDevice:
  DeviceId:             b2fe055ad313c84169f183e16de69bdffbd625d0
  Flags:                none
  Version:              0.0.0
  VersionFormat:        triplet
  PhysicalId:           PCI_SLOT_NAME=0000:03:00.0
  LogicalId:            drm_dp_aux2
  BackendId:            /sys/devices/pci0000:00/0000:00:01.0/0000:01:00.0/0000:02:00.0/0000:03:00.0/drm/card1/card1-DP-3/drm_dp_aux2
  AcquiesceDelay:       2500
  PossiblePlugin:       kinetic_dp
  PossiblePlugin:       mediatek_scaler
  PossiblePlugin:       synaptics_mst
  Subsystem:            drm_dp_aux_dev
  DeviceFile:           /dev/drm_dp_aux2
  SysfsPath:            /sys/devices/pci0000:00/0000:00:01.0/0000:01:00.0/0000:02:00.0/0000:03:00.0/drm/card1/card1-DP-3/drm_dp_aux2
  DpcdIeeeOui:          0xe04c
  DpcdDevId:            Dp1.2
```

superm1, 2023-11-19:
We've just fixed some stuff like this in main recently. Can you try a git checkout by chance?

pschichtel, 2023-11-19:
Indeed... https://github.com/fwupd/fwupd/commit/e01d2adea766f914376e4b5a56128bfc63693e66 seems very applicable. Building right now.

pschichtel, 2023-11-19:
yes, on main it starts again.

closed 2023-11-19 by pschichtel