fwupd/fwupd #6407: Encrpted swap not detected

physkets, 2023-11-20:
**Describe the bug**
In spite of an encrypted SWAP partition, it is detected as "Unencrypted".
Note that this is different from https://github.com/fwupd/fwupd/issues/4969 because I do not have it in an LVM.

```
$ fwupdmgr security
Host Security ID: HSI:1! (v1.9.8)

HSI-1
✔ BIOS firmware updates:         Enabled
✔ Fused platform:                Locked
✔ Supported CPU:                 Valid
✔ TPM empty PCRs:                Valid
✔ TPM v2.0:                      Found
✔ UEFI bootservice variables:    Locked
✔ UEFI platform key:             Valid
✔ UEFI secure boot:              Enabled

HSI-2
✔ BIOS rollback protection:      Enabled
✔ IOMMU:                         Enabled
✔ Platform debugging:            Locked
✔ TPM PCR0 reconstruction:       Valid
✘ SPI write protection:          Disabled

HSI-3
✔ Pre-boot DMA protection:       Enabled
✔ Suspend-to-idle:               Enabled
✔ Suspend-to-ram:                Disabled
✘ SPI replay protection:         Not supported

HSI-4
✘ Encrypted RAM:                 Not supported
✘ Processor rollback protection: Disabled

Runtime Suffix -!
✔ Linux kernel:                  Untainted
✔ Linux kernel lockdown:         Enabled
✔ fwupd plug-ins:                Untainted
✘ Linux swap:                    Unencrypted

This system has HSI runtime issues.
 » https://fwupd.github.io/hsi.html#hsi-runtime-suffix
```

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.14
compile   org.freedesktop.Passim        0.1.4
compile   com.hughsie.libjcat           0.1.14
compile   org.freedesktop.fwupd         1.9.8
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.7
runtime   org.freedesktop.Passim        0.1.4
runtime   org.freedesktop.gusb          0.4.7
runtime   com.hughsie.libjcat           0.1.14
runtime   org.kernel                    6.6.1-arch1-1
runtime   org.freedesktop.fwupd         1.9.8
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): `pacman`

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.
[... 136 lines cut ...]
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:          d96de5c124b60ed6241ebcb6bb2c839cb5580786
      Summary:            UEFI ESRT device
      Current version:    117571840
      Minimum Version:    117571840
      Vendor:             DMI:LENOVO
      Update State:       Success
      GUID:               ebfd9b0e-34a6-482c-9712-d4e84a2feb64
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
```

</details>

**Additional questions**

- Operating system and version: Archlinux
- Have you tried rebooting? Yes
- Is this a regression? Don't know; it has always been this way.

physkets, 2023-11-20:
My swap is encrypted like so:

```shell
$ lsblk
NAME          MAJ:MIN RM   SIZE RO TYPE  MOUNTPOINTS
nvme0n1       259:0    0 476.9G  0 disk  
├─nvme0n1p1   259:1    0     1G  0 part  /boot
├─nvme0n1p2   259:2    0   459G  0 part  
│ └─cryptroot 254:0    0   459G  0 crypt /
└─nvme0n1p3   259:3    0  16.9G  0 part  
  └─swap      254:1    0  16.9G  0 crypt [SWAP]
```

also,

```shell
$ cat /proc/swaps 
Filename                                Type            Size            Used            Priority
/dev/dm-1                               partition       17761600        0               -2
```

physkets, 2023-11-20:
[udisksctl_dump.txt](https://github.com/fwupd/fwupd/files/13412098/udisksctl_dump.txt)

Anything odd there?

hughsie, 2023-11-20:
> Anything odd there?

I see:

	/org/freedesktop/UDisks2/block_devices/dm_2d1:
	  org.freedesktop.UDisks2.Block:
	    Configuration:              [('fstab', {'fsname': <b'/dev/mapper/swap'>, 'dir': <b'none'>, 'type': <b'swap'>, 'opts': <b'defaults'>, 'freq': <0>, 'passno': <0>})]
	    CryptoBackingDevice:        '/'

and the code is:

	gboolean
	fu_volume_is_encrypted(FuVolume *self)
	{
		g_autoptr(GVariant) val = NULL;

		g_return_val_if_fail(FU_IS_VOLUME(self), FALSE);

		if (self->proxy_blk == NULL)
			return FALSE;
		val = g_dbus_proxy_get_cached_property(self->proxy_blk, "CryptoBackingDevice");
		if (val == NULL)
			return FALSE;
		if (g_strcmp0(g_variant_get_string(val, NULL), "/") == 0)
			return FALSE;
		return TRUE;
	}

but CryptoBackingDevice is `/` -- it's supposed to be a link to a block device -- e.g. `/org/freedesktop/UDisks2/block_devices/nvme1n1p4`

physkets, 2023-11-20:
How do I change that?

hughsie, 2023-11-20:
>  Did I do something wrong?

No idea I'm afraid.

physkets, 2023-11-22:
So if that entry isn't necessarily filled in every case where the swap is encrypted, shouldn't fwupd consider such a scenario?

Or should I file a bug report with whichever component is responsible for this?

hughsie, 2023-11-22:
> So if that entry isn't necessarily filled in every case where the swap is encrypted, shouldn't fwupd consider such a scenario

I'm not entirely sure. I don't know what the CryptoBackingDevice of / actually means -- it probably means udisks doesn't know what partition is backing it. Do you get an actual partition if you use something like a Fedora LiveUSB?

physkets, 2023-11-24:
Can I manually modify whatever `udisks` is reading?

As you can see from the output of `lsblk` in an earlier comment, it is a valid partition.

hughsie, 2023-11-24:
> Can I manually modify whatever udisks is reading?

No idea I'm afraid. Does the Fedora LiveUSB get the correct value?

FML128, 2023-12-09:
I have the same issue with a similar setup.

physkets, 2024-01-03:
[image]

This is what I see when using the Fedora Live USB.

mhalano, 2024-01-03:
@physkets can you test with fwupdmgt 1.9.11?

superm1, 2024-01-03:
It should be the same. We didn't make any changes in this area.

mhalano, 2024-01-03:
I'm saying that because I got a regression in the part related to encrypted ram. [Here](https://github.com/fwupd/fwupd/issues/6603) the issue.

physkets, 2024-01-04:
I think I have the same result with the latest version:

```bash
$ fwupdmgr security 
Host Security ID: HSI:1! (v1.9.11)

HSI-1
✔ BIOS firmware updates:         Enabled
✔ Fused platform:                Locked
✔ Supported CPU:                 Valid
✔ TPM empty PCRs:                Valid
✔ TPM v2.0:                      Found
✔ UEFI bootservice variables:    Locked
✔ UEFI platform key:             Valid
✔ UEFI secure boot:              Enabled

HSI-2
✔ BIOS rollback protection:      Enabled
✔ IOMMU:                         Enabled
✔ Platform debugging:            Locked
✔ TPM PCR0 reconstruction:       Valid
✘ SPI write protection:          Disabled

HSI-3
✔ Pre-boot DMA protection:       Enabled
✔ Suspend-to-idle:               Enabled
✔ Suspend-to-ram:                Disabled
✘ SPI replay protection:         Not supported
✘ CET Platform:                  Not supported

HSI-4
✔ SMAP:                          Enabled
✘ Processor rollback protection: Disabled
✘ Encrypted RAM:                 Not supported

Runtime Suffix -!
✔ fwupd plug-ins:                Untainted
✔ Linux kernel lockdown:         Enabled
✔ Linux kernel:                  Untainted
✘ Linux swap:                    Unencrypted
```

ksimuk, 2024-01-15:
Udisks can't detect plain encryption in partitions (no luks header) https://github.com/storaged-project/udisks/issues/916, and looks like nobody interested in fixing either.

hughsie, 2024-01-15:
It's open source! Can you help?

sjuk, 2024-10-20:
Unfortunately, I had the same experience. It cost me a lot of time to research because I thought the error was mine. Perhaps it would be possible to provide a prominent notice that the currently implemented method cannot reliably detect swap encryption. Thanks

hughsie, 2025-11-05:
> Wouldn't it be possible to use these methods to detect that the device is encrypted ?

Maybe; this is the code we have now: https://github.com/fwupd/fwupd/blob/main/plugins/linux-swap/fu-linux-swap.c#L116