fwupd/fwupd #6418: 1.9.9: test suites is failing `5/34 fwupdplugin-self-test` and `34/34 fu-self-test` units

kloczek, 2023-11-20:
**Describe the bug**
test suite is failing in two units.

**Steps to Reproduce**
Source code configures with:
```spec
%meson \
        -D bluez=enabled \
        -D build=all \
        -D consolekit=enabled \
        -D curl=enabled \
        -D docs=enabled \
        -D firmware-packager=true \
        -D gudev=enabled \
        -D gusb=enabled \
        -D introspection=enabled \
        -D launchd=disabled \
        -D libarchive=enabled \
        -D lvfs=true \
        -D lzma=enabled \
        -D man=true \
        -D plugin_acpi_phat=enabled \
        -D plugin_bcm57xx=enabled \
        -D plugin_cfu=enabled \
        -D plugin_cpu=enabled \
        -D plugin_emmc=enabled \
        -D plugin_ep963x=enabled \
        -D plugin_fastboot=enabled \
        -D plugin_flashrom=enabled \
        -D plugin_gpio=enabled \
        -D plugin_igsc=enabled \
        -D plugin_intel_me=enabled \
        -D plugin_intel_spi=enabled \
        -D plugin_intel_spi=true \
        -D plugin_logitech_bulkcontroller=enabled \
        -D plugin_modem_manager=enabled \
        -D plugin_msr=enabled \
        -D plugin_mtd=enabled \
        -D plugin_nitrokey=enabled \
        -D plugin_nvme=enabled \
        -D plugin_parade_lspcon=enabled \
        -D plugin_pixart_rf=enabled \
        -D plugin_powerd=enabled \
        -D plugin_realtek_mst=enabled \
        -D plugin_redfish=enabled \
        -D plugin_scsi=enabled \
        -D plugin_synaptics_mst=enabled \
        -D plugin_synaptics_rmi=enabled \
        -D plugin_tpm=enabled \
        -D plugin_uefi_capsule=enabled \
        -D plugin_uefi_capsule_splash=enabled \
        -D plugin_uefi_capsule_splash=true \
        -D plugin_uefi_pk=enabled \
        -D plugin_uf2=enabled \
        -D plugin_upower=enabled \
        -D polkit=enabled \
        -D supported_build=enabled \
        -D systemd=enabled \
        -D tests=true \
        %{nil}
```

**Expected behavior**
test suile should not fail.

**fwupd version information**
1.9.9.

- Operating system and version: Linux x86/64 kernel 6.6.0
- Have you tried rebooting? No
- Is this a regression? Yes

Prev version 1.9.8 in the same build env was ok.

hughsie, 2023-11-20:
What are the failures?

kloczek, 2023-11-21:
Sorry forgot to add that.
<details>

```console
+ cd fwupd-1.9.9
+ dbus-run-session -- /usr/bin/meson test -C x86_64-redhat-linux-gnu --num-processes 48 --print-errorlogs
ninja: Entering directory `/home/tkloczko/rpmbuild/BUILD/fwupd-1.9.9/x86_64-redhat-linux-gnu'
ninja: no work to do.
 1/34 fwupd-exported-api                     OK               0.02s
dbus-daemon[357880]: [session uid=1000 pid=357880] Activating service name='org.gtk.vfs.Daemon' requested by ':1.0' (uid=1000 pid=357888 comm="/home/tkloczko/rpmbuild/BUILD/fwupd-1.9.9/x86_64-r" label="kernel")
dbus-daemon[357880]: [session uid=1000 pid=357880] Successfully activated service 'org.gtk.vfs.Daemon'
 2/34 fwupd-context-test                     OK               0.04s
 3/34 fwupd-self-test                        OK               0.05s
 4/34 fwupd-thread-test                      OK               0.07s
dbus-daemon[357880]: [session uid=1000 pid=357880] Activating service name='org.gtk.vfs.Metadata' requested by ':1.4' (uid=1000 pid=357948 comm="/home/tkloczko/rpmbuild/BUILD/fwupd-1.9.9/x86_64-r" label="kernel")
dbus-daemon[357880]: [session uid=1000 pid=357880] Successfully activated service 'org.gtk.vfs.Metadata'
 5/34 fwupdplugin-self-test                  FAIL             0.38s   killed by signal 11 SIGSEGV
>>> G_TEST_BUILDDIR=/home/tkloczko/rpmbuild/BUILD/fwupd-1.9.9/x86_64-redhat-linux-gnu/libfwupdplugin G_TEST_SRCDIR=/home/tkloczko/rpmbuild/BUILD/fwupd-1.9.9/libfwupdplugin MALLOC_PERTURB_=247 LD_LIBRARY_PATH=/home/tkloczko/rpmbuild/BUILD/fwupd-1.9.9/x86_64-redhat-linux-gnu/libfwupdplugin:/home/tkloczko/rpmbuild/BUILD/fwupd-1.9.9/x86_64-redhat-linux-gnu/libfwupd /home/tkloczko/rpmbuild/BUILD/fwupd-1.9.9/x86_64-redhat-linux-gnu/libfwupdplugin/fwupdplugin-self-test
――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――― ✀  ――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――
Listing only the last 100 lines from a long log.
#   efi_image_offset: 0x0
#   pci_header_offset: 0x1c
#   expansion_header_offset: 0x200
# FuStruct-DEBUG: OpromPci:
#   vendor_id: 0x8086
#   device_id: 0x0
#   device_list_pointer: 0x0
#   structure_length: 0x0
#   structure_revision: 0x0
#   class_code: 0x0
#   image_length: 0x2
#   image_revision: 0x0
#   code_type: 0x1
#   indicator: 0x80
#   max_runtime_image_length: 0x0
#   conf_util_code_header_pointer: 0x0
#   dmtf_clp_entry_point_pointer: 0x0
ok 72 /fwupd/firmware{oprom}
# FuCommon-DEBUG: mapped file /home/tkloczko/rpmbuild/BUILD/fwupd-1.9.9/libfwupdplugin/tests/firmware.dfu of size 0x98
# FuStruct-DEBUG: DfuFtr:
#   release: 0xdead
#   pid: 0x4321
#   vid: 0x1234
#   ver: 0x100
#   len: 0x10
#   crc: 0xc6e75934
# FuCommon-DEBUG: mapped file /home/tkloczko/rpmbuild/BUILD/fwupd-1.9.9/libfwupdplugin/tests/firmware.bin of size 0x88
ok 73 /fwupd/firmware{dfu}
# FuCommon-DEBUG: mapped file /home/tkloczko/rpmbuild/BUILD/fwupd-1.9.9/libfwupdplugin/tests/firmware.dfu of size 0x98
# FuStruct-DEBUG: DfuFtr:
#   release: 0xdead
#   pid: 0x4321
#   vid: 0x1234
#   ver: 0x100
#   len: 0x10
#   crc: 0xc6e75934
ok 74 /fwupd/firmware{dfu-patch}
# FuCommon-DEBUG: mapped file /home/tkloczko/rpmbuild/BUILD/fwupd-1.9.9/libfwupdplugin/tests/firmware.dfuse of size 0x278
# FuStruct-DEBUG: DfuFtr:
#   release: 0x8642
#   pid: 0x5678
#   vid: 0x1234
#   ver: 0x11a
#   len: 0x10
#   crc: 0x15715d50
# FuStruct-DEBUG: DfuseHdr:
#   image_size: 0x268
#   targets: 0x2
# FuStruct-DEBUG: DfuseImage:
#   alt_setting: 0x0
[... 191 lines cut ...]
#   time: 0x0
#   fattr: 0x0 [none]
# FuStruct-DEBUG: CabFile:
#   usize: 0x1c4
#   uoffset: 0x1f40
#   index: 0x0
#   date: 0x0
#   time: 0x0
#   fattr: 0x0 [none]
―――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――

Summary of Failures:

 5/34 fwupdplugin-self-test          FAIL             0.38s   killed by signal 11 SIGSEGV
32/34 linux-swap-self-test           FAIL             0.23s   killed by signal 6 SIGABRT
34/34 fu-self-test                   FAIL             0.48s   killed by signal 11 SIGSEGV

Ok:                 31
Expected Fail:      0
Fail:               3
Unexpected Pass:    0
Skipped:            0
Timeout:            0
```
</details>

hughsie, 2023-11-21:
Can you get a backtrace of the ./src/fu-self-test one please?

superm1, 2023-12-21:
https://github.com/fwupd/fwupd/commit/c80bff37553c6c782d341a63870566605f4c6a06 might fix this, can you cherry pick and try?

kloczek, 2023-12-22:
Sorry I forgot about this ticket.
Will try to test patch ASP.
Will back in minutes.

kloczek, 2023-12-22:
<details>
<summary>Just tested that with that patch and result still is negative</summary>

```console
warning: Downloading https://github.com/fwupd/fwupd//releases/download/1.9.10/fwupd-1.9.10.tar.xz to /home/tkloczko/rpmbuild/SOURCES/fwupd-1.9.10.tar.xz
warning: Downloading https://github.com/fwupd/fwupd//commit/c80bff37.patch#/fwupd-Add-missing-build-dependency-to-fwupdplugin-self-test.patch to /home/tkloczko/rpmbuild/SOURCES/fwupd-Add-mi
ssing-build-dependency-to-fwupdplugin-self-test.patch
warning: source_date_epoch_from_changelog set but %changelog is missing
Executing(%prep): /bin/sh -e /var/tmp/rpm-tmp.QTPwGj
+ umask 022
+ cd /home/tkloczko/rpmbuild/BUILD
+ cd /home/tkloczko/rpmbuild/BUILD
+ rm -rf fwupd-1.9.10
+ /usr/lib/rpm/rpmuncompress -x /home/tkloczko/rpmbuild/SOURCES/fwupd-1.9.10.tar.xz
+ STATUS=0
+ '[' 0 -ne 0 ']'
+ cd fwupd-1.9.10
+ rm -rf /home/tkloczko/rpmbuild/BUILD/fwupd-1.9.10/SPECPARTS
+ /usr/bin/mkdir -p /home/tkloczko/rpmbuild/BUILD/fwupd-1.9.10/SPECPARTS
+ /usr/bin/chmod -Rf a+rX,u+w,g-w,o-w .
+ /usr/lib/rpm/rpmuncompress /home/tkloczko/rpmbuild/SOURCES/fwupd-Add-missing-build-dependency-to-fwupdplugin-self-test.patch
+ /usr/bin/patch -p1 -s --fuzz=0 --no-backup-if-mismatch -f
+ /usr/lib/rpm/rpmuncompress /home/tkloczko/rpmbuild/SOURCES/fwupd-remove_hardcoding_compiler_warning_flags.patch
+ /usr/bin/patch -p1 -s --fuzz=0 --no-backup-if-mismatch -f
+ /usr/lib/rpm/rpmuncompress /home/tkloczko/rpmbuild/SOURCES/fwupd-do_not_install_some_docs.patch
+ /usr/bin/patch -p1 -s --fuzz=0 --no-backup-if-mismatch -f
+ /usr/lib/rpm/rpmuncompress /home/tkloczko/rpmbuild/SOURCES/fwupd-reuse_gtk-doc_path.patch
+ /usr/bin/patch -p1 -s --fuzz=0 --no-backup-if-mismatch -f

[..]

+ cd fwupd-1.9.10
+ dbus-run-session -- /usr/bin/meson test -C x86_64-redhat-linux-gnu --num-processes 48 --print-errorlogs
ninja: Entering directory `/home/tkloczko/rpmbuild/BUILD/fwupd-1.9.10/x86_64-redhat-linux-gnu'
ninja: no work to do.
 1/34 fwupd-exported-api             OK               0.02s
 2/34 fwupd-self-test                OK               0.03s
 3/34 fwupd-context-test             OK               0.02s
 4/34 fwupd-thread-test              OK               0.04s
 5/34 fwupdplugin-self-test          FAIL             0.27s   killed by signal 11 SIGSEGV
>>> UBSAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1:print_stacktrace=1 ASAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1 LD_LIBRARY_PATH=/home/tkloczko/rpmbuild/BUILD/fwupd-1.9.10/x86_64-redhat-linux-gnu/libfwupdplugin:/home/tkloczko/rpmbuild/BUILD/fwupd-1.9.10/x86_64-redhat-linux-gnu/libfwupd G_TEST_BUILDDIR=/home/tkloczko/rpmbuild/BUILD/fwupd-1.9.10/x86_64-redhat-linux-gnu/libfwupdplugin MALLOC_PERTURB_=34 G_TEST_SRCDIR=/home/tkloczko/rpmbuild/BUILD/fwupd-1.9.10/libfwupdplugin /home/tkloczko/rpmbuild/BUILD/fwupd-1.9.10/x86_64-redhat-linux-gnu/libfwupdplugin/fwupdplugin-self-test
――――――――――――――――――――――――――――――――――――― ✀  ―――――――――――――――――――――――――――――――――――――
Listing only the last 100 lines from a long log.
#   efi_image_offset: 0x0
#   pci_header_offset: 0x1c
#   expansion_header_offset: 0x200
# FuStruct-DEBUG: OpromPci:
#   vendor_id: 0x8086
#   device_id: 0x0
#   device_list_pointer: 0x0
#   structure_length: 0x0
#   structure_revision: 0x0
#   class_code: 0x0
#   image_length: 0x2
#   image_revision: 0x0
#   code_type: 0x1
#   indicator: 0x80
#   max_runtime_image_length: 0x0
#   conf_util_code_header_pointer: 0x0
#   dmtf_clp_entry_point_pointer: 0x0
ok 73 /fwupd/firmware{oprom}
# FuCommon-DEBUG: mapped file /home/tkloczko/rpmbuild/BUILD/fwupd-1.9.10/libfwupdplugin/tests/firmware.dfu of size 0x98
# FuStruct-DEBUG: DfuFtr:
#   release: 0xdead
#   pid: 0x4321
#   vid: 0x1234
#   ver: 0x100
#   len: 0x10
#   crc: 0xc6e75934
# FuCommon-DEBUG: mapped file /home/tkloczko/rpmbuild/BUILD/fwupd-1.9.10/libfwupdplugin/tests/firmware.bin of size 0x88
[... 197 lines cut ...]
#   date: 0x0
#   time: 0x0
#   fattr: 0x0 [none]
# FuStruct-DEBUG: CabFile:
#   usize: 0x204
#   uoffset: 0x1f40
#   index: 0x0
#   date: 0x0
#   time: 0x0
#   fattr: 0x0 [none]
――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――

Summary of Failures:

 5/34 fwupdplugin-self-test          FAIL             0.27s   killed by signal 11 SIGSEGV
34/34 fu-self-test                   FAIL             0.38s   killed by signal 11 SIGSEGV

Ok:                 32
Expected Fail:      0
Fail:               2
Unexpected Pass:    0
Skipped:            0
Timeout:            0
```
</details>

superm1, 2023-12-22:
Okay can you please get the back trace then?

kloczek, 2023-12-22:
Call traces

<details>
<summary>fu-self-test</summary>

```console
[tkloczko@pers-jacek x86_64-redhat-linux-gnu]$ dbus-run-session -- /usr/bin/meson test --gdb fwupdplugin-self-test
ninja: Entering directory `/home/tkloczko/rpmbuild/BUILD/fwupd-1.9.10/x86_64-redhat-linux-gnu'
ninja: no work to do.
1/1 fwupdplugin-self-test RUNNING
>>> MALLOC_PERTURB_=1 UBSAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1:print_stacktrace=1 G_TEST_BUILDDIR=/home/tkloczko/rpmbuild/BUILD/fwupd-1.9.10/x86_64-redhat-linux-gnu/libfwupdplugin LD_LIBRARY_PATH=/home/tkloczko/rpmbuild/BUILD/fwupd-1.9.10/x86_64-redhat-linux-gnu/libfwupd:/home/tkloczko/rpmbuild/BUILD/fwupd-1.9.10/x86_64-redhat-linux-gnu/libfwupdplugin G_TEST_SRCDIR=/home/tkloczko/rpmbuild/BUILD/fwupd-1.9.10/libfwupdplugin ASAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1 gdb --quiet --args /home/tkloczko/rpmbuild/BUILD/fwupd-1.9.10/x86_64-redhat-linux-gnu/libfwupdplugin/fwupdplugin-self-test
――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――― ✀  ――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――
Reading symbols from /home/tkloczko/rpmbuild/BUILD/fwupd-1.9.10/x86_64-redhat-linux-gnu/libfwupdplugin/fwupdplugin-self-test...
(gdb) r
Starting program: /home/tkloczko/rpmbuild/BUILD/fwupd-1.9.10/x86_64-redhat-linux-gnu/libfwupdplugin/fwupdplugin-self-test
warning: Unable to open "librpm.so.9" (librpm.so.9: cannot open shared object file: No such file or directory), missing debuginfos notifications will not be displayed
Missing separate debuginfo for /lib64/libxmlb.so.2.
The debuginfo package for this file is probably broken.
Missing separate debuginfo for /lib64/libgio-2.0.so.0.
The debuginfo package for this file is probably broken.
Missing separate debuginfo for /lib64/libgobject-2.0.so.0.
The debuginfo package for this file is probably broken.
Missing separate debuginfo for /lib64/libglib-2.0.so.0.
The debuginfo package for this file is probably broken.
Missing separate debuginfo for /lib64/libc.so.6.
The debuginfo package for this file is probably broken.
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
Missing separate debuginfo for /lib64/libjcat.so.1.
The debuginfo package for this file is probably broken.
Missing separate debuginfo for /lib64/libjson-glib-1.0.so.0.
The debuginfo package for this file is probably broken.
Missing separate debuginfo for /lib64/libcurl.so.4.
The debuginfo package for this file is probably broken.
Missing separate debuginfo for /lib64/libgusb.so.2.
The debuginfo package for this file is probably broken.
Missing separate debuginfo for /lib64/libgudev-1.0.so.0.
The debuginfo package for this file is probably broken.
Missing separate debuginfo for /lib64/libgmodule-2.0.so.0.
The debuginfo package for this file is probably broken.
Missing separate debuginfo for /lib64/libz.so.1.
The debuginfo package for this file is probably broken.
Missing separate debuginfo for /lib64/liblzma.so.5.
The debuginfo package for this file is probably broken.
Missing separate debuginfo for /lib64/libarchive.so.13.
The debuginfo package for this file is probably broken.
Missing separate debuginfo for /lib64/libcbor.so.0.10.
The debuginfo package for this file is probably broken.
Missing separate debuginfo for /lib64/libzstd.so.1.
The debuginfo package for this file is probably broken.
Missing separate debuginfo for /lib64/libmount.so.1.
The debuginfo package for this file is probably broken.
Missing separate debuginfo for /lib64/libselinux.so.1.
The debuginfo package for this file is probably broken.
Missing separate debuginfo for /lib64/libffi.so.8.
The debuginfo package for this file is probably broken.
Missing separate debuginfo for /lib64/libpcre2-8.so.0.
The debuginfo package for this file is probably broken.
Missing separate debuginfo for /lib64/libgnutls.so.30.
The debuginfo package for this file is probably broken.
Missing separate debuginfo for /lib64/libhogweed.so.6.
The debuginfo package for this file is probably broken.
Missing separate debuginfo for /lib64/libgpgme.so.11.
The debuginfo package for this file is probably broken.
Missing separate debuginfo for /lib64/libgpg-error.so.0.
The debuginfo package for this file is probably broken.
Missing separate debuginfo for /lib64/libnghttp2.so.14.
The debuginfo package for this file is probably broken.
Missing separate debuginfo for /lib64/libidn2.so.0.
The debuginfo package for this file is probably broken.
[... 6050 lines cut ...]
#     <length>0xc</length>
#     <handle>0xb100</handle>
#     <string idx="00">
#     </string>
#   </item>
#   <item>
#     <type>0xb2</type>
#     <length>0x54</length>
#     <handle>0xb200</handle>
#     <string idx="00">
#     </string>
#   </item>
#   <item>
#     <type>0xcb</type>
#     <length>0x51</length>
#     <handle>0xcb00</handle>
#     <string idx="00">MEI1</string>
#     <string idx="01">MEI2</string>
#     <string idx="02">MEI3</string>
#   </item>
#   <item>
#     <type>0xcd</type>
#     <length>0x1a</length>
#     <handle>0xcd00</handle>
#     <string idx="00">Reference C

kloczek, 2023-12-22:
Please let me know if you want to have call trace with other librariues symbols.

kloczek, 2023-12-22:
I've updated comment with call traces because I've copied the same unit call traces two times.
Sorry.

superm1, 2023-12-22:
Both stacktraces point to something caused by the cleanup process used by ea506bd371fc19a3ea49856fcde43dd64c044ba6.

It's odd that you're hitting it but no one else has.  What version of zlib is on your system?    At least on my local system I've got zlib-1.2.13-4.fc39.i686.
I think it might be useful to install debug symbols for zlib to understand this as well.

kloczek, 2023-12-23:
I'm using zlib-ng 2.1.5.
I've switched to use only zlib-ng almost two years ago.
Initially it was few issues and even zlib-ng was fixed after some of my reports. From more than half year I do not remember any other issues.
Using zlib-ng brings some benefits but because zlib-ng maintainer is trying to optimise that library to the limits of physical possibilities I fully understand that only by this time to time it may be some new side effect.

Nevertheless I'm quite busy working on finishing distribution on which I'm working so if you will have any requests to perform any additional diagnostics please let me know and I'll try to provide results ASP.

Other details about build methodology: all my builds are performed inside LXC zones.
Each one is spawned to build only single package where inside zone are installed only build dependencies..

kloczek, 2023-12-23:
Just in case 
<details>
<summary>Here is my fwupd.spec file</summary>

```spec
# BUG: test suite is failing https://github.com/fwupd/fwupd/issues/6418
# PRE-INST: gcc pandoc
# TODO: remove Provides: dbxtool when no longer needed
# TODO: RFE port TLS code from gnutls to openssl https://github.com/fwupd/fwupd/issues/3128
%ifarch x86_64
%bcond_without  redfish
%endif
%ifarch x86_64 aarch64
%bcond_without  uefi
%endif
%bcond_with     check
%global         build_cflags    %{build_cflags} -Os

Summary:        Firmware update daemon
Name:           fwupd
Version:        1.9.10
Release:        2%{?dist}
License:        LGPL-2.1-or-later (https://spdx.org/licenses/LGPL-2.1-or-later.html)
URL:            https://github.com/fwupd/fwupd/
VCS:            https://github.com/fwupd/fwupd/
Source:         %{VCS}/releases/download/%{version}/%{name}-%{version}.tar.xz
Patch:          %{name}-remove_hardcoding_compiler_warning_flags.patch
Patch:          %{name}-do_not_install_some_docs.patch
Patch:          %{name}-reuse_gtk-doc_path.patch
BuildRequires:  efivar-devel
BuildRequires:  gcc
BuildRequires:  gettext-devel
BuildRequires:  meson
BuildRequires:  pandoc
BuildRequires:  pkgconfig
BuildRequires:  pkgconfig(bash-completion)
BuildRequires:  pkgconfig(efiboot)
BuildRequires:  pkgconfig(efivar)               >= 33
BuildRequires:  pkgconfig(flashrom)
BuildRequires:  pkgconfig(fwupd-efi)
BuildRequires:  pkgconfig(gi-docgen)
BuildRequires:  pkgconfig(gio-2.0)              >= 2.45.8
BuildRequires:  pkgconfig(gio-unix-2.0)         >= 2.45.8
BuildRequires:  pkgconfig(gmodule-2.0)
BuildRequires:  pkgconfig(gnutls)               >= 3.4.4.1
BuildRequires:  pkgconfig(gobject-introspection-1.0)
BuildRequires:  pkgconfig(gtk-doc)
BuildRequires:  pkgconfig(gudev-1.0)            >= 232
BuildRequires:  pkgconfig(gusb)                 >= 0.3.5
BuildRequires:  pkgconfig(jcat)                 >= 0.1.0
BuildRequires:  pkgconfig(json-glib-1.0)        >= 1.1.1
BuildRequires:  pkgconfig(libarchive)
BuildRequires:  pkgconfig(libcbor)
BuildRequires:  pkgconfig(libcurl)              >= 7.56.0
BuildRequires:  pkgconfig(libdrm_amdgpu)        >= 2.4.113
BuildRequires:  pkgconfig(liblzma)
BuildRequires:  pkgconfig(libprotobuf-c)
BuildRequires:  pkgconfig(libsmbios_c)          >= 2.4.0
BuildRequires:  pkgconfig(libsoup-2.4)          >= 2.51.92
BuildRequires:  pkgconfig(libsystemd)
BuildRequires:  pkgconfig(mbim-glib)            >= 1.22.0
BuildRequires:  pkgconfig(mm-glib)              >= 1.10.0
BuildRequires:  pkgconfig(passim)
BuildRequires:  pkgconfig(polkit-gobject-1)     >= 0.103
BuildRequires:  pkgconfig(qmi-glib)             >= 1.22.0
BuildRequires:  pkgconfig(sqlite3)
BuildRequires:  pkgconfig(systemd)              >= 211
BuildRequires:  pkgconfig(tss2-esys)            >= 2.0
BuildRequires:  pkgconfig(udev)
BuildRequires:  pkgconfig(umockdev-1.0)
[... 154 lines cut ...]
%{_localstatedir}/lib/fwupd
%{_mandir}/man[158]/*
%{_modulesloaddir}/*.conf
%{_sysconfdir}/grub.d/*fwupd
%{_sysconfdir}/pki/fwupd-metadata
%{_systemd_util_dir}/system-shutdown/fwupd.shutdown
%{_sysusersdir}/*.conf
%{_udevrulesdir}/*.rules
%{_unitdir}/*.{timer,service,wants}
%ghost %{_localstatedir}/lib/fwupd/gnupg

%files devel
%{_datadir}/gir-*/*.gir
%{_datadir}/gtk-doc/html/*
%{_datadir}/vala/vapi/*.{deps,vapi}
%{_includedir}/fwupd-*
%{_libdir}/lib*.so
%{_libdir}/pkgconfig/*.pc

%files -n tests-%{name}
%config(noreplace)%{_sysconfdir}/fwupd/remotes.d/fwupd-tests.conf
%{_datadir}/installed-tests/*
%{_libexecdir}/installed-tests/*
```
</details>

superm1, 2023-12-23:
This does look to me to be a zlib-ng bug then.  I did a quick try to swap out zlib with zlib-ng just for fwupd, but it's not going to be enough for me to reproduce it because other system libraries need to be complied against it too.  I can't do this without other stuff potentially exploding.

For example:
```
./build/src/fu-self-test: /home/supermario/src/zlib-ng/libz.so.1: version `ZLIB_1.2.2' not found (required by /lib64/libgio-2.0.so.0)
./build/src/fu-self-test: /home/supermario/src/zlib-ng/libz.so.1: version `ZLIB_1.2.3.3' not found (required by /lib64/libxml2.so.2)
./build/src/fu-self-test: /home/supermario/src/zlib-ng/libz.so.1: version `ZLIB_1.2.2.3' not found (required by /lib64/libxml2.so.2)
```

So I suggest what you do is get the debug symbols for zlib-ng installed, reproduce this again and then file a bug with zlib-ng providing that backtrace to them.  You can link the bug here.

kloczek, 2023-12-23:
```console
[tkloczko@pers-jacek SRPMS]$ objdump -x /usr/lib64/libz.so.1

/usr/lib64/libz.so.1:     file format elf64-x86-64
/usr/lib64/libz.so.1
architecture: i386:x86-64, flags 0x00000150:
HAS_SYMS, DYNAMIC, D_PAGED
start address 0x0000000000000000

Program Header:
    LOAD off    0x0000000000000000 vaddr 0x0000000000000000 paddr 0x0000000000000000 align 2**12
         filesz 0x0000000000002348 memsz 0x0000000000002348 flags r--
    LOAD off    0x0000000000003000 vaddr 0x0000000000003000 paddr 0x0000000000003000 align 2**12
         filesz 0x000000000001625d memsz 0x000000000001625d flags r-x
    LOAD off    0x000000000001a000 vaddr 0x000000000001a000 paddr 0x000000000001a000 align 2**12
         filesz 0x000000000000709c memsz 0x000000000000709c flags r--
    LOAD off    0x0000000000021ac0 vaddr 0x0000000000022ac0 paddr 0x0000000000022ac0 align 2**12
         filesz 0x0000000000000530 memsz 0x0000000000000548 flags rw-
 DYNAMIC off    0x0000000000021ce0 vaddr 0x0000000000022ce0 paddr 0x0000000000022ce0 align 2**3
         filesz 0x0000000000000220 memsz 0x0000000000000220 flags rw-
    NOTE off    0x00000000000002e0 vaddr 0x00000000000002e0 paddr 0x00000000000002e0 align 2**3
         filesz 0x0000000000000040 memsz 0x0000000000000040 flags r--
    NOTE off    0x0000000000000320 vaddr 0x0000000000000320 paddr 0x0000000000000320 align 2**2
         filesz 0x0000000000000024 memsz 0x0000000000000024 flags r--
     TLS off    0x0000000000021ac0 vaddr 0x0000000000022ac0 paddr 0x0000000000022ac0 align 2**4
         filesz 0x0000000000000088 memsz 0x0000000000000088 flags r--
0x6474e553 off    0x00000000000002e0 vaddr 0x00000000000002e0 paddr 0x00000000000002e0 align 2**3
         filesz 0x0000000000000040 memsz 0x0000000000000040 flags r--
EH_FRAME off    0x000000000001ec00 vaddr 0x000000000001ec00 paddr 0x000000000001ec00 align 2**2
         filesz 0x0000000000000694 memsz 0x0000000000000694 flags r--
   STACK off    0x0000000000000000 vaddr 0x0000000000000000 paddr 0x0000000000000000 align 2**4
         filesz 0x0000000000000000 memsz 0x0000000000000000 flags rw-
   RELRO off    0x0000000000021ac0 vaddr 0x0000000000022ac0 paddr 0x0000000000022ac0 align 2**0
         filesz 0x0000000000000530 memsz 0x0000000000000540 flags r--

Dynamic Section:
  NEEDED               libc.so.6
  NEEDED               ld-linux-x86-64.so.2
  SONAME               libz.so.1
  INIT                 0x0000000000003000
  FINI                 0x0000000000019250
  INIT_ARRAY           0x0000000000022b48
  INIT_ARRAYSZ         0x0000000000000008
  FINI_ARRAY           0x0000000000022b50
  FINI_ARRAYSZ         0x0000000000000008
  GNU_HASH             0x0000000000000348
  STRTAB               0x00000000000012f8
  SYMTAB               0x00000000000006f8
  STRSZ                0x0000000000000624
  SYMENT               0x0000000000000018
  PLTGOT               0x0000000000022f00
  PLTRELSZ             0x00000000000001f8
  PLTREL               0x0000000000000007
  JMPREL               0x0000000000002150
  RELA                 0x0000000000001ca0
  RELASZ               0x00000000000004b0
  RELAENT              0x0000000000000018
  VERDEF               0x0000000000001a20
  VERDEFNUM            0x000000000000000f
  FLAGS                0x0000000000000008
  FLAGS_1              0x0000000000000001
  VERNEED              0x0000000000001c30
  VERNEEDNUM           0x0000000000000002
  VERSYM               0x000000000000191c
  RELACOUNT            0x000000000000002d

Version definitions:
1 0x01 0x09d5f4e1 libz.so.1
2 0x00 0x0827e5c0 ZLIB_1.2.0
3 0x00 0x07e5cb32 ZLIB_1.2.0.2
        ZLIB_1.2.0
4 0x00 0x07e5cb38 ZLIB_1.2.0.8
        ZLIB_1.2.0.2
5 0x00 0x0827e5c2 ZLIB_1.2.2
        ZLIB_1.2.0.8
6 0x00 0x07e5cd33 ZLIB_1.2.2.3
        ZLIB_1.2.2
7 0x00 0x07e5cd34 ZLIB_1.2.2.4
        ZLIB_1.2.2.3
8 0x00 0x07e5ce33 ZLIB_1.2.3.3
        ZLIB_1.2.2.4
9 0x00 0x07e5ce34 ZLIB_1.2.3.4
        ZLIB_1.2.3.3
10 0x00 0x07e5ce35 ZLIB_1.2.3.5
        ZLIB_1.2.3.4
11 0x00 0x07e5d031 ZLIB_1.2.5.1
        ZLIB_1.2.3.5
12 0x00 0x07e5d032 ZLIB_1.2.5.2
        ZLIB_1.2.5.1
13 0x00 0x07e5c231 ZLIB_1.2.7.1
        ZLIB_1.2.5.2
14 0x00 0x0827e5c9 ZLIB_1.2.9
        ZLIB_1.2.7.1
15 0x00 0x027e5cc2 ZLIB_1.2.12
        ZLIB_1.2.9

Version References:
  required from ld-linux-x86-64.so.2:
    0x0d696913 0x00 19 GLIBC_2.3
  required from libc.so.6:
    0x06969194 0x00 20 GLIBC_2.14
    0x0d696914 0x00 18 GLIBC_2.4
    0x09691a75 0x00 17 GLIBC_2.2.5
    0x09691974 0x00 16 GLIBC_2.3.4
```
As you see libz DSO have those versioned sections of symbols.

hughsie, 2023-12-23:
Looks to me like a double-free in inflateEnd -- so zlib-ng -- I'm guessing if you run the self tests with valgrind that also gets upset?

kloczek, 2023-12-23:
<details>
<summary>fu-self-test over valgrind</summary>

```console
[tkloczko@pers-jacek x86_64-redhat-linux-gnu]$ dbus-run-session -- /usr/bin/meson test --wrap='valgrind' fu-self-test
ninja: Entering directory `/home/tkloczko/rpmbuild/BUILD/fwupd-1.9.10/x86_64-redhat-linux-gnu'
ninja: no work to do.
1/1 fu-self-test        OK              80.82s

Ok:                 1
Expected Fail:      0
Fail:               0
Unexpected Pass:    0
Skipped:            0
Timeout:            0

Full log written to /home/tkloczko/rpmbuild/BUILD/fwupd-1.9.10/x86_64-redhat-linux-gnu/meson-logs/testlog-valgrind.txt
```
</details>

<details>
<summary>fwupdplugin-self-test over valgrind</summary>

```console
[tkloczko@pers-jacek x86_64-redhat-linux-gnu]$ dbus-run-session -- /usr/bin/meson test --wrap='valgrind' fwupdplugin-self-test
ninja: Entering directory `/home/tkloczko/rpmbuild/BUILD/fwupd-1.9.10/x86_64-redhat-linux-gnu'
ninja: no work to do.
1/1 fwupdplugin-self-test        OK              10.99s

Ok:                 1
Expected Fail:      0
Fail:               0
Unexpected Pass:    0
Skipped:            0
Timeout:            0

Full log written to /home/tkloczko/rpmbuild/BUILD/fwupd-1.9.10/x86_64-redhat-linux-gnu/meson-logs/testlog-valgrind.txt
```
</details>

If you have any proposition to run other diagnostics please let me know.

PS. it well known that sometimes valgrind instead exposing it masks bugs.

hughsie, 2023-12-24:
> If you have any proposition to run other diagnostics please let me know.

It looks like a zlib-ng bug to me.

KungFuJesus, 2023-12-29:
This will be closed by: https://github.com/fwupd/fwupd/pull/6581
I assume a backport to the stable branch is forthcoming.

KungFuJesus, 2023-12-29:
Relevant log portion:
```
ok 48 /fwupd/efi-load-option
# GLib-DEBUG: setenv()/putenv() are not thread-safe and should not be used after threads are created
not ok /fwupd/efivar - FuSelfTest:ERROR:../libfwupdplugin/fu-self-test.c:3337:fu_efivar_func: assertion failed (total >= 0x2000): (2048 >= 8192)
Bail out!
----------------------------------- stderr -----------------------------------
**
FuSelfTest:ERROR:../libfwupdplugin/fu-self-test.c:3337:fu_efivar_func: assertion failed (total >= 0x2000): (2048 >= 8192)
==============================================================================
```

hughsie, 2023-12-29:
> fu_efivar_func: assertion failed (total >= 0x2000):

That should be fixed by https://github.com/fwupd/fwupd/commit/153ffd3e57904bd59df35695cd30db405b7c2f23 -- can you pull and try again please.

kloczek, 2023-12-29:
Just tested PR and looks like issue has been solved 👍 😄 
```console
+ cd fwupd-1.9.10
+ dbus-run-session -- /usr/bin/meson test -C x86_64-redhat-linux-gnu --num-processes 48 --print-errorlogs
ninja: Entering directory `/home/tkloczko/rpmbuild/BUILD/fwupd-1.9.10/x86_64-redhat-linux-gnu'
ninja: no work to do.
 1/34 fwupd-exported-api             OK               0.02s
 2/34 fwupd-self-test                OK               0.03s
 3/34 fwupd-context-test             OK               0.02s
 4/34 fwupd-thread-test              OK               0.04s
 5/34 fwupdplugin-self-test          OK               0.21s
 6/34 uefi-self-test                 OK               0.24s
 7/34 acpi-dmar-self-test            OK               0.24s
 8/34 acpi-facp-self-test            OK               0.23s
 9/34 acpi-ivrs-self-test            OK               0.23s
10/34 acpi-phat-self-test            OK               0.22s
11/34 ata-self-test                  OK               0.21s
12/34 bcm57xx-self-test              OK               0.21s
13/34 ccgx-self-test                 OK               0.20s
14/34 ccgx-dmc-self-test             OK               0.20s
15/34 fu-dfu-self-test               OK               0.19s
16/34 elantp-self-test               OK               0.18s
17/34 lenovo-thinklmi-self-test      OK               0.18s
18/34 linux-swap-self-test           OK               0.17s
19/34 logitech-hidpp-self-test       OK               0.16s
20/34 mtd-self-test                  OK               0.16s
21/34 nitrokey-self-test             OK               0.15s
22/34 nvme-self-test                 OK               0.15s
23/34 pxi-self-test                  OK               0.14s
24/34 redfish-self-test              OK               0.14s
25/34 synaptics-mst-self-test        OK               0.13s
26/34 synaptics-prometheus-self-test OK               0.12s
27/34 synaptics-rmi-self-test        OK               0.12s
28/34 tpm-self-test                  OK               0.10s
29/34 uefi-dbx-self-test             OK               0.10s
30/34 uf2-self-test                  OK               0.09s
31/34 wacom-usb-self-test            OK               0.06s
32/34 vli-self-test                  OK               0.09s
33/34 thunderbolt-self-test          OK               3.38s
34/34 fu-self-test                   OK               3.50s

Ok:                 34
Expected Fail:      0
Fail:               0
Unexpected Pass:    0
Skipped:            0
Timeout:            0
```

hughsie, 2023-12-29:
Great, thanks for all the hard work!

closed 2023-12-29 by hughsie