fwupd/fwupd #6426: 1.9.9: Pipe error [-9] / Bad address [14] when probing Genesys hub 05e3:0610

paride, 2023-11-21:
**Describe the bug**

When (re)starting fwupd I get the following error in the logs:
```
Nov 21 16:28:27 ossimoro systemd[1]: Starting fwupd.service - Firmware update daemon...
Nov 21 16:28:27 ossimoro fwupd[79139]: 15:28:27.649 FuEngine             failed to add device usb:01:00:03: failed to get static tool info from device: USB error on device 05e3:0610 : Pipe error [-9]
Nov 21 16:28:33 ossimoro fwupd[79139]: 15:28:33.378 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:08.1/0000:64:00.0: ioctl error: Bad address [14]
```

This happens when I have the Genesys 05e3:0610 USB hub plugged in, which is the USB hub internal to my Lenovo ThinkVision monitor (connected via usb-c).

The error appears to be harmless, however fwupd 1.9.8 used to become unresponsive when hitting it.

**Steps to Reproduce**

Just (re)start fwupd.

**fwupd version information**
Please provide the version of the daemon and client.

```
compile   com.hughsie.libxmlb           0.3.14
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.9.9
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   org.freedesktop.gusb          0.4.5
runtime   org.freedesktop.fwupd         1.9.9
runtime   org.kernel                    6.5.0-10-generic
```

Installed from apt (Ubuntu `noble-proposed` pocket).

<details>

<summary>**fwupd device information**</summary>

```
$ fwupdmgr get-devices --show-all-devices
LENOVO 21K5CTO1WW
│
├─Unknown Device:
│     Device ID:          348dee50b7a5bea9eb8cf66e1e1d42290f02cb13
│     Serial Number:      V90B4RX5
│     GUID:               a66cd6e8-0d6e-54b3-a71a-03a547b64697 ← DRM\VEN_LEN&DEV_61E9
│   
├─Unknown Device:
│     Device ID:          e51f39efc9f77b8dd62fb3942b7c30b0dd64752d
│     Serial Number:      16843009
│     GUID:               b8c6e8be-ccd7-5943-8fee-61be4c9bbbb4 ← DRM\VEN_LEN&DEV_61AF
│   
├─AMD Ryzen 7 PRO 7840U w/ Radeon 780M Graphics:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x0a704103
│ │   Vendor:             Advanced Micro Devices, Inc.
│ │   GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│ │                       2a1df9de-fe02-59c8-8af1-32c8bb011f94 ← CPUID\PRO_0&FAM_19&MOD_74
│ │                       5a4bca32-6547-584f-bb36-920d9adede25 ← CPUID\PRO_0&FAM_19&MOD_74&STP_1
│ │   Device Flags:       • Internal device
│ │ 
│ ├─GPIO controller:
│ │     Device ID:        f685512aa07369c9e77742acef941d779d31e766
│ │     GUID:             37b440a9-2473-5087-a39b-db84f32a8ed8 ← GPIO\ID_AMDI0030:00
│ │   
│ ├─Secure Processor:
│ │     Device ID:        c54ab0237d7a8db8c717b68e0be78e4374a2a079
│ │     Vendor:           Advanced Micro Devices, Inc. (PCI:0x1022)
│ │     GUIDs:            9eb6a793-7f97-5fb5-b49c-139d2ce3ee46 ← PCI\VEN_1022&DEV_15C7
│ │                       ccdf3a32-442f-5c0b-ad12-2e7709eb8444 ← PCI\VEN_1022&DEV_15C7&SUBSYS_17AA50D9
│ │     Device Flags:     • Internal device
[... 195 lines cut ...]
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─Unifying Receiver:
      Device ID:          1f578d9fc240a4af740d8dbdb051667d8c25b011
      Summary:            Miniaturised USB wireless receiver
      Current version:    RQR24.11_B0036
      Bootloader Version: BOT03.02_B0009
      Vendor:             HIDRAW:0x046D|USB:0x046D
      Install Duration:   30 seconds
      GUIDs:              cc4cbfa9-bf9d-540b-b92b-172ce31013c1
                          279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
      Device Flags:       • Updatable
                          • Signed Payload
    
```

</details>

**Additional questions**

- Operating system and version: Ubuntu 24.04 (devel)
- Have you tried rebooting? Yes.
- Is this a regression? I can't tell: this is a new machine.

hughsie, 2023-11-21:
@adamgene do you know what might be happening here? Thanks!

adamgene, 2023-11-22:
Maybe host control has wrong. @paride have you seem the error that monitor connecting on Ubuntu 22.04? And could you provide the model name of the monitor?

paride, 2023-11-22:
@adamgene the monitor is a Lenovo ThinkVision p27h-20, daisy-chanined to a p27h-10 via DisplayPort, so that shouldn't really matter. I don't have a Jammy system at hand, but I can easily downgrade fwupd to the Jammy version should that help. Or I can boot a live Jammy system and test it out, let me know what could be more useful.

paride, 2023-11-22:
@adamgene looks like there are a couple of other bug reports mentioning this issue with the  Genesys 05e3:0610, so I think we have something.

* https://github.com/fwupd/fwupd/issues/5117

> Sep 15 14:11:39 fwupd[2139]: 12:11:39:0928 FuEngine             failed to add device usb:03:00:04:01:02:01:04: failed to get static tool info from device: USB error on device 05e3:0610 : Pipe Error [-9]

---

- https://github.com/fwupd/fwupd/issues/5871

> failed to get static tool info from device: USB error on device 05e3:0610 : Pipe error [-9]

hughsie, 2023-11-22:
@paride can you attach the output of `sudo fwupdtool get-devices --plugins genesys -vv` please.

paride, 2023-11-22:
@hughsie here:

<details>

```
$ sudo fwupdtool get-devices --plugins genesys -vv
09:08:46.952 FuDebug              verbose to debug (on console 0)
09:08:46.953 FuEngine             starting fwupd 1.9.9...
09:08:46.953 FuMain               locked /run/lock/fwupdtool
09:08:46.954 GLib-GIO             Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
09:08:46.957 FuMain               failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
Loading…                 [ -                                     ]Loading…                 [ -                                     ]09:08:46.957 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for 'gio-vfs'
09:08:46.958 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
09:08:46.958 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
09:08:46.958 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x77
09:08:46.958 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
09:08:46.958 FuConfig             Failed to open file "/var/etc/fwupd/fwupd.conf": No such file or directory
09:08:46.958 FuConfig             ::configuration loaded
09:08:46.958 FuRemoteList         loading /usr/share/fwupd/metainfo
09:08:46.958 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
09:08:46.958 XbSilo               file: d7260f34-de69-5d01-7b1a-d183d48e9922, current:d7260f34-de69-5d01-7b1a-d183d48e9922, cached: (null)
09:08:46.958 XbSilo               loading silo with file contents
09:08:46.959 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
09:08:46.959 FuRemoteList         ::remote_list changed
09:08:46.959 FuEngine             priority local and dell-esrt is not download remote, so bumping
09:08:46.959 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
09:08:46.960 FuRemoteList         ::remote_list changed
09:08:46.960 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
09:08:46.960 FuRemoteList         ::remote_list changed
09:08:46.960 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
09:08:46.960 FuRemoteList         ::remote_list changed
09:08:46.960 FuEngine             priority local and vendor-directory is not download remote, so bumping
09:08:46.960 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
09:08:46.960 FuRemoteList         ::remote_list changed
09:08:46.960 FuEngine             priority local and vendor is not download remote, so bumping
09:08:46.960 FuRemoteList         ordering lvfs-testing=lvfs+1
09:08:46.960 FuRemoteList         enabled remotes: dell-esrt[1000], vendor-directory[1000], lvfs[0]
Loading…: 1%
09:08:46.960 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
09:08:46.960 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
09:08:46.961 Jcat                 loaded 1 certificates
09:08:46.961 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
09:08:46.961 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
09:08:46.961 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
09:08:46.961 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
09:08:46.961 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
09:08:46.961 Jcat                 loaded 1 certificates
09:08:46.961 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
09:08:46.961 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2459 bytes
09:08:46.961 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
09:08:46.961 FuEngine             client certificate now exists: nothing to do
Loading…: 2%
09:08:46.961 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
09:08:46.961 FuHistory            got schema version of 9
Loading…: 3%
Loading…: 4%
09:08:46.964 FuPlugin             load(/usr/lib/x86_64-linux-gnu/fwupd-1.9.9/libfu_plugin_modem_manager.so)
09:08:46.964 FuPluginTest         init
09:08:46.964 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
09:08:46.964 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
09:08:46.964 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
09:08:46.964 XbSilo               file: 868417d9-52cc-10df-14f9-2152b943ac91, current:868417d9-52cc-10df-14f9-2152b943ac91, cached: (null)
09:08:46.964 XbSilo               loading silo with file contents
Loading…: 5%
09:08:46.965 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
09:08:46.965 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
09:08:46.965 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x77
09:08:46.965 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
09:08:46.965 FuConfig             Failed to open file "/var/etc/fwupd/fwupd.conf": No such file or directory
09:08:46.965 FuConfig             ::configuration loaded
[... 814 lines cut ...]
  DoneSetup:            true
  CanInvalidate:        false
FuUdevBackend:
[... cut ...]
  Enabled:              true
  DoneSetup:            true
  CanInvalidate:        false
FuBluezBackend:
  Name:                 bluez
  Enabled:              true
  DoneSetup:            true
  CanInvalidate:        false
FuGenesysPlugin:
  Name:                 genesys
  Flags:                ready

09:08:47.072 FuEngine             resetting update motd timeout

No detected devices
09:08:47.073 FuMain               No detected devices

09:08:47.075 FuPluginTest         destroy
```

</details>

hughsie, 2023-11-22:
So this seems to be failing:

	static_buf =
	    g_usb_device_get_string_descriptor_bytes_full(usb_device,
							  static_idx,
							  G_USB_DEVICE_LANGID_ENGLISH_UNITED_STATES,
							  64,
							  error);

adamgene, 2023-11-22:
> So this seems to be failing:

That's right. Maybe the `Power Management for USB` cause? I'm not sure.

adamgene, 2023-11-22:
> Or I can boot a live Jammy system and test it out, let me know what could be more useful.

@paride Please test on Jammy system. Also test connecting each of the two monitors to the host without daisy-chanined.

hughsie, 2023-11-27:
@superm1 can you try 1.8.0 and if that works, then bisect? _No_ is a valid answer :)

superm1, 2023-11-27:
Sure.  It bisects down to https://github.com/fwupd/fwupd/commit/32d6c7f8dd...
I confirmed reverting that on main no errors.  But also no genesys device for the "good one".  
With it there the 2 errors above.

superm1, 2023-11-27:
Well wait a minute; look at my `lsusb` output (hubs.txt).  The two other hubs are missing binary object stores.  Is that the problem?

hughsie, 2023-11-27:
I'm guessing the working hub has new firmware (with a BOS) and the other has older firmware. @adamgene would know for sure.

superm1, 2023-11-27:
The BOS is crucial AFAICT, right?
So if that's the reason for this failure; wouldn't it be better to look for a BOS in `probe()`?  If it's missing presumably `setup()` shouldn't run.

adamgene, 2023-11-28:
> The two other hubs are missing binary object stores. Is that the problem?

BOS is available in USB 3.x hub. And I saw the `hubs.txt`, there are two hubs are USB 2.0 hub (`bcdUSB 2.00`); USB 3.x hub should has two hubs: USB 3.x(`bcdUSB 3.x0`) and USB 2.1(`bcdUSB 2.10`).

```
# lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 10000M
    |__ Port 2: Dev 6, If 0, Class=Hub, Driver=hub/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/12p, 480M
    |__ Port 2: Dev 14, If 0, Class=Hub, Driver=hub/9p, 480M

# lsusb -v -d 05e3:
Bus 002 Device 006: ID 05e3:0629 Genesys Logic, Inc. USB3.2 Hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.20
  bDeviceClass            9 Hub
  bDeviceSubClass         0
  bDeviceProtocol         3
  bMaxPacketSize0         9
  idVendor           0x05e3 Genesys Logic, Inc.
  idProduct          0x0629
  bcdDevice           89.04
  iManufacturer           1 GenesysLogic
  iProduct                2 USB3.2 Hub
  iSerial                 0
...

Bus 001 Device 014: ID 05e3:0630 Genesys Logic, Inc. USB2.1 Hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass            9 Hub
  bDeviceSubClass         0
  bDeviceProtocol         2 TT per port
  bMaxPacketSize0        64
  idVendor           0x05e3 Genesys Logic, Inc.
  idProduct          0x0630
  bcdDevice           89.04
  iManufacturer           1 GenesysLogic
  iProduct                2 USB2.1 Hub
  iSerial                 0
...
```

Can you see the USB 3 hubs on another `BUS`? Maybe is bus 2. And I will check the USB 2 hub product inside my office.

superm1, 2023-11-28:
This is the only other genesys USB device inside this laptop on another bus.

```shell
$ sudo lsusb -d 05e3:0625 -vvv

Bus 002 Device 002: ID 05e3:0625 Genesys Logic, Inc. [unknown]
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.20
  bDeviceClass            9 Hub
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         3
  bMaxPacketSize0         9
  idVendor           0x05e3 Genesys Logic, Inc.
  idProduct          0x0625 [unknown]
  bcdDevice           34.02
  iManufacturer           1 GenesysLogic
  iProduct                2 USB3.2 Hub
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x001f
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes           19
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Feedback
        wMaxPacketSize     0x0002  1x 2 bytes
        bInterval               8
        bMaxBurst               0
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x0049
  bNumDeviceCaps          5
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x00000006
      BESL Link Power Management (LPM) Supported
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x000e
      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   1
      Lowest fully-functional device speed is Full Speed (12Mbps)
    bU1DevExitLat           8 micro seconds
    bU2DevExitLat         190 micro seconds
  SuperSpeedPlus USB Device Capability:
    bLength                28
    bDescriptorType        16
    bDevCapabilityType     10
    bmAttributes         0x00000023
      Sublink Speed Attribute count 4
      Sublink Speed ID count 2
    wFunctionalitySupport   0x1100
      Min functional Speed Attribute ID: 0
      Min functional RX lanes: 1
      Min functional TX lanes: 1
    bmSublinkSpeedAttr[0]   0x00050030
      Speed Attribute ID: 0 5Gb/s Symmetric RX SuperSpeed
    bmSublinkSpeedAttr[1]   0x000500b0
      Speed Attribute ID: 0 5Gb/s Symmetric TX SuperSpeed
    bmSublinkSpeedAttr[2]   0x000a4031
      Speed Attribute ID: 1 10Gb/s Symmetric RX SuperSpeedPlus
    bmSublinkSpeedAttr[3]   0x000a40b1
      Speed Attribute ID: 1 10Gb/s Symmetric TX SuperSpeedPlus
  ** UNRECOGNIZED:  03 10 0b
  Container ID Device Capability:
    bLength                20
    bDescriptorType        16
    bDevCapabilityType      4
    bReserved               0
    ContainerID             {fcde0235-5d71-9740-bab1-525bb2cac9ce}
Hub Descriptor:
  bLength              12
  bDescriptorType      42
  nNbrPorts             3
  wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection
  bPwrOn2PwrGood       60 * 2 milli seconds
  bHubContrCurrent    576 milli Ampere
  bHubDecLat          0.4 micro seconds
  wHubDelay          2308 nano seconds
  DeviceRemovable    0x00
 Hub Port Status:
   Port 1: 0000.02a0 5Gbps power Rx.Detect
   Port 2: 0000.02a0 5Gbps power Rx.Detect
   Port 3: 0000.02a0 5Gbps power Rx.Detect
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
```

adamgene, 2023-11-28:
GL's pure USB2 hub also has the same VID:PID(`05e3:0610`), and the `bcdUSB` may customize to `2.10`. Therefore the `failed to get static tool info from device:` is true; they do not support get tool info.

How can we modify the error log to skip this situation? Or should I check something in `_probe`?

superm1, 2023-11-28:
> How can we modify the error log to skip this situation? Or should I check something in _probe?

So are you meaning they aren't updatable, or they are updatable?
* If they're not updatable I think we should look for this in `probe()`.
* If they're updatable but this failure is expected it should be caught and special cased to ignore in the case of this type of hub.

adamgene, 2023-11-28:
> So are you meaning they aren't updatable, or they are updatable?

They aren't updatable.
 - In `probe()` I should drop `bcdUSB` small than 0x210.
 - In `setup()` I should validate a token likes [hid](https://github.com/fwupd/fwupd/blob/main/plugins/genesys/fu-genesys-hubhid-device.c#L372C1-L372C1).