fwupd/fwupd #6430: fwupd refuses to work with 2 or more HID descriptors

GamePad64, 2023-11-22:
I am making a device with 2 HID descriptors: one for its core functions and one for CFU update. fwupd refuses to work, because this line prohibits enumerating devices with 2 or more HID descriptors

https://github.com/fwupd/fwupd/blob/f920c576143450dad27376ccd936dee4b07573a8/libfwupdplugin/fu-hid-device.c#L110

hughsie, 2023-11-22:
Hmm, looks like we need to parse all the descriptors then -- I've only come across devices with one HID descriptor. Do you know offhand how we know if the HID descriptor is the one we should parse with CFU?

Also, if you're playing with CFU: see https://github.com/fwupd/fwupd/tree/main/plugins/cfu#implementation-notes -- it's a really difficult protocol to implement correctly.

GamePad64, 2023-11-22:
I am using ESP-IDF, so ESP32 OTA machinery is available to me. It manages two partitions, versioning and bootloader settings to switch versions. My device is HID compliant, so I have made a thin CFU wrapper around OTA.
1. I have tried to add CFU descriptor into existing one. Got this issue: https://github.com/hughsie/libgusb/issues/114
2. Now, the issue is fixed and I get another one: 
```
11:26:47.907 FuEngine             failed to add device usb:01:00:01: table invalid, too many duplicate local:usage tokens
```
Which is produced by this line:
https://github.com/fwupd/fwupd/blob/9b27a9f504172989b03551de33505ef8706e4286/libfwupdplugin/fu-hid-descriptor.c#L111
My descriptor contains a LOT of usages, and it is completely valid HID descriptor (works correctly in Windows and Linux apps).
3. I have tried to make this device composite and add a second USB interface (programmatically, of course :) ) with a separate HID descriptor. But I got an error above.

I know, that CFU is pretty quirky, but implementing DFU (with its multiple attach-detach cycles) seems to be much more complicated.
Please, let know if there is any simpler to implement, but standardized firmware update protocol over USB, and I will consider it instead of CFU

hughsie, 2023-11-22:
> My descriptor contains a LOT of usages

Can you share it here please? Sounds like I need to chill the parser out a bit.

GamePad64, 2023-11-22:
> how we know if the HID descriptor is the one we should parse with CFU

Maybe, it could be the first one, that contains all the usages, defined in quirks (or not defined, but "default" ones).

GamePad64, 2023-11-22:
> Can you share it here please?

Sure, here it is:
```
0x05, 0x59,        // Usage Page (0x59)
0x09, 0x01,        // Usage (0x01)
0xA1, 0x01,        // Collection (Application)
0x85, 0x01,        //   Report ID (1)
0x09, 0x02,        //   Usage (0x02)
0xA1, 0x02,        //   Collection (Logical)
0x09, 0x03,        //     Usage (0x03)
0x15, 0x00,        //     Logical Minimum (0)
0x27, 0xFF, 0xFF, 0x00, 0x00,  //     Logical Maximum (65534)
0x75, 0x10,        //     Report Size (16)
0x95, 0x01,        //     Report Count (1)
0xB1, 0x02,        //     Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x09, 0x04,        //     Usage (0x04)
0x09, 0x05,        //     Usage (0x05)
0x09, 0x06,        //     Usage (0x06)
0x09, 0x07,        //     Usage (0x07)
0x09, 0x08,        //     Usage (0x08)
0x15, 0x00,        //     Logical Minimum (0)
0x27, 0xFF, 0xFF, 0xFF, 0x7F,  //     Logical Maximum (2147483646)
0x75, 0x20,        //     Report Size (32)
0x95, 0x05,        //     Report Count (5)
0xB1, 0x03,        //     Feature (Const,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0xC0,              //   End Collection
0x85, 0x02,        //   Report ID (2)
0x09, 0x20,        //   Usage (0x20)
0xA1, 0x02,        //   Collection (Logical)
0x09, 0x21,        //     Usage (0x21)
0x15, 0x00,        //     Logical Minimum (0)
0x27, 0xFF, 0xFF, 0x00, 0x00,  //     Logical Maximum (65534)
0x75, 0x10,        //     Report Size (16)
0x95, 0x01,        //     Report Count (1)
0xB1, 0x02,        //     Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0xC0,              //   End Collection
0x85, 0x03,        //   Report ID (3)
0x09, 0x22,        //   Usage (0x22)
0xA1, 0x02,        //   Collection (Logical)
0x09, 0x21,        //     Usage (0x21)
0x15, 0x00,        //     Logical Minimum (0)
0x27, 0xFF, 0xFF, 0x00, 0x00,  //     Logical Maximum (65534)
0x75, 0x10,        //     Report Size (16)
0x95, 0x01,        //     Report Count (1)
0xB1, 0x02,        //     Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x09, 0x23,        //     Usage (0x23)
0x09, 0x24,        //     Usage (0x24)
0x09, 0x25,        //     Usage (0x25)
0x09, 0x27,        //     Usage (0x27)
0x09, 0x26,        //     Usage (0x26)
0x15, 0x00,        //     Logical Minimum (0)
0x27, 0xFF, 0xFF, 0xFF, 0x7F,  //     Logical Maximum (2147483646)
0x75, 0x20,        //     Report Size (32)
0x95, 0x05,        //     Report Count (5)
0xB1, 0x02,        //     Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x09, 0x28,        //     Usage (0x28)
0x09, 0x29,        //     Usage (0x29)
0x09, 0x2A,        //     Usage (0x2A)
0x09, 0x2B,        //     Usage (0x2B)
0x09, 0x2C,        //     Usage (0x2C)
0x09, 0x2D,        //     Usage (0x2D)
0x15, 0x00,        //     Logical Minimum (0)
0x26, 0xFF, 0x00,  //     Logical Maximum (255)
0x75, 0x08,        //     Report Size (8)
0x95, 0x06,        //     Report Count (6)
0xB1, 0x02,        //     Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0xC0,              //   End Collection
0x85, 0x04,        //   Report ID (4)
0x09, 0x50,        //   Usage (0x50)
[... 566 lines cut ...]
0x75, 0x20,        //   Report Size (32)
0x95, 0x04,        //   Report Count (4)
0x09, 0x8A,        //   Usage (0x8A)
0x91, 0x02,        //   Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x85, 0x1B,        //   Report ID (27)
0x75, 0x20,        //   Report Size (32)
0x95, 0x04,        //   Report Count (4)
0x09, 0x8E,        //   Usage (0x8E)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x85, 0x1C,        //   Report ID (28)
0x75, 0x08,        //   Report Size (8)
0x95, 0x3C,        //   Report Count (60)
0x09, 0x61,        //   Usage (0x61)
0x92, 0x02, 0x01,  //   Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile,Buffered Bytes)
0x85, 0x1D,        //   Report ID (29)
0x75, 0x20,        //   Report Size (32)
0x95, 0x04,        //   Report Count (4)
0x09, 0x66,        //   Usage (0x66)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              // End Collection

// 1368 bytes

```
It implements four instances of HID LampArray (each in its own "Application" collection). And then CFU in its own "Application".

hughsie, 2023-11-22:
> Sure, here it is:

Ohh as a blob please -- then I can chuck it in the fwupd parser.

GamePad64, 2023-11-22:
[hid.bin.zip](https://github.com/fwupd/fwupd/files/13439124/hid.bin.zip)

GamePad64, 2023-11-22:
These usages are part of this spec:
https://www.usb.org/sites/default/files/hutrr84_-_lighting_and_illumination_page.pdf

Page 3 defines the values of these usages: `RedUpdateChannel`, `GreenUpdateChannel`, `BlueUpdateChannel`, `IntensityUpdateChannel`
Page 13 defines the format of `LampMultiUpdateReport`
Pages 18-20 contain an example of HID descriptor, implementing this spec. On page 19 there is a big chunk of usages, similar to mine :)

GamePad64, 2023-11-22:
And yes, it is a really weird spec (why not split channels into arrays and use REPORT_COUNT for this), but this is a standard now. And it is supported by Windows 11 :)

GamePad64, 2023-11-22:
> I've only come across devices with one HID descriptor

I have checked just some things, plugged in my desktop right now. It seems, that composite HID devices are pretty common these days.

```
# lsusb -vd 046d:c52b

Bus 001 Device 006: ID 046d:c52b Logitech, Inc. Unifying Receiver
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x046d Logitech, Inc.
  idProduct          0xc52b Unifying Receiver
  bcdDevice           12.10
  iManufacturer           1 Logitech
  iProduct                2 USB Receiver
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          4 RQR12.10_B0032
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower               98mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      59
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               8
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
[... 343 lines cut ...]
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      33
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               6
Device Status:     0x0000
  (Bus Powered)

```

hughsie, 2023-11-22:
Does https://github.com/fwupd/fwupd/pull/6431 help?

GamePad64, 2023-11-22:
It works perfectly now for two HID descriptors: device is discovered correctly (and I will go this way for my device).

But it doesn't work with a single descriptor situation, where CFU reports are appended to other reports.
It tries to do HID::GetReport(0x0305)  (report type 3, id 5), but the report numbers, used in CFU process are these (the same as default for fwupd):

CfuVersionGetReport = 0x62
CfuOfferSetReport = 0x8A
CfuOfferGetReport = 0x8E
CfuContentSetReport = 0x61
CfuContentGetReport = 0x66

hughsie, 2023-11-22:
> Also it doesn't work with a single descriptor situation

What error message do you get with the wip/hughsie/hid branch?

GamePad64, 2023-11-22:
```
15:07:51.453 FuHidDevice          HID::GetReport [wValue=0x0305, wIndex=0]:05 00 00 
15:07:51.457 FuHidDevice          HID::GetReport [wValue=0x0305, wIndex=0]:05 
15:07:51.946 FuEngine             failed to add device usb:01:00:01: read 1, requested 3 bytes
```

And in device log I see:
```
I (3224) usb: HID get(report_id: 0x5, report_type: 0x3)
```
But there is no such report_id with this report_type, that could be received with GetReport

hughsie, 2023-11-22:
> 14:53:11.150 FuEngine             failed to add device usb:01:00:01: failed to get HID report descriptor: USB error on device 303a:81c1 : Input/Output Error [-1]

That's a device error, no? we're asking for the descriptor and getting EIO.

hughsie, 2023-11-22:
> But there is no such report_id with this report_type, that could be received with GetReport

It's probably getting this from the first HID descriptor, i.e.:

	0x85, 0x05,        //   Report ID (5)
	0x09, 0x60,        //   Usage (0x60)
	0xA1, 0x02,        //   Collection (Logical)
	0x09, 0x55,        //     Usage (0x55)
	0x15, 0x00,        //     Logical Minimum (0)
	0x25, 0x08,        //     Logical Maximum (8)
	0x75, 0x08,        //     Report Size (8)
	0x95, 0x01,        //     Report Count (1)
	0xB1, 0x02,        //     Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
	0x09, 0x61,        //     Usage (0x61)
	0x09, 0x62,        //     Usage (0x62)

GamePad64, 2023-11-22:
But the device enumerates correctly both in Windows and in Linux. HID recommendation happens when the device is powered on and I see it in Windows Device Manager and in lsusb output.
I am afk now, so I can't scan thorough the code, but my guess is that fwupd tries to get second HID descriptor through endpoint 1 and gets EIO.

hughsie, 2023-11-22:
> but my guess is that fwupd tries to get second HID descriptor through endpoint 1 and gets EIO

Is that wrong? The GUsb code looks like this:

	if (!g_usb_device_control_transfer(self,
					   G_USB_DEVICE_DIRECTION_DEVICE_TO_HOST,
					   G_USB_DEVICE_REQUEST_TYPE_STANDARD,
					   G_USB_DEVICE_RECIPIENT_INTERFACE,
					   LIBUSB_REQUEST_GET_DESCRIPTOR,
					   LIBUSB_DT_REPORT << 8,
					   g_usb_interface_get_number(intf),
					   buf2,
					   buf2sz,
					   &actual_length,
					   5000,
					   NULL,
					   error)) {
		g_prefix_error(error, "failed to get HID report descriptor: ");
		return NULL;
	}

GamePad64, 2023-11-22:
It seems it is.

I've googled some info about control requests and found that wValue must have interface index as it's low byte:

https://stackoverflow.com/questions/52106252/how-to-request-usb-hid-report-descriptor-from-hid-devices-using-usbdeviceconnect
https://www.beyondlogic.org/usbnutshell/usb6.shtml

`LIBUSB_DT_REPORT << 8` is just a high byte, we need interface number here

hughsie, 2023-11-22:
Okay, that would make sense, but then I'm confused what vale we use for the "Request index"

hughsie, 2023-11-22:
Maybe just this?

	-                                          LIBUSB_DT_REPORT << 8,
	-                                          g_usb_interface_get_number(intf),
	+                                          (LIBUSB_DT_REPORT << 8) |
	+                                              g_usb_interface_get_number(intf),
	+                                          g_usb_interface_get_index(intf),

Not at my desk until tomorrow, but I can test then.

hughsie, 2023-11-23:
> but I can test then

That didn't work on any of the HID devices I have here either. On the flip side, I did get `LIBUSB_TRANSFER_STALL` rather than `LIBUSB_ERROR_IO` so maybe that's progress.

closed 2023-12-04 by hughsie