fwupd/fwupd #6436: Embargo conf requires a login - we do not have one

ctsdownloads, 2023-11-22:
**Describe the bug**
Create an embargo conf, upon fresh, getting Access denied server error.
**Steps to Reproduce**
Create /etc/fwupd/remotes.d/our-embargo.conf, drop in:
```
[fwupd Remote]
Enabled=true
Title=Embargoed for Us
#Keyring=gpg
MetadataURI=TheExpectedURLhere.xml.gz
ReportURI=https://fwupd.org/lvfs/firmware/report
OrderBefore=lvfs,fwupd
```
Also tried:

```
[fwupd Remote]
Enabled=true
Title=Embargoed for Us
#Keyring=gpg
MetadataURI=TheExpectedURLhere.xml.gz
ReportURI=https://fwupd.org/lvfs/firmware/report
OrderBefore=lvfs,fwupd
#Username=
#Password=
```

Save the file, run fwupdmgr refresh or  fwupdmgr refresh --force as we have on previous releases.
**Expected behavior**
Successfully run fwupdmgr refresh without server errors.
**fwupd version information**
Please provide the version of the daemon and client.

On Fedora 39, fully updated:
```shell
fwupdmgr 1.9.9
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
`dnf`

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```
N/A
</details>

**Additional questions**

- Operating system and version: Fedora 39
- Have you tried rebooting? Yes
- Is this a regression? No, appears to be a new feature. Previous to 1.9.9, our embargo conf worked fine. It's the attempt at authorizing a login that is causing the issue. Ideally, we'd like to see a flag to skip this.

hughsie, 2023-11-23:
Hi! Sorry about this. I sent an email announcing the changes here: https://lists.linuxfoundation.org/pipermail/lvfs-announce/2023-November/000087.html

We can opt out your vendor (Framework?) if you want, but I'd much rather `nrp`, `ktl` or `dhs` create you an LVFS account so that you can use your username+passtoken -- as it's much more secure. Let me know what you'd like to do.

superm1, 2023-11-23:
The side effect of going authentication is going to be that any partners you've shared the embargo conf need to have an explicit relationship described in LVFS in order to access the repo.

@hughsie can LVFS actually express an IHV relationship yet?

hughsie, 2023-11-23:
> can LVFS actually express an IHV relationship yet

Yup, you can even have multiple ODMs for one component now.