fwupd/fwupd #6437: fwupd is still breaking fastboot

thestinger, 2023-11-23:
Previous issue: https://github.com/fwupd/fwupd/issues/3187

New issue filed by an Arch Linux user: https://github.com/GrapheneOS/os-issue-tracker/issues/2708

superm1, 2023-11-23:
@codemodify can you please add output for `sudo fwupdtool get-devices --verbose --verbose` that reproduces this?

thestinger, 2023-11-23:
There's both fastboot mode and fastbootd mode. It might only be breaking one of those, unclear. I can try to get the necessary info for multiple generations of Pixels later.

thestinger, 2023-11-26:
Pixel 8 in fastboot mode:

```
Loading…: 16%
01:29:32.809 FuEngine             usb:01:00:02 added FuUsbDevice:
  Guid:                 4980b2e3-f8f0-5434-8a41-031e35500953 <- USB\VID_18D1&PID_4EE0 ?
  Flags:                none
  VendorId:             USB:0x18D1
  Version:              1.0
  VersionFormat:        bcd
  VersionRaw:           0x00000100
  Guid[quirk]:          0357022b-3c79-5f77-aa9d-fc0ec3e8c934 <- USB\VID_18D1
  Guid[quirk]:          4bd986bc-4aaf-5baf-8bc2-9f7928182387 <- USB\CLASS_FF
  Guid[quirk]:          54f7aa0f-6450-5d0c-ac85-0302809a7f6b <- USB\CLASS_FF&SUBCLASS_42
  Guid[quirk]:          e49bd1ea-2f0e-5709-a60c-63e72f607ba2 <- USB\CLASS_FF&SUBCLASS_42&PROT_03
  PhysicalId:           usb:01:00:02
  AcquiesceDelay:       2500
  PossiblePlugin:       fastboot
  ParentPhysicalIds:    usb:01:00,usb:01
  UsbDeviceClass:       interface-desc

01:29:32.809 FuDevice             using 301046452a49d84af6356d23e43a684b8f10660f for usb:01:00:02
01:29:32.849 FuPluginFastboot     writing:        67 65 74 76 61 72 3a 70 72 6f 64 75 63 74 
01:29:32.849 FuUsbDevice          re-binding kernel driver as not waiting for replug
01:29:32.849 FuEngine             failed to add device usb:01:00:02: failed to do bulk transfer: USB error on device 18d1:4ee0 : Input/Output Error [-1]
Loading…: 17%
01:29:32.849 FuEngine             usb:01:00:04 added FuUsbDevice:
  Flags:                none
  PhysicalId:           usb:01:00:04
  AcquiesceDelay:       2500
  UsbDeviceClass:       interface-desc
```

Pixel 8 in fastbootd mode:

```
Loading…: 16%
01:31:21.907 FuEngine             usb:01:00:02 added FuUsbDevice:
  Guid:                 4980b2e3-f8f0-5434-8a41-031e35500953 <- USB\VID_18D1&PID_4EE0 ?
  Flags:                none
  VendorId:             USB:0x18D1
  Version:              5.15
  VersionFormat:        bcd
  VersionRaw:           0x00000515
  Guid[quirk]:          0357022b-3c79-5f77-aa9d-fc0ec3e8c934 <- USB\VID_18D1
  Guid[quirk]:          4bd986bc-4aaf-5baf-8bc2-9f7928182387 <- USB\CLASS_FF
  Guid[quirk]:          54f7aa0f-6450-5d0c-ac85-0302809a7f6b <- USB\CLASS_FF&SUBCLASS_42
  Guid[quirk]:          e49bd1ea-2f0e-5709-a60c-63e72f607ba2 <- USB\CLASS_FF&SUBCLASS_42&PROT_03
  PhysicalId:           usb:01:00:02
  AcquiesceDelay:       2500
  PossiblePlugin:       fastboot
  ParentPhysicalIds:    usb:01:00,usb:01
  UsbDeviceClass:       interface-desc

01:31:21.907 FuDevice             using 301046452a49d84af6356d23e43a684b8f10660f for usb:01:00:02
01:31:21.939 FuPluginFastboot     writing:        67 65 74 76 61 72 3a 70 72 6f 64 75 63 74 
01:31:21.942 FuPluginFastboot     read:           4f 4b 41 59 73 68 69 62 61 
01:31:21.942 FuDevice             301046452a49d84af6356d23e43a684b8f10660f device overwriting name value: Pixel 8->Fastboot shiba
01:31:21.942 FuPluginFastboot     writing:        67 65 74 76 61 72 3a 76 65 72 73 69 6f 6e 
01:31:21.944 FuPluginFastboot     read:           4f 4b 41 59 30 2e 34 
01:31:21.944 FuPluginFastboot     fastboot version 0.4
01:31:21.944 FuPluginFastboot     writing:        67 65 74 76 61 72 3a 76 65 72 73 69 6f 6e 2d 62 6f 6f 74 6c 6f 61 64 65 72 
01:31:21.947 FuPluginFastboot     read:           4f 4b 41 59 72 69 70 63 75 72 72 65 6e 74 2d 31 34 2e 30 2d 31 30 38 33 39 34 33 36 
01:31:21.947 FuDevice             changing verfmt for 301046452a49d84af6356d23e43a684b8f10660f: bcd->pair
01:31:21.947 FuDevice             ripcurrent-14.0-10839436 is not a valid pair (guessed plain)
01:31:21.947 FuPluginFastboot     writing:        67 65 74 76 61 72 3a 73 65 72 69 61 6c 6e 6f 
01:31:21.950 FuPluginFastboot     read:           4f 4b 41 59 33 37 31 33 31 46 44 4a 48 30 30 35 30 4b 
01:31:21.950 FuPluginFastboot     writing:        67 65 74 76 61 72 3a 73 65 63 75 72 65 
01:31:21.951 FuPluginFastboot     read:           4f 4b 41 59 79 65 73 
01:31:21.951 FuPlugin             emit added from fastboot: 301046452a49d84af6356d23e43a684b8f10660f
[... 367 lines cut ...]
01:31:21.957 FuBackend            0 devices with subsystem thunderbolt
01:31:21.957 FuBackend            0 devices with subsystem mei
01:31:21.957 FuBackend            10 devices with subsystem i2c
01:31:21.958 FuBackend            2 devices with subsystem drm
01:31:21.958 FuBackend            0 devices with subsystem drm_dp_aux_dev
01:31:21.960 FuBackend            10 devices with subsystem hidraw
01:31:21.960 FuBackend            0 devices with subsystem mtd
01:31:21.963 FuBackend            42 devices with subsystem pci
01:31:21.964 FuBackend            0 devices with subsystem firmware-attributes
01:31:21.964 FuBackend            2 devices with subsystem gpio
01:31:21.964 FuBackend            0 devices with subsystem video4linux
01:31:21.964 FuBackend            0 devices with subsystem i2c-dev
01:31:21.964 FuBackend            0 devices with subsystem serio
01:31:21.964 FuBackend            2 devices with subsystem nvme
01:31:21.966 FuBackend            25 devices with subsystem platform
01:31:21.966 FuEngine             /sys/devices/LNXSYSTM:00/LNXSYBUS:00/MSFT0101:00/tpm/tpm0 added FuUdevDevice:
  Summary:              TPM 2.0 Device
  Flags:                none
  BackendId:            /sys/devices/LNXSYSTM:00/LNXSYBUS:00/MSFT0101:00/tpm/tpm0
  AcquiesceDelay:       2500
  PossiblePlugin:       tpm
  Subsystem:            tpm
  DeviceFile:           /dev/tpm0
  SysfsPath:            /sys/devices/LNXSYSTM:00/LNXSYBUS:00/MSFT0101:00/tpm/tpm0
```

thestinger, 2023-11-26:
It's likely that fwupd has always been causing this issue with fastbootd mode, which has been around since dynamic partitions were introduced in Android 10.

GVFS had a bug breaking fastboot and we thought it was all being caused by GVFS. Could have reported this to you a long time ago, probably years ago, but we didn't realize there was another common service causing a problem. It has caused a lot of harm because we didn't figure it out there was a 2nd major source of this problem earlier.

Many users trying to install GrapheneOS from a Linux distribution with fwupd have run into this. Many hundreds of people have asked about it. It's likely thousands and thousands of people have run into it. CLI install can work despite this bug seemingly because `fastboot` tends to beat `fwupd` to the race while Chromium does not. It leads to people wrongly thinking web install doesn't work as well. We've generally had to get people to use macOS, Windows, Android or ChromeOS to run the web installer to work around this. A lot of people gave up. We've added instructions for working around this to our install guides, so it shouldn't be a problem going forward since most users on platforms with fwupd can run a CLI command:

https://grapheneos.org/install/cli#working-around-fwupd-bug-on-linux-distributions
https://grapheneos.org/install/web#working-around-fwupd-bug-on-linux-distributions

It impacts us more because our web installer is the recommended installation method and Chromium almost always loses the race as opposed to CLI fastboot which can win fairly often and succeed at flashing since the device only boots to fastbootd mode once. We're going to need to keep this workaround in our guides for years due to Debian and Ubuntu. We removed our workaround for the GVFS issue because the workaround didn't work reliably anyway.

It's likely there's other software beyond GVFS and fwupd causing these issues, but at least we figured out part of it. We've seen issues on other operating systems caused by GPG related tools including macOS. Combine that with USB drivers and hardware (ports, hubs, cables) often being unreliable and it has been quite the mess...

hughsie, 2023-11-26:
> It's likely thousands and thousands of people have run into it

Okay, that might be true, but I think it might be better to work on a solution together rather than using all that emotional language. Lets unpick this a bit. What's essentially happening is that three things are all trying to access the fastboot device at the same time, correct? There are I think 4 ways to do this:

* An _arbiter_, usually in hardware
* Retrying on EBUSY
* Co-operative inhibits
* Using fwupd to deploy the image

We could use a software _arbiter_, which something that controls access to the device and that other services and applications can use. That thing might be fastbootd,  I don't really know if fastbootd provides the kind of access we need in fwupd, but I'm not against just using that rather than accessing the hardware directly. I don't think we can assume fastbootd is installed and running in all distros tho.

Another way of a few different things all accessing the hardware at the same time is by retrying on EBUSY or EAGAIN -- then the thing with the hardware access gets the device, and everything else just polls the device until it's free. This is what fwupd does with some other devices -- it's the blunt chisel approach but works surprisingly well.

Another way is for fwupd to say to fastbootd (or fastboots to fwupd) "don't touch the device please" and for the *other thing* to ignore the device while the first process is accessing the device. This means one thing has to trust the other thing to always exist, and one project has to agree to be the manager.

Of course the last way is to actually use fwupd to deploy the firmware rather than using fastboot CLI -- which would maybe be quite easy as you can use a simple DBus method call and just pass a file descriptor with the payload. Its also means something else (polkit) handles the authorisation elevation for you too. Of course we might need to teach the fwupd plugin a few new tricks like unlocking.

Also, ChromeOS also has fwupd installed by default, although I think perhaps the fastboot plugin isn't enabled on production hardware.

thestinger, 2023-11-26:
You connecting to the fastboot device after we've already been using it for installation. Installing takes multiple reboots, and doing it safely takes multiple extra reboots. It's also not simply blocking GrapheneOS and others from using it for the standard, intended purpose but rather is interrupting the process halfway into it and leaving users with a soft bricked device if they don't realize what happened, reboot back into fastboot mode from fastbootd mode, disable fwupd and start the process over again.

Our web installer is not supposed to involve any CLI commands, but currently has to include a command for stopping your software: https://grapheneos.org/install/web#working-around-fwupd-bug-on-linux-distributions. We'll have to add this to other guides too. Since it starts breaking things against after each reboot we would be telling people to disable and uninstall it if they didn't need it for best effort updates of some of their firmware...

Why is fwupd using fastboot in the first place? This isn't how it's meant to be used. Fastboot was created for booting into a special mode to flash operating systems onto devices for installation and development. It's not supposed to be used for updates and it's not safe to use it for that purpose. Android devices are supposed to be updated via their update system which is done via recovery for installing an update via USB. The standard protocol for sideloading an update verifies the signature and does downgrade protection. Flashing via fastboot is only possible when unlocked and can easily brick the device. It will break devices without A/B partitioning if anything cuts it off. It can still break devices with A/B partitioning because you do not know what's on the other slot. Flashing something that doesn't boot will cause rollback to the other slot which may not boot if it's a downgrade. This isn't the safe update system.

Sideloading an update is robust against bricking, particularly on A/B update devices where even the power fully cutting out at any point during writing out the update won't break anything because the active slot hasn't been switched to what was flashed, and even after it has it will fall back to the known good previous active slot. Fastboot has none of these guarantees. GrapheneOS uses a very careful approach to flashing with fastboot taking into account A/B slots and rollback by flashing the inactive slot, making it active, booting into it and then flashing the previous slot from there.

I don't know why fwupd is using fastboot in the first place and it seems like the wrong choice was made. If you were using the regular update sideloading protocol then you would be interacting with the ADB service which handles multiplexing. The fastboot protocol really isn't meant to be used automatically and isn't as robust. It's for development and installing a different OS, not updates.

This approach is breaking the official fastboot tool, the official factory images for Pixels and similar factory images from other vendors, the official Android flashing tool and other things based on fastboot. It's not only breaking it for GrapheneOS. We're simply the ones who had a user figure out what was wrong with their setup.

Google doesn't currently seem aware that their ecosystem is being broken this way, but if I can't get anywhere then I'm going to let them know and get them to deal with it instead. I'll also contact distributions about it. I'm confident I can convince distributions to disable the broken functionality.

We can't control how Chromium-based browsers connect to USB devices via WebUSB. You may not agree, but you're breaking things for a ton of people and I believe that I can convince distributions this behavior is wrong even if I can't convince you.

hughsie, 2023-11-26:
I think we need a code of conduct document.

thestinger, 2023-11-26:
> I don't think we can assume fastbootd is installed and running in all distros tho.

fastbootd is not a service in the OS being used for flashing. fastbootd is a service running on the device being flashed. It's what you're connecting to with the fastboot protocol, in the middle of the GrapheneOS installation process or installing any other OS on Pixel or any other modern Android devices doing things the standard way. Installing flashes the firmware via fastboot mode to the inactive slot, makes it active and reboots into it, repeats that again, then flashes the core OS, reboots into fastbootd mode, flashes the high level OS, reboots back into fastboot mode and finishes up including flashing verified boot key, etc. before the user locks. **It's extremely dangerous for anything to be connecting to fastboot automatically since it can be in the middle of doing critical things and it does not have any inherent multiplexing support**. It's not the protocol used for updates. ADB has a daemon for multiplexing and has a safe protocol for updates which uses a safe update client (update_engine) on the other side.

I don't think you understand how much trouble this has caused for us. Many users have had their devices soft bricked because of this, and some of them didn't realize it was a soft brick.

thestinger, 2023-11-26:
> I think we need a code of conduct document.

I'm not sure how anything that I've said here is at all inappropriate. However, you do have multiple coworkers participating in an extreme harassment campaign towards me and perhaps you're aware of that. Please delete this issue. I genuinely do not feel safe participating here after seeing this play out.

thestinger, 2023-11-26:
Please delete this issue as soon as possible. I don't feel safe participating in an issue tracker associated with IBM. Someone working on Arch Linux will file a new issue to replace this.

hughsie, 2023-11-26:
I have no idea what's going on here. Please stop modifying and deleting comment in issues, it's hugely confusing.

hughsie, 2023-11-26:
I've locked and closed this issue as there were several breaches of the code of conduct and anything of use was deleted: https://github.com/fwupd/fwupd/blob/main/CODE_OF_CONDUCT.md

closed 2023-11-26 by hughsie