fwupd/fwupd #6438: Create a Guide for Usage on Windows

SonarBeserk, 2023-11-23:
**Is your feature request related to a problem? Please describe.**
When attempting to install fwupd on Windows there appears to be no Icons / Start Menu  or Path updates made by the installer which normally would be how I identify how to run a tool. I see very little mention of Windows in general other than issues such as https://github.com/fwupd/fwupd/issues/6156 and I wish to update the firmware of my keyboard.

**Describe the solution you'd like**
A guide for install / usage on Windows that matches the guide for Linux. A Windows specific webpage would be very helpful if there are caveats or special details that needed to be added. Ideally high enough up in the wiki sidebar to be found by new users or in the readme, ideally both if it makes sense.

**Describe alternatives you've considered**
Ive attempted to find a desktop icon to launch the tool, a start menu icon, tried to run the exe files by opening fwupdmgr.exe directly from the bin folder, and added the bin to the PATH and running it through that method.

**Additional context**
OS: Windows 10 22H2
Terminal: Cmd / Windows Terminal / Git Bash from Git for Windows 2.43.0

- Error when running fwupdmgr.exe by double clicking in Explorer:
```bash
The code execution cannot proceed because libfwupdplugin.dll was not found. Reinstalling the program my fix this problem.
```

- Error when running fwupdmgr.exe through CMD:
```bash
C:/Program Files (x86)/fwupd/bin/fwupdmgr.exe: error while loading shared libraries: libfwupdutil.dll: cannot open shared object file: No such file or directory
```

- No errors but no output when running though Git Bash

hughsie, 2023-11-23:
I think the tl;dr is that you have to open a command prompt with admin access, then cd to `C:/Program Files (x86)/fwupd/` then do something like `fwupdtool.exe refresh` -- the Windows tools are really designed for factory use and are not really for public consumption. I did ponder building gnome-firmware to Windows and to use that as a GUI, but that's not something my employer (Red Hat) would approve. I think we could do a small quickstart in docs/ or something.

superm1, 2023-11-23:
> I think the tl;dr is that you have to open a command prompt with admin access, then cd to `C:/Program Files (x86)/fwupd/` then do something like `fwupdtool.exe refresh` -- the Windows tools are [...]

I have successfully built gnome firmware and gotten it working for win32. But I did it without a cross build because Fedora is missing dependencies (iirc).

hughsie, 2023-11-23:
And of course, now I'm trying to write a document -- it doesn't work. Initially it was failing because the rpath wasn't being used, but you can just move the fwupd-1.9.9/* into bin and that allows it to run -- but getting the device list shows me a error like `entry point g_string_free_and_steal could not be located in libjson-glib-1.0-0.dll` -- but weirdly I couldn't reproduce in wine on Linux. I can't really spend much time on this now, but if anyone wants to work it out I'd be very grateful.

hughsie, 2023-11-27:
Lets try that in https://github.com/fwupd/fwupd/pull/6455

closed 2024-01-02 by hughsie

necros2k7, 2024-03-11:
so it`s not working on Win as of now right?