fwupd/fwupd #6462: Failed to update Colohug (v1) from 1.1.6 to 1.2.6

PhracturedBlue, 2023-11-28:
**Describe the bug**
Trying to update a very old colorhug device (v1 fw 1.1.6) to the latest available fails.

**Steps to Reproduce**
```
sudo fwupdmgr update
WARNING: UEFI firmware can not be updated in legacy BIOS mode
See https://github.com/fwupd/fwupd/wiki/PluginFlag:legacy-bios for more information.
Devices with no available firmware updates:
 • ColorHug
 No updatable devices
```
```
sudo fwupdtool --plugins colorhug update
Loading…                 [**************                         ]14:45:17.666 FuDevice             1.1.6 is not a valid bcd (guessed triplet)
Loading…                 [************************************** ]
Devices with no available firmware updates:
 • ColorHug
```
```
 sudo fwupdtool install c6fbb716abbb204d98f12edf1f146b6406f39b1eade741b353c15a86f5da8278-hughski-colorhug-1.2.6.cab
Loading…                 [*******                                ]14:52:46.979 FuDevice             1.1.6 is not a valid bcd (guessed triplet)
Loading…                 [*******************                    ]
WARNING: UEFI firmware can not be updated in legacy BIOS mode
See https://github.com/fwupd/fwupd/wiki/PluginFlag:legacy-bios for more information.
Waiting…                 [*******************                    ]14:52:47.692 FuEngine             failed to attach device after failed update: failed to attach using colorhug: failed to boot to runtime: Unknown command
```
```
sudo fwupdtool get-details $PWD/c6fbb716abbb204d98f12edf1f146b6406f39b1eade741b353c15a86f5da8278-hughski-colorhug-1[123/1818]
Loading…                 [**************                         ]14:44:16.345 FuDevice             1.1.6 is not a valid bcd (guessed triplet)
Loading…                 [************************************** ]
WARNING: UEFI firmware can not be updated in legacy BIOS mode
See https://github.com/fwupd/fwupd/wiki/PluginFlag:legacy-bios for more information.
Decompressing…           [ -                                     ]MSI MS-7522
│

└─ColorHug:
  │   Device ID:          9f9d275b28405fc5bd4f8b940a95a345aa743334
  │   Description:
  │   Updating the firmware on your ColorHug device improves performance and adds new features.
  │   Current version:    1.1.6
  │   Vendor:             Hughski Ltd. (USB:0x04D8)
  │   Install Duration:   8 seconds
  │   GUIDs:              40338ceb-b966-4eae-adae-9c32edfcc484
  │                       6bc5ff27-d631-5660-9991-6d24954c6f90
  │                       afdcc391-6c33-5914-b4d2-b4dd71fe9c5a ← USB\VID_273F&PID_1000
  │                       4ad44bb7-c600-5432-870c-559bdcedfd07 ← USB\VID_04D8&PID_F8DA
  │                       e6292a9b-89cb-54d3-8fd5-fef9bbb09bdb ← USB\VID_04D8&PID_F8DA&REV_0002
  │   Device Flags:       • Updatable
  │                       • Is in bootloader mode
  │                       • Device can recover flash failures
  │                       • End of life
  │                       • Unsigned Payload
  │
  └─ColorHug:
        New version:      1.2.6
        Summary:          Firmware for the Hughski ColorHug Colorimeter
        License:          GPL-2.0+
        Size:             16.4 kB
        Urgency:          Low
        Source:           https://github.com/hughski/colorhug1-firmware/releases/tag/1.2.6
        Release Flags:    • Trusted payload
                          • Trusted metadata
        Description:
        This release fixes prevents the firmware returning an error when the remote SHA1 hash was never sent.
```

**fwupd version information**
```
runtime   org.freedesktop.fwupd         1.8.12
compile   com.hughsie.libxmlb           0.3.10
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    6.1.0-9-amd64
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.8.12
runtime   org.freedesktop.gusb          0.3.10
```
Installed from Debian bookworm

hughsie, 2023-11-28:
> sudo fwupdtool install c6fbb716abbb204d98f12edf1f146b6406f39b1eade741b353c15a86f5da8278-hughski-colorhug-1.2.6.cab

Can you attach the output of `sudo fwupdtool install c6fbb716abbb204d98f12edf1f146b6406f39b1eade741b353c15a86f5da8278-hughski-colorhug-1.2.6.cab --plugins colorhug -vv` please?

PhracturedBlue, 2023-11-28:
Sure, here:
[log.txt](https://github.com/fwupd/fwupd/files/13491184/log.txt)

hughsie, 2023-11-28:
@PhracturedBlue wow, that's one of the super-early ColorHugs! That's the device that kicked off the fwupd and the LVFS projects. Can you try https://github.com/fwupd/fwupd/pull/6464 for me please?

PhracturedBlue, 2023-11-28:
Thank you for your help.  I am happy that you are willing to help with such old hardware.
I uninstalled the distro version, and I believe I built the new version properly, however 'get-devices' can no longer see the colorhug:
fwupdtool get-devices -vv
[log2.txt](https://github.com/fwupd/fwupd/files/13493268/log2.txt)

Just to be sure I built properly, I built 'main' and then I could see the CologHug in 'get-devices' (but of course the 'install' fails as before)

hughsie, 2023-11-28:
> however 'get-devices' can no longer see the colorhug

Can you pull the branch and try again please -- logic thinko.

PhracturedBlue, 2023-11-28:
Thanks.  It at least tried to upload the firmware now, but it failed in the process, and left the device in the bootloader.  I needed to replug to revert.

fwupdtool install c6fbb716abbb204d98f12edf1f146b6406f39b1eade741b353c15a86f5da8278-hughski-colorhug-1.2.6.cab -vv 2
[log3.txt](https://github.com/fwupd/fwupd/files/13493340/log3.txt)

get-devices after failure (before replug):
[log4.txt](https://github.com/fwupd/fwupd/files/13493343/log4.txt)

hughsie, 2023-11-29:
@PhracturedBlue can you try with todays fixup please. That seemed to work for me when I forced all the VIDPIDs to USB\VID_04D8&PID_F8DA

PhracturedBlue, 2023-11-29:
It didn't help.  Here are the logs with your latest changes:
[log5.txt](https://github.com/fwupd/fwupd/files/13501541/log5.txt)

But now I may have made things worse (or maybe better?).
After the 1st failure, while it was in the bootloader mode, I tried to flash it again.  It still failed (here are the logs):
[log6.txt](https://github.com/fwupd/fwupd/files/13501550/log6.txt)

But after failure, 'get-devices' reported he firmware as 1.2.6.
I tested some of the colorhug utilities, and they seem to be able to communicate with it (I didn't do much more than read the sensor and twiddle the LED).
**HOWEVER:**  Now when I replug the device, it always comes back to the bootloader.  So I need to reload the firmware on every plugin.

PhracturedBlue, 2023-11-29:
...I noticed that from a replug, I can run a 'boot-flash' command to boot into the 1.2.6 firmware, so I don't need to reflash each time.  It just doesn't immediately boot into the firmware on its own. Maybe this is the intended behavior for this firmware?  In any case, it seems like it is sufficient for me to use the device.  If this isn't the intended mode of operation, and/or you want me to continue to do further testing, let me know, I'm happy to help.

hughsie, 2023-11-29:
Okay, I think I see what's happening. We're not setting the target Counterpart GUID. One more try with the latest fixup please?

PhracturedBlue, 2023-11-29:
I don't see any difference with the latest change.  But as I said, I'm now starting from the bootloader, not from the loaded firmware.  i.e. starting from the bootloader, I can load 1.2.6, but it doesn't boot into the bootloader on a replug

[log7.txt](https://github.com/fwupd/fwupd/files/13502224/log7.txt)

PhracturedBlue, 2023-11-29:
Ok  I installed 1.2.0, and then without replugging, I tried to install 1.2.6.  This brought me back to the bootloader, but did not attempt to install 1.2.6 afterwards:
[log8.txt](https://github.com/fwupd/fwupd/files/13502434/log8.txt)

PhracturedBlue, 2023-11-29:
One more comment:
I am able to flash the firmware via the ` colorhug-cmd flash-firmware` command (after extracting it from the cab) and it boots properly on a replug.  I have tested with both 1.2.6 and 1.2.0.  I currently have 1.2.0 on the device so I can continue testing fwupd (I verified 1.2.6 upgrade with fwupd still doesn't work...just goes back to bootloader and hangs there), but I am probably the only person who will ever be in this situation, so I'm not sure if you want to continue to use your bandwidth solving this issue.
I'm happy to continue helping to get it work if you want.

hughsie, 2023-11-29:
I'm having trouble reproducing this issue -- colorhug #0001 got the booloader upgraded and so I don't have anything to test this with. I'll hunt down in my archives -- what bootloader version does fwupdmgr report when it's in bootloader mode? it'll be something like 0.x.x

ghaushe-ampere, 2023-11-29:
It reports 0.1.2

PhracturedBlue, 2023-11-29:
It reports 0.1.2