fwupd/fwupd #6467: Add release flag for requests

rishabh-agr, 2023-11-29:
**Is your feature request related to a problem? Please describe.**
I think adding a release flag that states if that release has any requests or not would be very valuable. As an extension it might be valuable to add 2 release flags, generic requests and non-generic requests. I feature flag for this was recently added.

**Why?**
We want to be able to filter releases based on requests so we can identify which devices we can support and add them to our test cases

hughsie, 2023-11-29:
@rishabh-chromium I think https://github.com/fwupd/fwupd/commit/f51a04530df60a5b7d4603d20a4bc6bbe7da336a gets us some way there -- i.e. I don't think we need a public flag like `FWUPD_DEVICE_FLAG_REQUIRES_NON_GENERIC_REQUEST` as the device just won't be marked as updatable if the session software doesn't opt-in with the new feature.

@superm1 I guess something like `FWUPD_DEVICE_FLAG_REQUIRES_INTERACTION` might makes sense -- it's also something I could do with in the device tests -- at the moment we hardcode a manual `"interactive": true` for some of the devices we need to skip in CI and it would be nice to get rid of that. Of course, we could just lean into the feature flags feature even more and mark all interactive devices as non-updatable if the existing `FWUPD_FEATURE_FLAG_REQUESTS` is not selected by the client. That's certainly cleaner, but would need a `FU_DEVICE_INTERNAL_FLAG_REQUIRES_INTERACTION` sprinkled in a few plugins -- I also think we should drop all the "set a device message using the plugin request" compat junk at the same time if we do that just for our own sanity.

Ideas?

rishabh-chromium, 2023-12-01:
I guess this should be a Release flag instead of a device flag since a release could or could not have requests for any one device...?

hughsie, 2023-12-01:
I moved away from the idea of a per-release flag as it also depends on the `from -> to`  and it all gets very messy -- I've got a PR coming up -- I think we just need to say what interaction the device *could* require.

rishabh-chromium, 2023-12-01:
Gotcha. Cant remember if we ended up deciding on an Enum or a bit flag

closed 2023-12-01 by hughsie