fwupd/fwupd #6477: Firmware update fails because of not enough storage

schiessle, 2023-12-01:
**Describe the bug**

I try to update the firmware of my Thinkpad L13 on Fedora 39

[image]

if I run "fwupdmgr update" I get the following error:

```/run/media/root/PQSERVICE does not have sufficient space, required 90,1 MB, got 51,2 MB```

I don't really know where this mountpoint comes from. I can't see it by looking at the mounted drives. The EFI partition also has enough storage:

[image]

[image]

**Steps to Reproduce**
run fwupdmgr update

**Expected behavior**
Firmware update gets applied without an error

**fwupd version information**

```shell
fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.9 
compile   com.hughsie.libxmlb           0.3.14 
compile   com.hughsie.libjcat           0.1.14 
runtime   org.freedesktop.fwupd-efi     1.4 
compile   org.freedesktop.gusb          0.4.8 
runtime   com.hughsie.libjcat           0.1.14 
runtime   org.freedesktop.gusb          0.4.8 
runtime   org.freedesktop.fwupd         1.9.9 
runtime   org.kernel                    6.6.2-201.fc39.x86_64
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
Installed via dnf

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices

LENOVO 21FNCTO1WW
│
├─ELAN067F:00 04F3:3209:
│     Device ID:          80db34198c1165629a06c6c1a55a976b25b219f1
│     Summary:            Touchpad
│     Current version:    0x0004
│     Bootloader Version: 0x0003
│     Vendor:             ELAN Microelectronics (HIDRAW:0x04F3)
│     GUIDs:              38897f13-3a41-5b8a-9cc8-9370a2b41291 ← HIDRAW\VEN_04F3&DEV_3209
│                         4b69bffd-d456-54ae-baee-e11b517fa3bd ← HIDRAW\VEN_04F3&DEV_3209&MOD_0037
│                         3313b7ba-c61b-5183-a9c4-3411cc892998 ← ELANTP\ICTYPE_14&MOD_0037
│                         a97baf40-d178-5cf6-81e6-874542d698e2 ← ELANTP\ICTYPE_14&MOD_0037&DRIVER_HID
│     Device Flags:       • Internal device
│                         • Updatable
│   
├─AMD Ryzen 7 PRO 7730U with Radeon Graphics:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x0a50000d
│ │   Vendor:             Advanced Micro Devices, Inc.
│ │   GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│ │                       79759cdc-94db-5098-be7b-eb02521fbbec ← CPUID\PRO_0&FAM_19&MOD_50
[... 248 lines cut ...]
      Update State:       Success
      Problems:           • Device requires AC power to be connected
      GUID:               36ac1b96-a134-4030-b71f-8fdbd944b258
      Device Flags:       • Internal device
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
                          • Updatable
    
────────────────────────────────────────────────
Devices that were not updated correctly:
 • System Firmware (0.1.10 → 0.1.11)
Devices that have been updated successfully:
 • UEFI Device Firmware (0.1.6 → 0.1.07)
 • UEFI dbx (190 → 371)

```

</details>

**Additional questions**

- Operating system and version: Fedora 39
- Have you tried rebooting? yes
- Is this a regression? probably yes (previous firmware updates on the same device worked)

schiessle, 2023-12-01:
I just mounted the second EFI partition. The 4GB are almost completely filled up, this might be the root of the issue?

/dev/nvme0n1p2   4186104   4136076     50028  99% /mnt/test

Question is, what could I do about it?

hughsie, 2023-12-01:
How did you install Fedora 39? Do you know why you have two EFI partitions? If you use `sudo fwupdtool esp-list` can you show us what's on each one?

schiessle, 2023-12-01:
I did a normal installation with the Fedora 39 iso from a usb stick, auto-partitioning. The laptop was shipped with Ubuntu, maybe the two partitions where created already by Lenovo?

This is the content of /org/freedesktop/UDisks2/block_devices/nvme0n1p2 (seems to be indeed like a lot of Ubuntu stuff, something like a "rescue" system from Lenovo?):

<details>

```
/run/media/root/PQSERVICE/.disk/base_installable
/run/media/root/PQSERVICE/.disk/casper-uuid-oem
/run/media/root/PQSERVICE/.disk/cd_type
/run/media/root/PQSERVICE/.disk/info
/run/media/root/PQSERVICE/.disk/ubuntu_dist_channel
/run/media/root/PQSERVICE/.oem/client_type
/run/media/root/PQSERVICE/.oem/locale
/run/media/root/PQSERVICE/EFI/boot/grubx64.efi
/run/media/root/PQSERVICE/EFI/boot/mmx64.efi
/run/media/root/PQSERVICE/EFI/boot/shimx64.efi
/run/media/root/PQSERVICE/EFI/fedora/fw/fwupd-d8353e6e-c244-4f95-9826-eb1f1d006365.cap
/run/media/root/PQSERVICE/boot/grub/cedartraillist.txt
/run/media/root/PQSERVICE/boot/grub/common.cfg
/run/media/root/PQSERVICE/boot/grub/efi
/run/media/root/PQSERVICE/boot/grub/fonts/unicode.pf2
/run/media/root/PQSERVICE/boot/grub/grub.cfg
/run/media/root/PQSERVICE/boot/grub/grub.cfg.old
/run/media/root/PQSERVICE/boot/grub/grubenv
/run/media/root/PQSERVICE/boot/grub/i386-pc/915resolution.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/acpi.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/adler32.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/affs.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/afs.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/afsplitter.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/ahci.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/all_video.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/aout.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/archelp.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/at_keyboard.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/ata.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/backtrace.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/bfs.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/biosdisk.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/bitmap.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/bitmap_scale.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/blocklist.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/boot.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/bsd.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/bswap_test.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/btrfs.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/bufio.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/cat.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/cbfs.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/cbls.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/cbmemc.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/cbtable.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/cbtime.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/chain.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/cmdline_cat_test.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/cmosdump.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/cmostest.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/cmp.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/cmp_test.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/command.lst
/run/media/root/PQSERVICE/boot/grub/i386-pc/configfile.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/cpio.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/cpio_be.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/cpuid.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/crc64.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/crypto.lst
/run/media/root/PQSERVICE/boot/grub/i386-pc/crypto.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/cryptodisk.mod
/run/media/root/PQSERVICE/boot/grub/i386-pc/cs5536.mod
[... 927 lines cut ...]
/boot/efi/EFI/BOOT/mmx64.efi
/boot/efi/EFI/BOOT/BOOTIA32.EFI
/boot/efi/EFI/BOOT/fbia32.efi
/boot/efi/EFI/ubuntu/BOOTX64.CSV
/boot/efi/EFI/ubuntu/grub.cfg
/boot/efi/EFI/ubuntu/grubx64.efi
/boot/efi/EFI/ubuntu/mmx64.efi
/boot/efi/EFI/ubuntu/shimx64.efi
/boot/efi/EFI/fedora/grub.cfg
/boot/efi/EFI/fedora/BOOTIA32.CSV
/boot/efi/EFI/fedora/BOOTX64.CSV
/boot/efi/EFI/fedora/gcdia32.efi
/boot/efi/EFI/fedora/gcdx64.efi
/boot/efi/EFI/fedora/grubia32.efi
/boot/efi/EFI/fedora/grubx64.efi
/boot/efi/EFI/fedora/mmia32.efi
/boot/efi/EFI/fedora/mmx64.efi
/boot/efi/EFI/fedora/shim.efi
/boot/efi/EFI/fedora/shimia32.efi
/boot/efi/EFI/fedora/shimx64.efi
/boot/efi/modules.log
/boot/efi/System/Library/CoreServices/SystemVersion.plist
/boot/efi/mach_kernel
```
</details>

hughsie, 2023-12-01:
> I did a normal installation with the Fedora 39 iso from a usb stick, auto-partitioning

It might be easier to wipe the disk and start afresh to be honest. You can certainly just set the ESP in fwupd.conf, but you're also wasting a lot of space on the recovery partition... :)

schiessle, 2023-12-01:
Could I wipe everything? Both EFI partitions and they will be recreated during installation from scratch without loosing any important firmware data?

hughsie, 2023-12-01:
> Could I wipe everything?

It's what I do, but you probably want to back things up first just in case.

superm1, 2023-12-01:
Maybe we want to make an exclusion for that partition label for ESP detection. It looks like a recovery image.

hughsie, 2023-12-01:
Yes, that's probably a good idea. @schiessle can you attach the output of `udisksctl dump` please.

schiessle, 2023-12-01:
Sure, this is the output of `udisksctl dump`

<details>

```
/org/freedesktop/UDisks2/Manager:
  org.freedesktop.UDisks2.Manager:
    DefaultEncryptionType:      luks2
    SupportedEncryptionTypes:   luks1
                                luks2
    SupportedFilesystems:       ext2
                                ext3
                                ext4
                                xfs
                                vfat
                                ntfs
                                f2fs
                                nilfs2
                                exfat
                                btrfs
                                udf
                                swap
    Version:                    2.10.1
  org.freedesktop.UDisks2.Manager.NVMe:
    HostID:             
    HostNQN:            nqn.2014-08.org.nvmexpress:uuid:75c9ce4c-2f90-11b2-a85c-ac77b0cba5b0

/org/freedesktop/UDisks2/block_devices/nvme0n1:
  org.freedesktop.UDisks2.Block:
    Configuration:              []
    CryptoBackingDevice:        '/'
    Device:                     /dev/nvme0n1
    DeviceNumber:               66304
    Drive:                      '/org/freedesktop/UDisks2/drives/KBG5AZNT1T02_LA_KIOXIA_934PC90VQAXU'
    HintAuto:                   false
    HintIconName:               
    HintIgnore:                 false
    HintName:                   
    HintPartitionable:          true
    HintSymbolicIconName:       
    HintSystem:                 true
    Id:                         by-id-nvme-KBG5AZNT1T02_LA_KIOXIA_934PC90VQAXU
    IdLabel:                    
    IdType:                     
    IdUUID:                     
    IdUsage:                    
    IdVersion:                  
    MDRaid:                     '/'
    MDRaidMember:               '/'
    PreferredDevice:            /dev/nvme0n1
    ReadOnly:                   false
    Size:                       1024209543168
    Symlinks:                   /dev/disk/by-diskseq/1
                                /dev/disk/by-id/nvme-KBG5AZNT1T02_LA_KIOXIA_934PC90VQAXU
                                /dev/disk/by-id/nvme-KBG5AZNT1T02_LA_KIOXIA_934PC90VQAXU_1
                                /dev/disk/by-id/nvme-eui.8ce38e04046793e2
                                /dev/disk/by-path/pci-0000:01:00.0-nvme-1
    UserspaceMountOptions:      
  org.freedesktop.UDisks2.NVMe.Namespace:
    EUI64:                      8ce38e04046793e2
    FormatPercentRemaining:     -1
    FormattedLBASize:           (512, 0, 0x03)
    LBAFormats:                 [(512, 0, 0x03), (4096, 0, 0x02)]
    NGUID:                      
    NSID:                       1
    NamespaceCapacity:          2000409264
    NamespaceSize:              2000409264
    NamespaceUtilization:       2000409264
    UUID:                       
    WWN:                        eui.8ce38e04046793e2
[... 263 lines cut ...]
    TimeMediaDetected:          1701471919142004
    Vendor:                     
    WWN:                        
  org.freedesktop.UDisks2.NVMe.Controller:
    ControllerID:                       0
    FGUID:                              
    NVMeRevision:                       1.4
    SanitizePercentRemaining:           -1
    SanitizeStatus:                     never_sanitized
    SmartCriticalWarning:               
    SmartPowerOnHours:                  18
    SmartSelftestPercentRemaining:      -1
    SmartSelftestStatus:                success
    SmartTemperature:                   297
    SmartUpdated:                       1701471920
    State:                              live
    SubsystemNQN:                       nqn.2019-10.com.kioxia:KBG5AZNT1T02 LA KIOXIA:934PC90VQAXU
    UnallocatedCapacity:                0
```

</details>

Btw, I thought about what would be the best way for me to solve the issue. As a lot of work went already in the configuration of my system I would prefer to avoid a complete wipe.

Therefore I thought it would be better to set the ESP in fwupd.conf and if this works, only delete nvme0n1p2 and resize my other partition. Do you think this approach makes sense? What would I have to add to the fwupd.conf so that nvme0n1p1 is used for the firmware?

superm1, 2023-12-01:
I think we should be able to exclude your disk partition from the automatic detection.

schiessle, 2023-12-01:
> I think we should be able to exclude your disk partition from the automatic detection.

This would be great. It would mean that I could also just wait for one of the next updates of fwupd, where this change is included, and than the firmware update would work again. Right?

superm1, 2023-12-02:
Would you be able to check if the detection helps on https://github.com/fwupd/fwupd/pull/6482? 

You can check out that branch, build it, and then run `sudo ./build/src/fwupdtool esp-list` and it should show the right ESP.

superm1, 2023-12-02:
And if still doesn't work, please provide output for `sudo ./build/src/fwupdtool esp-list --verbose --verbose`

mrhpearson, 2023-12-02:
If this is our Ubuntu preload then I think the 2nd partition is the recovery image partition....though I have no idea why it's marked as an EFI partition.
Realistically it should be safe to delete. You should be able to get the preload image via the Lenovo digital download service if you brought with the preload; and if not you're welcome to private message me and I'll get you a link to the download in case you ever want to recover the system.

I'll follow up with the team in China next week - maybe this is an issue with our preload process? It's a bit curious as I run dual-booted Ubuntu/Fedora systems all the time and haven't hit this...but my systems haven't been thru the formal release process as they're all prototypes..

superm1, 2023-12-02:
I don't expect it's a problem with your preload. I think it's because of the algorithm change that happened in newer fwupd to try to pick among multiple candidates for ESP.

It "looks" better because it is bigger. Hopefully that change can just exclude anything tagged a with the recovery partition name is enough. Or maybe @hughsie has a better idea how to skip it.

In my opinion it's better we caught this now before people take 22.04 preloads with older fwupd and upgrade to 24.04 and then hit it 6 months from now.

schiessle, 2023-12-02:
I just compiled it. The output of `sudo ./build/src/fwupdtool esp-list` still shows both partitions:

```
 sudo ./build/src/fwupdtool esp-list
0.	Cancel
1.	/org/freedesktop/UDisks2/block_devices/nvme0n1p2
2.	/org/freedesktop/UDisks2/block_devices/nvme0n1p1
```

```
$ sudo ./build/src/fwupdtool esp-list
0.	Cancel
1.	/org/freedesktop/UDisks2/block_devices/nvme0n1p2
2.	/org/freedesktop/UDisks2/block_devices/nvme0n1p1
Choose volume [0-2]: ^CTerminated
schiesbn@potato:~/Repos/fwupd$ sudo ./build/src/fwupdtool esp-list --verbose --verbose
09:45:07.582 FuDebug              verbose to debug (on console 1)
09:45:07.582 FuEngine             starting fwupd 1.9.10…
09:45:07.593 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
09:45:07.594 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1, type: (null), internal: 1, fs: 
09:45:07.596 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p3, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: ext4
09:45:07.598 FuVolume             device /org/freedesktop/UDisks2/block_devices/zram0, type: (null), internal: 1, fs: 
09:45:07.600 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p4, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: btrfs
09:45:07.602 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
09:45:07.611 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
09:45:07.612 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1, type: (null), internal: 1, fs: 
09:45:07.614 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p3, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: ext4
09:45:07.616 FuVolume             device /org/freedesktop/UDisks2/block_devices/zram0, type: (null), internal: 1, fs: 
09:45:07.617 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p4, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: btrfs
09:45:07.619 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
09:45:07.619 FuContext            no volumes of type ebd0a0a2-b9e5-4433-87c0-68b6b72699c7
0.	Cancel
1.	/org/freedesktop/UDisks2/block_devices/nvme0n1p2
2.	/org/freedesktop/UDisks2/block_devices/nvme0n1p1
```

schiessle, 2023-12-02:
> Sure -- we'll show both but we should pick the right one now.

Ah, ok. Than this was my missunderstanding

> Can you try a firmware update please?

Seems not possible to update the firmware with the compiled version:

```
schiesbn@potato:~/Repos/fwupd$ sudo  ./build/src/fwupdmgr update
[sudo] password for schiesbn: 
WARNING: This package has not been validated, it may not work properly.
Unsupported daemon version 1.9.9, client version is 1.9.9-85-g6434dcfa9
```

Any hint how I could get it running? Would it be safe to try "--force" ?

superm1, 2023-12-02:
Try ./build/src/fwupdtool update

schiessle, 2023-12-10:
I just received the update to fwupd 1.9.10 from Fedora which should contain the fix according to the release notes, right?

But I still see the same error :(

My current versions:

```
$ sudo fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.10
compile   com.hughsie.libxmlb           0.3.14
compile   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         1.9.10
runtime   org.kernel                    6.6.4-200.fc39.x86_64
```

superm1, 2023-12-10:
What are the new logs showing you?

schiessle, 2023-12-11:
This is the esp list:

````
$ sudo fwupdtool esp-list --verbose --verbose
13:37:21.119 FuDebug              verbose to debug (on console 1)
13:37:21.119 FuEngine             starting fwupd 1.9.10…
13:37:21.129 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
13:37:21.133 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p3, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: ext4
13:37:21.136 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p4, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: btrfs
13:37:21.138 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
13:37:21.147 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
13:37:21.150 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p3, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: ext4
13:37:21.154 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p4, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: btrfs
13:37:21.156 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
13:37:21.156 FuContext            no volumes of type ebd0a0a2-b9e5-4433-87c0-68b6b72699c7
0.	Cancel
1.	/org/freedesktop/UDisks2/block_devices/nvme0n1p2
2.	/org/freedesktop/UDisks2/block_devices/nvme0n1p1
````

The fwupd journal shows this when I try to update the firmware:

```
Dez 11 14:39:23 potato fwupd[3519]: 13:39:23.034 FuEngine             HSI attribute org.fwupd.hsi.EncryptedRam (from pci_psp) had unknown result
Dez 11 14:39:23 potato fwupd[3519]: 13:39:23.034 FuEngine             HSI attribute org.fwupd.hsi.PlatformFused (from pci_psp) had unknown result
Dez 11 14:39:23 potato fwupd[3519]: 13:39:23.034 FuEngine             HSI attribute org.fwupd.hsi.PlatformDebugLocked (from pci_psp) had unknown result
Dez 11 14:39:23 potato fwupd[3519]: 13:39:23.034 FuEngine             HSI attribute org.fwupd.hsi.Amd.RollbackProtection (from pci_psp) had unknown result
Dez 11 14:39:23 potato fwupd[3519]: 13:39:23.034 FuEngine             HSI attribute org.fwupd.hsi.Amd.SpiReplayProtection (from pci_psp) had unknown result
Dez 11 14:39:23 potato fwupd[3519]: 13:39:23.034 FuEngine             HSI attribute org.fwupd.hsi.Amd.SpiWriteProtection (from pci_psp) had unknown result
```

The update itself shows the same error es before:

```
$ sudo fwupdmgr update
Devices with no available firmware updates: 
 • ELAN067F:00 04F3:3209
 • Fingerprint Sensor
 • Integrated Camera
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • KBG5AZNT1T02 LA KIOXIA
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 0.1.10 to 0.1.11?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ ThinkPad L13 Gen 4/ L13 Yoga Gen 4 System Firmware Version 0.1.11            ║
║                                                                              ║
║ • MR00487 - Changed fan error is not displayed (CP20230710_0006)             ║
║ • MR00286 - Removed fan error skip flag                                      ║
║ • delete the old EC ESRT GUID in ESRTPLATFORMENTRY variable                  ║
║ • Support Time out function with F1/F2 string prompt. (MFG Linux Preload     ║
║ issue)                                                                       ║
║ • Update EDID-format_-Fenrir2 intel & AMD-V05-20230605-Add COST LCD by       ║
║ CP20230420_0006.xlsx                                                         ║
║ • Fixed Ubuntu check_tsc_sync_source failed                                  ║
║ • CP20230627_0009 Remove MSC mode(Mods) for CS22/CS23 AMD Products           ║
║ • Fixed BIOS don’t verify FvMain and FvABT region                            ║
║ • UNDI driver update Realtek(USB3 LAN v2.040)                                ║
║ • add Secure Boot DBX update Q1-2023                                         ║
║ • Improve NIST backup process                                                ║
║ • Fix a buffer overflow issue                                                ║
║ • CP20230412_0002 Asynchronize Samsung PM9B1 LTS SSD                         ║
║ • CP20230627_0009 Remove MSC mode(Mods) for CS22/CS23 AMD Products           ║
║ • Phoenix 002-006 OpenSSL 1.1.1 is out of date                               ║
║ • OpenSSL 1.1.1t Update                                                      ║
║ • Integer Overflow in various edk2 CreateHob() instances                     ║
║ • update to PI1000E                                                          ║
║ • Modify flash order from "Update BIOS -> NIST backup -> Update EC" to       ║
║ "Update BIOS -> Update EC -> NIST backup"                                    ║
║                                                                              ║
║ 21FNCTO1WW must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Decompressing…           [                                       ]
/run/media/root/PQSERVICE does not have sufficient space, required 90,1 MB, got 51,2 MB
```

Anything else I can provide to help?

superm1, 2023-12-11:
Ah I see what's wrong.  I used the wrong thing to trigger it.  I used this:
```
  org.freedesktop.UDisks2.Partition:
    Flags:              0
    IsContained:        false
    IsContainer:        false
    Name:               "Recovery Partition"
```

But I should have used this:
```
    IdLabel:                    PQSERVICE
```

I'll send up a PR.

schiessle, 2024-01-12:
unfortunately I have to come back to the issue. Today I received the update to fwupd 1.9.11 which should contain the latest fix, right? I still face the same problems :(

current version:

````
$ sudo fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.11
compile   com.hughsie.libxmlb           0.3.14
compile   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         1.9.11
runtime   org.kernel                    6.6.9-200.fc39.x86_64
````

`fwupdtool esp-list --verbose --verbose` creates now a huge list of errors:

<details>

````
$ sudo fwupdtool esp-list --verbose --verbose
23:03:09.296 FuDebug              verbose to debug (on console 1)
23:03:09.296 FuEngine             starting fwupd 1.9.11…
23:03:09.297 FuMain               locked /run/lock/fwupdtool
Loading…                 [ -                                     ]23:03:09.306 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
23:03:09.306 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
23:03:09.306 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x77
23:03:09.306 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
23:03:09.306 FuConfig             Failed to open file “/var/etc/fwupd/fwupd.conf”: No such file or directory
23:03:09.306 FuConfig             ::configuration loaded
Loading…                 [                                       ]23:03:09.306 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
23:03:09.306 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
23:03:09.306 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
23:03:09.306 Jcat                 loaded 1 certificates
23:03:09.306 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
23:03:09.306 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
23:03:09.306 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
23:03:09.306 Jcat                 loaded 1 certificates
23:03:09.306 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2459 bytes
23:03:09.306 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
23:03:09.306 FuEngine             client certificate now exists: nothing to do
Loading…                 [                                       ]23:03:09.306 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
23:03:09.307 FuHistory            got schema version of 9
Loading…                 [*                                      ]23:03:09.311 FuPlugin             load(/usr/lib64/fwupd-1.9.11/libfu_plugin_modem_manager.so)
23:03:09.312 FuPluginTest         init
23:03:09.312 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
23:03:09.312 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
23:03:09.312 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
23:03:09.313 XbSilo               file: f2a7904e-dda5-d47b-4ffb-43f3a1488085, current:f2a7904e-dda5-d47b-4ffb-43f3a1488085, cached: (null)
23:03:09.313 XbSilo               loading silo with file contents
Loading…                 [*                                      ]23:03:09.313 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
23:03:09.313 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
23:03:09.313 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x77
23:03:09.313 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
23:03:09.313 FuConfig             Failed to open file “/var/etc/fwupd/fwupd.conf”: No such file or directory
23:03:09.313 FuConfig             ::configuration loaded
23:03:09.313 FuStruct             SmbiosEp64:
  anchor_str: _SM3_
  entry_point_csum: 0xbf
  entry_point_len: 0x18
  smbios_major_ver: 0x3
  smbios_minor_ver: 0x3
  smbios_docrev: 0x0
  entry_point_rev: 0x1
  reserved0: 0x0
  structure_table_len: 0xa60
  structure_table_addr: 0xc5e28000
23:03:09.313 FuStruct             SmbiosStructure:
[... 781 lines cut ...]
FuDellPlugin:
  Name:                 dell
  Order:                2
FuUefiRecoveryPlugin:
  Name:                 uefi_recovery
  Flags:                require-hwid
  Order:                2

23:03:10.744 FuEngine             resetting update motd timeout
23:03:10.755 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
23:03:10.758 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p3, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: ext4
23:03:10.761 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p4, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: btrfs
23:03:10.762 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
23:03:10.768 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
[... cut ...]
23:03:10.773 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p4, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: btrfs
23:03:10.774 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
23:03:10.774 FuContext            no volumes of type ebd0a0a2-b9e5-4433-87c0-68b6b72699c7

0.	Cancel
1.	/org/freedesktop/UDisks2/block_devices/nvme0n1p2
2.	/org/freedesktop/UDisks2/block_devices/nvme0n1p1
Choose volume [0-2]:     
````
</details>

hughsie, 2024-01-13:
@schiessle can you grab us `sudo fwupdtool get-devices --plugins uefi-capsule -vv` please.

schiessle, 2024-01-13:
@hughsie this is the output:

<details>

````
$ sudo fwupdtool get-devices --plugins uefi-capsule -vv
[sudo] password for schiesbn: 
16:46:34.355 FuDebug              verbose to debug (on console 1)
16:46:34.356 FuEngine             starting fwupd 1.9.11…
16:46:34.356 FuMain               locked /run/lock/fwupdtool
Loading…                 [ -                                     ]16:46:34.370 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
16:46:34.370 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
16:46:34.370 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x77
16:46:34.370 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
16:46:34.370 FuConfig             Failed to open file “/var/etc/fwupd/fwupd.conf”: No such file or directory
16:46:34.370 FuConfig             ::configuration loaded
16:46:34.370 FuRemoteList         loading /usr/share/fwupd/metainfo
16:46:34.371 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
16:46:34.371 XbSilo               file: 025dd757-8036-da01-5042-0f16bf1690f1, current:025dd757-8036-da01-5042-0f16bf1690f1, cached: (null)
16:46:34.371 XbSilo               loading silo with file contents
16:46:34.371 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
16:46:34.371 FuRemoteList         ::remote_list changed
16:46:34.371 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
16:46:34.371 FuRemoteList         ::remote_list changed
16:46:34.371 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
16:46:34.371 FuRemoteList         ::remote_list changed
16:46:34.371 FuEngine             priority local and vendor-directory is not download remote, so bumping
16:46:34.371 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
16:46:34.371 FuRemoteList         ::remote_list changed
16:46:34.371 FuEngine             priority local and vendor is not download remote, so bumping
16:46:34.371 FuRemoteList         ordering lvfs-testing=lvfs+1
16:46:34.371 FuRemoteList         enabled remotes: vendor-directory[1000], lvfs[0]
Loading…                 [                                       ]16:46:34.371 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
16:46:34.371 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
16:46:34.372 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
16:46:34.372 Jcat                 loaded 1 certificates
16:46:34.372 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
16:46:34.372 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
16:46:34.372 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
16:46:34.372 Jcat                 loaded 1 certificates
16:46:34.372 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2459 bytes
16:46:34.372 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
16:46:34.372 FuEngine             client certificate now exists: nothing to do
Loading…                 [                                       ]16:46:34.372 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
16:46:34.373 FuHistory            got schema version of 9
Loading…                 [*                                      ]16:46:34.378 FuPlugin             load(/usr/lib64/fwupd-1.9.11/libfu_plugin_modem_manager.so)
16:46:34.380 FuPluginTest         init
16:46:34.380 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
16:46:34.381 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
16:46:34.381 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
16:46:34.381 XbSilo               file: f2a7904e-dda5-d47b-4ffb-43f3a1488085, current:f2a7904e-dda5-d47b-4ffb-43f3a1488085, cached: (null)
16:46:34.381 XbSilo               loading silo with file contents
Loading…                 [*                                      ]16:46:34.382 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
16:46:34.382 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
16:46:34.382 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x77
16:46:34.382 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
16:46:34.382 FuConfig             Failed to open file “/var/etc/fwupd/fwupd.conf”: No such file or directory
16:46:34.382 FuConfig             ::configuration loaded
16:46:34.382 FuStruct             SmbiosEp64:
  anchor_str: _SM3_
  entry_point_csum: 0xbf
  entry_point_len: 0x18
  smbios_major_ver: 0x3
  smbios_minor_ver: 0x3
  smbios_docrev: 0x0
  entry_point_rev: 0x1
  reserved0: 0x0
  structure_table_len: 0xa60
  structure_table_addr: 0xc5e28000
16:46:34.382 FuStruct             SmbiosStructure:
[... 1117 lines cut ...]

16:46:35.989 FuMain               FuUefiNvramDevice:
  DeviceId:             5bfc5b91c18ef8b751d3d052af77b69ebf738038
  Name:                 UEFI Device Firmware
  Guid:                 36ac1b96-a134-4030-b71f-8fdbd944b258
  Summary:              UEFI ESRT device
  Plugin:               uefi_capsule
  Protocol:             org.uefi.capsule
  Flags:                internal|updatable|require-ac|registered|needs-reboot|usable-during-update
[... cut ...]
  Version:              66305
  VersionLowest:        66305
  VersionFormat:        number
  VersionRaw:           0x00010301
  VersionLowestRaw:     0x00010301
  Created:              2024-01-13
  UpdateState:          success

16:46:35.990 FuPluginTest         destroy

16:46:35.992 FuProgress           raw timing data was { 0,079, 0,001, 0,008, 0,001, 0,009, 0,001 } -- steps were set as [ 63 1 8 1 26 1 ] but should have been [ 80 1 8 1 9 1 ] at ../plugins/uefi-capsule/fu-uefi-capsule-plugin.c:1110

````

</details>

hughsie, 2024-01-15:
> 16:46:35.942 FuPluginUefiCapsule  skipping partition '"Recovery_Partition"'
> 16:46:35.956 FuPluginUefiCapsule  skipping partition 'EFI_System_Partition'

Oopa, two bugs -- both fixed in https://github.com/fwupd/fwupd/pull/6675

curtrube, 2024-04-25:
Hello, I am facing this same issue running ubuntu 24.04 on a thinkpad x1 carbon.

`➜  ~ fwupdtool --version
Loading…                 [************************************** ]
compile   org.freedesktop.fwupd         1.9.16
compile   com.hughsie.libxmlb           0.3.17
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         1.9.16
runtime   org.kernel                    6.1.0-1028-oem
`

`➜  ~ sudo fwupdtool update
Loading…                 [************************************** ]
Devices with no available firmware updates: 
 • ELAN067B:00 04F3:31F8
 • Integrated RGB Camera
 • Prometheus IOTA Config
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • Embedded Controller
 • Intel Management Engine
 • MZVL21T0HDLU-00BLL
 • Prometheus
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 0.1.20 to 0.1.25?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Lenovo System Firmware Version 1.25                                          ║
║                                                                              ║
║ Important updates                                                            ║
║                                                                              ║
║ • Enhancement to address security vulnerabilities                            ║
║                                                                              ║
║ New functions or enhancements                                                ║
║                                                                              ║
║ • Support Lenovo Memory Self Repair(PPR) function.                           ║
║                                                                              ║
║ Problem fixes                                                                ║
║                                                                              ║
║ •                                                                            ║
║ Fixed an issue where system might take a minute to boot when user attached   ║
║                                                                              ║
║ Docking/Monitor with Realtek USB Ethernet device.                            ║
║                                                                              ║
║ • Fixed an issue where system might take a minute to boot with               ║
║ Kingstone Duo USB memory in AOU port.                                        ║
║                                                                              ║
║ LENOVO 21HMCTO1WW must remain plugged into a power source for the duration   ║
║ of the update to avoid damage.                                               ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: Y
Decompressing…           [*******************                    ]
/media/root/PQSERVICE does not have sufficient space, required 78.7 MB, got 50.1 MB
 • UEFI dbx
`

roberto-zamparelli, 2024-08-17:
Sorry to reopen this issue, but it is still there in 2025 on Lenovo Thinkpad X1 carbon, running Kubuntu 24.04. I have compiled the latest fwupd, in the hope to solve the issue, but I get:
 

> @ base) roberto:~/fwupd/venv$ sudo ./build/src/fwupdtool update
> [sudo] password for roberto: 
> Loading…                 [************************************** ]
> WARNING: This package has not been validated, it may not work properly.
> ╔══════════════════════════════════════════════════════════════════════════════╗
> ║ Upgrade System Firmware from 0.1.18 to 0.1.30?                               ║
> ╠══════════════════════════════════════════════════════════════════════════════╣
> ║ Lenovo System Firmware Version 1.30                                          ║
> ║                                                                              ║
> ║ Important updates                                                            ║
> ║                                                                              ║
> ║ • Enhancement to address security vulnerability                              ║
> ║                                                                              ║
> ║ New functions or enhancements                                                ║
> ║                                                                              ║
> ║ • Updated the Diagnostics module to version 04.35.000.                       ║
> ║ • Updated the CPU microcode.                                                 ║
> ║                                                                              ║
> ║ LENOVO 21HNS4PK00 must remain plugged into a power source for the duration   ║
> ║ of the update to avoid damage.                                               ║
> ╚══════════════════════════════════════════════════════════════════════════════╝
> Perform operation? [Y|n]: 
> Decompressing…           [*******************                    ] Less than one minute remaining…
> failed to write-firmware: /media/root/PQSERVICE does not have sufficient space, required 78,7 MB, got 50,1 MB

Context: the same two partitions:

> base) roberto:~/fwupd/venv$ fwupdtool esp-list
> Loading…                 [************************************** ]
> 0.      Cancel
> 1.      /org/freedesktop/UDisks2/block_devices/nvme0n1p2
> 2.      /org/freedesktop/UDisks2/block_devices/nvme0n1p1

Anything I can do to solve this? I was hoping that a firmware update could go in the direction of fixing the symantec fingerprint reader, which essentially does not work on Thinkpad X1.

superm1, 2024-08-17:
When you say latest, can you specify what commit hash?  We just made a fix for this (again!) which I'm surprised to hear is not working.

superm1, 2024-08-17:
And if you're trying something with that in place please attach output  for `fwupdtool esp-list -vv` and `udisksctl dump`.

roberto-zamparelli, 2024-08-17:
First, I am not a developer, so pardon any naive error I may have committed.

I followed the instruction in https://github.com/fwupd/fwupd/blob/main/docs/building.md and used:
git clone https://github.com/fwupd/fwupd.git less than 30 min ago.  

(base) roberto:~/fwupd$  venv/build/src/fwupdtool --version
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.0
compile   com.hughsie.libxmlb           0.3.18
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
runtime   com.hughsie.libxmlb           0.3.x
runtime   com.hughsie.libjcat           0.2.0
runtime   org.kernel                    6.1.0-1022-oem
runtime   org.freedesktop.fwupd         2.0.0
s:

Where do you read the commit hash? 

here are the other outputs. 

fwupdtool esp-list -vv
[esp-list.txt](https://github.com/user-attachments/files/16646527/esp-list.txt)

udisksctl dump
[udisksctl.txt](https://github.com/user-attachments/files/16646528/udisksctl.txt)

Can you read them?

superm1, 2024-08-18:
> fwupdtool esp-list -vv
[esp-list.txt](https://github.com/user-attachments/files/16646527/esp-list.txt)

```
21:24:35.291 FuEngine             starting fwupd 1.9.16...
```

Ummm You don't seem to be running the version from the venv.  Did you remember to run `source venv/bin/activate`?

superm1, 2024-08-19:
Using fwupdtool from the venv get me that `fwupdtool esp-list -vv` output again.

roberto-zamparelli, 2024-08-19:
Voilà

Roberto

12:39:06.867 FuDebug              verbose to debug (on console 0)
12:39:06.868 FuEngine             starting fwupd 2.0.0...
12:39:06.868 FuMain               locked /run/lock/fwupdtool
12:39:06.868 FuConfig             loading config /home/roberto/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf
12:39:06.868 FuConfig             loading config /home/roberto/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf
12:39:06.868 FuConfig             trying to load config values from /home/roberto/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf
12:39:06.868 FuCommon             mapped file /home/roberto/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf of size 0x33
12:39:06.868 FuConfig             trying to load config values from /home/roberto/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf
12:39:06.868 FuCommon             mapped file /home/roberto/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf of size 0x33
12:39:06.868 FuConfig             ::configuration loaded
12:39:06.868 Jcat                 reading /home/roberto/fwupd/venv/bin/../dist/etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
12:39:06.868 Jcat                 loaded 1 certificates
12:39:06.868 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
12:39:06.868 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
12:39:06.868 Jcat                 reading /home/roberto/fwupd/venv/bin/../dist/etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
12:39:06.868 Jcat                 loaded 1 certificates
12:39:06.868 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
12:39:06.868 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
12:39:06.868 Jcat                 reading /home/roberto/fwupd/venv/bin/../dist/lib/fwupd/pki/secret.key with 2459 bytes
12:39:06.868 Jcat                 reading /home/roberto/fwupd/venv/bin/../dist/lib/fwupd/pki/client.pem with 1383 bytes
12:39:06.868 FuEngine             client certificate now exists: nothing to do
12:39:06.868 FuHistory            trying to open database '/home/roberto/fwupd/venv/bin/../dist/lib/fwupd/pending.db'
12:39:06.868 FuHistory            got schema version of 13
12:39:06.869 FuPlugin             load(/home/roberto/fwupd/venv/bin/../dist/lib/x86_64-linux-gnu/fwupd-2.0.0/libfu_plugin_modem_manager.so)
12:39:06.871 FuQuirks             open database /home/roberto/fwupd/venv/bin/../dist/cache/fwupd/quirks.db
12:39:06.871 FuQuirks             mtimes unchanged: quirks,pci.ids:1711847705,usb.ids:1710796935,pnp.ids:1707542972, doing nothing
12:39:06.871 FuQuirks             loading quirks from /home/roberto/fwupd/venv/bin/../dist/share/fwupd/quirks.d
12:39:06.871 FuQuirks             loading quirks from /home/roberto/fwupd/venv/bin/../dist/lib/fwupd/quirks.d
12:39:06.871 XbSilo               attempting to load /home/roberto/fwupd/venv/dist/cache/fwupd/quirks.xmlb
12:39:06.871 XbSilo               file: 8dcd6abc-6188-2adc-1116-53f236cf25ae, current:8dcd6abc-6188-2adc-1116-53f236cf25ae, cached: (null)
12:39:06.871 XbSilo               loading silo with file contents
12:39:06.872 FuConfig             loading config /home/roberto/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf
12:39:06.872 FuConfig             loading config /home/roberto/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf
12:39:06.872 FuConfig             trying to load config values from /home/roberto/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf
12:39:06.872 FuCommon             mapped file /home/roberto/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf of size 0x33
12:39:06.872 FuConfig             trying to load config values from /home/roberto/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf
12:39:06.872 FuCommon             mapped file /home/roberto/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf of size 0x33
12:39:06.872 FuConfig             ::configuration loaded
12:39:06.872 FuStruct             FuStructSmbiosEp64:
  anchor_str: _SM3_
  entry_point_csum: 0x19
  entry_point_len: 0x18
  smbios_major_ver: 0x3
  smbios_minor_ver: 0x4
  smbios_docrev: 0x0
  entry_point_rev: 0x1
  reserved0: 0x0
  structure_table_len: 0x13a2
  structure_table_addr: 0x59a88000
12:39:06.872 FuStruct             FuStructSmbiosStructure:
  type: 0x86
  length: 0xd
  handle: 0x0
12:39:06.872 FuStruct             FuStructSmbiosStructure:
  type: 0x10
  length: 0x17
  handle: 0x1
12:39:06.872 FuStruct             FuStructSmbiosStructure:
  type: 0x11
  length: 0x5c
  handle: 0x2
12:39:06.872 FuStruct             FuStructSmbiosStructure:
  type: 0x11
  length: 0x5c
  handle: 0x3
12:39:06.872 FuStruct             FuStructSmbiosStructure:
[... 895 lines cut ...]
  Name:                 dell
  Flags:                none
Order:                  2
FuUefiRecoveryPlugin:
  Name:                 uefi_recovery
  Flags:                require-hwid
Order:                  2

12:39:06.969 FuEngine             resetting update motd timeout
[31m[1mWARNING[0m: This package has not been validated, it may not work properly.
12:39:06.989 FuVolume             Looking for volumes of type c12a7328-f81f-11d2-ba4b-00a0c93ec93b
12:39:06.994 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
12:39:07.005 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
[... cut ...]
12:39:07.065 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
12:39:07.076 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p3, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: ext4
12:39:07.081 FuContext            no volumes of type ebd0a0a2-b9e5-4433-87c0-68b6b72699c7
Request canceled
12:39:10.944 FuMain               Request canceled

12:39:10.945 FuPlugin             finalize(modem_manager)
12:39:10.945 FuPlugin             finalize(flashrom)
12:39:10.945 FuPluginTest         destroy

superm1, 2024-08-20:
Can you try the pull request I just published?  If this doesn't help, please attach (as a file) the output from both `fwupdtool get-devices -vv` and `fwupdtool esp-list -vv`

roberto-zamparelli, 2024-08-20:
Unfortunately, it does not work yet. From the fwupd dir I did:
git pull
./contrib/setup
source  venv/bin/activate
build-fwupd
sudo ./build/src/fwupdtool refresh
sudo ./build/src/fwupdtool update
[...]
WARNING: This package has not been validated, it may not work properly.
Upgrade System Firmware from 0.1.18 to 0.1.30?
[... info about the Lenovo System Firmer and its functionalities...]
Perform operation? [Y|n]:
Decompressing…           [*******************                    ]
failed to write-firmware: /media/root/PQSERVICE does not have sufficient
space, required 78,7 MB, got 50,1 MB

Attached: output to fwupdtool esp-list -vv 2> esp-list.txt and fwupdtool
get-devices -vv 2> get-devices.txt

20:38:16.123 FuDebug              verbose to debug (on console 0)
20:38:16.124 FuEngine             starting fwupd 2.0.0...
20:38:16.124 FuMain               locked /home/roberto/.cache/fwupd/fwupdtool
20:38:16.124 FuConfig             loading config /home/roberto/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf
20:38:16.124 FuConfig             not loading config /home/roberto/fwupd/venv/bin/../dist/var/etc/fwupd/fwupd.conf
20:38:16.124 FuConfig             skipping mode check for /home/roberto/fwupd/venv/bin/../dist/var/etc/fwupd/fwupd.conf as not writable
20:38:16.124 FuConfig             trying to load config values from /home/roberto/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf
20:38:16.124 FuCommon             mapped file /home/roberto/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf of size 0x33
20:38:16.124 FuConfig             trying to load config values from /home/roberto/fwupd/venv/bin/../dist/var/etc/fwupd/fwupd.conf
20:38:16.124 FuConfig             Failed to open file "/home/roberto/fwupd/venv/bin/../dist/var/etc/fwupd/fwupd.conf": No such file or directory
20:38:16.125 FuConfig             ::configuration loaded
20:38:16.125 Jcat                 reading /home/roberto/fwupd/venv/bin/../dist/etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
20:38:16.125 Jcat                 loaded 1 certificates
20:38:16.125 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
20:38:16.125 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
20:38:16.125 Jcat                 reading /home/roberto/fwupd/venv/bin/../dist/etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
20:38:16.125 Jcat                 loaded 1 certificates
20:38:16.125 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
20:38:16.125 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
20:38:16.125 Jcat                 reading /home/roberto/fwupd/venv/bin/../dist/var/lib/fwupd/pki/secret.key with 2455 bytes
20:38:16.125 Jcat                 reading /home/roberto/fwupd/venv/bin/../dist/var/lib/fwupd/pki/client.pem with 1383 bytes
20:38:16.125 FuEngine             client certificate now exists: nothing to do
20:38:16.125 FuHistory            trying to open database '/home/roberto/fwupd/venv/bin/../dist/var/lib/fwupd/pending.db'
20:38:16.126 FuHistory            got schema version of 13
20:38:16.128 FuPlugin             load(/home/roberto/fwupd/venv/bin/../dist/lib/x86_64-linux-gnu/fwupd-2.0.0/libfu_plugin_modem_manager.so)
20:38:16.132 FuQuirks             open database /home/roberto/fwupd/venv/bin/../dist/var/cache/fwupd/quirks.db
20:38:16.133 FuQuirks             mtimes unchanged: quirks,pci.ids:1711847705,usb.ids:1710796935,pnp.ids:1707542972, doing nothing
20:38:16.133 FuQuirks             loading quirks from /home/roberto/fwupd/venv/bin/../dist/share/fwupd/quirks.d
20:38:16.133 FuQuirks             loading quirks from /home/roberto/fwupd/venv/bin/../dist/var/lib/fwupd/quirks.d
20:38:16.133 XbSilo               attempting to load /home/roberto/fwupd/venv/dist/var/cache/fwupd/quirks.xmlb
20:38:16.133 XbSilo               file: e933b17a-d7d2-0b47-68a4-3fa357e90e10, current:e933b17a-d7d2-0b47-68a4-3fa357e90e10, cached: (null)
20:38:16.133 XbSilo               loading silo with file contents
20:38:16.135 FuConfig             loading config /home/roberto/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf
20:38:16.135 FuConfig             not loading config /home/roberto/fwupd/venv/bin/../dist/var/etc/fwupd/fwupd.conf
20:38:16.135 FuConfig             skipping mode check for /home/roberto/fwupd/venv/bin/../dist/var/etc/fwupd/fwupd.conf as not writable
20:38:16.135 FuConfig             trying to load config values from /home/roberto/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf
20:38:16.135 FuCommon             mapped file /home/roberto/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf of size 0x33
20:38:16.135 FuConfig             trying to load config values from /home/roberto/fwupd/venv/bin/../dist/var/etc/fwupd/fwupd.conf
20:38:16.135 FuConfig             Failed to open file "/home/roberto/fwupd/venv/bin/../dist/var/etc/fwupd/fwupd.conf": No such file or directory
20:38:16.135 FuConfig             ::configuration loaded
20:38:16.135 FuContext            failed to load smbios: Failed to open file "/sys/firmware/dmi/tables/smbios_entry_point": Permission denied
20:38:16.135 FuContext            failed to load fdt: cannot find /sys/firmware/fdt or override /home/roberto/fwupd/venv/bin/../dist/var/lib/fwupd/system.dtb
20:38:16.135 FuHwids              HardwareID-00 is not available, not available as 'BiosMajorRelease' unknown
[... cut ...]
  descriptor_type: 0x24
20:37:02.581 FuStruct             FuUsbBaseHdr:
  length: 0x1e
  descriptor_type: 0x24
20:37:02.581 FuStruct             FuUsbBaseHdr:
  length: 0x1e
  descriptor_type: 0x24
20:37:02.581 FuStruct             FuUsbBaseHdr:
  length: 0x6
  descriptor_type: 0x24
20:37:02.581 FuStruct             FuUsbBaseHdr:
  length: 0xd
  descriptor_type: 0x24
20:37:02.581 FuStruct             FuUsbBaseHdr:
  length: 0x12
  descriptor_type: 0x24
20:37:02.581 FuStruct             FuUsbBaseHdr:
  l

superm1, 2024-08-21:
OK, I've got another idea, can you try this PR now?

closed 2024-08-21 by superm1

roberto-zamparelli, 2024-08-21:
Hurray! The download error has vanished!
The update seems to have been successful, with a few odd or slightly
worrying messages during reboot (one about 2 minutes needed for "memory
training", the other "BIOS self healing backup progressing"). Running
test-fwupd I get all OKs except for:

4/45 fwupd-self-test                       FAIL             1.14s   killed
by signal 6 SIGABRT

Thanks a lot, this issue has been with me since I got the computer. Now I
can turn to tracking down the reason why the fingerprint scanner does not
work.

CorneelSoete, 2025-01-08:
On debian 12, i seem to have the same problem. I get the error saying 
/media/root/VDOYEFI does not have sufficient space, required 23.3 MB, got 5.0MB

**fwupd version information**
`compile   org.freedesktop.fwupd         1.8.12
compile   com.hughsie.libxmlb           0.3.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.fwupd         1.8.12
runtime   org.kernel                    6.1.0-28-amd64`

superm1, 2025-01-08:
Please update to a newer (non EOL) fwupd version if you're affected.

PR #6529 (2023-12-11): Match Lenovo `PQSERVICE` partition to skip ESP selection as well.
Fixes: #6477

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit 9acd75b9 (2023-12-11): Match Lenovo `PQSERVICE` partition to skip ESP selection as well.

Fixes: #6477
commit 59e8e885 (2023-12-11): Match Lenovo `PQSERVICE` partition to skip ESP selection as well.

Fixes: #6477