fwupd/fwupd #6494: redfish: show redfish plugin startup failed using snap install

vincent-tang1, 2023-12-05:
**Pre-condition**
Ubuntu 22.04

**Describe the question**

Using snap to install fwupd
```
#snap install fwupd --candidate
```
and  see version as below after installed done
```
#fwupdtool --version

Loading…      [ 5209.614749] audit: type=1326 audit(1701760823.819:75): auid=0 uid=0 gid=0 ses=3 subj=snap.fwupd.fwupdtool pid=5755 comm="fwupdtool" exe="/snap/fwupd/5438/bin/fwupdtool" sig=0 arch=c000003e syscall=314 compat=0 ip=0x7ffae716ba3d code=0x50000
Loading…                 [************************************** ]
compile   org.freedesktop.fwupd         1.9.10
compile   com.hughsie.libxmlb           0.3.6
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.fwupd         1.9.10
runtime   org.kernel                    5.15.0-67-generic
```

And run below get devices command using --verbose, as result, redfish plugin fail reason as 
```
root@FWA-6172-efi:~# fwupd.fwupdtool get-devices --verbose
Loading…      [ 5442.651824] audit: type=1326 audit(1701761056.851:76): auid=0 uid=0 gid=0 ses=3 subj=snap.fwupd.fwupdtool pid=5806 comm="fwupdtool" exe="/snap/fwupd/5438/bin/fwupdtool" sig=0 arch=c000003e syscall=314 compat=0 ip=0x7f4d72bafa3d code=0x50000
06:46:36.820 FuDebug              verbose to debug (on console 1)
06:46:36.820 FuEngine             starting fwupd 1.9.10
....
....
06:46:36.866 FuPluginRedfish      attempting to [re-]create user using IPMI
06:46:36.866 FuEngine             disabling plugin because: failed to startup using redfish: no BMC device found
...
...
```

For this fail log
```
06:46:36.866 FuEngine             disabling plugin because: failed to startup using redfish: no BMC device found
```
we can found coresponding code in ../plugins/redfish/fu-ipmi-device.c
```
static gboolean
fu_ipmi_device_probe(FuDevice *device, GError **error)
{
	FuIpmiDevice *self = FU_IPMI_DEVICE(device);
	const gchar *physical_ids[] = {"/dev/ipmi0", "/dev/ipmi/0", "/dev/ipmidev/0", NULL};

	/* look for the IPMI device */
	for (guint i = 0; physical_ids[i] != NULL; i++) {
		if (g_file_test(physical_ids[i], G_FILE_TEST_EXISTS)) {
			fu_device_set_physical_id(FU_DEVICE(self), physical_ids[i]);
			return TRUE;
		}
	}

	/* cannot continue */
	g_set_error_literal(error, G_IO_ERROR, G_IO_ERROR_NOT_SUPPORTED, "no BMC device found");
	return FALSE;
}
```

as you can see, this fail is  because "/dev/ipmi0", "/dev/ipmi/0" and "/dev/ipmidev/0" can not be found. but in fact, /dev/ipmi0 really exist in my system
```
root@FWA-6172-efi:~# ls -al /dev/ipmi0
crw------- 1 root root 235, 0 Dec  5 05:55 /dev/ipmi0
```

Seem like this application have not permission for /dev/.., 
Not sure if somthing permission controls need to be changed in snap side or source code?

**Note**
If i manually compile this code from trunk by myself, redfish plugin startup can be successfully and functionality can work
```
root@FWA-6172-efi:~# fwupdtool --version
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
compile   org.freedesktop.fwupd         1.9.10
compile   com.hughsie.libxmlb           0.3.6
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.9.7
compile   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.fwupd         1.9.10
runtime   org.kernel                    5.15.0-67-generic

root@FWA-6172-efi:~# fwupdtool get-devices
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
Advantech FWA-6172
│
├─BIOS:
│     Device ID:          562313e34c756a05a2e878861377765582bbf971
│     Summary:            Redfish host bus adapter
│     Current version:    1.00.0000
│     Vendor:             Advantech (REDFISH:ADVANTECH)
│     GUIDs:              8105ec9e-3100-52f9-8d10-fa27cf36d055 ← REDFISH\VENDOR_Advantech&SOFTWAREID_FWA-6172-BIOS
│                         84591cf7-9cee-5641-99d5-11af7ab46ecd ← REDFISH\VENDOR_Advantech&ID_BIOS
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│
├─BL:
│     Device ID:          92b9cdf108e011d817c6a97824eab088f9dd1194
│     Summary:            Redfish host bus adapter
│     Current version:    0.30.0000
│     Vendor:             Advantech (REDFISH:ADVANTECH)
│     GUIDs:              52cdda88-c2d2-5227-b62b-12cdc882b682 ← REDFISH\VENDOR_Advantech&SOFTWAREID_FWA-6172-BL
│                         ac7965d0-0781-5aa1-951d-4a263bb7a68b ← REDFISH\VENDOR_Advantech&ID_BL
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│
├─BMC:
│     Device ID:          62c1cd95692c5225826cf8568a460427ea3b1827
│     Summary:            Redfish host bus adapter
│     Current version:    0.73.0000
│     Vendor:             Advantech (REDFISH:ADVANTECH)
│     GUIDs:              25555b81-c4d7-5df8-843f-9f5c75b2bba4 ← REDFISH\VENDOR_Advantech&SOFTWAREID_1BMC
│                         69ec5e09-0466-5679-83f5-0dc75e3ff4d5 ← REDFISH\VENDOR_Advantech&ID_BMC
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
...
...
```

valentindavid, 2023-12-05:
Do we just rw access to `/dev/ipmi*`?

hughsie, 2023-12-05:
ioctl rw

vincent-tang1, 2023-12-06:
thanks for the snapd commit.

i would like to verify it, but do you know if having method to update more quickly snapd package into latest github master version from older snap version ?

superm1, 2023-12-15:
As snapd picked up fix, closing in fwupd to stop tracking here.  Please upgrade to newer snapd to fix issue, thanks!

vincent-tang1, 2024-01-04:
Sorry for lately testing!

Following is my test steps, but still can not work and have same permission issue, not sure if any incorrect operation from my side ? thanks

1.  install snap to latest master from github
```
# snap info snapd
name:      snapd
summary:   Daemon and tooling that enable snap packages
publisher: Canonical✓
store-url: https://snapcraft.io/snapd
contact:   https://github.com/snapcore/snapd/issues
license:   GPL-3.0
description: |
  Install, configure, refresh and remove snap packages. Snaps are
  'universal' packages that work across many different Linux systems,
  enabling secure distribution of the latest apps and utilities for
  cloud, servers, desktops and the internet of things.

  Start with 'snap list' to see installed snaps.
type:         snapd
snap-id:      PMrrV4ml8uWuEUDBT8dSGnKUYbevVhc4
tracking:     latest/edge
refresh-date: today at 08:56 UTC
channels:
  latest/stable:    2.60.4                  2023-10-10 (20290) 42MB -
  latest/candidate: 2.61.1                  2023-12-14 (20671) 42MB -
  latest/beta:      2.61.1                  2023-11-29 (20671) 42MB -
  latest/edge:      2.61.1+git1578.gd002397 2024-01-03 (20830) 42MB -
installed:          2.61.1+git1578.gd002397            (20830) 42MB snapd

# snap refresh --channel=edge

# snap --version
snap    2.61.1+git1578.gd002397
snapd   2.61.1+git1578.gd002397
series  16
ubuntu  22.04
kernel  5.15.0-46-generic
```

2. install fwupd from snap
```
# snap info fwupd
name:      fwupd
summary:   Firmware updates for Linux
publisher: Richard Hughes
store-url: https://snapcraft.io/fwupd
contact:   https://github.com/fwupd/fwupd
license:   unset
description: |
  The full fwupd firmware updating stack for Linux.  Can be used to perform updates on all supported
  devices via https://fwupd.org
commands:
  - fwupd.dbxtool
  - fwupd.fwupdmgr
  - fwupd.fwupdtool
services:
  fwupd: dbus, enabled, inactive
snap-id:      HpOj37PuyuaMUZY0NQhtwnp7oS5P8u5R
tracking:     latest/candidate
refresh-date: today at 09:09 UTC
channels:
  latest/stable:    1.9.10                2023-12-13 (5438) 24MB -
  latest/candidate: 1.9.11                2024-01-03 (5533) 24MB -
  latest/beta:      ↑
  latest/edge:      1.9.10-135-g170bc99a9 2024-01-03 (5532) 24MB -
installed:          1.9.11                           (5533) 24MB -

# snap install fwupd --candidate

# fwupd.fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.6
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.9.11
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.fwupd         1.9.11
runtime   org.kernel                    5.15.0-46-generic
```
3. after execute 'get device' command, this issue still occured
```
# fwupd.fwupdmgr get-devices --verbose

...
09:26:38.160 FuPluginRedfish      attempting to [re-]create user using IPMI
09:26:38.161 FuEngine             disabling plugin because: failed to startup using redfish: no BMC device found
...
...
```

hughsie, 2024-01-04:
> fwupd.fwupdmgr get-devices --verbose

Are there any daemon logs in the snap version? Maybe we can add `-vv` and get something useful, although I have no idea where we'd add that! :)

vincent-tang1, 2024-01-04:
> > fwupd.fwupdmgr get-devices --verbose
> 
> Are there any daemon logs in the snap version? Maybe we can add `-vv` and get something useful, although I have no idea where we'd add that! :)

seem like that i can not found related snap version information, you can also take a look about -vv log as following attachment
[get_device_v_v.log](https://github.com/fwupd/fwupd/files/13829060/get_device_v_v.log)

vincent-tang1, 2024-01-04:
after i enabled ManagerNetwork daemon, this error log have gone, you can take a look below again
[get_device_v_v_2.log](https://github.com/fwupd/fwupd/files/13829287/get_device_v_v_2.log)

but final redfish plugin module still halt in same permission deny place
```
11:21:44.995 FuPluginRedfish      attempting to [re-]create user using IPMI
11:21:44.995 FuEngine             disabling plugin because: failed to startup using redfish: no BMC device found
```

hughsie, 2024-01-04:
> no BMC device found

Ahh, this is because the daemon can't see "/dev/ipmi0", "/dev/ipmi/0" or "/dev/ipmidev/0" -- perhaps those are being blocked by the snap security policy?

vincent-tang1, 2024-01-04:
i am agree with u, seems like it is blocked by ths snap security policy,
actually, our ipmi device node is really existing as below
```
~# ls /dev/ipmi0 -al
crw------- 1 root root 235, 0 Jan  4 09:09 /dev/ipmi0
```

> [i/b/fwupd.go: give access to IPMI devices](https://github.com/valentindavid/snapd/commit/e0b94a4b69d0d082ebc8dcc483faee1f96705d68) …

but above fixing seems like it can not work if i really included this commit after upgraded below version 
```
# snap --version
snap    2.61.1+git1578.gd002397
snapd   2.61.1+git1578.gd002397
series  16
ubuntu  22.04
kernel  5.15.0-46-generic
```

vincent-tang1, 2024-01-05:
because this issue still existed, could i reopen this ticket ?
thanks

superm1, 2024-01-05:
I don't think we have action to fwupd do we? It's just waiting for snapd release.

vincent-tang1, 2024-01-05:
following is about current fwupd permission in snap, is it helpful for you?
```
# snap connections fwupd
Interface         Plug                    Slot               Notes
bluez             fwupd:bluez             :bluez             -
dbus              -                       fwupd:fwupd-dbus   -
fwupd             -                       fwupd:fwupd        -
fwupd             fwupd:fwupdmgr          -                  -
hardware-observe  fwupd:hardware-observe  :hardware-observe  -
modem-manager     fwupd:modem-manager     :modem-manager     -
network           fwupd:network           :network           -
opengl            fwupd:opengl            :opengl            -
polkit            fwupd:polkit            :polkit            -
raw-usb           fwupd:raw-usb           :raw-usb           -
shutdown          fwupd:shutdown          :shutdown          -
udisks2           fwupd:udisks2           :udisks2           -
upower-observe    fwupd:upower-observe    :upower-observe    -

```

superm1, 2024-01-05:
It's not a permission issue - it's a lack of the fwupd permission exported by snapd doesn't have permission to read `/dev/ipmi*` until a version with fix linked above.

If it still doesn't work with updated snapd with that change then we might have a second bug.

vincent-tang1, 2024-01-05:
> I don't think we have action to fwupd do we? It's just waiting for snapd release.

yes, i am just a little afriad of whether if this solution is valid. 

Because this commit already merged into snapd master, so we can upgrade snapd into latest master by below command to verify it (so, we should need not to wait for snapd formal release)
```
# snap refresh  snapd --channel=edge

# snap --version
snap    2.61.1+git1578.gd002397
snapd   2.61.1+git1578.gd002397
series  16
ubuntu  22.04
kernel  5.15.0-46-generic
```
not sure if my understanding is correct ?
or, if possible, could snapd owner confirm it?
thanks

superm1, 2024-01-05:
> In fact, after that, /dev/ipmi* permission issue still existed from fwupd observation.

You've restarted since making this change right?   If so; then yes we should re-open because it _should_ have been fixed by that upgrade you did.  Can you please share updated logs (kernel and fwupd) when reproducing with the updated snapd?

vincent-tang1, 2024-01-05:
yes, i'vd restarted it before.

sure, **fwupd** log, please see 
[get_device_v_v.log](https://github.com/fwupd/fwupd/files/13837273/get_device_v_v.log)

**kernel** log, please see
```
[  313.183542] audit: type=1400 audit(1704421758.854:14): apparmor="DENIED" operation="capable" profile="/snap/snapd/20840/usr/lib/snapd/snap-confine" pid=3211 comm="snap-confine" capability=12  capname="net_admin"
[  313.204729] audit: type=1400 audit(1704421758.854:15): apparmor="DENIED" operation="capable" profile="/snap/snapd/20840/usr/lib/snapd/snap-confine" pid=3211 comm="snap-confine" capability=38  capname="perfmon"
[  315.359965] audit: type=1326 audit(1704421761.030:16): auid=0 uid=0 gid=0 ses=4 subj=? pid=3211 comm="fwupdtool" exe="/snap/fwupd/5533/bin/fwupdtool" sig=0 arch=c000003e syscall=314 compat=0 ip=0x7f5bdb5eaa7d code=0x50000
```

thanks

superm1, 2024-01-05:
@valentindavid can you please look?

valentindavid, 2024-01-05:
I do not see any interesting error in the logs. It looks to me that the device node does not exist. Apparmor does not seem to have blocked anything. Also, I do not think that apparmor would make `g_file_test(..., G_FILE_TEST_EXISTS)` fail (I suppose it is just `access(..., F_OK)`). I do not think apparmor can block `access`.

Maybe give the logs of `snap run --strace="-f" fwupd.fwupdtool get-devices --verbose --verbose` to make sure what syscall fails.

Also `snap run --shell fwupd.fwupdtool`, then in the shell, then try to see if you can see `/dev/ipmi0` (with `test -e` or something like that).

vincent-tang1, 2024-01-05:
> snap run --strace="-f" fwupd.fwupdtool get-devices --verbose --verbose

see log [snap_get_device_v_v.log](https://github.com/fwupd/fwupd/files/13843114/snap_get_device_v_v.log)
from above log, it indicated that "Operation not permitted" in /dev/ipmi0
```
[pid  4368] access("/dev/ipmi0", F_OK)  = -1 EPERM (Operation not permitted)
```
about EPERM define as below ,  means permission is not enough ?
```
#define	EPERM 1		/* Not super-user */
```

>Also snap run --shell fwupd.fwupdtool, then in the shell, then try to see if you can >see /dev/ipmi0 (with test -e or something like that).
```
root@FWA-6172-efi:~# snap run --shell fwupd.fwupdtool
root@FWA-6172-efi:/root# ls -al /dev/ipmi0
crw------- 1 root root 235, 0 Jan  5 08:14 /dev/ipmi0       ===> can print /dev/ipmi0 file 

root@FWA-6172-efi:/root# if test -e /dev/ipmi0; then echo "test file success"; else echo "test file fail"; fi
test file success
```
from above, seems like  '/dev/ipmi0' file testing can be successful with test -e

valentindavid, 2024-01-05:
That might be the table for the BPF_PROG_TYPE_CGROUP_DEVICE that is missing the device. That could give `EPERM` maybe. I was missing several devices there. I have opened a PR https://github.com/snapcore/snapd/pull/13451

I am missing `/dev/wmi/dell-smbios` still. But I am not sure how to match that in udev.

vincent-tang1, 2024-01-08:
thanks for your comments,
so, once the PR is approved, i can verify it instantly, right ? or i still need wait other solutions you are still searching ?

valentindavid, 2024-01-08:
@vincent-tang1 Could you try the snap from the CI for the PR? See https://github.com/snapcore/snapd/actions/runs/7445855599?pr=13451 But I am not sure you have the permission to download it. So I have uploaded it to https://drive.google.com/file/d/1kK5Kl68fs3jhZaPOX0CkxyTRj7-3zTQ3/view?usp=sharing

vincent-tang1, 2024-01-09:
No problem,
i can downloaded the snap file you provided : **snapd_2.61.1+git58dc3be.58dc3be_amd64.snap**
could you give me a favor and guide me how to install the file (specific command) ? thanks

superm1, 2024-01-09:
> No problem,
> i can downloaded the snap file you provided : **snapd_2.61.1+git58dc3be.58dc3be_amd64.snap**
> could you give me a favor and guide me how to install the file (specific command) ? thanks

IIRC it should be something like `sudo snap install foo.snap --dangerous`.

vincent-tang1, 2024-01-09:
> @vincent-tang1 Could you try the snap from the CI for the PR? See https://github.com/snapcore/snapd/actions/runs/7445855599?pr=13451 But I am not sure you have the permission to download it. So I have uploaded it to https://drive.google.com/file/d/1kK5Kl68fs3jhZaPOX0CkxyTRj7-3zTQ3/view?usp=sharing

after applied, please see log
[get_device_v_v_0109.log](https://github.com/fwupd/fwupd/files/13868227/get_device_v_v_0109.log)

and your commit is useful and fixed the EPERM issue in /dev/ipmi0, but there is new permission deny issue as below
```
02:31:37.295 FuEngine             disabling plugin because: failed to startup using redfish: failed after 5 retries: error locking IPMI device: Permission denied
```

this error map into source code in fwupd as below
```
static gboolean
fu_ipmi_device_lock(GObject *device, GError **error)
{
    FuIpmiDevice *self = FU_IPMI_DEVICE(device);
    struct flock lock = {.l_type = F_WRLCK, .l_whence = SEEK_SET};
    if (fcntl(fu_udev_device_get_fd(FU_UDEV_DEVICE(self)), F_SETLKW, &lock) == -1) {
        g_set_error(error, G_IO_ERROR, G_IO_ERROR_FAILED, "error locking IPMI device: %m");
        return FALSE;
    }
    return TRUE;
}

```

seems like it is unable to lock ipmi device via fcntl call due to permission deny..

valentindavid, 2024-01-09:
Seems we need "rwk", not just "rw" then.

valentindavid, 2024-01-09:
Try again with https://drive.google.com/file/d/1FV7zFkDmXWjTZ8MnpNYj4Axlu3Trxyfm/view?usp=sharing

valentindavid, 2024-01-10:
^ @vincent-tang1

vincent-tang1, 2024-01-10:
after applied it, seems like it work for "get-devices" command which successfully list devices via redfish plugin, see log
[get_device_0110.log](https://github.com/fwupd/fwupd/files/13888913/get_device_0110.log)

but here found another permission issue if install locally CAB file as below, it show 
```
root@FWA-6172-efi:~/fwupd_scripts_and_cabs/lvfs_1bmc# fwupd.fwupdtool install firmware.cab
Loading…                 [********                               ]14:21:35.613 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/host10/target10:0:0/10:0:0:0/block/sdc: failed to open /dev/sdc: Operation not permitted
Loading…                 [************                           ]14:21:37.004 FuEngine             failed to add device /sys/devices/virtual/msr/msr0: failed to add device using on msr: failed to open /dev/cpu/0/msr: Operation not permitted
Writing…                 [*******************                    ]
Failed to open file “firmware.cab”: Permission denied
```
seems like when open CAB file under home folder, it shows permission denied, for detail debug log, see
[install_cab_v_v_0110.log](https://github.com/fwupd/fwupd/files/13889004/install_cab_v_v_0110.log)

superm1, 2024-01-10:
OK I think I know what's going on.  The snap doesn't have a home plug so it can't connect to the [home interface](https://forum.snapcraft.io/t/the-home-interface/7838).
1. Here's a PR that enables the home plug: https://github.com/fwupd/fwupd/pull/6650
2. You'll need to manually connect the two until we modify the store to do it for us.  Run this:
```shell
snap connect fwupd:home :home
```

Can you grab the artifact from that PR (or if it merges install the snap from the edge channel) and then connect the home plug?

vincent-tang1, 2024-01-11:
after applied your artifact, install CAB file using **"fwupdtool"** can be successful, see as below
```
root@FWA-6172-efi:~/fwupd_scripts_and_cabs/lvfs_1bmc# fwupd.fwupdtool install firmware.cab --allow-reinstall
Loading…                 [********                               ]01:43:52.263 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/host10/target10:0:0/10:0:0:0/block/sdc: failed to open /dev/sdc: Operation not permitted
Loading…                 [********                               ]01:43:52.363 FuPartialInputStream base stream is outside seekable range
01:43:52.363 FuPluginMtd          no version metadata found: failed to parse image: failed to read start signature: requested 0x4 and got 0x0
Loading…                 [************                           ]01:43:53.662 FuEngine             failed to add device /sys/devices/virtual/msr/msr0: failed to add device using on msr: failed to open /dev/cpu/0/msr: Operation not permitted
Loading…                 [*******************                    ]
WARNING: This package has not been validated, it may not work properly.
Restarting device…       [************************************** ]
```

but install CAB file using **"fwupmgr"**, it shows failed
```
root@FWA-6172-efi:~/fwupd_scripts_and_cabs/lvfs_1bmc# fwupd.fwupdmgr install firmware.cab --allow-reinstall
WARNING: This package has not been validated, it may not work properly.
invalid handle
```
seens like it shows **"invalid handle"** error.  i don't know what's going on
for same operations using "fwupdtool" and "fwupdmdr", but the former can work, but the latter can not
could you help this ?

**Note**
However, "fwupd.fwupmgr get-devices" command still can work and correctlly display

superm1, 2024-01-11:
Great! I'll merge that at least.  For` fwupdmgr` not working can you please also try `fwupd.fwupdmgr refresh` and `fwupd.fwupdmgr get-details foo.cab`?  I'd like to understand where the problem is (general file descriptor sharing or otherwise).

If those work, can you please modify the systemd unit for the daemon and add `--verbose --verbose`  to the `ExecStart` and then `systemctl daemon-reload` and `systemctl restart snap.fwupd.fwupd`.  After reproducing failure share journal output from the fwupd daemon.

vincent-tang1, 2024-01-11:
> Great! I'll merge that at least. For fwupdmgr not working can you please also try fwupd.fwupdmgr refresh and fwupd.fwupdmgr get-details foo.cab? I'd like to understand where the problem is (general file descriptor sharing or otherwise).

here is the result, but seems like having same error
```
root@FWA-6172-efi:~/fwupd_scripts_and_cabs/lvfs_1bmc# fwupd.fwupdmgr refresh
WARNING: This package has not been validated, it may not work properly.
Updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: 0 local devices supported

root@FWA-6172-efi:~/fwupd_scripts_and_cabs/lvfs_1bmc# fwupd.fwupdmgr get-details firmware.cab
WARNING: This package has not been validated, it may not work properly.
invalid handle
```

following my cab file, if you interesting,  
https://drive.google.com/file/d/1VjJs-VVn2-cPvB1_SxZ4QUke1FO4z_Gq/view?usp=drive_link
it generated by
https://github.com/fwupd/fwupd/blob/main/docs/only-trusted.md

superm1, 2024-01-11:
Can you please try same with [an artifact from 1_9_X backport](https://github.com/fwupd/fwupd/actions/runs/7483411742/artifacts/1161302782)?  If this works it means it's a problem with the new streaming changes from 2.0.0.

vincent-tang1, 2024-01-11:
after applied it, seems like still having same error
```
root@FWA-6172-efi:~/fwupd_scripts_and_cabs/lvfs_1bmc# snap list
Name    Version                    Rev    Tracking         Publisher   Notes
core22  20231123                   1033   latest/stable    canonical✓  base
fwupd   1.9.11-26-gb0c1a7046       x2     latest/stable    -           -
lxd     5.19-8635f82               26200  latest/stable/…  canonical✓  -
snapd   2.61.1+git1a18cdf.1a18cdf  x1     latest/edge      -           snapd

root@FWA-6172-efi:~/fwupd_scripts_and_cabs/lvfs_1bmc# fwupd.fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.6
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.9.12
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.fwupd         1.9.12
runtime   org.kernel                    6.5.0-10-generic

root@FWA-6172-efi:~/fwupd_scripts_and_cabs/lvfs_1bmc# fwupd.fwupdmgr get-details firmware.cab
WARNING: This package has not been validated, it may not work properly.
invalid handle
```
not sure fwupd version as 1.9.11-26-gb0c1a7046 in snap command and 1.9.12 in fwup command is your expectation ?

superm1, 2024-01-11:
Yes; that's correct version.  The 1.9.11-26-gb0c1a7046 is what you get from `git describe`.  1.9.12 is what is in `meson.build`.

OK this confirms it's not the new streaming stuff but there is some handling issue with passing the fd going on specifically coming out of home directory.  Can you put your CAB on LVFS and see that it works properly from there?

superm1, 2024-01-11:
I wonder if we need to also give the fwupd daemon the home plug in order to pass the fd around.

superm1, 2024-01-11:
Have a try with [this artifact](https://github.com/fwupd/fwupd/actions/runs/7483979776/artifacts/1161375784).  This is https://github.com/fwupd/fwupd/pull/6654 which adds the home plug to the daemon too.

vincent-tang1, 2024-01-11:
after applied it, it work and successfully installed cab file, see below
```
root@FWA-6172-efi:~/fwupd_scripts_and_cabs/create_cab_1bmc.0.75# fwupd.fwupdmgr install firmware.cab
WARNING: This package has not been validated, it may not work properly.
Waiting…                 [***************************************]
Successfully installed firmware
```

thanks

vincent-tang1, 2024-01-11:
> Have a try with [this artifact](https://github.com/fwupd/fwupd/actions/runs/7483979776/artifacts/1161375784). This is #6654 which adds the home plug to the daemon too.

@superm1 thanks, this PR can work

vincent-tang1, 2024-01-11:
>  Can you put your CAB on LVFS and see that it works properly from there?

@superm1 currently, locallym installing cab file can work, Next, i will find the time to validate LVFS online part.

closed 2024-01-13 by superm1

vincent-tang1, 2024-01-15:
> > [snapcore/snapd#13451](https://github.com/snapcore/snapd/pull/13451) (including 2 commits)
> 
> @valentindavid thanks, this PR can work

@valentindavid seems like it currently have not being merged into master (trunk), could you help to check it, thanks :)