fwupd/fwupd #6506: Documentation incorrect about Intel CET status

mhalano, 2023-12-06:
**Describe the bug**
[Here](https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.IntelCet.Active) in the documentation states that is necessary Intel CET to be active to get HSI:3 level. I don't know for sure, if it's a mistake in the documentation or in other place, but I got HSI:3 even I don't have Intel CET active.
```
fwupdmgr security 
Host Security ID: HSI:3! (v1.9.10)

HSI-1
✔ BIOS firmware updates:         Enabled
✔ MEI key manifest:              Valid
✔ csme manufacturing mode:       Locked
✔ csme override:                 Locked
✔ csme v0:15.0.45.2411:          Valid
✔ Platform debugging:            Disabled
✔ SPI write:                     Disabled
✔ SPI lock:                      Enabled
✔ SPI BIOS region:               Locked
✔ Supported CPU:                 Invalid
✔ TPM empty PCRs:                Valid
✔ TPM v2.0:                      Found
✔ UEFI bootservice variables:    Locked

HSI-2
✔ BIOS rollback protection:      Enabled
✔ Intel BootGuard ACM protected: Valid
✔ Intel BootGuard:               Enabled
✔ Intel BootGuard OTP fuse:      Valid
✔ Intel BootGuard verified boot: Valid
✔ Intel GDS mitigation:          Enabled
✔ IOMMU:                         Enabled
✔ Platform debugging:            Locked
✔ TPM PCR0 reconstruction:       Valid

HSI-3
✔ Intel BootGuard error policy:  Valid
✔ Intel CET Enabled:             Enabled
✔ Pre-boot DMA protection:       Enabled
✔ Suspend-to-idle:               Enabled
✔ Suspend-to-ram:                Disabled

HSI-4
✔ Intel SMAP:                    Enabled
✘ Encrypted RAM:                 Not supported

Runtime Suffix -!
✔ fwupd plugins:                 Untainted
✔ Linux swap:                    Disabled
✘ Intel CET Active:              Not supported
✘ Linux kernel lockdown:         Disabled
✘ Linux kernel:                  Tainted
✘ UEFI secure boot:              Disabled

This system has HSI runtime issues.
 » https://fwupd.github.io/hsi.html#hsi-runtime-suffix

Host Security Events
  2023-11-20 21:00:19:  ✔ Supported CPU changed: Valid → Invalid
  2023-09-16 22:51:13:  ✔ Pre-boot DMA protection is enabled
  2023-09-13 18:20:16:  ✘ Secure Boot disabled
  2023-09-13 18:20:16:  ✘ Pre-boot DMA protection is disabled
  2023-09-13 12:27:38:  ✔ TPM v2.0 changed: Not found → Found
  2023-09-13 12:11:31:  ✘ TPM v2.0 changed: Found → Not found
  2023-09-11 13:40:01:  ✘ Kernel lockdown disabled
  2023-09-11 13:40:01:  ✔ TPM v2.0 changed: Not found → Found
  2023-09-11 13:31:29:  ✘ Kernel is tainted
  2023-09-11 13:12:55:  ✔ Kernel is no longer tainted
  2023-09-11 13:12:55:  ✔ Kernel lockdown enabled
  2023-09-11 13:12:55:  ✔ Secure Boot enabled

Host Security ID attributes uploaded successfully, thanks!

[... 223 lines cut ...]
      Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
      Summary:            UEFI ESRT device
      Current version:    2451608321
      Minimum Version:    2451608321
      Vendor:             Dell (DMI:Dell Inc.)
      Update State:       Success
      GUID:               096da7a8-2137-4343-8d99-bdc6f640016d
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Supported on remote server
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
      Device Requests:    • Message
                          • Image
                          • Image (custom)
    

</details>

**Additional questions**

- Operating system and version: Ubuntu 23.10
- Have you tried rebooting? Yes
- Is this a regression? I don't know for sure

hughsie, 2023-12-07:
> Intel CET Enabled:             Enabled

It's weird, but that's actually correct. If your system has CET active (in use) you'd actually have `✔ Intel CET Active` in the runtime section -- IIRC, it was split up because the `Enabled` is something that your BIOS firmware vendor needs to fix, and `Active` is something that your *distro* needs to fix.

Maybe it's a bit misleading being `✔ Intel CET Enabled:             Enabled` -- maybe it should be `✔ Intel CET Enabled:             Supported` instead?

mhalano, 2023-12-07:
That makes sense, @hughsie. What is necessary to activate? I was testing the new 6.6 kernels that have CET/Shadow Stack enabled but still got not supported.

hughsie, 2023-12-07:
> What is necessary to activate

I'm pretty sure for CET you need to rebuild the entire distro with some extra CPU flags. I think the plan was to do that for Fedora 39, no idea for Ubuntu I'm afraid.

superm1, 2023-12-11:
So something else I notice here is that CET and SMAP are Intel only checks, but AMD also supports shadow stack and SMAP on recent processors.

Here's a Phoenix (Zen 4) APU:
```
  DeviceId:             4bde70ba4e39b28f9eab1628f9dd6e6244c03027
  Name:                 AMD Ryzen 5 7640U w/ Radeon 760M Graphics
  Guid:                 2a1df9de-fe02-59c8-8af1-32c8bb011f94 ← CPUID\PRO_0&FAM_19&MOD_74
  Guid:                 5a4bca32-6547-584f-bb36-920d9adede25 ← CPUID\PRO_0&FAM_19&MOD_74&STP_1
  Plugin:               cpu
  Flags:                internal|registered
  Vendor:               Advanced Micro Devices, Inc.
  VersionFormat:        hex
  Icon:                 computer
  Created:              2023-12-11
  Guid[quirk]:          22f9ecf4-588d-5c0a-8326-6ebff3655c6d ← CPUID\PRO_0&FAM_19
  PhysicalId:           cpu:0
  AcquiesceDelay:       50
  InternalFlags:        host-cpu
  HasSHSTK:             true
  HasIBT:               false
  HasTME:               false
  HasSMAP:              true
```

The lack of IBT is probably due to my kernel config.
```
CONFIG_CC_HAS_IBT=y
# CONFIG_X86_KERNEL_IBT is not set
```

superm1, 2023-12-19:
This should be fixed in the 1_9_X branch and also main. Please reopen if you disagree with the results.

closed 2023-12-19 by superm1

mhalano, 2023-12-19:
Thank you. I will wait the next release be available to test and I will let you know.

mhalano, 2024-01-03:
I just tested, since it's in Debian repositories, and now shows as supported and green. I hope with Glibc 2.39 supporting Intel CET we can see this option showing as enabled in a near future.