fwupd/fwupd #6535: Host security reports HSI-4 with the CPU supported attribute showing invalid

aleatoricLilac, 2023-12-13:
**Describe the bug**
`$ fwupdmgr security` displays the following which shows the HSI level at 4, yet with an reportedly unsupported CPU; It also seemed to have struggled with determining whether the CPU is actually supported.

```
Host Security ID: HSI:4 (v1.9.10)

HSI-1
✔ BIOS firmware updates:         Enabled
✔ Fused platform:                Locked
✔ Supported CPU:                 Invalid
✔ TPM empty PCRs:                Valid
✔ TPM v2.0:                      Found
✔ UEFI bootservice variables:    Locked
✔ UEFI platform key:             Valid
✔ UEFI secure boot:              Enabled

HSI-2
✔ SPI write protection:          Enabled
✔ BIOS rollback protection:      Enabled
✔ IOMMU:                         Enabled
✔ Platform debugging:            Locked
✔ TPM PCR0 reconstruction:       Valid

HSI-3
✔ SPI replay protection:         Enabled
✔ Pre-boot DMA protection:       Enabled
✔ Suspend-to-idle:               Enabled
✔ Suspend-to-ram:                Disabled

HSI-4
✔ Processor rollback protection: Enabled
✔ Encrypted RAM:                 Encrypted

Runtime Suffix -!
✔ fwupd plugins:                 Untainted
✔ Linux kernel lockdown:         Enabled
✔ Linux swap:                    Encrypted
✔ Linux kernel:                  Untainted

Host Security Events
  2023-11-22 14:58:27:  ✘ Supported CPU changed: Valid → Invalid
  2023-11-22 10:38:19:  ✔ Supported CPU changed: Invalid → Valid
  2023-11-22 10:16:00:  ✘ Supported CPU changed: Valid → Invalid
  2023-11-21 15:03:42:  ✔ Supported CPU changed: Invalid → Valid
  2023-11-21 14:59:56:  ✘ Supported CPU changed: Valid → Invalid
  2023-11-21 14:53:42:  ✔ Supported CPU changed: Invalid → Valid
  2023-11-21 14:42:25:  ✘ Supported CPU changed: Valid → Invalid
  2023-11-21 13:02:58:  ✔ Supported CPU changed: Invalid → Valid
  2023-11-21 12:43:04:  ✘ Supported CPU changed: Valid → Invalid

```

**Steps to Reproduce**
Run `fwupdmgr security` on a ThinkPad T14 Gen 3 AMD (machine type 21CFA000CD, processor Ryzen 7 PRO 6850U).

**Expected behavior**
As the attribute is in HSI-1, I expect an unsupported CPU to invalidate the rest of the report; on the other hand, with another Fedora installation on the same laptop before a warranty motherboard replacement the CPU was reported as supported, so logically I would expect it to be supported on essentially the same hardware as well?

**fwupd version information**
Please provide the version of the daemon and client.

```
compile   org.freedesktop.fwupd         1.9.10
compile   com.hughsie.libxmlb           0.3.14
compile   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         1.9.10
[... 302 lines cut ...]
│     GUID:               71599d14-9b31-5270-b3bd-74c494585820 ← UEFI\CRT_9AEF2123F4DE7C19AFABD909BB2C8CAC4411E07E
│   
└─Unifying Receiver:
      Device ID:          30f5fb521d2c42a8f2caee96524522cd0c85a24e
      Summary:            Miniaturised USB wireless receiver
      Current version:    RQR12.11_B0032
      Bootloader Version: BOT01.04_B0016
      Vendor:             HIDRAW:0x046D|USB:0x046D
      Install Duration:   30 seconds
      GUIDs:              9d131a0c-a606-580f-8eda-80587250b8d6
                          279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
      Device Flags:       • Updatable
                          • Supported on remote server
                          • Signed Payload
      Device Requests:    • Image

```

</details>

**Additional questions**

- Operating system and version: Fedora 39
- Have you tried rebooting? Yes, issue persists through rebooting.
- Is this a regression? No idea, as I'm not sure whether the report pre-mobo replacement was on an earlier version.

superm1, 2023-12-13:
It's fixed by https://github.com/fwupd/fwupd/commit/7d80e35bbc86ef7c9c112a615c02b49b8d4f48a1

aleatoricLilac, 2023-12-13:
Thanks, will check if it works correctly later.

closed 2023-12-13 by aleatoricLilac