fwupd/fwupd #6536: redfish: a requirement for adding OEM handling in manager-reset

vincent-tang1, 2023-12-13:
**Pre-condition**
use "manager-reset" flag in redfish.quirk file
```
+[REDFISH\VENDOR_Advantech&ID_BMC]
+Flags = manager-reset
+RedfishResetPreDelay = 100000
```

**Describe the question**
There is a requirement. In our server product, after BMC FW update finished, the BMC chip will auto-reboot to take a effect. So is it chance to merge below patch to ignore Redfish Action execution for Manager.Reset using HWID condition ('Advantech' manufacture) into the github community ?
```
diff --git a/plugins/redfish/fu-redfish-plugin.c b/plugins/redfish/fu-redfish-plugin.c
index 13592adf1..08766c99d 100644
--- a/plugins/redfish/fu-redfish-plugin.c
+++ b/plugins/redfish/fu-redfish-plugin.c
@@ -564,6 +564,7 @@ fu_redfish_plugin_cleanup(FuPlugin *plugin,
        g_autoptr(FuRedfishRequest) request = fu_redfish_backend_request_new(self->backend);
        g_autoptr(JsonBuilder) builder = json_builder_new();
        g_autoptr(GPtrArray) devices = NULL;
+       gboolean auto_bmc_reboot = FALSE;

        /* nothing to do */
        if (!fu_device_has_private_flag(device, FU_REDFISH_DEVICE_FLAG_MANAGER_RESET))
@@ -577,19 +578,27 @@ fu_redfish_plugin_cleanup(FuPlugin *plugin,
        fu_progress_add_step(progress, FWUPD_STATUS_DEVICE_RESTART, 18, "post-delay");
        fu_progress_add_step(progress, FWUPD_STATUS_DEVICE_BUSY, 9, "recoldplug");

-       /* ask the BMC to reboot */
-       json_builder_begin_object(builder);
-       json_builder_set_member_name(builder, "ResetType");
-       json_builder_add_string_value(builder, "ForceRestart");
-       json_builder_end_object(builder);
-       if (!fu_redfish_request_perform_full(request,
-                                            "/redfish/v1/Managers/1/Actions/Manager.Reset",
-                                            "POST",
-                                            builder,
-                                            FU_REDFISH_REQUEST_PERFORM_FLAG_NONE,
-                                            error)) {
-               g_prefix_error(error, "failed to reset manager: ");
-               return FALSE;
+       /* OEM specific for Advantech manufacture */
+       if (fu_context_has_hwid_guid(fu_plugin_get_context(plugin),
+                                    "18789130-a714-53c0-b025-fa93801d3995")) {
+               auto_bmc_reboot = TRUE;
+       }
+
+       if (!auto_bmc_reboot) {
+               /* ask the BMC to reboot */
+               json_builder_begin_object(builder);
+               json_builder_set_member_name(builder, "ResetType");
+               json_builder_add_string_value(builder, "ForceRestart");
+               json_builder_end_object(builder);
+               if (!fu_redfish_request_perform_full(request,
+                                                    "/redfish/v1/Managers/1/Actions/Manager.Reset",
+                                                    "POST",
+                                                    builder,
+                                                    FU_REDFISH_REQUEST_PERFORM_FLAG_NONE,
+                                                    error)) {
+                       g_prefix_error(error, "failed to reset manager: ");
+                       return FALSE;
+               }
        }
        fu_progress_step_done(progress);
```

If you think this patch is ok (can work from my side), maybe you can first review it and then change or modify any code you think better, thanks

**fwupd version information**
```shell
s# fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.10
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.9.11
runtime   org.freedesktop.fwupd-efi     1.9.11
compile   org.freedesktop.gusb          0.4.5
runtime   org.freedesktop.gusb          0.4.5
runtime   org.freedesktop.fwupd         1.9.11
runtime   org.kernel                    6.5.0-10-generic

```

vincent-tang1, 2023-12-13:
For more information and background, seem like that for BMC FW update need to set manager-reset flags because fwupd need to wait BMC reboot successfully and then update latest device list, otherwise using command "fwupdmgr get-device" will get incorrect information.

Because our server will auto reboot BMC after BMC FW finish the update (with backup flash), if running redfish action "/redfish/v1/Managers/1/Actions/Manager.Reset" will arise some problems and error. So not sure if it is ok for you about above solution using manufacture tag to seperate?

also wellcom any feedback and suggestion!

hughsie, 2023-12-13:
So what I think you want is something like:

    [REDFISH\VENDOR_Advantech&ID_BMC]
    Flags = no-manager-reset

i.e. you want the BMC to reset for all vendors, but **not** for Advantech, right?

vincent-tang1, 2023-12-14:
thanks your greate point,

yes, you are right, this function should be for all vendors.
could you help to check whether below patch is like what you think ? (workable from my side)
```
diff --git a/plugins/redfish/fu-redfish-device.c b/plugins/redfish/fu-redfish-device.c
index 5ef387dfd..249a401c7 100644
--- a/plugins/redfish/fu-redfish-device.c
+++ b/plugins/redfish/fu-redfish-device.c
@@ -877,6 +877,9 @@ fu_redfish_device_init(FuRedfishDevice *self)
        fu_device_register_private_flag(FU_DEVICE(self),
                                        FU_REDFISH_DEVICE_FLAG_MANAGER_RESET,
                                        "manager-reset");
+       fu_device_register_private_flag(FU_DEVICE(self),
+                                       FU_REDFISH_DEVICE_FLAG_NO_MANAGER_RESET,
+                                       "no-manager-reset");
 }

 static void
diff --git a/plugins/redfish/fu-redfish-device.h b/plugins/redfish/fu-redfish-device.h
index 8f789b276..bfc2ed49c 100644
--- a/plugins/redfish/fu-redfish-device.h
+++ b/plugins/redfish/fu-redfish-device.h
@@ -49,6 +49,8 @@ struct _FuRedfishDeviceClass {
  */
 #define FU_REDFISH_DEVICE_FLAG_WILDCARD_TARGETS (1 << 3)

+#define FU_REDFISH_DEVICE_FLAG_NO_MANAGER_RESET (1 << 4)
+
 FuRedfishBackend *
 fu_redfish_device_get_backend(FuRedfishDevice *self);
 gboolean
diff --git a/plugins/redfish/fu-redfish-plugin.c b/plugins/redfish/fu-redfish-plugin.c
index 13592adf1..c3491eb0a 100644
--- a/plugins/redfish/fu-redfish-plugin.c
+++ b/plugins/redfish/fu-redfish-plugin.c
@@ -566,7 +566,8 @@ fu_redfish_plugin_cleanup(FuPlugin *plugin,
        g_autoptr(GPtrArray) devices = NULL;

        /* nothing to do */
-       if (!fu_device_has_private_flag(device, FU_REDFISH_DEVICE_FLAG_MANAGER_RESET))
+       if (!fu_device_has_private_flag(device, FU_REDFISH_DEVICE_FLAG_MANAGER_RESET) &&
+               !fu_device_has_private_flag(device, FU_REDFISH_DEVICE_FLAG_NO_MANAGER_RESET))
                return TRUE;

        /* progress */
@@ -577,19 +578,21 @@ fu_redfish_plugin_cleanup(FuPlugin *plugin,
        fu_progress_add_step(progress, FWUPD_STATUS_DEVICE_RESTART, 18, "post-delay");
        fu_progress_add_step(progress, FWUPD_STATUS_DEVICE_BUSY, 9, "recoldplug");

-       /* ask the BMC to reboot */
-       json_builder_begin_object(builder);
-       json_builder_set_member_name(builder, "ResetType");
-       json_builder_add_string_value(builder, "ForceRestart");
-       json_builder_end_object(builder);
-       if (!fu_redfish_request_perform_full(request,
-                                            "/redfish/v1/Managers/1/Actions/Manager.Reset",
-                                            "POST",
-                                            builder,
-                                            FU_REDFISH_REQUEST_PERFORM_FLAG_NONE,
-                                            error)) {
-               g_prefix_error(error, "failed to reset manager: ");
-               return FALSE;
+       if (fu_device_has_private_flag(device, FU_REDFISH_DEVICE_FLAG_MANAGER_RESET)) {
+               /* ask the BMC to reboot */
+               json_builder_begin_object(builder);
+               json_builder_set_member_name(builder, "ResetType");
+               json_builder_add_string_value(builder, "ForceRestart");
+               json_builder_end_object(builder);
+               if (!fu_redfish_request_perform_full(request,
+                                                    "/redfish/v1/Managers/1/Actions/Manager.Reset",
+                                                    "POST",
+                                                    builder,
+                                                    FU_REDFISH_REQUEST_PERFORM_FLAG_NONE,
+                                                    error)) {
+                       g_prefix_error(error, "failed to reset manager: ");
+                       return FALSE;
+               }
        }
        fu_progress_step_done(progress);
```

If it is ok for you, could you help to commit it after you have better modification ?

Thanks!!

vincent-tang1, 2023-12-15:
Many thanks your help and commint in branch wip/hughsie/advantech-bmc
But seems like that there might be a accidental mistake.
After manually add following modidication, it can work from my side,
Could you please check this patch? thanks
```
diff --git a/plugins/redfish/fu-redfish-plugin.c b/plugins/redfish/fu-redfish-plugin.c
index 6cf28c903..69580d5f6 100644
--- a/plugins/redfish/fu-redfish-plugin.c
+++ b/plugins/redfish/fu-redfish-plugin.c
@@ -577,7 +577,7 @@ fu_redfish_plugin_cleanup(FuPlugin *plugin,
        fu_progress_add_step(progress, FWUPD_STATUS_DEVICE_BUSY, 9, "recoldplug");

        /* ask the BMC to reboot */
-       if (fu_device_has_private_flag(device, FU_REDFISH_DEVICE_FLAG_NO_MANAGER_RESET_REQUEST)) {
+       if (!fu_device_has_private_flag(device, FU_REDFISH_DEVICE_FLAG_NO_MANAGER_RESET_REQUEST)) {
                g_autoptr(JsonBuilder) builder = json_builder_new();
                json_builder_begin_object(builder);
                json_builder_set_member_name(builder, "ResetType");
```

By the way,
here is current redfish.quirk settings in this branch
[REDFISH\VENDOR_Advantech&ID_BMC]
Flags = manager-reset,no-manager-reset-request
RedfishResetPreDelay = 100000

hughsie, 2023-12-15:
> -       if (fu_device_has_private_flag(device, FU_REDFISH_DEVICE_FLAG_NO_MANAGER_RESET_REQUEST)) {
>+       if (!fu_device_has_private_flag(device, FU_REDFISH_DEVICE_FLAG_NO_MANAGER_RESET_REQUEST)) {

Doh, fixed!

vincent-tang1, 2023-12-15:
latest commit work from my side in this brnach

> > * ```
> >     if (fu_device_has_private_flag(device, FU_REDFISH_DEVICE_FLAG_NO_MANAGER_RESET_REQUEST)) {
> >   ```
> > 
> > 
> > * ```
> >     if (!fu_device_has_private_flag(device, FU_REDFISH_DEVICE_FLAG_NO_MANAGER_RESET_REQUEST)) {
> >   ```
> 
> Doh, fixed!

after verified this new commit in the branch, it work from my side.
could you help apply it into trunk? thanks!

hughsie, 2023-12-15:
Can you mark the PR as reviewed please.

vincent-tang1, 2023-12-15:
> Can you mark the PR as reviewed please.

mark done..

closed 2023-12-15 by hughsie