fwupd/fwupd #6537: Refresh embargo remote get 401 response

luswdev, 2023-12-13:
**Describe the bug**
I have copy aver-embargo.conf into my computer (/usr/local/etc/fwupd/remotes.d), then enable it.
But when I want to refresh, I will got 401 response.

**Steps to Reproduce**

Command: `fwupdmgr refresh`

Response:

```
WARNING: UEFI firmware can not be updated in legacy BIOS mode
See https://github.com/fwupd/fwupd/wiki/PluginFlag:legacy-bios for more information.
WARNING: This package has not been validated, it may not work properly.
Updating aver-embargo
Downloading…             [***************************************]
Failed to download metadata for aver-embargo: Failed to download, server response was 401: Unauthorized Access
```

fwupd version information:

```
compile   com.hughsie.libxmlb           0.3.6
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.9.11
compile   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.fwupd         1.9.11
runtime   org.kernel                    6.2.0-37-generic
```

My environment:
PC with windows 10
FWUPD install at VMWare with Ubuntu 22.04

luswdev, 2023-12-13:
Hi! I have already set username and password, and still get 401 error.

hughsie, 2023-12-13:
Can you attach the `fwupdmgr refresh -vv` output please? Make sure the user token isn't visible in the dump.

luswdev, 2023-12-13:
This is the output
```
$ fwupdmgr refresh -vv
(fwupdmgr:21931): GLib-DEBUG: 17:37:25.135: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:21931): GLib-GIO-DEBUG: 17:37:25.160: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:21931): dconf-DEBUG: 17:37:25.160: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:21931): dconf-DEBUG: 17:37:25.161: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:21931): dconf-DEBUG: 17:37:25.161: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:21931): dconf-DEBUG: 17:37:25.171: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:21931): dconf-DEBUG: 17:37:25.171: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:21931): GLib-GIO-DEBUG: 17:37:25.171: _g_io_module_get_default: Found default implementation gnome (GProxyResolverGnome) for ‘gio-proxy-resolver’
(fwupdmgr:21931): dconf-DEBUG: 17:37:25.172: watch_established: "/system/proxy/" (establishing: 1)
(fwupdmgr:21931): dconf-DEBUG: 17:37:25.172: watch_established: "/system/proxy/http/" (establishing: 1)
(fwupdmgr:21931): dconf-DEBUG: 17:37:25.173: watch_established: "/system/proxy/https/" (establishing: 1)
(fwupdmgr:21931): dconf-DEBUG: 17:37:25.174: watch_established: "/system/proxy/ftp/" (establishing: 1)
(fwupdmgr:21931): dconf-DEBUG: 17:37:25.174: watch_established: "/system/proxy/socks/" (establishing: 1)
(fwupdmgr:21931): Fwupd-DEBUG: 17:37:25.937: Emitting ::status-changed() [idle]
WARNING: UEFI firmware can not be updated in legacy BIOS mode
See https://github.com/fwupd/fwupd/wiki/PluginFlag:legacy-bios for more information.
WARNING: This package has not been validated, it may not work properly.
Updating aver-embargo
Fwupd-INFO: 17:37:25.951: downloading https://******fwupd.org/downloads/firmware-f3ea84f25a94a6414b717d4e92ec608155e1736ae1a03a968d451bf7bdc2e50f.xml.gz.jcat/auth
(fwupdmgr:21931): Fwupd-DEBUG: 17:37:25.952: Emitting ::status-changed() [downloading]
Downloading…             [                         |             ]Fwupd-INFO: 17:37:26.933: download progress: 100%
Downloading…             [***************************************](fwupdmgr:21931): Fwupd-DEBUG: 17:37:26.933: Emitting ::status-changed() [idle]
Fwupd-INFO: 17:37:26.933: status-code was 401

Failed to download metadata for aver-embargo: Failed to download, server response was 401: Unauthorized Access
```

hughsie, 2023-12-13:
> .xml.gz.jcat

Can you try changing that to .xml.xz.jcat in the remote please?

luswdev, 2023-12-14:
Hi, I have no idea how to change .gz to .xz
This is auto gen from LVFS, does there have any document to change it?

luswdev, 2023-12-14:
I tried to change .gz into .xz and still get 401.

output:

```
Updating aver-embargo
Fwupd-INFO: 10:40:14.653: downloading https://**************@fwupd.org/downloads/firmware-f3ea84f25a94a6414b717d4e92ec608155e1736ae1a03a968d451bf7bdc2e50f.xml.xz.jcat/auth
(fwupdmgr:51900): Fwupd-DEBUG: 10:40:14.653: Emitting ::status-changed() [downloading]
Downloading…             [                         |             ]Fwupd-INFO: 10:40:15.649: download progress: 100%
(fwupdmgr:51900): Fwupd-DEBUG: 10:40:15.649: Emitting ::status-changed() [idle]
Fwupd-INFO: 10:40:15.649: status-code was 401

Failed to download metadata for aver-embargo: Failed to download, server response was 401: Unauthorized Access
```

luswdev, 2023-12-14:
> Hi! I wrote about this here: https://lists.linuxfoundation.org/pipermail/lvfs-announce/2023-November/000087.html
> 
> The fix is to add to the remote.conf:
> 
> ```
> Username=
> Password=
> ```
> 
> ..where username is the email address you use to log into the LVFS, and password is the user **token** generated from https://fwupd.org/lvfs/profile -- does that help?

Sorry I didn't notice that password should fill in token, I've change password to token and now can refresh successfully.

closed 2023-12-14 by luswdev