fwupd/fwupd #6538: proxmox UEFI BIOS update fail: Secure boot is enabled, but shim isn't installed to /boot/efi/EFI/debian/shimx64.efi

EuroTrash2, 2023-12-13:
**Describe the bug**
In a Proxmox 8.1 system, shimx64.efi is located in /boot/efi/EFI/proxmox/shimx64.efi
Apparently, fwupdmgr ignores the "proxmox" folder and downloads the update to a newly created "debian" one.
Then it fails updating, complaining that the shimx64.efi is not present there.

**Steps to Reproduce**
```shell
# tree /boot/efi/EFI
/boot/efi/EFI
├── BOOT
│   ├── BOOTx64.efi
│   ├── fbx64.efi
│   ├── grubx64.efi
│   └── mmx64.efi
└── proxmox
    ├── BOOTX64.CSV
    ├── fbx64.efi
    ├── grub.cfg
    ├── grubx64.efi
    ├── mmx64.efi
    └── shimx64.efi

3 directories, 10 files
# fwupdmgr update
Devices with no available firmware updates:
 • SSD 970 EVO Plus 2TB
 • ST16000NM001G-2KK103
 • TPM 2.0
 • UEFI dbx
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 1.26.0 to 1.27.0?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Fixes & Enhancements - Firmware updates to address security                  ║
║ vulnerabilities.                                                             ║
║                                                                              ║
║ OptiPlex 7060 must remain plugged into a power source for the duration of    ║
║ the update to avoid damage.                                                  ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]:
Decompressing…           [                                       ]
Secure boot is enabled, but shim isn't installed to /boot/efi/EFI/debian/shimx64.efi
# tree /boot/efi/EFI
/boot/efi/EFI
├── BOOT
│   ├── BOOTx64.efi
│   ├── fbx64.efi
│   ├── grubx64.efi
│   └── mmx64.efi
├── debian
│   └── fw
│       └── fwupd-df35f706-c014-49e1-81e7-029945a45b12.cap
└── proxmox
    ├── BOOTX64.CSV
    ├── fbx64.efi
    ├── grub.cfg
    ├── grubx64.efi
    ├── mmx64.efi
    └── shimx64.efi

5 directories, 11 files
```
**Expected behavior**
BIOS update is placed in the correct proxmox folder and executed at reboot.

**fwupd version information**
```shell
# fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.12
compile   com.hughsie.libxmlb           0.3.10
compile   com.hughsie.libjcat           0.1.9
[... 835 lines cut ...]
├── memtest86+ia32.bin
├── memtest86+ia32.efi
├── memtest86+x64.bin
├── memtest86+x64.efi
├── pve
│   ├── initrd.img -> /boot/pve/initrd.img-6.5
│   ├── initrd.img-6.5 -> /boot/initrd.img-6.5.11-7-pve
│   ├── vmlinuz -> /boot/pve/vmlinuz-6.5
│   └── vmlinuz-6.5 -> /boot/vmlinuz-6.5.11-7-pve
├── System.map-6.5.11-4-pve
├── System.map-6.5.11-7-pve
├── vmlinuz-6.5.11-4-pve
└── vmlinuz-6.5.11-7-pve

13 directories, 638 files
```

**Additional questions**

- Operating system and version: Proxmox VE 8.1
- Have you tried rebooting? yes
- Is this a regression? don't think so.
- Are you using an NVMe disk? yes
- Is secure boot enabled? yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? no

superm1, 2023-12-13:
This should be a proxmox packaging bug.  This is controlled by the `efi_os_dir` [meson parameter](https://github.com/fwupd/fwupd/blob/1.9.10/meson_options.txt#L476).

superm1, 2023-12-13:
Alternatively if that parameter isn't to be used, the right value should be populated in `/etc/os-release`.
https://github.com/fwupd/fwupd/blob/c54b96987a860db7de0a811a00ed77ef8ce52272/plugins/uefi-capsule/fu-uefi-common.c#L226

EuroTrash2, 2023-12-13:
Many thanks, I can confirm I was able to update the BIOS by changing temporarily `/etc/os-release` before running `fwupdmgr`:

`sed -i "s/ID=proxmox/ID=debian/" /etc/os-release`

superm1, 2023-12-13:
Looking over all the code that's really weird IMO that that helped.  Could this be a systemd confinement issue?

EuroTrash2, 2023-12-13:
> Looking over all the code that's really weird IMO that that helped. Could this be a systemd confinement issue?

Sorry I don't know how to test that, but this is the content of /boot/efi/EFI after BIOS update, in case it helps:
```shell
# tree /boot/efi/EFI/
/boot/efi/EFI/
├── BOOT
│   ├── BOOTx64.efi
│   ├── fbx64.efi
│   ├── grubx64.efi
│   └── mmx64.efi
├── debian
│   └── fw
├── Dell
│   └── Bios
│       └── Recovery
│           └── BIOS_CUR.RCV
└── proxmox
    ├── BOOTX64.CSV
    ├── fbx64.efi
    ├── fw
    │   └── fwupd-df35f706-c014-49e1-81e7-029945a45b12.cap
    ├── fwupdx64.efi
    ├── grub.cfg
    ├── grubx64.efi
    ├── mmx64.efi
    └── shimx64.efi

9 directories, 13 files
```

So once the `/etc/os-release` check passed, it seems that `fwupd` put the firmware update under the `proxmox` folder anyway

superm1, 2023-12-13:
> So once the /etc/os-release check passed, it seems that fwupd put the firmware update under the proxmox folder anyway

Then something funny is going on we should understand.  Can you please share the original `/etc/os-release` that triggered the issue?

EuroTrash2, 2023-12-13:
before changing `ID=proxmox` to `ID=debian`:

```shell
# cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=proxmox
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

superm1, 2023-12-13:
Maybe it's an issue with `fwupd_get_os_release`

superm1, 2023-12-13:
Do you have a `/usr/lib/os-release` or does the daemon happen to run with `FWUPD_SYSCONFDIR` in the environment set to something?

superm1, 2023-12-13:
Or by chance does your `/etc/os-release` have funky line endings?

EuroTrash2, 2023-12-13:
Apparently I have a `/usr/lib/os-release` which is not the same as `/etc/os-release`

```shell
# cat /usr/lib/os-release
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

superm1, 2023-12-13:
Does adding quotes around `proxmox` in `/etc/os-release` help things?

EuroTrash2, 2023-12-14:
There are no funky line endings in `/etc/os-release`.
This was an install of Proxmox 8.1 from scratch from the official ISO, albeit now I've tried to reproduce the issue on another machine and I cannot seem to reproduce it. The `/etc/os-release` now shows correct ID of `debian`. I'm going through the command history of the host that gave me the issue to see if there's any command that might have changed `/etc/os-release`. Unfortunately the repro machine I've just setup has already its BIOS up to date. Not sure if worth chasing a downgrade.

EuroTrash2, 2023-12-14:
can't reproduce. I really have no idea what changed `/etc/os-release`.

```shell
# cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

# cat /usr/lib/os-release
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```
(after `apt install tree`)

```shell
# tree /boot/efi/EFI/
/boot/efi/EFI/
├── BOOT
│   ├── BOOTx64.efi
│   ├── fbx64.efi
│   ├── grubx64.efi
│   └── mmx64.efi
└── proxmox
    ├── BOOTX64.CSV
    ├── fbx64.efi
    ├── grub.cfg
    ├── grubx64.efi
    ├── mmx64.efi
    └── shimx64.efi

# fwupdmgr refresh
Updating lvfs
Downloading…             [*************************************  ]
Successfully downloaded new metadata: 1 local device supported
root@wonker:~# fwupdmgr get-updates
Devices with no available firmware updates:
 • SSD 970 EVO Plus 2TB
 • TPM 2.0
 • UEFI dbx
Devices with the latest available firmware version:
 • System Firmware
No updates available

# tree /boot/efi/EFI/
/boot/efi/EFI/
├── BOOT
│   ├── BOOTx64.efi
│   ├── fbx64.efi
│   ├── grubx64.efi
│   └── mmx64.efi
└── proxmox
    ├── BOOTX64.CSV
    ├── fbx64.efi
    ├── grub.cfg
    ├── grubx64.efi
    ├── mmx64.efi
    └── shimx64.efi

3 directories, 10 files
```

closed 2023-12-14 by EuroTrash2