fwupd/fwupd #6543: FuPluginIntelMe      failed to get public key using /fpf/OemCred: generic failure [0xb]

ckujau, 2023-12-14:
**Describe the bug**

When running `fwupdtool` the following message is printed for some commands:
```
FuPluginIntelMe      failed to get public key using /fpf/OemCred: generic failure [0xb]
```

Note: this error has been mentioned in older bug reports already, but all of them appear to be closed: #5745 #6060 #6227 #5704 

**Steps to Reproduce**
```
# fwupdtool --no-timestamp --verbose get-bios-setting 
[....]
FuPluginIntelMe      failed to get public key using /fpf/OemCred: generic failure [0xb]

# echo #?  
0
```

**Expected behavior**
No error should be shown, or the error message should be clearer :-)

**fwupd version information**
```
# fwupdmgr --version 
compile   org.freedesktop.fwupd         1.9.10
compile   com.hughsie.libxmlb           0.3.14
compile   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         1.9.10
runtime   org.kernel                    6.6.4-200.fc39.x86_64
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): `dnf`

<details>

<summary><b>fwupd device information</b></summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
LENOVO 20HDCTO1WW
│
├─USB2.0 Hub:
│ │   Device ID:          983c3cffc6fd36d32b00b62928d30721eaeb93db
│ │   Summary:            USB 3.x hub
│ │   Current version:    51.63
│ │   Vendor:             VIA Labs, Inc. (USB:0x17EF)
│ │   Install Duration:   15 seconds
│ │   GUIDs:              8ee94f0e-9b44-596a-bdd9-6f90401664cc ← USB\VID_17EF&PID_3080
│ │                       0987e3c9-b1ee-5763-ac6e-51329b034e4b ← USB\VID_17EF&PID_3080&HUB_20
│ │                       0b31f05d-23d6-5ac9-a7e7-2a6b2795230f ← USB\VID_17EF&PID_3080&SPI_5E32
│ │                       26ea2e97-397a-5975-9e58-a3858a6e25e8 ← USB\VID_17EF&PID_3080&SPI_5E32&REV_5163
│ │   Device Flags:       • Updatable
│ │                       • Cryptographic hash verification is available
│ │                       • Device stages updates
│ │                       • Device can recover flash failures
│ │                       • Unsigned Payload
│ │   Device Requests:    • Image
│ │ 
│ └─USB2.0 Hub:
│   │   Device ID:        691357701be9715ae7179d965f4fbfefdeb6e828
│   │   Summary:          USB 3.x hub
│   │   Current version:  51.33
│   │   Vendor:           VIA Labs, Inc. (USB:0x17EF)
│   │   Install Duration: 15 seconds
[... 265 lines cut ...]
│     Device Requests:    • Image
│   
└─VMM5322:
      Device ID:          d762543f8c20f636e6fff031a000078d3e10c600
      Summary:            Multi-stream transport device
      Current version:    5.06.01
      Vendor:             Synaptics (DRM_DP_AUX_DEV:0x06CB, DRM_DP_AUX_DEV:0x8086)
      GUIDs:              e6b5bc25-d512-51b0-b23c-25fdf483b15d ← MST-595
                          ef6a2415-266b-5469-b991-8f6633b96344 ← MST-dock-595
                          3857608c-c51a-5b70-9e52-c95b66ad5022 ← MST-dock-vmm5322-595
                          e0b67325-e8ff-5171-93e9-aaf098589166 ← MST-panamera-vmm5322-595
                          50f72ec0-b147-56d7-b1a4-6af9256b2d6a ← MST-panamera-595
      Device Flags:       • Updatable
                          • Device stages updates
                          • Unsigned Payload
      Device Requests:    • Image
```

</details>

**Additional questions**

- Operating system and version: Fedora Linux 39
- Have you tried rebooting? no
- Is this a regression? Possibly, yes. I've used `fwupdmgr` before and did not notice this error message.

ckujau, 2023-12-14:
output of `fwupdtool security --force --verbose`: https://pastebin.ubuntu.com/p/nQXr85m6Yg/plain/

hughsie, 2023-12-14:
I think we need the `strace` output like described here: https://blogs.gnome.org/hughsie/2023/05/09/msi-and-insecure-kms/

AlexLJordan, 2024-02-28:
@hughsie Hi, I'm having the same issue.
I don't quite unterstand the article or what to do, but it mentions csme and i do have csme v0:11.8.50.3425:  invalid in my `fwupd security` output...
What does this mean?

hughsie, 2024-02-29:
> i do have csme v0:11.8.50.3425

This version is critically insecure and you should ask your OEM for an update.

AlexLJordan, 2024-03-01:
> > i do have csme v0:11.8.50.3425
> 
> This version is critically insecure and you should ask your OEM for an update.

Lenovo has a lot of updates available for my machine. I'm seeing a lot of updates on fwupd like this one: https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR0RET.firmware

But as far as I can tell, fwupd is not detecting my machine and thus not updating everything...?

Do I have to boot some kind of windows and install the updates that way?

hughsie, 2024-03-04:
> fwupd is not detecting my machine and thus not updating everything

What does `fwupdmgr get-devices` say?

AlexLJordan, 2024-03-20:
> > fwupd is not detecting my machine and thus not updating everything
> 
> What does `fwupdmgr get-devices` say?

Curiously enough, it's reporting this:

```
Devices that were not updated correctly:
 - Intel Management Engine (184.50.3425 → 184.94.4494)
 - System Firmware (0.1.12 → 0.1.35)
```

Which I'm gonna attempt to update now, maybe that fixes it!

superm1, 2024-08-03:
> I'm unable to update my firmware on Dell G3 3779 with  `sudo fwupdtool update --force` and got the same error:
> ```
> $ sudo fwupdtool update --force --disable-ssl-strict
> УВАГА: Ігноруємо строгі перевірки SSL. Щоб зробити ігнорування у майбутньому автоматичним, експортуйте до вашого середовища змінну DISABLE_SSL_STRICT
> Загрузка…                [*********************                  ]16:43:04.139 FuPluginIntelMe      failed to get public key using /fpf/OemCred: generic failure [0xb]
> Загрузка…                [************************************** ]
> Устройства без доступных обновлений прошивки:
>  • Internal SPI Controller
>  • SC311 SATA 256GB
>  • ST2000LM007-1R8174
>  • System Firmware
>  • TPM
> Устройства с последней доступной версией прошивки:
>  • UEFI dbx
> 
> ```

This error won't stop you from updating firmware. You probably have your own different issue.

superm1, 2024-08-04:
> I just can't figure out what's wrong with my OS (or settings) so I can not update to latest firmware (which is `1.25.0`) with consecutive `fwupdmgr refresh --force && fwupdmgr update` .
> I still got old firmware version:
> ```
> $ sudo dmidecode -s bios-release-date
> 07/18/2018
> ```  
> while I have the needed file structure:
> ```
> $ tree /esp
> /esp
> └── EFI
> ```
> but the update didn't apply in fact:
> ```
> $ fwupdmgr refresh --force
> Обновление lvfs
> Получение...             [      \                                ]
> Успешно загружены новые метаданные:Передбачено підтримку 1 локального пристрою
> [roman@roman-pc ~]$ fwupdmgr update
> Устройства без доступных обновлений прошивки:
>  • Internal SPI Controller
>  • SC311 SATA 256GB
>  • ST2000LM007-1R8174
>  • System Firmware
>  • TPM
> Устройства с последней доступной версией прошивки:
>  • UEFI dbx
> ```
> P.S. I also tried the straightforward way with USB stick with `<firmware>.exe` on it, while it's successfully recognized on boot - it's going  to a black screen for a second and then just emits **_"Firmware update failed. Rebooting ..."_**
> 

Please open your own bug report.

closed 2024-11-11 by hughsie