fwupd/fwupd #6547: Upgrade TPM 1.2 to TPM 2.0 for Dell 3420

aodhan-domhnaill, 2023-12-17:
**Describe the question**

I don't run Windows, so I don't have a good way of upgrading the TPM from 1.2 to 2.0 on a Dell Precision 3420. Is it possible to do this with this tool?

I've tried running the various commands, but nothing seems to work. It simply upgrades, but stays TPM v1.2

**fwupd version information**

```shell
# fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.12
compile   com.hughsie.libxmlb           0.3.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.fwupd         1.8.12
runtime   org.kernel                    6.1.0-16-amd64
```

superm1, 2023-12-17:
This feature is removed from fwupd 1.9.x but older fwupd can support it.

Look at the plugins/dell/README.md on the 1_8_X branch for instructions.

aodhan-domhnaill, 2023-12-17:
Link for future generations https://github.com/fwupd/fwupd/tree/1_8_X/plugins/dell

aodhan-domhnaill, 2023-12-17:
@superm1 Do you know what to do if the TPM 2.0 version isn't showing up?

```    
root@debian:~# fwupdmgr get-devices
└─TPM 1.2:
      Device ID:          a3487e128cf1413519bce8e9a1ab3f5981e61458
      Summary:            UEFI ESRT device
      Current version:    5.81.2.1
      Vendor:             Dell Inc. (PCI:0x1028)
      Update State:       Success
      GUIDs:              52a6f885-b7bc-58e0-b8ec-0e0d8170bb0a ← 06c7-1.2
                          ff71992e-52f7-5eea-94ef-883e56e034c6 ← system-tpm
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Supported on remote server
                          • Needs a reboot after installation
```

superm1, 2023-12-17:
It looks like this model wasn't on the list that should allow it:
https://github.com/fwupd/fwupd/blob/1_8_X/plugins/dell/fu-dell-plugin.c#L102

You can try adding `067c` to that list.