fwupd/fwupd #6548: Removal of libsmbios breaks UEFI updates on Dell Optiplex 5090?

kaott, 2023-12-18:
**Describe the bug**
1.9.10 will not update UEFI on a Dell Optiplex 5090. I downgraded to 1.9.3 and it works as expected. I was previously on 1.20.1 UEFI version (this seems to have been pulled from the Dell website) trying was trying to upgrade to 1.21.1. Previous upgrades were possible on older versions, I don't know the version numbers for fwupd that were working, but I upgraded the UEFI in March, April, and June in 2023. I am speculating that the reason it no longer works is due to the removal of libsmbios. 1.9.3 was the last version I could build easily. 

**Steps to Reproduce**
On 1.9.10:
```shell
fwupdmgr refresh
fwupdmgr get-updates
fwupdmgr update
```
The refresh seems to work, but get-updates and update indicate there is no update.

**Expected behavior**
Same steps on 1.9.3 but it works as expected and the UEFI was updated.

**fwupd version information**
Not working: 
```shell
fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.14
compile   com.hughsie.libjcat           0.1.14
compile   org.freedesktop.fwupd         1.9.10
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.gusb          0.4.8
runtime   org.kernel                    6.6.7-0-lts
runtime   org.freedesktop.fwupd         1.9.10
```

Working:
```shell
fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.3
compile   com.hughsie.libxmlb           0.3.14
compile   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.1.14
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.8
runtime   org.kernel                    6.6.7-0-lts
runtime   org.freedesktop.fwupd         1.9.3
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
1.9.10 was installed via `apk`, I built 1.9.3 from source.

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
Dell Inc. OptiPlex 5090
│
├─Unknown Device:
│     Device ID:          5f7dee69f5c6dc34f6af5a8ecff65b51ce478670
│     Serial Number:      KH0NG7BJ2TAS
│     GUID:               2b6547cb-2ab1-5598-ab9b-0fbf12dca61b ← DRM\VEN_DEL&DEV_A0DB
│   
├─Unknown Device:
│     Device ID:          bbc98c073ac3282642278424d95dc59fcd055153
│     GUID:               c9ddd069-c7f2-58c5-b382-0487e683463a ← GPIO\ID_INT34C6:00
│   
├─CometLake-S GT2 [UHD Graphics 630]:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    03
[... 465 lines cut ...]
│       ├── xzio.mod
│       ├── zfs.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       └── zstd.mod
├── initramfs-lts
└── vmlinuz-lts

12 directories, 292 files
```

**Additional questions**

- Operating system and version:
Alpine Linux (edge)
- Have you tried rebooting?
Yes.
- Is this a regression?
Yes.
- Are you using an NVMe disk?
Yes.
- Is secure boot enabled?
No.
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?
No.

hughsie, 2023-12-18:
> 9e83f146-512a-47af-9540-f0d07c39cfc9

Although that hardware is supported on the LVFS, it's got a CHID requirement of  f85245a6-9a9a-5002-82b1-3c920efa0fc9 

What does `sudo fwupdtool hwids` say please.

kaott, 2023-12-18:
```shell
fwupdtool hwids
Loading…                 [*                                      ]
Computer Information
--------------------
BiosVendor: Dell Inc.
BiosVersion: 1.21.1
BiosMajorRelease: 1
BiosMinorRelease: 21
FirmwareMajorRelease: ff
FirmwareMinorRelease: ff
Manufacturer: Dell Inc.
Family: OptiPlex
ProductName: OptiPlex 5090
ProductSku: A54
EnclosureKind: 3
BaseboardManufacturer: Dell Inc.
BaseboardProduct: X4H68
Hardware IDs
------------
{0146fd5a-d826-568a-afff-4620972cbf98}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{f5e1e7f8-4465-5b7a-9297-e438c7df2ff2}   <- Manufacturer + Family + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{367125cc-8dc2-5c23-952b-7e21577c2d1f}   <- Manufacturer + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{d12f4078-5326-5798-b648-e08623932905}   <- Manufacturer + Family + ProductName + ProductSku + BaseboardManufacturer + BaseboardProduct
{3ad5970d-cbf1-5e03-a593-bcdbcc8493c8}   <- Manufacturer + Family + ProductName + ProductSku
{7d61445f-0d7e-5192-b911-10204a8deb36}   <- Manufacturer + Family + ProductName
{fd0c1620-2c35-5a25-9e75-9083e85aa4f6}   <- Manufacturer + ProductSku + BaseboardManufacturer + BaseboardProduct
{ac6492ac-71a8-59c3-a783-605b7eceb02c}   <- Manufacturer + ProductSku
{f13f99cb-8cca-5269-94d9-10f21fb99041}   <- Manufacturer + ProductName + BaseboardManufacturer + BaseboardProduct
{f85245a6-9a9a-5002-82b1-3c920efa0fc9}   <- Manufacturer + ProductName
{09c92b57-cff5-585a-abd2-76234853eb90}   <- Manufacturer + Family + BaseboardManufacturer + BaseboardProduct
{91251071-a01b-5067-9dca-81de39f555f9}   <- Manufacturer + Family
{529aea4b-4247-5705-89e2-edc2edd08c4e}   <- Manufacturer + EnclosureKind
{ca5baaaf-7d84-584f-a5b6-f6371cc39f7a}   <- Manufacturer + BaseboardManufacturer + BaseboardProduct
{85d38fda-fc0e-5c6f-808f-076984ae7978}   <- Manufacturer
```

Thinking on it more, I do think the removal of libsmbios is unrelated. The now pulled 1.20.1 UEFI was pushed out mid November and that would have fwupd version 1.9.7 or 1.9.8 that worked successfully to update it.

hughsie, 2023-12-18:
@CragW do you know what the CHID of  f85245a6-9a9a-5002-82b1-3c920efa0fc9 is all about? fwupd seems to be doing the right thing here.

CragW, 2023-12-19:
> f85245a6-9a9a-5002-82b1-3c920efa0fc9

It's the product name, i.e.: Dell Inc. Optiplex 5090. 

> {f85245a6-9a9a-5002-82b1-3c920efa0fc9}   <- Manufacturer + ProductName

The requirement on CHID should be matched in this case. 

More info needed to understand why the update was excluded.

hughsie, 2023-12-19:
> The requirement on CHID should be matched in this case.

Ohh, you're right! The daemon should have marked that as supported. @kaott can you attach the output of `sudo /usr/libexec/fwupd/fwupd -vv` please.

kaott, 2023-12-19:
Attached is the output of `fwupd -vv`

[fwlog.txt](https://github.com/fwupd/fwupd/files/13718189/fwlog.txt)

hughsie, 2023-12-19:
Ohh I see why:

    16:36:54.979 FuEngine             failed to set release for component: requirement 'hardware' also needs org.freedesktop.fwupd >= 1.0.1

I think I need something on the LVFS to fix up firmwares that have CHIDs and don't have the fwupd requirement of 1.0.1 -- in the meantime I've added the requirement on the LVFS firmware -- it should take affect in the next 4 hours (and then you'll need to do `fwupdmgr refresh --force`) to get the new metadata.

Link for @CragW : https://fwupd.org/lvfs/components/75065/requires

hughsie, 2023-12-19:
I've pushed https://gitlab.com/fwupd/lvfs-website/-/commit/565af0bf48161a1606700daecfcd39ab4472bdf9 and I'll deploy it in a few mins.

hughsie, 2023-12-19:
[image]

hughsie, 2023-12-19:
Okay; all done -- give it 4 hours and please try to refresh.

superm1, 2023-12-21:
closing as it was actually an LVFS issue.

closed 2023-12-21 by superm1